SzczepanLeon/esphome-components #104: Add driver for aquastream

G-R-O, 2024-06-20:
Hello,

Can you please add the driver aquastream?

Thanks

https://github.com/wmbusmeters/wmbusmeters/blob/master/drivers/src/aquastream.xmq

SzczepanLeon, 2024-09-20:
Please try version 4.0.0

closed 2024-09-20 by SzczepanLeon

G-R-O, 2025-02-27:
Hi, 
I finally got round to install a new ESP32 with version 4.0. But the driver aquastream is not implemented.
I'm using ESPHome 2025.2.1

[17:55:09][C][wmbus:394]:        State Class: 'measurement'
[17:55:09][C][wmbus:394]:        Unit of Measurement: 'dBm'
[17:55:09][C][wmbus:394]:        Accuracy Decimals: 0
[17:55:09][C][wmbus:393]:     Field: 'status'
[17:55:09][C][wmbus:394]:      Name: 'Watermeter Status'
[17:55:09][C][wmbus:394]:        State Class: ''
[17:55:09][C][wmbus:394]:        Unit of Measurement: ''
[17:55:11][D][mbus:035]: Received T1 A frame
[17:55:11][W][wmbus:088]: Selected driver aquastream doesn't exist, using 
[17:55:11][I][wmbus:104]: Unknown! [0x10049257] RSSI: -78dBm T: 4644B42557920410050E7237329305B425010749003025BA2C46593B9DA78BDB681A1927A8D4CE7FDF62D31B8211B574E024F73FE138AA482971F491E466986148CC5C7AF929CF (71) T1 A
[17:55:11][W][wmbus:111]: Link modes not defined in driver Unknown!. Processing anyway.
[17:55:12][W][wmbus:119]: Can't find driver for T: 4644B42557920410050E7237329305B425010749003025BA2C46593B9DA78BDB681A1927A8D4CE7FDF62D31B8211B574E024F73FE138AA482971F491E466986148CC5C7AF929CF (71)
[17:55:31][D][time:051]: Synchronized time: 2025-02-27 18:55:31
[17:55:31][W][system:328]: Synchronized clock
[17:55:36][D][sensor:094]: 'Boot counter': Sending state 10.00000  with 0 decimals of accuracy

But when I analyse my telegram by https://wmbusmeters.org/analyze I get the correct answer

telegram=|4644B42557920410050E7237329305B4250107D80030252F2F_04139ED9020084101300000000046D0C333B3244133BB20200426C3F3102FD17000002FD740C142F2F2F2F2F2F2F|

Auto driver    : aquastream
Similar driver : aquastream 22/22
Using driver   : aquastream 00/00
000   : 46 length (70 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : b425 dll-mfct (IMT)
004   : 57920410 dll-id (10049257)
008   : 05 dll-version
009   : 0e dll-type (Bus/System component)
010   : 72 tpl-ci-field (EN 13757-3 Application Layer (long tplh))
011   : 37329305 tpl-id (05933237)
015   : b425 tpl-mfct (IMT)
017   : 01 tpl-version
018   : 07 tpl-type (Water meter)
019   : d8 tpl-acc-field
020   : 00 tpl-sts-field (OK)(OK)
021   : 3025 tpl-cfg 2530 (synchronous AES_CBC_IV nb=3 cntn=0 ra=0 hc=0 )
023   : 2f2f decrypt check bytes (OK)
025   : 04 dif (32 Bit Integer/Binary Instantaneous value)
026   : 13 vif (Volume l)
027 C!: 9ED90200 ("total_m3":186.782)
031   : 84 dif (32 Bit Integer/Binary Instantaneous value)
032   : 10 dife (subunit=0 tariff=1 storagenr=0)
033   : 13 vif (Volume l)
034 C!: 00000000 ("total_tariff1_m3":0)
038   : 04 dif (32 Bit Integer/Binary Instantaneous value)
039   : 6D vif (Date and time type)
040 C!: 0C333B32 ("meter_datetime":"2025-02-27 19:12")
044   : 44 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
045   : 13 vif (Volume l)
046 C!: 3BB20200 ("target_m3":176.699)
050   : 42 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
051   : 6C vif (Date type G)
052 C!: 3F31 ("target_date":"2025-01-31")
054   : 02 dif (16 Bit Integer/Binary Instantaneous value)
055   : FD vif (Second extension FD of VIF-codes)
056   : 17 vife (Error flags (binary))
057 C!: 0000 ("status":"OK")
059   : 02 dif (16 Bit Integer/Binary Instantaneous value)
060   : FD vif (Second extension FD of VIF-codes)
061   : 74 vife (Reserved)
062 C!: 0C14 ("battery_y":14.060274)
064   : 2F skip
065   : 2F skip
066   : 2F skip
067   : 2F skip
068   : 2F skip
069   : 2F skip
070   : 2F skip

{
    "_":"telegram",
    "media":"water",
    "meter":"aquastream",
    "name":"",
    "id":"05933237",
    "battery_y":14.060274,
    "target_m3":176.699,
    "target_date":"2025-01-31",
    "total_m3":186.782,
    "total_tariff1_m3":0,
    "meter_datetime":"2025-02-27 19:12",
    "status":"OK",
    "timestamp":"2025-02-27T17:28:33Z"
}

G-R-O, 2025-02-28:
Checked everything on issue #135 but didn't find the problem. What do I need to check?

I realized that - noted on the very bottom of your main page - not all drivers of wmbusmeters are loaded.
`Supported drivers: **almost** all from wmbusmeters version 1.17.1 (without drivers from [file](https://github.com/wmbusmeters/wmbusmeters/blob/1.17.1/src/generated_database.cc))`

Why is the aquastream not included ? There is no file like driver_aquastream.cpp
After some testing I can now get the "total" and "battery" with the drivers topaseskr and gwfwater

PS: The list of available drivers in the log is not complete. the log is cut as it is probably too long

SzczepanLeon, 2025-02-28:
Now I see -  aquastream driver in in new format so it is not available in component.

But ...
You can try to decode it with another driver. Just test it on wmbusmeters page

G-R-O, 2025-02-28:
I can get part with different drivers but now everything.
I created the driver but can't get it to use/compile.

[driver_aquastream.cpp.txt](https://github.com/user-attachments/files/19030414/driver_aquastream.cpp.txt)

G-R-O, 2025-02-28:
Ahhhhh. 
You were completely right. Works perfectly with hydrus. :-)
I had the idea that the texts need to match..
Thanks for the help

SzczepanLeon, 2025-02-28:
No problem. 
Hydrus is almost perfect - only status is missing.

I have plans to implement that new drivers but I don't have time.

G-R-O, 2025-03-01:
I can get the status with hydrus like this
Haven't tryied timestamp yet

text_sensor:
  #--------------------------------------------------------------------------------------------------------------------
  # AQS-W8 status
  #--------------------------------------------------------------------------------------------------------------------
  - platform: wmbus
    meter_id: $wm_id
    type: $wm_type
    key: $wm_key
    sensors:
      - name: "AQS-W8 Status"
        field: "status"

SzczepanLeon, 2025-03-01:
Because there is no status in hydrus driver.
[image]

But **time_stamp** should be OK.