SzczepanLeon/esphome-components #107: [please explain] receiving many packets

luka6000, 2024-07-01:
Hello,

I can see a lot of logs like this
```
[16:55:59][D][mbus:034]: Processing T1 A frame
[16:56:03][D][mbus:034]: Processing T1 A frame
[16:56:11][D][mbus:034]: Processing T1 A frame
[16:56:15][D][mbus:034]: Processing T1 A frame
[16:56:15][D][wmbus:182]: Meter ID [0x90510343] RSSI: -63 dBm LQI: 128 Frame: T1 A not found in configuration T: 2F446850430351909572A2069F2F3205100EAF00000000101010100E1010130B050D11101710130A0F0E0C0D12161B13 (48)
```
Why there is no information about datagrams received at 55:59, 56:03 and 56:11?
Are those frames lost?
Is it worth debugging?

SzczepanLeon, 2024-07-01:
Set log to VERBOSE for more information.

luka6000, 2024-07-01:
ok, a few examples below.
Apart from some "decode 3 out of 6 failed" messages, it's like some datagrams are decoded with mbus and not processed by wmbus component.

```
**21:21:36][D][mbus:034]: Processing T1 A frame**
[21:21:36][V][mbus:045]: Frame: 3A971C6AC65639358B64D95695968E7294E698E59A9693A99B459C3565B2D0E5965965965965B25B25B28F259334E5B234D34D35334E34D35959935634D34B34B34E5A934D59A5B15A53565B45B45A35A3D34972 (84) [RAW]
[21:21:36][V][mbus:052]: Frame: 2F4468502703519095624F7AA2069F2FAC04100EC2000000000E0E0EBE07120E1111171211150510111313120F11060D09100C0C0B0BCC9E (56) [with CRC]
[21:21:36][V][mbus:095]: Validating CRC for Block1
[21:21:36][V][crc:031]:     calculated: 0x4F7A, read: 0x4F7A
[21:21:36][V][mbus:115]: Validating CRC for Block2
[21:21:36][V][crc:031]:     calculated: 0xBE07, read: 0xBE07
[21:21:36][V][mbus:115]: Validating CRC for Block3
[21:21:36][V][crc:031]:     calculated: 0x060D, read: 0x060D
[21:21:38][V][mbus:115]: Validating CRC for Block4
[21:22:06][V][rxLoop:167]: Have 84 bytes from CC1101 Rx, RSSI: -49 dBm LQI: 128
**[21:22:06][D][mbus:034]: Processing T1 A frame**
[21:22:06][V][mbus:045]: Frame: 3A971C6AC65670D58B64D9569594CE8E69A598E59A9693A93725AC3565B266658D59659659638B38B372C63B133993633693633933743663745AC37237437237436635338E4E3A7135638B38D3743713653A659C (84) [RAW]
[21:22:06][V][mbus:052]: Frame: 2F446850410351909572BAA9A2069F2F1E08100E5A0100000023231EDB87251B1F1B271C1A1C081E1C1E1C1A17227BFD1023211C1D192A04 (56) [with CRC]
[21:22:06][V][mbus:095]: Validating CRC for Block1
[21:22:06][V][crc:031]:     calculated: 0xBAA9, read: 0xBAA9
[21:22:06][V][mbus:115]: Validating CRC for Block2
[21:22:06][V][crc:031]:     calculated: 0xDB87, read: 0xDB87
[21:22:06][V][mbus:115]: Validating CRC for Block3
[21:22:06][V][crc:031]:     calculated: 0x7BFD, read: 0x7BFD
[21:22:06][V][mbus:115]: Validating CRC for Block4
[21:22:06][V][crc:031]:     calculated: 0x2A04, read: 0x2A04
[21:22:06][V][mbus:062]: Frame: 2F446850410351909572A2069F2F1E08100E5A0100000023231E251B1F1B271C1A1C081E1C1E1C1A17221023211C1D19 (48) [without CRC]
[21:22:09][V][rxLoop:167]: Have 96 bytes from CC1101 Rx, RSSI: -65 dBm LQI: 128
**[21:22:09][D][mbus:034]: Processing T1 A frame**
[21:22:09][V][mbus:045]: Frame: 2D371C6AC6564E5B1A4CBB1C393D0BA63A7498E34E9693A98DCC5358D2EC8D63B1596596B0E596596596D166A35965965965965963717258CD96C4D39636B14E56B4CA5A4D2D94E68DACB1A4E69C35CD0D59659659659659659659659698BB31 (96) [RAW]
[21:22:09][V][mbus:052]: Frame: 374468507986738427C3FBFCA2129F2FB4D70138B02D000082000000C06B00000000001D49B198779B6D796CE9F1357AB6EDF26414C10000000000000000A38D (64) [with CRC]
[21:22:09][V][mbus:095]: Validating CRC for Block1
[21:22:09][V][crc:031]:     calculated: 0xFBFC, read: 0xFBFC
[21:22:09][V][mbus:115]: Validating CRC for Block2
[21:22:09][V][crc:031]:     calculated: 0xC06B, read: 0xC06B
[21:22:09][V][mbus:115]: Validating CRC for Block3
[21:22:09][V][crc:031]:     calculated: 0x357A, read: 0x357A
[21:22:09][V][mbus:115]: Validating CRC for Block4
[21:22:09][V][crc:031]:     calculated: 0xA38D, read: 0xA38D
[21:22:09][V][mbus:062]: Frame: 374468507986738427C3A2129F2FB4D70138B02D00008200000000000000001D49B198779B6D796CE9F1B6EDF26414C10000000000000000 (56) [without CRC]
[21:22:09][D][wmbus:097]: Using driver 'vario451' for ID [0x84738679] RSSI: -65 dBm LQI: 128 Frame: T1 A T: 374468507986738427C3A2129F2FB4D70138B02D00008200000000000000001D49B198779B6D796CE9F1B6EDF26414C10000000000000000 (56)
[21:22:09][V][wmbus:121]: Publishing 'total_heating_gj' = 66.9160
[21:22:09][D][sensor:094]: 'heat meter raw': Sending state 66.91600 GJ with 3 decimals of accuracy
[21:22:09][V][wmbus:247]: Will send RTLWMBUS telegram to 192.168.0.9:7011 via UDP
```

```
**[21:22:40][D][mbus:013]: Processing C1 A frame**
[21:22:40][V][mbus:017]: Frame: 364468502703519095370FB372125823632423FE06BD00202551415B82530115B38EBDFA0258113331965E01F313880D038E0A42128C06F269E9F17BD7A612 (63) [with CRC]
[21:22:40][V][mbus:095]: Validating CRC for Block1
[21:22:40][V][crc:031]:     calculated: 0x0FB3, read: 0x0FB3
[21:22:40][V][mbus:115]: Validating CRC for Block2
[21:22:40][V][crc:031]:     calculated: 0x8253, read: 0x8253
[21:22:40][V][mbus:115]: Validating CRC for Block3
[21:22:40][V][crc:031]:     calculated: 0x880D, read: 0x880D
[21:22:40][V][mbus:115]: Validating CRC for Block4
[21:22:40][V][crc:031]:     calculated: 0xA612, read: 0xA612
[21:22:40][V][mbus:062]: Frame: 3644685027035190953772125823632423FE06BD00202551415B0115B38EBDFA0258113331965E01F313038E0A42128C06F269E9F17BD7 (55) [without CRC]
[21:22:40][V][wmbus:247]: Will send RTLWMBUS telegram to 192.168.0.9:7011 via UDP
[21:22:44][V][rxLoop:167]: Have 96 bytes from CC1101 Rx, RSSI: -68 dBm LQI: 128
**[21:22:44][D][mbus:034]: Processing T1 A frame**
[21:22:44][V][mbus:045]: Frame: 2D371C6AC6564E5B1A4CBB1C393D0BA63A7498E34E9693A98DCC5358D2EC8D63B1596596B0E596596596D166A35965965965965963717258CD96C4D39636B14E56B4CA5A4D2D94E68DACB1A4E69C35CD0D59659659659659659659659698BB31 (96) [RAW]
[21:22:44][V][mbus:052]: Frame: 374468507986738427C3FBFCA2129F2FB4D70138B02D000082000000C06B00000000001D49B198779B6D796CE9F1357AB6EDF26414C10000000000000000A38D (64) [with CRC]
[21:22:44][V][mbus:095]: Validating CRC for Block1
[21:22:44][V][crc:031]:     calculated: 0xFBFC, read: 0xFBFC
[21:22:44][V][mbus:115]: Validating CRC for Block2
[21:22:44][V][crc:031]:     calculated: 0xC06B, read: 0xC06B
[21:22:44][V][mbus:115]: Validating CRC for Block3
[... cut ...]
[21:24:27][V][mbus:062]: Frame: 2F446850270351909562A2069F2FAC04100EC2000000000E0E0E120E1111171211150510111313120F1109100C0C0B0B (48) [without CRC]
[21:24:27][D][wmbus:097]: Using driver 'mkradio4' for ID [0x90510327] RSSI: -50 dBm LQI: 128 Frame: T1 A T: 2F446850270351909562A2069F2FAC04100EC2000000000E0E0E120E1111171211150510111313120F1109100C0C0B0B (48)
[21:24:27][V][wmbus:121]: Publishing 'total_water_m3' = 139.0000
[21:24:27][D][sensor:094]: 'hot water raw': Sending state 139.00000 m³ with 3 decimals of accuracy
[21:24:27][V][wmbus:247]: Will send RTLWMBUS telegram to 192.168.0.9:7011 via UDP
```

closed 2024-09-24 by SzczepanLeon