SzczepanLeon/esphome-components #112: qheat Can't get value(s) from telegram for ID

sapientior, 2024-08-04:
I have issues decoding Telegrams using "qheat" as driver.
I always get "Can't get value(s) from telegram for ID [xxx]" as error.
The telegram definetly is fine and can be decoded properly. The Value for "total_energy_consumption_kwh" is correct (checked on the display of the meter) Here is one example:
[https://wmbusmeters.org/analyze](https://wmbusmeters.org/analyze/49449344426266963508780DFF5F350082A10000100007B06EFFFF15000000FF2C030000001F371500000003000000000000000000FEFF000000000000000000000E002F046D0F100438:qheat)

The logs of my ESP32-WROOM32 look like this:
```
[17:10:19][V][rxLoop:167]: Have 86 bytes from CC1101 Rx, RSSI: -78 dBm LQI: 129
[17:10:19][D][mbus:013]: Processing C1 A frame
[17:10:19][V][mbus:017]: Frame: 49449344426266963508306E780DFF5F350082A10000100007B06EFFD405FF15000000FF2C030000001F3715000050E30003000000000000000000FEFF0000004F3A000000000000000E002F046D0F1004389E2B (84) [with CRC]
[17:10:19][V][mbus:095]: Validating CRC for Block1
[17:10:19][V][crc:031]:     calculated: 0x306E, read: 0x306E
[17:10:19][V][mbus:115]: Validating CRC for Block2
[17:10:19][V][crc:031]:     calculated: 0xD405, read: 0xD405
[17:10:19][V][mbus:115]: Validating CRC for Block3
[17:10:19][V][crc:031]:     calculated: 0x50E3, read: 0x50E3
[17:10:19][V][mbus:115]: Validating CRC for Block4
[17:10:19][V][crc:031]:     calculated: 0x4F3A, read: 0x4F3A
[17:10:19][V][mbus:115]: Validating CRC for Block5
[17:10:19][V][crc:031]:     calculated: 0x9E2B, read: 0x9E2B
[17:10:19][V][mbus:062]: Frame: 49449344426266963508780DFF5F350082A10000100007B06EFFFF15000000FF2C030000001F371500000003000000000000000000FEFF000000000000000000000E002F046D0F100438 (74) [without CRC]
[17:10:19][D][wmbus:097]: Using driver 'qheat' for ID [0x96666242] RSSI: -78 dBm LQI: 129 Frame: C1 A T: 49449344426266963508780DFF5F350082A10000100007B06EFFFF15000000FF2C030000001F371500000003000000000000000000FEFF000000000000000000000E002F046D0F100438 (74)
[17:10:19][D][wmbus:161]: Can't get value(s) from telegram for ID [0x96666242]
```

Is it an issue in your implementation or is it my code? With the same device, I am also monitoring my warm and cold water meters successfully.

code on my ESPHome:

```
substitutions:
  log_level: "VERBOSE"    
  log_wmbus: "VERBOSE"    
  log_baudrate: "0"  

esphome:
  name: wassermeter
  comment: "Wasserzähler ESP32, CUL - CC1101, qwater"
  friendly_name: Wassermeter

preferences:
  flash_write_interval: 10min

esp32:
  board: esp32dev
  framework:
    type: arduino

logger:
  id: appslogger
  level: ${log_level}
  baud_rate: ${log_baudrate}
  logs:
    wmbus: ${log_wmbus}
    wMBus-lib: ${log_wmbus}
    sensor: WARN
    sensor.filter: WARN
    text_sensor: WARN
    api.service: ERROR
    json: ERROR
    mqtt: WARN
    scheduler: ERROR
    internal_temperature: ERROR
    wifi: WARN
    component: ERROR
    api: WARN

api:
  id: espapi_wmbus_esp32
  port: 6053
  reboot_timeout: 0s
  # https://esphome.io/components/api.html?highlight=api
  encryption:
    key: "***"
  
ota:
  - platform: esphome
    password: "***"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  ap:
    ssid: "Wassermeter Fallback Hotspot"
    password: "rN5I8rnYuOjp"

  manual_ip:
    static_ip: 172.30.0.80
    gateway: 172.30.0.254
    subnet: 255.255.255.0

time:
  - platform: sntp
    id: time_sntp
    timezone: Europe/Berlin
    servers:
      - 172.30.0.254

external_components:
  # uses the latest version from SzczepanLeon
  - source: github://SzczepanLeon/esphome-components@main
    refresh: 0s
    components: [wmbus]

# ============================================================ 
# Hardware related setup
#
## ------------------------------------------------------------------
##           WMBUS CC1101 --> ESP32-WROOM-32D
## ------------------------------------------------------------------
##                                                                                                      ANT
##                                                                                                       |
##   ╭――x――x――x――x――x――x――x――x――x――x――x――x――x――x――x――x――x――x――x―╮                         ╭―――――――――――x――o――x――――――――╮
##   |                                                          |                         |         CC1101           |
##   |                      ESP32-WROOM-32D                     |                         |     G  G  M     M        |
## - | USB                                                      | -- ANT                  |  C  D  D  I  C  O  G  V  |
##   |                                                          |                         |  S  O  O  S  L  S  N  C  |
##   |                 16 17 5  18 19                        23 |                         |  N  0  2  O  K  I  D  C  |
##   ╰――o――o――x――x――x――o――o――o――o――o――o――o――o――o――o――o――o――o――o―╯                         ╰――o――o――o――o――o――o――o――o――╯
##      | GND         RX2 |  |  |  |                          |                              1  2  3  4  5  6  7  8
##     3.3V|           | TX2 |  |  |                          |                              |  |  |  |  |  |  |  |
##      |  |           |  |  |  |  |                          |                              |  |  |  |  |  |  |  |   
##      |  |           |  |  |  |  |                          6                              5 RX2 |  19 18 23 | 3.3
##      8  7           2  3  1  5  4                                                              TX2         GND 
##                                    
## ------------------------------------------------------------------
wmbus:
  mosi_pin: GPIO23
  miso_pin: GPIO19
  clk_pin: GPIO18
  cs_pin: GPIO5
  gdo0_pin: GPIO16
  gdo2_pin: GPIO17
  log_unknown: true

## ---------------------------------------------------
## SENSORS
## ---------------------------------------------------
sensor:
  - platform: wmbus
    meter_id: 0x51423803
    type: qwater
    mode: C1
    total_water_m3:
      name: "Warmwasser"
      id: warmwasser
      unit_of_measurement: "m³"
      state_class: total_increasing
      device_class: "water"
      accuracy_decimals: 3
      icon: mdi:counter
    lqi:
      name: "Warmwasser lqi"
      entity_category: "diagnostic"
    rssi:
      name: "Warmwasser RSSI"
      entity_category: "diagnostic"
  - platform: wmbus
    meter_id: 0x51423792
    type: qwater
    mode: C1
    total_water_m3:
      name: "Kaltwasser"
      id: kaltwasser
      unit_of_measurement: "m³"
      state_class: total_increasing
      device_class: "water"
      accuracy_decimals: 3
      icon: mdi:counter
    lqi:
      name: "Kaltwasser lqi"
      entity_category: "diagnostic"
    rssi:
      name: "Kaltwasser RSSI"
      entity_category: "diagnostic"
  - platform: wmbus
    meter_id: 0x96666242
    type: qheat
    mode: C1
    total_energy_consumption_kwh:
      name: "Heizung Bad"
      id: heizung_bad_gesamt
      unit_of_measurement: "kWh"
      state_class: total_increasing
      device_class: "energy"
      accuracy_decimals: 0
      icon: mdi:counter
    lqi:
      name: "Heizung Bad lqi"
      entity_category: "diagnostic"
    rssi:
      name: "Heizung Bad RSSI"
      entity_category: "diagnostic"

```

ndandanov, 2024-09-16:
I have identical problem:

```
[00:54:13][V][rxLoop:167]: Have 173 bytes from CC1101 Rx, RSSI: -89 dBm LQI: 133
[00:54:13][D][mbus:013]: Processing C1 A frame
[00:54:13][V][mbus:017]: Frame: 9644A732511518700704DF487A6E00002002744503046D02170F390C478E0656590000446D3B17FF2C4C0692540007670084016D3B171F388C010656590000CC924E0106565900008C020656590000CC02063071565900008C030656590000CC03065659E0E800008C040606590000CC0406205800002C828C050692540000CC0506925100008C069B8F0672490000CC0606724900008C070672EAFD4900003C22000000000F801000E574 (171) [with CRC]
[00:54:13][V][mbus:095]: Validating CRC for Block1
[00:54:13][V][crc:031]:     calculated: 0xDF48, read: 0xDF48
[00:54:13][V][mbus:115]: Validating CRC for Block2
[00:54:13][V][crc:031]:     calculated: 0x478E, read: 0x478E
[00:54:13][V][mbus:115]: Validating CRC for Block3
[00:54:13][V][crc:031]:     calculated: 0x0767, read: 0x0767
[00:54:13][V][mbus:115]: Validating CRC for Block4
[00:54:13][V][crc:031]:     calculated: 0x924E, read: 0x924E
[00:54:13][V][mbus:115]: Validating CRC for Block5
[00:54:13][V][crc:031]:     calculated: 0x3071, read: 0x3071
[00:54:13][V][mbus:115]: Validating CRC for Block6
[00:54:13][V][crc:031]:     calculated: 0xE0E8, read: 0xE0E8
[00:54:13][V][mbus:115]: Validating CRC for Block7
[00:54:13][V][crc:031]:     calculated: 0x2C82, read: 0x2C82
[00:54:13][V][mbus:115]: Validating CRC for Block8
[00:54:13][V][crc:031]:     calculated: 0x9B8F, read: 0x9B8F
[00:54:13][V][mbus:115]: Validating CRC for Block9
[00:54:13][V][crc:031]:     calculated: 0xEAFD, read: 0xEAFD
[00:54:13][V][mbus:115]: Validating CRC for Block10
[00:54:13][V][crc:031]:     calculated: 0xE574, read: 0xE574
[00:54:13][V][mbus:062]: Frame: 9644A7325115187007047A6E00002002744503046D02170F390C0656590000446D3B17FF2C4C069254000084016D3B171F388C010656590000CC0106565900008C020656590000CC0206565900008C030656590000CC0306565900008C040606590000CC0406205800008C050692540000CC0506925100008C060672490000CC0606724900008C0706724900003C22000000000F801000 (151) [without CRC]
[00:54:13][D][wmbus:097]: Using driver 'qheat' for ID [0x70181551] RSSI: -89 dBm LQI: 133 Frame: C1 A T: 9644A7325115187007047A6E00002002744503046D02170F390C0656590000446D3B17FF2C4C069254000084016D3B171F388C010656590000CC0106565900008C020656590000CC0206565900008C030656590000CC0306565900008C040606590000CC0406205800008C050692540000CC0506925100008C060672490000CC0606724900008C0706724900003C22000000000F801000 (151)
[00:54:13][D][wmbus:161]: Can't get value(s) from telegram for ID [0x70181551]
```

This occurs repeatedly and no telegram has been detected successfully using the `qheat` driver.

On the other hand, the telegrams have correct CRC values, and [are decoded successfully with `wmbusmeters`.](https://wmbusmeters.org/analyze/9644A7325115187007047A6E00002002744503046D02170F390C0656590000446D3B17FF2C4C069254000084016D3B171F388C010656590000CC0106565900008C020656590000CC0206565900008C030656590000CC0306565900008C040606590000CC0406205800008C050692540000CC0506925100008C060672490000CC0606724900008C0706724900003C22000000000F801000)

SzczepanLeon, 2024-09-16:
Wait for version 4.x with all drivers from wmbusmeters.

ndandanov, 2024-09-16:
Thanks @SzczepanLeon, this sounds great! When is the expected release?

If there is any help that would come in handy, please let me know :-)

SzczepanLeon, 2024-09-20:
Please try version 4.0.0

closed 2024-09-20 by SzczepanLeon

ndandanov, 2024-09-21:
Thanks! I migrated my configuration to `v4` corresponding with the README.

Unfortunately, none of the three meters work, even the two `qwater` meters which used to work with the latest main before `v4` (`fe7e6162fcb12ceb9d47ab2866ea58a16a30cb32`).

I also encounter very frequent disconnects while observing the log:
```
WARNING water-meter-esp @ 192.168.17.240: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for water-meter-esp @ 192.168.17.240
WARNING Disconnected from API
WARNING Can't connect to ESPHome API for water-meter-esp @ 192.168.17.240: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.17.240', port=6053))]: [Errno 111] Connect call failed ('192.168.17.240', 6053) (SocketAPIError)
INFO Trying to connect to water-meter-esp @ 192.168.17.240 in the background
```

Somet

SzczepanLeon, 2024-09-21:
Please open new issue with detailed I formation. What HW, what ESPHome version, serial logs (VERBOSE).