SzczepanLeon/esphome-components #113: Apator08, (TPL) unknown CI field [A0]

solarssk, 2024-08-15:
Hey!
I have a little issue with my meter. 

Bellow configuration:
```
esphome:
  name: "XXX"
  comment: "XXX"
  area: "XXX"  

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
#  level: DEBUG  

web_server:
  port: 80
  local: true  

# Enable Home Assistant API
api:
  encryption:
    key: "XXX"

ota:
  - platform: esphome
    password: "XXX"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  use_address: XXX

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "XXX Fallback Hotspot"
    password: "XXX"

captive_portal:

one_wire:
  - platform: gpio
    pin: GPIO17

sensor:
  - platform: dallas_temp
    name: "Rack temperature"
    address: "0x0d0000006b11b828"
    update_interval: 120s  
  - platform: wmbus
    meter_id: "XXX"
    add_prefix: false
    type: apator162
    key: "00000000000000000000000000000000"
    lqi:
     name: "Cold water link quality"
     id: "cold_water_meter_lqi"
     entity_category: "diagnostic"
     unit_of_measurement: "lqi"
     state_class: "measurement"
    rssi:
     name: "Cold water signal strength"
     id: "cold_water_meter_rssi"
     unit_of_measurement: "dbm"
     entity_category: "diagnostic"
    total_water_m3:
     name: "Cold water meter"
     id: "cold_water_meter_m3"
     unit_of_measurement: "m³"
     state_class: total_increasing
     device_class: "water"
     accuracy_decimals: 3
     icon: "mdi:counter"
     filters:
      - offset: -99.07
  - platform: wmbus
    meter_id: "0x0005794e"
    add_prefix: false
    type: apator08
    key: "00000000000000000000000000000000"
    lqi:
     name: "Hot water link quality"
     id: "hot_water_meter_lqi"
     entity_category: "diagnostic"
     unit_of_measurement: "lqi"
     state_class: "measurement"
    rssi:
     name: "Hot water signal strength"
     id: "hot_water_meter_rssi"
     unit_of_measurement: "dbm"
     entity_category: "diagnostic"
    total_water_m3:
     name: "Hot water meter"
     id: "hot_water_meter_m3"
     unit_of_measurement: "m³"
     state_class: total_increasing
     device_class: "water"
     accuracy_decimals: 3
     icon: "mdi:counter"    

time:
  - platform: sntp
    id: time_sntp

external_components:
  - source: github://SzczepanLeon/esphome-components@main
    components: [ wmbus ]

wmbus:
#  frequency: 434.475
  mosi_pin: GPIO32
  clk_pin: GPIO33
  miso_pin: GPIO19
  gdo2_pin: GPIO21
  gdo0_pin: GPIO22
  cs_pin: GPIO23

#  led_pin: GPIO0
#  led_blink_time: "1s"
  log_unknown: "true"  

#  clients:
#    - name: "wmbusmeters"
#      ip_address: "10.0.0.1"
#      port: 7227    
```

and  my issue

```
[18:05:57][D][mbus:034]: Processing T1 A frame
[18:05:57][D][wmbus:097]: Using driver 'apator08' for ID [0x0005794E] RSSI: -64 dBm LQI: 128 Frame: T1 A T: 734414864E7905000303A03E0D1800D8613407E2F20106FF07CA231700C45317003D8117006EA7170079CD1700F4EB1700A40318001B5016000D701600679716003EBC160008F31600030000000037DE1020000000000000242C31080A0512280000000000002C0000000000002F000000000000 (116)
[18:05:57][E][utils:214]: (TPL) unknown CI field [A0]
```

with apator162 there is no problems.

Do you have any idea what can I preform to make this parameters available ?

Thanks!

SzczepanLeon, 2024-08-15:
Please remove key from apator08 section.

solarssk, 2024-08-15:
Thanks! Now it works :) 

have a nice day!

closed 2024-08-15 by solarssk