SzczepanLeon/esphome-components #115: ESP bootloops

bladecell, 2024-09-09:
I am trying to make the watermeter, but my ESP bootloops. I have double checked the CC1101 wiring, but I am not sure if that could cause it. I am looking for a direction to look for to find the issue. I am attaching the config yaml, ESP home log and serial log from esp.

<details>
  <summary>yaml file</summary>
  
  ```yaml
substitutions:
  log_level: "VERBOSE"    
  log_wmbus: "VERBOSE"    
  log_baudrate: "115200"  

esphome:
  name: wassermeter
  comment: "Wasserzähler ESP32, CUL - CC1101"
  friendly_name: Wassermeter

preferences:
  flash_write_interval: 10min

esp32:
  board: esp32dev
  framework:
    type: arduino

logger:
  id: appslogger
  level: ${log_level}
  baud_rate: ${log_baudrate}
  logs:
    wmbus: ${log_wmbus}
    wMBus-lib: ${log_wmbus}
    sensor: WARN
    sensor.filter: WARN
    text_sensor: WARN
    api.service: ERROR
    json: ERROR
    mqtt: WARN
    scheduler: ERROR
    internal_temperature: ERROR
    wifi: WARN
    component: ERROR
    api: WARN

api:
  encryption:
    key: !secret api

ota:
  - platform: esphome
    password: !secret ota

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  manual_ip:
    static_ip: 192.168.6.128
    gateway: 192.168.6.1
    subnet: 255.255.255.0

time:
  - platform: sntp
    id: time_sntp

external_components:
  # uses the latest version from SzczepanLeon
  - source: github://SzczepanLeon/esphome-components@main
    refresh: 0s
    components: [wmbus]

[... 143 lines cut ...]
========================= [SUCCESS] Took 5.89 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.6.128
INFO Uploading /data/build/wassermeter/.pioenvs/wassermeter/firmware.bin (978960 bytes)
Uploading: [============================================================] 100% Done...

INFO Upload took 9.30 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.6.128 using esphome API
INFO Successfully connected to wassermeter @ 192.168.6.128 in 7.143s
WARNING Can't connect to ESPHome API for wassermeter @ 192.168.6.128: Handshake timed out after 30.0s (TimeoutAPIError)
INFO Trying to connect to wassermeter @ 192.168.6.128 in the background
WARNING wassermeter @ 192.168.6.128: Connection error occurred: wassermeter @ 192.168.6.128: Connection lost
INFO Successfully connected to wassermeter @ 192.168.6.128 in 7.237s
WARNING wassermeter @ 192.168.6.128: Connection error occurred: wassermeter @ 192.168.6.128: Connection lost
INFO Successfully connected to wassermeter @ 192.168.6.128 in 3.196s
WARNING wassermeter @ 192.168.6.128: Connection error occurred: wassermeter @ 192.168.6.128: Connection lost
INFO Successfully connected to wassermeter @ 192.168.6.128 in 0.027s
WARNING wassermeter @ 192.168.6.128: Connection error occurred: wassermeter @ 192.168.6.128: Connection lost
INFO Successfully connected to wassermeter @ 192.168.6.128 in 15.226s
WARNING wassermeter @ 192.168.6.128: Connection error occurred: wassermeter @ 192.168.6.128: Connection lost
  ```
  
</details>

szymon3, 2024-09-14:
I can confirm the same issue on my end using following pins connection
[image]

ivoh95, 2024-09-15:
I can confirm the same issue with a minimal yaml. ESP bootloops due to WDT being triggered if the connections are correct, if the cc1101 is not connected it marks the component as failed and does not trigger the wdt. 

Some interesting observations, the first boot after applying power gives the following logs:
```
[23:19:21][D][rxLoop:031]: Set CC1101 frequency to 0.001MHz [21 09 5A]
[23:19:21][E][rxLoop:049]: CC1101 initialization FAILED!
[23:19:21][E][component:119]: Component wmbus was marked as failed.
[23:19:21][E][component:164]: Component wmbus set Error flag: unspecified
[23:19:21][E][wmbus:021]: CC1101 initialization failed.

```

Any reboots after first applying power trigger the WDT at the stage of setting frequency. It seems the frequency may be being set incorrectly which locks up the cc1101 module and causes the WDT to trigger on any subsequent communication? Is the frequency actually being set incorrectly to 0.001MHz?

Yaml used:
```
esphome:
  name: esp-water
  friendly_name: esp-water

esp32:
  board: esp32dev
  framework:
    type: arduino

logger:
  level: VERBOSE
ota:
  platform: esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

time:
  - platform: sntp
    id: time_sntp

external_components:
  - source: github://SzczepanLeon/esphome-components@main
    components: [ wmbus]

wmbus:
  frequency: 868.950
  mosi_pin: GPIO13 
  miso_pin: GPIO5 
  clk_pin:  GPIO2 
  cs_pin:   GPIO14 
  gdo0_pin: GPIO15 
  gdo2_pin: GPIO16

```

bladecell, 2024-09-17:
My problem was deffective CC1101 board, new one works fine

ivoh95, 2024-09-17:
> My problem was deffective CC1101 board, new one works fine

I'm suspecting the same issue on my end. I've ordered a new one from a different vendor to confirm. Thanks for following up!

szymon3, 2024-09-17:
Me too, awaiting delivery

closed 2024-09-20 by SzczepanLeon