SzczepanLeon/esphome-components #117: Error while building version 4.0

Maciek99, 2024-09-21:
```
src/esphome/components/wmbus/Telegram.cpp: In member function 'bool Telegram::parseELL(std::vector<unsigned char>::iterator&)':
src/esphome/components/wmbus/Telegram.cpp:1215:10: warning: statement has no effect [-Wunused-value]
 1215 |         ("ELL V not yet handled");
      |         ~^~~~~~~~~~~~~~~~~~~~~~~~
Compiling .pioenvs/apator162_odczyt_wody/src/esphome/components/wmbus/formula.cpp.o
src/esphome/components/wmbus/driver_apator162.cpp:45:5: warning: '{anonymous}::Driver::Driver(MeterInfo&, DriverInfo&)' defined but not used [-Wunused-function]
   45 |     Driver::Driver(MeterInfo &mi, DriverInfo &di) : MeterCommonImplementation(mi, di)
      |     ^~~~~~
Compiling .pioenvs/apator162_odczyt_wody/src/esphome/components/wmbus/manufacturer_specificities.cpp.o
Compiling .pioenvs/apator162_odczyt_wody/src/esphome/components/wmbus/mbus.cpp.o
Compiling .pioenvs/apator162_odczyt_wody/src/esphome/components/wmbus/meters.cpp.o
Compiling .pioenvs/apator162_odczyt_wody/src/esphome/components/wmbus/rf_cc1101.cpp.o
Compiling .pioenvs/apator162_odczyt_wody/src/esphome/components/wmbus/translatebits.cpp.o
Compiling .pioenvs/apator162_odczyt_wody/src/esphome/components/wmbus/types.cpp.o
Compiling .pioenvs/apator162_odczyt_wody/src/esphome/components/wmbus/units.cpp.o
src/esphome/components/wmbus/meters.cpp: In member function 'virtual std::string MeterCommonImplementation::unixTimestampOfUpdate()':
src/esphome/components/wmbus/meters.cpp:582:37: warning: format '%lu' expects argument of type 'long unsigned int', but argument 4 has type 'time_t' {aka 'long long int'} [-Wformat=]
  582 |     snprintf(ut, sizeof(ut) - 1, "%lu", datetime_of_update_);
      |                                   ~~^   ~~~~~~~~~~~~~~~~~~~
      |                                     |   |
      |                                     |   time_t {aka long long int}
      |                                     long unsigned int
      |                                   %llu
Compiling .pioenvs/apator162_odczyt_wody/src/esphome/components/wmbus/utils.cpp.o
Compiling .pioenvs/apator162_odczyt_wody/src/esphome/components/wmbus/utils_my.cpp.o
src/esphome/components/wmbus/utils.cpp: In function 'double addMonths(double, int)':
src/esphome/components/wmbus/utils.cpp:615:15: error: aggregate 'tm time' has incomplete type and cannot be defined
  615 |     struct tm time;
      |               ^~~~
src/esphome/components/wmbus/utils.cpp:616:5: error: 'localtime_r' was not declared in this scope; did you mean 'locale_t'?
  616 |     localtime_r( &ut,&time);
      |     ^~~~~~~~~~~
      |     locale_t
src/esphome/components/wmbus/utils.cpp:618:20: error: 'mktime' was not declared in this scope; did you mean 'mktemp'?
  618 |     return (double)mktime(&time);
      |                    ^~~~~~
      |                    mktemp
src/esphome/components/wmbus/utils.cpp: In function 'void addMonths(tm*, int)':
src/esphome/components/wmbus/utils.cpp:623:37: error: invalid use of incomplete type 'struct tm'
  623 |     bool is_last_day_in_month = date->tm_mday == get_days_in_month(date->tm_year, date->tm_mon);
      |                                     ^~
In file included from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/cwchar:44,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/postypes.h:40,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/char_traits.h:40,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/string:40,
                 from src/esphome/core/log.h:5,
                 from src/esphome/components/wmbus/utils.h:6,
                 from src/esphome/components/wmbus/utils.cpp:5:
/data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/wchar.h:82:8: note: forward declaration of 'struct tm'
   82 | struct tm;
      |        ^~
src/esphome/components/wmbus/utils.cpp:623:72: error: invalid use of incomplete type 'struct tm'
  623 |     bool is_last_day_in_month = date->tm_mday == get_days_in_month(date->tm_year, date->tm_mon);
      |                                                                        ^~
In file included from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/cwchar:44,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/postypes.h:40,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/char_traits.h:40,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/string:40,
                 from src/esphome/core/log.h:5,
                 from src/esphome/components/wmbus/utils.h:6,
                 from src/esphome/components/wmbus/utils.cpp:5:
/data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/wchar.h:82:8: note: forward declaration of 'struct tm'
   82 | struct tm;
      |        ^~
src/esphome/components/wmbus/utils.cpp:623:87: error: invalid use of incomplete type 'struct tm'
  623 |     bool is_last_day_in_month = date->tm_mday == get_days_in_month(date->tm_year, date->tm_mon);
      |                                                                                       ^~
In file included from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/cwchar:44,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/postypes.h:40,
[... 134 lines cut ...]
*** [.pioenvs/apator162_odczyt_wody/src/esphome/components/wmbus/utils.cpp.o] Error 1
In file included from src/esphome/components/web_server_base/web_server_base.h:11,
                 from src/esphome/components/captive_portal/captive_portal.h:11,
                 from src/esphome/components/wmbus/wmbus.cpp:11:
.piolibdeps/apator162_odczyt_wody/ESPAsyncWebServer-esphome/src/ESPAsyncWebServer.h:280:68: warning: 'warning_x' attribute directive ignored [-Wattributes]
  280 |     const String& ASYNCWEBSERVER_REGEX_ATTRIBUTE pathArg(size_t i) const;
      |                                                                    ^~~~~
src/esphome/components/wmbus/wmbus.cpp:14:25: error: expected constructor, destructor, or type conversion before '(' token
   14 | SET_LOOP_TASK_STACK_SIZE(16 * 1024);
      |                         ^
In file included from src/esphome/components/wmbus/wmbus.h:3,
                 from src/esphome/components/wmbus/wmbus.cpp:1:
src/esphome/components/wmbus/wmbus.cpp: In member function 'virtual void esphome::wmbus::WMBusComponent::dump_config()':
src/esphome/components/wmbus/wmbus.cpp:314:43: error: 'class EspClass' has no member named 'getEfuseMac'
  314 |     ESP_LOGCONFIG(TAG, "  MAC: %08X", ESP.getEfuseMac());
      |                                           ^~~~~~~~~~~
src/esphome/core/log.h:97:90: note: in definition of macro 'esph_log_config'
   97 |   esp_log_printf_(ESPHOME_LOG_LEVEL_CONFIG, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
      |                                                                                          ^~~~~~~~~~~
src/esphome/components/wmbus/wmbus.cpp:314:5: note: in expansion of macro 'ESP_LOGCONFIG'
  314 |     ESP_LOGCONFIG(TAG, "  MAC: %08X", ESP.getEfuseMac());
      |     ^~~~~~~~~~~~~
*** [.pioenvs/apator162_odczyt_wody/src/esphome/components/wmbus/wmbus.cpp.o] Error 1
========================= [FAILED] Took 36.95 seconds =========================
```

SzczepanLeon, 2024-09-21:
More info needed. What ESPHome version?

Maciek99, 2024-09-21:
2024.9.0

Maciek99, 2024-09-21:
Here is my whole config
```yaml
substitutions:
  name: apator162_odczyt_wody
  friendly_name: Apator162 Odczyt Wody 

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.9.0
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: dev

esp8266:
  board: esp01_1m

# Enable logging
logger:
  level: VERBOSE
  

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
- platform: esphome

# Allow provisioning Wi-Fi via serial
improv_serial:

wifi:
  networks: 
    - ssid: !secret wifi_ssid
      password: !secret wifi_password
      priority: 2
    - ssid: !secret wifi2_ssid
      password: !secret wifi2_password
      priority: 5
  manual_ip:
    static_ip: 192.168.1.118
    gateway: 192.168.1.1
    subnet: 255.255.255.0
  ap:
    ssid: "Apator162-Odczyt-Wody-ESPHome"
    password: !secret ap_password

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

dashboard_import:
  package_import_url: github://esphome/example-configs/esphome-web/esp8266.yaml@main
  import_full_config: true

web_server:
  version: 3

time:
  - platform: homeassistant
    id: homeassistant_time

external_components:
  - source: github://SzczepanLeon/esphome-components@main
    refresh: 0d
    components: [ wmbus ]

button:
  - platform: restart
    name: "${friendly_name} Reset"
    id: apator162restart

  
wmbus:
  mosi_pin: GPIO13
  miso_pin: GPIO12
  clk_pin:  GPIO14
  cs_pin:   GPIO15
  gdo0_pin: GPIO5
  gdo2_pin: GPIO4

  sync_mode: true
  log_all: True

sensor:
  - platform: wmbus
    meter_id: <meter ID here>
    type: apator162
    key: "00000000000000000000000000000000"
    sensors:
      - name: "LQI"
        field: "lqi"
      - name: "RSSI"
        field: "rssi"
      - name: "Odczyt Wody"
        field: "total"
        accuracy_decimals: 5
```

mykindofit, 2024-09-21:
I have similar problems with esphome 2024.9.0:

`INFO ESPHome 2024.9.0
INFO Reading configuration /config/esphome/water-meter.yaml...
INFO Updating https://github.com/SzczepanLeon/esphome-components.git@main
INFO Detected timezone 'Europe/Warsaw'
INFO Generating C++ source...
INFO Compiling app...
Processing water-meter (board: d1_mini_pro; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 16MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.6
|-- ArduinoJson @ 6.18.5
|-- SPI @ 1.0
|-- SmartRC-CC1101-Driver-Lib @ 2.5.7
Compiling .pioenvs/water-meter/src/esphome/components/wmbus/utils.cpp.o
Compiling .pioenvs/water-meter/src/esphome/components/wmbus/wmbus.cpp.o
src/esphome/components/wmbus/utils.cpp: In function 'double addMonths(double, int)':
src/esphome/components/wmbus/utils.cpp:615:15: error: aggregate 'tm time' has incomplete type and cannot be defined
  615 |     struct tm time;
      |               ^~~~
src/esphome/components/wmbus/utils.cpp:616:5: error: 'localtime_r' was not declared in this scope; did you mean 'locale_t'?
  616 |     localtime_r( &ut,&time);
      |     ^~~~~~~~~~~
      |     locale_t
src/esphome/components/wmbus/utils.cpp:618:20: error: 'mktime' was not declared in this scope; did you mean 'mktemp'?
  618 |     return (double)mktime(&time);
      |                    ^~~~~~
      |                    mktemp
src/esphome/components/wmbus/utils.cpp: In function 'void addMonths(tm*, int)':
src/esphome/components/wmbus/utils.cpp:623:37: error: invalid use of incomplete type 'struct tm'
  623 |     bool is_last_day_in_month = date->tm_mday == get_days_in_month(date->tm_year, date->tm_mon);
      |                                     ^~
In file included from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/cwchar:44,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/postypes.h:40,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/char_traits.h:40,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/string:40,
                 from src/esphome/core/log.h:5,
                 from src/esphome/components/wmbus/utils.h:6,
                 from src/esphome/components/wmbus/utils.cpp:5:
/data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/wchar.h:82:8: note: forward declaration of 'struct tm'
   82 | struct tm;
      |        ^~
src/esphome/components/wmbus/utils.cpp:623:72: error: invalid use of incomplete type 'struct tm'
  623 |     bool is_last_day_in_month = date->tm_mday == get_days_in_month(date->tm_year, date->tm_mon);
      |                                                                        ^~
In file included from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/cwchar:44,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/postypes.h:40,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/char_traits.h:40,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/string:40,
                 from src/esphome/core/log.h:5,
                 from src/esphome/components/wmbus/utils.h:6,
                 from src/esphome/components/wmbus/utils.cpp:5:
/data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/wchar.h:82:8: note: forward declaration of 'struct tm'
   82 | struct tm;
      |        ^~
src/esphome/components/wmbus/utils.cpp:623:87: error: invalid use of incomplete type 'struct tm'
  623 |     bool is_last_day_in_month = date->tm_mday == get_days_in_month(date->tm_year, date->tm_mon);
      |                                                                                       ^~
In file included from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/cwchar:44,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/postypes.h:40,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/char_traits.h:40,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/string:40,
                 from src/esphome/core/log.h:5,
[... 146 lines cut ...]
src/esphome/core/log.h:97:90: note: in definition of macro 'esph_log_config'
   97 |   esp_log_printf_(ESPHOME_LOG_LEVEL_CONFIG, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
      |                                                                                          ^~~~~~~~~~~
src/esphome/components/wmbus/wmbus.cpp:314:5: note: in expansion of macro 'ESP_LOGCONFIG'
  314 |     ESP_LOGCONFIG(TAG, "  MAC: %08X", ESP.getEfuseMac());
      |     ^~~~~~~~~~~~~
*** [.pioenvs/water-meter/src/esphome/components/wmbus/wmbus.cpp.o] Error 1
========================= [FAILED] Took 132.18 seconds =========================
 `

I have sensors declared as in this example:
`sensor:       
  - platform: wmbus
    meter_id: 0x24032111
    type: bmeters
    key: "00000000000000000000000000000000"
    sensors:
      - name: "Cold water usage"
        id: "cold_water"
        field: "total"
        accuracy_decimals: 2
        unit_of_measurement: "m3"
        device_class: "water"
        state_class: "total_increasing"
        icon: "mdi:water"`

mikroprofesor, 2024-09-21:
I've observed same problem, version 4.0.0 and ESP9266 (D1 mini).

thecem, 2024-09-23:
D1 Mini with 3.2.3 working, but since 4.0.0 and 4.0.1 not:

<details>

<summary>Log</summary>

```
Compiling .pioenvs/aa28-heizung-sensoren-esphome/src/esphome/components/wmbus/dvparser.cpp.o
src/esphome/components/wmbus/Telegram.cpp: In member function 'bool Telegram::parseELL(std::vector<unsigned char>::iterator&)':
src/esphome/components/wmbus/Telegram.cpp:1215:10: warning: statement has no effect [-Wunused-value]
 1215 |         ("ELL V not yet handled");
      |         ~^~~~~~~~~~~~~~~~~~~~~~~~
Compiling .pioenvs/aa28-heizung-sensoren-esphome/src/esphome/components/wmbus/formula.cpp.o
Compiling .pioenvs/aa28-heizung-sensoren-esphome/src/esphome/components/wmbus/manufacturer_specificities.cpp.o
Compiling .pioenvs/aa28-heizung-sensoren-esphome/src/esphome/components/wmbus/mbus.cpp.o
Compiling .pioenvs/aa28-heizung-sensoren-esphome/src/esphome/components/wmbus/meters.cpp.o
Compiling .pioenvs/aa28-heizung-sensoren-esphome/src/esphome/components/wmbus/rf_cc1101.cpp.o
Compiling .pioenvs/aa28-heizung-sensoren-esphome/src/esphome/components/wmbus/translatebits.cpp.o
Compiling .pioenvs/aa28-heizung-sensoren-esphome/src/esphome/components/wmbus/types.cpp.o
src/esphome/components/wmbus/meters.cpp: In member function 'virtual std::string MeterCommonImplementation::unixTimestampOfUpdate()':
src/esphome/components/wmbus/meters.cpp:582:37: warning: format '%lu' expects argument of type 'long unsigned int', but argument 4 has type 'time_t' {aka 'long long int'} [-Wformat=]
  582 |     snprintf(ut, sizeof(ut) - 1, "%lu", datetime_of_update_);
      |                                   ~~^   ~~~~~~~~~~~~~~~~~~~
      |                                     |   |
      |                                     |   time_t {aka long long int}
      |                                     long unsigned int
      |                                   %llu
Compiling .pioenvs/aa28-heizung-sensoren-esphome/src/esphome/components/wmbus/units.cpp.o
Compiling .pioenvs/aa28-heizung-sensoren-esphome/src/esphome/components/wmbus/utils.cpp.o
Compiling .pioenvs/aa28-heizung-sensoren-esphome/src/esphome/components/wmbus/utils_my.cpp.o
src/esphome/components/wmbus/utils.cpp: In function 'double addMonths(double, int)':
src/esphome/components/wmbus/utils.cpp:615:15: error: aggregate 'tm time' has incomplete type and cannot be defined
  615 |     struct tm time;
      |               ^~~~
src/esphome/components/wmbus/utils.cpp:616:5: error: 'localtime_r' was not declared in this scope; did you mean 'locale_t'?
  616 |     localtime_r( &ut,&time);
      |     ^~~~~~~~~~~
      |     locale_t
src/esphome/components/wmbus/utils.cpp:618:20: error: 'mktime' was not declared in this scope; did you mean 'mktemp'?
  618 |     return (double)mktime(&time);
      |                    ^~~~~~
      |                    mktemp
src/esphome/components/wmbus/utils.cpp: In function 'void addMonths(tm*, int)':
src/esphome/components/wmbus/utils.cpp:623:37: error: invalid use of incomplete type 'struct tm'
  623 |     bool is_last_day_in_month = date->tm_mday == get_days_in_month(date->tm_year, date->tm_mon);
      |                                     ^~
In file included from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/cwchar:44,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/postypes.h:40,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/char_traits.h:40,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/string:40,
                 from src/esphome/core/log.h:5,
                 from src/esphome/components/wmbus/utils.h:6,
                 from src/esphome/components/wmbus/utils.cpp:5:
/data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/wchar.h:82:8: note: forward declaration of 'struct tm'
   82 | struct tm;
      |        ^~
src/esphome/components/wmbus/utils.cpp:623:72: error: invalid use of incomplete type 'struct tm'
  623 |     bool is_last_day_in_month = date->tm_mday == get_days_in_month(date->tm_year, date->tm_mon);
      |                                                                        ^~
In file included from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/cwchar:44,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/postypes.h:40,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/char_traits.h:40,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/string:40,
                 from src/esphome/core/log.h:5,
                 from src/esphome/components/wmbus/utils.h:6,
                 from src/esphome/components/wmbus/utils.cpp:5:
/data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/wchar.h:82:8: note: forward declaration of 'struct tm'
   82 | struct tm;
      |        ^~
src/esphome/components/wmbus/utils.cpp:623:87: error: invalid use of incomplete type 'struct tm'
[... 140 lines cut ...]
                 from src/esphome/components/captive_portal/captive_portal.h:11,
                 from src/esphome/components/wmbus/wmbus.cpp:11:
[... cut ...]
   97 |   esp_log_printf_(ESPHOME_LOG_LEVEL_CONFIG, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
      |                                                                                          ^~~~~~~~~~~
src/esphome/components/wmbus/wmbus.cpp:314:5: note: in expansion of macro 'ESP_LOGCONFIG'
  314 |     ESP_LOGCONFIG(TAG, "  MAC: %08X", ESP.getEfuseMac());
      |     ^~~~~~~~~~~~~
*** [.pioenvs/aa28-heizung-sensoren-esphome/src/esphome/components/wmbus/wmbus.cpp.o] Error 1
========================= [FAILED] Took 35.50 seconds =========================
```

</details>

Mareckil, 2024-09-24:
same here esp8266 Node MCU, esphome 2024.9.1

ozziemandias1987, 2024-09-25:
I have the same warnings/errors as Maciek99 had. ESPHome 2024.9.1, board NodeMCUv2. Works fine with @3.2.3, won't compile with @main as source.

dam08, 2024-09-27:
Hello, same here with Wemos D1 mini and ESPHome 2024.9.1
Compile works in 3.2.3 but not with 4.x.x

Thanks for your help/work ! :-)

`INFO ESPHome 2024.9.1
INFO Reading configuration /config/esphome/compteur-eau.yaml...
INFO Updating https://github.com/SzczepanLeon/esphome-components.git@main
INFO Generating C++ source...
INFO Compiling app...
Processing compteur-eau (board: d1_mini; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- AsyncMqttClient-esphome @ 2.0.0
|-- noise-c @ 0.1.6
|-- ArduinoJson @ 6.18.5
|-- Syslog @ 2.0.0
|-- SPI @ 1.0
|-- SmartRC-CC1101-Driver-Lib @ 2.5.7
Compiling .pioenvs/compteur-eau/src/esphome/components/wmbus/utils.cpp.o
Compiling .pioenvs/compteur-eau/src/esphome/components/wmbus/utils_my.cpp.o
Compiling .pioenvs/compteur-eau/src/esphome/components/wmbus/wmbus.cpp.o
src/esphome/components/wmbus/utils.cpp: In function 'double addMonths(double, int)':
src/esphome/components/wmbus/utils.cpp:615:15: error: aggregate 'tm time' has incomplete type and cannot be defined
  615 |     struct tm time;
      |               ^~~~
src/esphome/components/wmbus/utils.cpp:616:5: error: 'localtime_r' was not declared in this scope; did you mean 'locale_t'?
  616 |     localtime_r( &ut,&time);
      |     ^~~~~~~~~~~
      |     locale_t
src/esphome/components/wmbus/utils.cpp:618:20: error: 'mktime' was not declared in this scope; did you mean 'mktemp'?
  618 |     return (double)mktime(&time);
      |                    ^~~~~~
      |                    mktemp
src/esphome/components/wmbus/utils.cpp: In function 'void addMonths(tm*, int)':
src/esphome/components/wmbus/utils.cpp:623:37: error: invalid use of incomplete type 'struct tm'
  623 |     bool is_last_day_in_month = date->tm_mday == get_days_in_month(date->tm_year, date->tm_mon);
      |                                     ^~
In file included from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/cwchar:44,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/postypes.h:40,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/char_traits.h:40,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/string:40,
                 from src/esphome/core/log.h:5,
                 from src/esphome/components/wmbus/utils.h:6,
                 from src/esphome/components/wmbus/utils.cpp:5:
/data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/wchar.h:82:8: note: forward declaration of 'struct tm'
   82 | struct tm;
      |        ^~
src/esphome/components/wmbus/utils.cpp:623:72: error: invalid use of incomplete type 'struct tm'
  623 |     bool is_last_day_in_month = date->tm_mday == get_days_in_month(date->tm_year, date->tm_mon);
      |                                                                        ^~
In file included from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/cwchar:44,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/postypes.h:40,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/char_traits.h:40,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/string:40,
                 from src/esphome/core/log.h:5,
                 from src/esphome/components/wmbus/utils.h:6,
                 from src/esphome/components/wmbus/utils.cpp:5:
/data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/wchar.h:82:8: note: forward declaration of 'struct tm'
   82 | struct tm;
      |        ^~
src/esphome/components/wmbus/utils.cpp:623:87: error: invalid use of incomplete type 'struct tm'
  623 |     bool is_last_day_in_month = date->tm_mday == get_days_in_month(date->tm_year, date->tm_mon);
      |                                                                                       ^~
[... 134 lines cut ...]
      |     ^~~~~~~~
*** [.pioenvs/compteur-eau/src/esphome/components/wmbus/utils.cpp.o] Error 1
In file included from src/esphome/components/web_server_base/web_server_base.h:11,
                 from src/esphome/components/captive_portal/captive_portal.h:11,
                 from src/esphome/components/wmbus/wmbus.cpp:11:
.piolibdeps/compteur-eau/ESPAsyncWebServer-esphome/src/ESPAsyncWebServer.h:280:68: warning: 'warning_x' attribute directive ignored [-Wattributes]
  280 |     const String& ASYNCWEBSERVER_REGEX_ATTRIBUTE pathArg(size_t i) const;
      |                                                                    ^~~~~
src/esphome/components/wmbus/wmbus.cpp:14:25: error: expected constructor, destructor, or type conversion before '(' token
   14 | SET_LOOP_TASK_STACK_SIZE(32 * 1024);
      |                         ^
In file included from src/esphome/components/wmbus/wmbus.h:3,
                 from src/esphome/components/wmbus/wmbus.cpp:1:
src/esphome/components/wmbus/wmbus.cpp: In member function 'virtual void esphome::wmbus::WMBusComponent::dump_config()':
[... cut ...]
   97 |   esp_log_printf_(ESPHOME_LOG_LEVEL_CONFIG, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
      |                                                                                          ^~~~~~~~~~~
src/esphome/components/wmbus/wmbus.cpp:316:5: note: in expansion of macro 'ESP_LOGCONFIG'
  316 |     ESP_LOGCONFIG(TAG, "  MAC: %08X", ESP.getEfuseMac());
      |     ^~~~~~~~~~~~~
*** [.pioenvs/compteur-eau/src/esphome/components/wmbus/wmbus.cpp.o] Error 1
========================== [FAILED] Took 4.38 seconds ==========================`

SzczepanLeon, 2024-10-03:
There is small fix in 4.0.8, but probably this is not enough.

LukaszGalos, 2024-10-03:
Just tried to install latest version, now I have other errors
```
INFO ESPHome 2024.9.2
INFO Reading configuration /config/esphome/water.yaml...
INFO Detected timezone 'Europe/Warsaw'
INFO Generating C++ source...
INFO Compiling app...
Processing water (board: nodemcuv2; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.6
|-- SPI @ 1.0
|-- SmartRC-CC1101-Driver-Lib @ 2.5.7
Compiling .pioenvs/water/src/esphome/components/wmbus/Telegram.cpp.o
Compiling .pioenvs/water/src/esphome/components/wmbus/address.cpp.o
src/esphome/components/wmbus/Telegram.cpp: In member function 'bool Telegram::parseELL(std::vector<unsigned char>::iterator&)':
src/esphome/components/wmbus/Telegram.cpp:1215:10: warning: statement has no effect [-Wunused-value]
 1215 |         ("ELL V not yet handled");
      |         ~^~~~~~~~~~~~~~~~~~~~~~~~
Compiling .pioenvs/water/src/esphome/components/wmbus/driver_apator162.cpp.o
src/esphome/components/wmbus/driver_apator162.cpp:45:5: warning: '{anonymous}::Driver::Driver(MeterInfo&, DriverInfo&)' defined but not used [-Wunused-function]
   45 |     Driver::Driver(MeterInfo &mi, DriverInfo &di) : MeterCommonImplementation(mi, di)
      |     ^~~~~~
Compiling .pioenvs/water/src/esphome/components/wmbus/dvparser.cpp.o
Compiling .pioenvs/water/src/esphome/components/wmbus/formula.cpp.o
Compiling .pioenvs/water/src/esphome/components/wmbus/manufacturer_specificities.cpp.o
Compiling .pioenvs/water/src/esphome/components/wmbus/meters.cpp.o
Compiling .pioenvs/water/src/esphome/components/wmbus/translatebits.cpp.o
src/esphome/components/wmbus/meters.cpp: In member function 'virtual std::string MeterCommonImplementation::unixTimestampOfUpdate()':
src/esphome/components/wmbus/meters.cpp:582:37: warning: format '%lu' expects argument of type 'long unsigned int', but argument 4 has type 'time_t' {aka 'long long int'} [-Wformat=]
  582 |     snprintf(ut, sizeof(ut) - 1, "%lu", datetime_of_update_);
      |                                   ~~^   ~~~~~~~~~~~~~~~~~~~
      |                                     |   |
      |                                     |   time_t {aka long long int}
      |                                     long unsigned int
      |                                   %llu
Compiling .pioenvs/water/src/esphome/components/wmbus/units.cpp.o
Compiling .pioenvs/water/src/esphome/components/wmbus/utils.cpp.o
Compiling .pioenvs/water/src/esphome/components/wmbus/wmbus.cpp.o
src/esphome/components/wmbus/wmbus.cpp:19:2: warning: #warning "ToDo - increase loop task stack" [-Wcpp]
   19 | #warning "ToDo - increase loop task stack"
      |  ^~~~~~~
Compiling .pioenvs/water/src/esphome/core/application.cpp.o
In file included from src/esphome/components/web_server_base/web_server_base.h:11,
                 from src/esphome/components/captive_portal/captive_portal.h:11,
                 from src/esphome/components/wmbus/wmbus.cpp:11:
.piolibdeps/water/ESPAsyncWebServer-esphome/src/ESPAsyncWebServer.h:280:68: warning: 'warning_x' attribute directive ignored [-Wattributes]
  280 |     const String& ASYNCWEBSERVER_REGEX_ATTRIBUTE pathArg(size_t i) const;
      |                                                                    ^~~~~
Compiling .pioenvs/water/src/esphome/core/bytebuffer.cpp.o
Compiling .pioenvs/water/src/esphome/core/color.cpp.o
Compiling .pioenvs/water/src/esphome/core/component.cpp.o
Compiling .pioenvs/water/src/esphome/core/component_iterator.cpp.o
Compiling .pioenvs/water/src/esphome/core/controller.cpp.o
Compiling .pioenvs/water/src/esphome/core/entity_base.cpp.o
Compiling .pioenvs/water/src/esphome/core/helpers.cpp.o
Compiling .pioenvs/water/src/esphome/core/log.cpp.o
Compiling .pioenvs/water/src/esphome/core/ring_buffer.cpp.o
Compiling .pioenvs/water/src/esphome/core/scheduler.cpp.o
Compiling .pioenvs/water/src/esphome/core/string_ref.cpp.o
Compiling .pioenvs/water/src/esphome/core/time.cpp.o
Compiling .pioenvs/water/src/esphome/core/util.cpp.o
Compiling .pioenvs/water/src/main.cpp.o
Generating LD script .pioenvs/water/ld/local.eagle.app.v6.common.ld
[... 218 lines cut ...]
Compiling .pioenvs/water/FrameworkArduino/reboot_uart_dwnld.cpp.o
Compiling .pioenvs/water/FrameworkArduino/spiffs/spiffs_cache.cpp.o
Compiling .pioenvs/water/FrameworkArduino/spiffs/spiffs_check.cpp.o
Compiling .pioenvs/water/FrameworkArduino/spiffs/spiffs_gc.cpp.o
Compiling .pioenvs/water/FrameworkArduino/spiffs/spiffs_hydrogen.cpp.o
Compiling .pioenvs/water/FrameworkArduino/spiffs/spiffs_nucleus.cpp.o
Compiling .pioenvs/water/FrameworkArduino/spiffs_api.cpp.o
Compiling .pioenvs/water/FrameworkArduino/sqrt32.cpp.o
Compiling .pioenvs/water/FrameworkArduino/stdlib_noniso.cpp.o
Compiling .pioenvs/water/FrameworkArduino/time.cpp.o
Compiling .pioenvs/water/FrameworkArduino/uart.cpp.o
Compiling .pioenvs/water/FrameworkArduino/umm_malloc/umm_info.c.o
Compiling .pioenvs/water/FrameworkArduino/umm_malloc/umm_integrity.c.o
Compiling .pioenvs/water/FrameworkArduino/umm_malloc/umm_local.c.o
Compiling .pioenvs/water/FrameworkArduino/umm_malloc/umm_malloc.cpp.o
Compiling .pioenvs/water/FrameworkArduino/umm_malloc/umm_poison.c.o
Compiling .pioenvs/water/FrameworkArduino/wpa2_eap_patch.cpp.o
[... cut ...]
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: address 0x40002840 of .pioenvs/water/firmware.elf section `.bss' is not within region `dram0_0_seg'
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: address 0x40002840 of .pioenvs/water/firmware.elf section `.bss' is not within region `dram0_0_seg'
collect2: error: ld returned 1 exit status
*** [.pioenvs/water/firmware.elf] Error 1
========================= [FAILED] Took 72.42 seconds =========================
```

SzczepanLeon, 2024-10-03:
> Just tried to install latest version, now I have other errors

This is expected behaviour. But I don't have time right now  to fix it.

closed 2024-10-03 by SzczepanLeon

brzozow, 2024-10-05:
With a new version I have following error at the end of compilation log:

```
Linking .pioenvs/liczniki/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/liczniki/firmware.elf section `.rodata' will not fit in region `dram0_0_seg'
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: address 0x4001cd18 of .pioenvs/liczniki/firmware.elf section `.bss' is not within region `dram0_0_seg'
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: address 0x4001cd18 of .pioenvs/liczniki/firmware.elf section `.bss' is not within region `dram0_0_seg'
collect2: error: ld returned 1 exit status
*** [.pioenvs/liczniki/firmware.elf] Error 1
```

SzczepanLeon, 2024-10-05:
> With a new version I have following error at the end of compilation log:

Known issue:
https://github.com/SzczepanLeon/esphome-components/issues/131

From 4.0 ESP8266 is not supported.

Mareckil, 2024-10-06:
good to know :(. But ... tough life.

SzczepanLeon, 2024-10-06:
You can stay on v3.x

cenobitedk, 2024-10-10:
> From 4.0 ESP8266 is not supported.

Should be probably be stated in the readme