SzczepanLeon/esphome-components #118: How to correctly set unit_of_measurement for Home Assistant

Flo2410, 2024-09-21:
I'm using the v4.0.0 with the hydrus driver.
To get the total water amount and its temperature I have following config.

```yaml
sensor:
  - platform: wmbus
    type: hydrus
    meter_id: !secret watermeter_id
    key: !secret water_meter_key
    sensors:
      - name: "Total"
        field: "total"
        id: "waterdisplay"
        unit_of_measurement: "m3"
        state_class: total_increasing
        device_class: "water"
        accuracy_decimals: 3
        icon: mdi:counter
      - name: "Water Temerature"
        field: "flow_temperature"
        id: "watertemperature"
        unit_of_measurement: "c"
        state_class: total_increasing
        device_class: "temperature"
        accuracy_decimals: 1
        icon: mdi:water-thermometer
```

This works fine, but I have to use `m3` and `c` as the units instead of `m³` and `°C` how Home Assistant would expect them.
This leads to the issue that I can't use the total water amount in the energy dashboard.

Using `m³` gives following error message in the log:
```
[12:42:37][W][wmbus:130]: Can't get proper unit from 'm³'
```

I assume that the units are used to complete the field name for the decoded values.
But unfortunatly this breaks compatibility with Home Assistant.

```json
{
    "media":"water",
    "meter":"hydrus",
    "name":"",
    "id":"71179486",
    "flow_temperature_c":10.5,
    "remaining_battery_life_y":13.3692,
    "target_m3":35.791,
    "target_datetime":"2024-01-31 23:59",
    "total_m3":37.512,
    "status":"OK",
    "timestamp":"2024-02-13T11:25:04Z"
}
```

SzczepanLeon, 2024-09-22:
Will be fixed soon. For now You can use liters - should be OK.

SzczepanLeon, 2024-09-22:
Please try version **4.0.1**.
It is quick fix - not tested, but maybe it will be enough.

Flo2410, 2024-09-23:
Thx, seems to work.

closed 2024-09-23 by SzczepanLeon