SzczepanLeon/esphome-components #119: After updating to version 4.0, no readings from the sensors.

Mario-W-Git, 2024-09-22:
HI, 
after updating to version 4.0, no readings from the sensors ( Fields without unit not supported yet! )

Logs:
[12:20:10][I][wmbus:085]: izar [0x2140711b] RSSI: -52dBm T: 1944304C1B7140219801A21119001377CF78E40269F95FA8CFF8 (26) T1 A
[12:20:10][D][meters.cpp:1993]: (meter) created ESPHome izar 2140711b not-encrypted
[12:20:10][D][meters.cpp:922]: (meter) ESPHome(0) izar  handling telegram from 2140711b.M=SAP.V=00.T=07
[12:20:10][W][wmbus:116]: Fields without unit not supported yet!

Snippet of my code:

sensor:

  - platform: wmbus
    meter_id: 0x2180611B
    type: izar
    sensors:
      
    - field: "total_water_m3"
      name: "Całkowite zużycie"
      icon: "mdi:water"
      device_class: "water"
      accuracy_decimals: 3

Did I do something incorrectly?

SzczepanLeon, 2024-09-22:
Please remove _m3.from field, and add unit of measurement as m3 (or l).

Mario-W-Git, 2024-09-22:
I changed it to something like this but it still doesn't work.
In the logs : [20:00:58][W][wmbus:126]: Can't get requested field 'total_water' with unit 'm3'

`sensor:

  - platform: wmbus
    meter_id: 0x2180611B
    type: izar
    sensors:
      
    - field: "total_water"
      name: "Całkowite zużycie"
      icon: "mdi:water"
      device_class: "water"
      accuracy_decimals: 3
      unit_of_measurement: "m3"`
 
Logs:
  
level: VERY_VERBOSE

`[20:00:58][V][rxLoop:167]: Have 45 bytes from CC1101 Rx, RSSI: -52 dBm LQI: 128
[20:00:58][V][rxLoop:172]: Synchronus mode enabled.
[20:00:58][D][mbus:035]: Received T1 A frame
[20:00:58][V][mbus:046]: Frame: 36571C2D67343634CD71638D96C58DA69C5A98E38D36559634B4D37195A52F2A695A3B312D9D19C9CA698DAC9A (45) [RAW]
[20:00:58][VV][3of6:095]: Decode 3 out of 6 OK.
[20:00:58][V][mbus:053]: Frame: 1944304C1B7140219801FFD6A2211900137745093EFF0B8D35C5E4FFB6E6 (30) [with CRC]
[20:00:58][V][mbus:096]: Validating CRC for Block1
[20:00:58][V][crc:031]:     calculated: 0xFFD6, read: 0xFFD6
[20:00:58][V][mbus:116]: Validating CRC for Block2
[20:00:58][V][crc:031]:     calculated: 0xB6E6, read: 0xB6E6
[20:00:58][V][mbus:063]: Frame: 1944304C1B7140219801A2211900137745093EFF0B8D35C5E4FF (26) [without CRC]
[20:00:58][VV][wmbus:050]: Have data from RF ...
[20:00:58][V][Telegram.cpp:2136]: (diehl) preprocess necessary SAP_PRIOS
[20:00:58][V][manufacturer_specificities.cpp:156]: (diehl) Pre-processing: setting device type to water meter for SAP PRIOS
[20:00:58][V][Telegram.cpp:1131]: (wmbus) parseDLL @0 26
[20:00:58][V][Telegram.cpp:1178]: (wmbus) parseELL @10 16
[20:00:58][V][Telegram.cpp:1333]: (wmbus) parseNWL @10 16
[20:00:58][V][Telegram.cpp:1391]: (wmbus) parseAFL @10 16
[20:00:58][V][Telegram.cpp:2078]: (wmbus) parseTPL @10 16
[20:00:58][I][wmbus:085]: izar [0x2140711b] RSSI: -52dBm T: 1944304C1B7140219801A2211900137745093EFF0B8D35C5E4FF (26) T1 A
[20:00:58][D][meters.cpp:1993]: (meter) created ESPHome izar 2140711b not-encrypted
[20:00:58][V][Telegram.cpp:2136]: (diehl) preprocess necessary SAP_PRIOS
[20:00:58][V][manufacturer_specificities.cpp:156]: (diehl) Pre-processing: setting device type to water meter for SAP PRIOS
[20:00:58][V][Telegram.cpp:1131]: (wmbus) parseDLL @0 26
[20:00:58][V][Telegram.cpp:1178]: (wmbus) parseELL @10 16
[20:00:58][V][Telegram.cpp:1333]: (wmbus) parseNWL @10 16
[20:00:58][V][Telegram.cpp:1391]: (wmbus) parseAFL @10 16
[20:00:58][V][Telegram.cpp:2078]: (wmbus) parseTPL @10 16
[20:00:58][V][meters.cpp:635]: (meter) ESPHome: for me? 2140711b.M=SAP.V=00.T=07,2140711b.M=SAP.V=00.T=07 in 2140711b
[20:00:58][V][meters.cpp:699]: (meter) ESPHome: yes for me
[20:00:58][D][meters.cpp:922]: (meter) ESPHome(0) izar  handling telegram from 2140711b.M=SAP.V=00.T=07
[20:00:58][V][Telegram.cpp:2136]: (diehl) preprocess necessary SAP_PRIOS
[20:00:58][V][manufacturer_specificities.cpp:156]: (diehl) Pre-processing: setting device type to water meter for SAP PRIOS
[20:00:58][V][Telegram.cpp:1131]: (wmbus) parseDLL @0 26
[20:00:58][V][Telegram.cpp:545]: (telegram) DLL L=19 C=44 (from meter SND_NR) M=4c30 (SAP) A=2140711b VER=00 TYPE=07 (Water meter) (driver izar) DEV=ESPHome wM-Bus RSSI=-52
[20:00:58][V][Telegram.cpp:1178]: (wmbus) parseELL @10 16
[20:00:58][V][Telegram.cpp:1333]: (wmbus) parseNWL @10 16
[20:00:58][V][Telegram.cpp:1391]: (wmbus) parseAFL @10 16
[20:00:58][V][Telegram.cpp:2078]: (wmbus) parseTPL @10 16
[20:00:58][V][Telegram.cpp:634]: (telegram) TPL CI=a2
[20:00:58][V][meters.cpp:979]: (meters) skipping field without matcher prefix(Text)[0]...
[20:00:58][V][meters.cpp:979]: (meters) skipping field without matcher serial_number(Text)[1]...
[20:00:58][V][meters.cpp:979]: (meters) skipping field without matcher total(Volume)[2]...
[20:00:58][V][meters.cpp:979]: (meters) skipping field without matcher last_month_total(Volume)[3]...
[20:00:58][V][meters.cpp:979]: (meters) skipping field without matcher last_month_measure_date(Text)[4]...
[20:00:58][V][meters.cpp:979]: (meters) skipping field without matcher remaining_battery_life(Time)[5]...
[20:00:58][V][meters.cpp:979]: (meters) skipping field without matcher current_alarms(Text)[6]...
[20:00:58][V][meters.cpp:979]: (meters) skipping field without matcher previous_alarms(Text)[7]...
[20:00:58][V][meters.cpp:979]: (meters) skipping field without matcher transmit_period(Time)[8]...
[20:00:58][V][meters.cpp:979]: (meters) skipping field without matcher manufacture_year(Text)[9]...
[20:00:58][V][driver_izar.cpp:200]: (izar) Decoded PRIOS data: 4BE7AD05003D9C05001F38
[20:00:58][W][wmbus:126]: Can't get requested field 'total_water' with unit 'm3'
[20:00:58][W][component:237]: Component wmbus took a long time for an operation (396 ms).
[20:00:58][W][component:238]: Components should block for at most 30 ms.`

Could you advise me how to fix it?

Thank You.

Mario-W-Git, 2024-09-22:
I guess I have a problem with the syntax of the code. I tried to add and remove units from fields and add or remove unit_of_measurent: from sensors but without success, the only sensor I managed to read was transmit_period.

- field: "transmit_period"
      name: "Okres transmisji"
      unit_of_measurement: "s"

For other sensors I receive in the logs:

[wmbus:116]: Fields without unit not supported yet!
or
[wmbus:126]: Can't get requested field 'total_water' with unit 'm3

Mario-W-Git, 2024-09-22:
In small steps I managed to read the next sensors that change the field: "total_water" to field: "total"
I will continue with other sensors.

SzczepanLeon, 2024-09-22:
```
{
    "media":"water",
    "meter":"izar",
    "name":"",
    "id":"2140711b",
    "last_month_total_m3":367.677,
    "remaining_battery_life_y":12.5,
    "total_m3":372.199,
    "transmit_period_s":8,
    "current_alarms":"no_alarm",
    "last_month_measure_date":"2024-08-31",
    "manufacture_year":"2021",
    "prefix":"C21FA",
    "previous_alarms":"no_alarm",
    "serial_number":"000475",
    "timestamp":"2024-09-22T20:16:55Z"
}
```

As You can see [above](https://wmbusmeters.org/analyze/1944304C1B7140219801A2211900137745093EFF0B8D35C5E4FF) there is no field "**total_water**". Proper name is "**total**".

closed 2024-09-23 by SzczepanLeon