SzczepanLeon/esphome-components #120: Compile error under VERBOSE logger settings

ThomDietrich, 2024-09-22:
Hey there, thanks for the 4.0 update! I ran into this error and just want to quickly raise it.

With:
```
logger:
  level: VERY_VERBOSE
  logs:
    wmbus: VERY_VERBOSE
    wMBus-lib: DEBUG
```

You get:
```
Building in release mode
Compiling .pioenvs\wasserzaehler\src\esphome\components\wmbus\wmbus.cpp.o
Compiling .pioenvs\wasserzaehler\src\main.cpp.o
Archiving .pioenvs\wasserzaehler\lib599\libFS.a
Archiving .pioenvs\wasserzaehler\lib84a\libUpdate.a
Compiling .pioenvs\wasserzaehler\lib65b\ESPAsyncWebServer-esphome\WebAuthentication.cpp.o
Compiling .pioenvs\wasserzaehler\lib65b\ESPAsyncWebServer-esphome\WebHandlers.cpp.o
Compiling .pioenvs\wasserzaehler\lib65b\ESPAsyncWebServer-esphome\WebRequest.cpp.o
Compiling .pioenvs\wasserzaehler\lib65b\ESPAsyncWebServer-esphome\WebResponses.cpp.o
Compiling .pioenvs\wasserzaehler\lib65b\ESPAsyncWebServer-esphome\WebServer.cpp.o
Compiling .pioenvs\wasserzaehler\liba8b\DNSServer\DNSServer.cpp.o
Compiling .pioenvs\wasserzaehler\lib74b\ESPmDNS\ESPmDNS.cpp.o
Compiling .pioenvs\wasserzaehler\libd3c\SPI\SPI.cpp.o
Compiling .pioenvs\wasserzaehler\liba9c\SmartRC-CC1101-Driver-Lib\ELECHOUSE_CC1101_SRC_DRV.cpp.o
Compiling .pioenvs\wasserzaehler\FrameworkArduino\Esp.cpp.o
Archiving .pioenvs\wasserzaehler\libd3c\libSPI.a
Compiling .pioenvs\wasserzaehler\FrameworkArduino\FirmwareMSC.cpp.o
Compiling .pioenvs\wasserzaehler\FrameworkArduino\FunctionalInterrupt.cpp.o
Compiling .pioenvs\wasserzaehler\FrameworkArduino\HWCDC.cpp.o
In file included from src/esphome/components/wmbus/dvparser.h:23,
                 from src/esphome/components/wmbus/meters.h:22,
                 from src/esphome/components/wmbus/wmbus.cpp:4:
src/esphome/components/wmbus/units.h:102:7: error: expected identifier before '(' token
     X(HZ,hz,"Hz",Frequency,"hz")                                        \
       ^~
src/esphome/components/wmbus/units.h:129:53: note: in definition of macro 'X'
 #define X(cname,lcname,hrname,quantity,explanation) cname,
                                                     ^~~~~
src/esphome/components/wmbus/units.h:130:5: note: in expansion of macro 'LIST_OF_UNITS'
     LIST_OF_UNITS
     ^~~~~~~~~~~~~
src/esphome/components/wmbus/units.h:102:7: error: expected '}' before '(' token
     X(HZ,hz,"Hz",Frequency,"hz")                                        \
       ^~
src/esphome/components/wmbus/units.h:129:53: note: in definition of macro 'X'
 #define X(cname,lcname,hrname,quantity,explanation) cname,
                                                     ^~~~~
src/esphome/components/wmbus/units.h:130:5: note: in expansion of macro 'LIST_OF_UNITS'
     LIST_OF_UNITS
     ^~~~~~~~~~~~~
src/esphome/components/wmbus/units.h:128:1: note: to match this '{'
 {
 ^
src/esphome/components/wmbus/units.h:102:7: error: expected unqualified-id before numeric constant
     X(HZ,hz,"Hz",Frequency,"hz")                                        \
       ^~
src/esphome/components/wmbus/units.h:129:53: note: in definition of macro 'X'
 #define X(cname,lcname,hrname,quantity,explanation) cname,
                                                     ^~~~~
src/esphome/components/wmbus/units.h:130:5: note: in expansion of macro 'LIST_OF_UNITS'
     LIST_OF_UNITS
     ^~~~~~~~~~~~~
src/esphome/components/wmbus/units.h:102:7: error: expected ')' before numeric constant
     X(HZ,hz,"Hz",Frequency,"hz")                                        \
       ^~
src/esphome/components/wmbus/units.h:129:53: note: in definition of macro 'X'
 #define X(cname,lcname,hrname,quantity,explanation) cname,
                                                     ^~~~~
src/esphome/components/wmbus/units.h:130:5: note: in expansion of macro 'LIST_OF_UNITS'
     LIST_OF_UNITS
     ^~~~~~~~~~~~~
src/esphome/components/wmbus/units.h:133:1: error: expected declaration before '}' token
 };
 ^
Archiving .pioenvs\wasserzaehler\liba8b\libDNSServer.a
*** [.pioenvs\wasserzaehler\src\esphome\components\wmbus\wmbus.cpp.o] Error 1
In file included from src/esphome\components\wmbus\dvparser.h:23,
                 from src/esphome.h:91,
                 from src/main.cpp:3:
src/esphome\components\wmbus\units.h:102:7: error: expected identifier before '(' token
     X(HZ,hz,"Hz",Frequency,"hz")                                        \
       ^~
src/esphome\components\wmbus\units.h:129:53: note: in definition of macro 'X'
 #define X(cname,lcname,hrname,quantity,explanation) cname,
                                                     ^~~~~
src/esphome\components\wmbus\units.h:130:5: note: in expansion of macro 'LIST_OF_UNITS'
     LIST_OF_UNITS
     ^~~~~~~~~~~~~
src/esphome\components\wmbus\units.h:102:7: error: expected '}' before '(' token
     X(HZ,hz,"Hz",Frequency,"hz")                                        \
       ^~
src/esphome\components\wmbus\units.h:129:53: note: in definition of macro 'X'
 #define X(cname,lcname,hrname,quantity,explanation) cname,
                                                     ^~~~~
src/esphome\components\wmbus\units.h:130:5: note: in expansion of macro 'LIST_OF_UNITS'
     LIST_OF_UNITS
     ^~~~~~~~~~~~~
src/esphome\components\wmbus\units.h:128:1: note: to match this '{'
 {
 ^
src/esphome\components\wmbus\units.h:102:7: error: expected unqualified-id before numeric constant
     X(HZ,hz,"Hz",Frequency,"hz")                                        \
       ^~
src/esphome\components\wmbus\units.h:129:53: note: in definition of macro 'X'
 #define X(cname,lcname,hrname,quantity,explanation) cname,
                                                     ^~~~~
src/esphome\components\wmbus\units.h:130:5: note: in expansion of macro 'LIST_OF_UNITS'
     LIST_OF_UNITS
     ^~~~~~~~~~~~~
src/esphome\components\wmbus\units.h:102:7: error: expected ')' before numeric constant
     X(HZ,hz,"Hz",Frequency,"hz")                                        \
       ^~
src/esphome\components\wmbus\units.h:129:53: note: in definition of macro 'X'
 #define X(cname,lcname,hrname,quantity,explanation) cname,
                                                     ^~~~~
src/esphome\components\wmbus\units.h:130:5: note: in expansion of macro 'LIST_OF_UNITS'
     LIST_OF_UNITS
     ^~~~~~~~~~~~~
src/esphome\components\wmbus\units.h:133:1: error: expected declaration before '}' token
 };
 ^
*** [.pioenvs\wasserzaehler\src\main.cpp.o] Error 1

```

SzczepanLeon, 2024-09-22:
This error is visible also on clean build?

ThomDietrich, 2024-09-22:
Hey @SzczepanLeon yes, more precisely, the following produces the error:

```
esphome clean wasserzaehler-hydrus-2-0.yaml
rm -rf .esphome/build/wasserzaehler
esphome run wasserzaehler-hydrus-2-0.yaml
```

machintrucbidule, 2024-09-23:
Same issue here, since esphome 2024.9.1, also after a clean build

ThomDietrich, 2024-09-23:
@machintrucbidule could you share your `esphome:` and `esp32:` sections?

machintrucbidule, 2024-09-23:
> @machintrucbidule could you share your `esphome:` and `esp32:` section?

Sorry can't copy paste from home assistant mobile app, but:
[image]

SzczepanLeon, 2024-09-23:
Can you remove **framework** section from YAML and rebuild?

machintrucbidule, 2024-09-23:
It was already not there

ThomDietrich, 2024-09-23:
Isn't `arduino` the default?
I did remove it and the resulting `esphome config` and error did not change

ThomDietrich, 2024-09-23:
```make
$ esphome run wasserzaehler-hydrus-2-0.yaml
INFO ESPHome 2024.9.0
INFO Reading configuration wasserzaehler-hydrus-2-0.yaml...
INFO Cloning https://github.com/SzczepanLeon/esphome-components.git@main
INFO Fetching main
WARNING GPIO8 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Core config, version or integrations changed, cleaning build files...
INFO Compiling app...
Processing wasserzaehler (board: esp32-c3-devkitm-1; framework: espidf; platform: platformio/espressif32@5.4.0)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Library Manager: Installing LSatan/SmartRC-CC1101-Driver-Lib @ 2.5.7
INFO Installing LSatan/SmartRC-CC1101-Driver-Lib @ 2.5.7
Unpacking  [####################################]  100%
Library Manager: SmartRC-CC1101-Driver-Lib@2.5.7 has been installed!
INFO SmartRC-CC1101-Driver-Lib@2.5.7 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c3-devkitm-1.html
PLATFORM: Espressif 32 (5.4.0) > Espressif ESP32-C3-DevKitM-1
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf @ 3.40407.240606 (4.4.7)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40400.0 (4.4.0)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 2.35.0-20220830
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
Warning: the 'src_filter' option cannot be used with ESP-IDF. Select source files to build in the project CMakeLists.txt file.

Reading CMake configuration...
Library Manager: Installing SPI
Warning! Could not find the package with 'SPI' requirements for your system 'windows_amd64'
Generating assembly for certificate bundle...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
Library Manager: Installing SPI
Warning! Could not find the package with 'SPI' requirements for your system 'windows_amd64'
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pioenvs\wasserzaehler\src\esphome\components\copy\sensor\copy_sensor.o
Compiling .pioenvs\wasserzaehler\src\esphome\components\esp32\core.o
Compiling .pioenvs\wasserzaehler\src\esphome\components\esp32\gpio.o
Compiling .pioenvs\wasserzaehler\src\esphome\components\esp32\preferences.o
Compiling .pioenvs\wasserzaehler\src\esphome\components\mdns\mdns_component.o
Compiling .pioenvs\wasserzaehler\src\esphome\components\mdns\mdns_esp32.o
Compiling .pioenvs\wasserzaehler\src\esphome\components\mdns\mdns_esp8266.o
Compiling .pioenvs\wasserzaehler\src\esphome\components\mdns\mdns_host.o
Compiling .pioenvs\wasserzaehler\src\esphome\components\mdns\mdns_libretiny.o
Compiling .pioenvs\wasserzaehler\src\esphome\components\mdns\mdns_rp2040.o
Compiling .pioenvs\wasserzaehler\src\esphome\components\network\util.o
Compiling .pioenvs\wasserzaehler\src\esphome\components\sensor\automation.o
Compiling .pioenvs\wasserzaehler\src\esphome\components\sensor\filter.o
Compiling .pioenvs\wasserzaehler\src\esphome\components\sensor\sensor.o
Compiling .pioenvs\wasserzaehler\src\esphome\components\sntp\sntp_component.o
Compiling .pioenvs\wasserzaehler\src\esphome\components\time\automation.o
Compiling .pioenvs\wasserzaehler\src\esphome\components\time\real_time_clock.o
Compiling .pioenvs\wasserzaehler\src\esphome\components\wifi\wifi_component.o
Compiling .pioenvs\wasserzaehler\src\esphome\components\wifi\wifi_component_esp32_arduino.o
Compiling .pioenvs\wasserzaehler\src\esphome\components\wifi\wifi_component_esp8266.o
Compiling .pioenvs\wasserzaehler\src\esphome\components\wifi\wifi_component_esp_idf.o
Compiling .pioenvs\wasserzaehler\src\esphome\components\wifi\wifi_component_libretiny.o
Compiling .pioenvs\wasserzaehler\src\esphome\components\wifi\wifi_component_pico_w.o
Compiling .pioenvs\wasserzaehler\src\esphome\components\wifi_signal\wifi_signal_sensor.o
[... 29 lines cut ...]
     ^~~~~~
cc1plus.exe: some warnings being treated as errors
*** [.pioenvs\wasserzaehler\src\esphome\components\wmbus\Telegram.o] Error 1
src/esphome/components/wmbus/driver_apator08.cpp:43:5: warning: '{anonymous}::Driver::Driver(MeterInfo&, DriverInfo&)' defined but not used [-Wunused-function]
     Driver::Driver(MeterInfo &mi, DriverInfo &di) : MeterCommonImplementation(mi, di)
     ^~~~~~
src/esphome/components/wmbus/driver_apator162.cpp:45:5: warning: '{anonymous}::Driver::Driver(MeterInfo&, DriverInfo&)' defined but not used [-Wunused-function]
     Driver::Driver(MeterInfo &mi, DriverInfo &di) : MeterCommonImplementation(mi, di)
     ^~~~~~
src/esphome/components/wmbus/driver_apator172.cpp:41:5: warning: '{anonymous}::Driver::Driver(MeterInfo&, DriverInfo&)' defined but not used [-Wunused-function]
     Driver::Driver(MeterInfo &mi, DriverInfo &di) : MeterCommonImplementation(mi, di)
     ^~~~~~
[... cut ...]
     ^~~~~~
src/esphome/components/wmbus/driver_aventieshca.cpp:37:5: warning: '{anonymous}::Driver::Driver(MeterInfo&, DriverInfo&)' defined but not used [-Wunused-function]
     Driver::Driver(MeterInfo &mi, DriverInfo &di) : MeterCommonImplementation(mi, di)
     ^~~~~~
src/esphome/components/wmbus/driver_aventieswm.cpp:37:5: warning: '{anonymous}::Driver::Driver(MeterInfo&, DriverInfo&)' defined but not used [-Wunused-function]
     Driver::Driver(MeterInfo &mi, DriverInfo &di) :
     ^~~~~~
```

machintrucbidule, 2024-09-23:
For the moment I came back to 3.2.3 by changing :

```
external_components:
  - source: github://SzczepanLeon/esphome-components@main
    components: [ wmbus ]
```

to

```
external_components:
  - source: github://SzczepanLeon/esphome-components@3.2.3
    components: [ wmbus ]
```

It compiles and works again

wonsky007, 2024-09-23:
Hi! 

I've also got a problem with compiling fw after updating ESPHome to 2024.9.1 even on default conf:

> INFO ESPHome 2024.9.1
> INFO Reading configuration /config/esphome/wmbus-reader.yaml...
> INFO Updating https://github.com/SzczepanLeon/esphome-components.git@main
> INFO Unable to import component wmbus.text_sensor: No module named 'esphome.components.wmbus.text_sensor'
> Failed config
> 
> text_sensor.wmbus: [source /config/esphome/wmbus-reader.yaml:96]
>   
>   Platform not found: 'text_sensor.wmbus'.
>   platform: wmbus
>   name: Text debug for Apator 16-2

Temporary workaround:

> external_components:
>   - source: github://SzczepanLeon/esphome-components@3.2.3
>     components: [ wmbus ]

evilmumi, 2024-09-23:
> > I've also got a problem with compiling fw after updating ESPHome to 2024.9.1 even on default conf:
> 
> This is another topic. This is not compilation problem. Please migrate YAML to new structure. And remove Apator text_sensor.

would you be so kind and share a link "how-to"?

SzczepanLeon, 2024-09-23:
> would you be so kind and share a link \"how-to\"?

Just read main page.

SzczepanLeon, 2024-09-23:
Not really. I don't have time to open computer and check this issue.
Maybe next week (or during cloudy weekend).

evilmumi, 2024-09-24:
> > would you be so kind and share a link "how-to"?
> 
> Just read main page.

which main? would some one be so kind please

SzczepanLeon, 2024-09-24:
> which main? would some one be so kind please

https://github.com/SzczepanLeon/esphome-components

ThomDietrich, 2024-09-24:
I have some c/cpp experience from the past but the macro magic in question is far beyond my understanding. I find it odd that one line right in the middle of everything is supposed to be the problem!? `X(HZ,hz,"Hz",Frequency,"hz")`
Did you make sense of that?

SzczepanLeon, 2024-10-01:
@spitfire Please share YAML. Probably You compiled with all drivers - do You need all of them?

spitfire, 2024-10-01:
```
substitutions:
  name: "stalowe-przedpokoj-esp32"
  friendly_name: przedpokój-esp32

esphome:
  name: ${name}
  friendly_name: ${friendly_name}

esp32:
  board: az-delivery-devkit-v4

external_components:
  - source: github://SzczepanLeon/esphome-components@main
  # - source: github://SzczepanLeon/esphome-components@3.2.3
    components: [ wmbus ]

logger:
  # level: WARN
  # level: VERBOSE
  # level: VERY_VERBOSE
  # logs:
  #   api.connection: INFO
  #   api.service: ERROR
  #   api.socket: ERROR
  #   json: ERROR
  #   scheduler: ERROR
  #   wmbus: VERY_VERBOSE

<<: !include templates/spitfire/common/api.yaml

<<: !include templates/spitfire/common/ota.yaml

<<: !include templates/spitfire/common/wifi.yaml

<<: !include templates/spitfire/common/webserver.yaml

<<: !include templates/spitfire/common/time.yaml

bluetooth_proxy:
  active: true

esp32_ble_tracker:

wmbus:
  all_drivers: False
  log_all: False

  mosi_pin: GPIO13
  clk_pin:  GPIO14
  miso_pin: GPIO12
  gdo2_pin: GPIO4
  gdo0_pin: GPIO5
  cs_pin:   GPIO15

binary_sensor: 
  - !include templates/spitfire/common/status.yaml

button:
  - !include templates/spitfire/common/restart.yaml

sensor:
  - !include templates/spitfire/common/esp32-internal_temperature.yaml

  - !include templates/spitfire/common/wifi_signal_db.yaml
    
  - !include templates/spitfire/common/wifi_signal_percent.yaml

  - !include templates/spitfire/common/uptime.yaml

  - platform: wmbus
    meter_id: 0x18226019
    type: bmeters
    key: "00000000000000000000000000000000"
    ##start wMBus 4.x config
    sensors:
      - name: "Wodomierz ciepła woda RSSI"
        field: "rssi"
        accuracy_decimals: 0
        unit_of_measurement: "dBm"
        device_class: "signal_strength"
        state_class: "measurement"
        entity_category: "diagnostic"
      - name:  "Ciepła woda"
        id: "ciepla_woda"
        field: "total"
        accuracy_decimals: 3
        unit_of_measurement: "m³"
        device_class: "water"
        state_class: "total_increasing"
        icon: "mdi:water"
     ##end wMBus 4.x config

    ##start wMBus 3.x config
    # lqi:
    #   name: "Wodomierz ciepła woda lqi"
    # rssi:
    #   name: "Wodomierz ciepła woda RSSI"
    # total_water_m3:
    #   name:  "Ciepła woda"
    #   id: "ciepla_woda"
    ##start wMBus 3.x config
    
  - platform: wmbus
    meter_id: 0x18147130 #0x18247718
    type: bmeters
    key: "00000000000000000000000000000000"
    ##start wMBus 4.x config
    sensors:
      - name: "Wodomierz zimna woda RSSI"
        field: "rssi"
        accuracy_decimals: 0
        unit_of_measurement: "dBm"
        device_class: "signal_strength"
        state_class: "measurement"
        entity_category: "diagnostic"
      - name:  "Zimna woda"
        id: "zimna_woda"
        field: "total"
        accuracy_decimals: 3
        unit_of_measurement: "m³"
        device_class: "water"
        state_class: "total_increasing"
        icon: "mdi:water"
    ##end wMBus 4.x config

    ##start wMBus 3.x config
    # lqi:
    #   name: "Wodomierz zimna woda lqi"
    # rssi:
    #   name: "Wodomierz zimna woda RSSI"
    # total_water_m3:
    #   name:  "Zimna woda"
    #   id: "zimna_woda"
    ##start wMBus 3.x config

  # - platform: wmbus
  #   meter_id: 0x00B8DE97
  #   type: bmeters
  #   key: "00000000000000000000000000000000"
  #   lqi:
  #     name: "Nieznany lqi"
  #   rssi:
  #     name: "Nieznany RSSI"
  #   total_water_m3:
  #     name:  "Nieznany woda"
  #     id: "nieznany_0x00B8DE97"

light:
  - platform: status_led
    name: "Switch state"
    pin: GPIO2
```
`Error: The program size (1904393 bytes) is greater than maximum allowed (1835008 bytes)`

ThomDietrich, 2024-10-01:
Is it necessary to spam this thread with unrelated issues? I don't want to be the party pooper but this could have been its own ticket 😇

Btw looking at your code I wonder if you want to rethink your multi-file design with packages. See an example [here](https://github.com/ThomDietrich/esphome-projects/blob/1fba36be1c6dd68f4bf82915e4f4108e6ec72341/haus-uv-astra.yaml#L34).

spitfire, 2024-10-01:
> Is it necessary to spam this thread with unrelated issues? I don't want to be the party pooper but this could have been its own ticket 😇

@SzczepanLeon I can create another issue for it if you think it's necessary 

> Btw looking at your code I wonder if you want to rethink your multi-file design with packages. See an example [here](https://github.com/ThomDietrich/esphome-projects/blob/1fba36be1c6dd68f4bf82915e4f4108e6ec72341/haus-uv-astra.yaml#L34).

No thanks, I'm good

My current concern is that pretty much the same configuration works with 3.x and doesn't with 4.x

SzczepanLeon, 2024-10-01:
>  I can create another issue for it if you think it's necessary

No - because this is not an issue.

spitfire, 2024-10-01:
> No - because this is not an issue.

It does not fit (while the previous version did) on a device that's designed to run on with not much else going on on it. Am I "holding it wrong"?

SzczepanLeon, 2024-10-01:
This is software on COTS HW, so it is up to you how to fit it.
You can remove BLE and try again, or change HW or change partitions layout. But all solutions are not related to my component.

SzczepanLeon, 2024-10-01:
You can always stay on v3.x

SzczepanLeon, 2024-10-02:
I don't have time to check it.
But first thing I will do is to add lots of brackets "()" to those DEFINEs.

spitfire, 2024-10-02:
> Back to the issue.
> 
> I tried to update the toolchain beyond `toolchain-riscv32-esp @ 8.4.0+2021r2-patch5` but that just brings other issues with it. Do you have a hunch what the compiler error stands for? I might play around with the source code some time later...

have you tried cleaning up your build files? It helped me with weird build issues like that.

closed 2024-10-03 by SzczepanLeon

ThomDietrich, 2024-10-03:
Hey! I can confirm that the code is compiling now. Thank you!!

Fyi

```
RAM:   [=         ]  12.1% (used 39668 bytes from 327680 bytes)
Flash: [======    ]  62.6% (used 1148608 bytes from 1835008 bytes)
```

ThomDietrich, 2024-10-04:
@spitfire we are only one step further sadly. My ESP32C3 goes into a boot loop. I will try with another ESP32 I have lying around. Could you please share which board/model you used when testing the v4.0? Thanks

spitfire, 2024-10-04:
> @spitfire we are only one step further sadly. My ESP32C3 goes into a boot loop. I will try with another ESP32 I have lying around. Could you please share which board/model you used when testing the v4.0? Thanks

I was testing it with an ESP8266 and this ESP32 board: https://www.amazon.pl/AZDelivery-ESP32-Development-następca-kompatybilny/dp/B074RGW2VQ

SzczepanLeon, 2024-10-04:
Please be precise. Include logs and create new issue ...

ThomDietrich, 2024-10-07:
```
[00:05:20]ESP-ROM:esp32c3-api1-20210207
[00:05:20]Build:Feb  7 2021
[00:05:20]rst:0x3 (RTC_SW_SYS_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
[00:05:20]Saved PC:0x4038217c
WARNING Decoded 0x4038217c: esp_restart_noos_dig at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:44 (discriminator 1)
[00:05:20]SPIWP:0xee
[00:05:20]mode:DIO, clock div:1
[00:05:20]load:0x3fcd6100,len:0x38c
[00:05:20]load:0x403ce000,len:0x6ac
[00:05:20]load:0x403d0000,len:0x24e4
[00:05:20]entry 0x403ce000
```