SzczepanLeon/esphome-components #125: [V4.0]  Apator NA-1 driver fail

kzajac83, 2024-09-23:
Hello

New driver Apator NA-1 cause disconnect after processing msg.  

In logs telegram from Apator NA-1 however disconnect occur after every telegram with correct CRC and build in `apatorna1 `driver.

```
INFO ESPHome 2024.9.1
INFO Reading configuration /config/esphome/wmbus-reader.yaml...
INFO Updating https://github.com/SzczepanLeon/esphome-components.git@main
INFO Detected timezone 'Europe/Warsaw'
INFO Starting log output from 192.168.13.108 using esphome API
WARNING Can't connect to ESPHome API for wmbus-reader-v2 @ 192.168.13.108: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.13.108', port=6053))]: [Errno 111] Connect call failed ('192.168.13.108', 6053) (SocketAPIError)
INFO Trying to connect to wmbus-reader-v2 @ 192.168.13.108 in the background
INFO Successfully connected to wmbus-reader-v2 @ 192.168.13.108 in 0.013s
INFO Successful handshake with wmbus-reader-v2 @ 192.168.13.108 in 0.098s
[21:37:16][I][app:100]: ESPHome version 2024.9.1 compiled on Sep 23 2024, 21:11:58
[21:37:16][I][app:102]: Project esphome.wmbus_reader_v2 version 1.0
[21:37:16][C][status_led:019]: Status LED:
[21:37:16][C][status_led:020]:   Pin: GPIO15
[21:37:16][C][wifi:600]: WiFi:
[21:37:16][C][wifi:428]:   Local MAC: 80:65:99:E2:30:6C
[21:37:16][C][wifi:433]:   SSID: [redacted]
[21:37:16][C][wifi:436]:   IP Address: 192.168.13.108
[21:37:16][C][wifi:440]:   BSSID: [redacted]
[21:37:16][C][wifi:441]:   Hostname: 'wmbus-reader-v2'
[21:37:16][C][wifi:443]:   Signal strength: -76 dB ▂▄▆█
[21:37:16][V][wifi:445]:   Priority: 0.0
[21:37:16][C][wifi:447]:   Channel: 8
[21:37:16][C][wifi:448]:   Subnet: 255.255.255.0
[21:37:16][C][wifi:449]:   Gateway: 192.168.13.1
[21:37:16][C][wifi:450]:   DNS1: 0.0.0.0
[21:37:16][C][wifi:451]:   DNS2: 0.0.0.0
[21:37:16][C][logger:185]: Logger:
[21:37:16][C][logger:186]:   Level: VERBOSE
[21:37:16][C][logger:188]:   Log Baud Rate: 115200
[21:37:16][C][logger:189]:   Hardware UART: USB_CDC
[21:37:16][C][homeassistant.time:010]: Home Assistant Time:
[21:37:16][C][homeassistant.time:011]:   Timezone: 'CET-1CEST,M3.5.0,M10.5.0/3'
[21:37:16][C][mdns:116]: mDNS:
[21:37:16][C][mdns:117]:   Hostname: wmbus-reader-v2
[21:37:16][V][mdns:118]:   Services:
[21:37:16][V][mdns:120]:   - _esphomelib, _tcp, 6053
[21:37:16][V][mdns:122]:     TXT: friendly_name = wmbus reader v2
[21:37:16][V][mdns:122]:     TXT: version = 2024.9.1
[21:37:16][V][mdns:122]:     TXT: mac = 806599e2306c
[21:37:16][V][mdns:122]:     TXT: platform = ESP32
[21:37:16][V][mdns:122]:     TXT: board = lolin_s2_mini
[21:37:16][V][mdns:122]:     TXT: network = wifi
[21:37:16][V][mdns:122]:     TXT: api_encryption = Noise_NNpsk0_25519_ChaChaPoly_SHA256
[21:37:16][V][mdns:122]:     TXT: project_name = esphome.wmbus_reader_v2
[21:37:16][V][mdns:122]:     TXT: project_version = 1.0
[21:37:16][V][mdns:122]:     TXT: package_import_url = github://MariuszWoszczynski/wmbus-reader-v2/wmbus-reader-v2.yaml@main?full_config
[21:37:16][C][esphome.ota:073]: Over-The-Air updates:
[21:37:16][C][esphome.ota:074]:   Address: 192.168.13.108:3232
[21:37:16][C][esphome.ota:075]:   Version: 2
[21:37:16][C][safe_mode:018]: Safe Mode:
[21:37:16][C][safe_mode:020]:   Boot considered successful after 60 seconds
[21:37:16][C][safe_mode:021]:   Invoke after 10 boot attempts
[21:37:16][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[21:37:16][C][api:139]: API Server:
[21:37:16][C][api:140]:   Address: 192.168.13.108:6053
[21:37:16][C][api:142]:   Using noise encryption: YES
[21:37:16][C][wmbus:297]: wM-Bus v4.0.1-1.17.1:
[21:37:16][C][wmbus:314]:   MAC: E2996580
[21:37:16][C][wmbus:315]:   CC1101 frequency: 868.950 MHz
[21:37:16][C][wmbus:316]:   CC1101 SPI bus:
[21:37:16][C][wmbus:320]:     MOSI Pin: GPIO35
[21:37:16][C][wmbus:321]:     MISO Pin: GPIO37
[21:37:16][C][wmbus:322]:     CLK Pin:  GPIO36
[... 184 lines cut ...]
        field: "total_heating"
        accuracy_decimals: 3
        unit_of_measurement: "kwh"
        device_class: "energy"
        state_class: "total_increasing"
      - name: "HYDROCAL m3"
        field: "total_heating"
        accuracy_decimals: 3
        unit_of_measurement: "m³"
        device_class: "water"
        state_class: "total_increasing"
  - platform: wmbus
    meter_id: 0x04913581
    type: apatorna1
    sensors:
      - name: "cold water from Apator NA-1"
        field: "total"
        accuracy_decimals: 3
        unit_of_measurement: "m³"
        device_class: "water"
        state_class: "total_increasing"
        icon: "mdi:water"
        # filters:
        #   - offset: 123.0
```

czesiu20, 2024-09-23:
Hello, 
the same for apator08.

Every time a frame from apator08 is received my ESP device crashes.
```
[21:31:36][V][Telegram.cpp:1131]: (wmbus) parseDLL  @0103
[21:31:36][V][Telegram.cpp:1178]: (wmbus) parseELL @10 93
[21:31:36][V][Telegram.cpp:1333]: (wmbus) parseNWL @10 93
WARNING wmbus-reader @ 10.0.0.135: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for wmbus-reader @ 10.0.0.135
WARNING Disconnected from API
INFO Successfully connected to wmbus-reader @ 10.0.0.135 in 0.116s
INFO Successful handshake with wmbus-reader @ 10.0.0.135 in 0.091s
[21:32:14][D][api:102]: Accepted 10.0.0.200
[21:32:14][V][api.connection:1427]: Hello from client: 'Home Assistant 2024.9.2' | 10.0.0.200 | API Version 1.10
```

Full log attached: [logs_wmbus-reader_run.txt](https://github.com/user-attachments/files/17103266/logs_wmbus-reader_run.txt)

I can see and decode the fame from the log:
```
{
    "media":"water",
    "meter":"apator08",
    "name":"",
    "id":"000b9ce7",
    "total_m3":61.097,
    "timestamp":"2024-09-23T19:12:48Z"
}
```

Other device unismart is working well:
[image]

I removed configuration for apator08 and compiled, with no luck.
Then, I removed configuration for unismart, it didn't help. 
At this stage only configuration of wmbus left, no sensors configured, still crashes.
Finally, I commented out configuration for wmbus and the hang-ups have stopped.

I reverted to version 3 where everything is OK.

kzajac83, 2024-09-23:
Well, seems you have right. Stay only `hydroclima3 `driver in my cause:

`[22:21:33][C][wmbus:331]:   Available drivers: hydrocalm3`

and again the same

```
22:24:02][D][api.connection:1445]: Home Assistant 2024.9.2 (192.168.15.20): Connected successfully
[22:24:02][D][time:051]: Synchronized time: 2024-09-23 22:24:02
[22:24:05][D][mbus:035]: Received T1 A frame
WARNING wmbus-reader-v2 @ 192.168.13.108: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for wmbus-reader-v2 @ 192.168.13.108
WARNING Disconnected from API
WARNING Can't connect to ESPHome API for wmbus-reader-v2 @ 192.168.13.108: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.13.108', port=6053))]: [Errno 111] Connect call failed ('192.168.13.108', 6053) (SocketAPIError)
INFO Trying to connect to wmbus-reader-v2 @ 192.168.13.108 in the background
```

So problem is bigger in V4.0

SzczepanLeon, 2024-09-23:
Please try 

```
external_components:
  - source: github://SzczepanLeon/esphome-components@ana1
    components: [ wmbus ]
    refresh: 0d
```

kzajac83, 2024-09-23:
```
INFO ESPHome 2024.9.1
INFO Reading configuration /config/esphome/wmbus-reader.yaml...
INFO Updating https://github.com/SzczepanLeon/esphome-components.git@ana1

...
....
....

[D][mbus:035]: Received T1 A frame
[V][mbus:046]: Frame: 6B271C58D59A3A5396B0E59659959366CD0B4E665A596696B19956B1998B94B38DD1A713C934CD66C94ECA36533724D3C5A98E3B1C9A69334E71A3A9D344ECCA64D398E9A5993B0B70B5A935AC9AC6969AB315AC2D99A659C5ACA56C4D38D6A673138BA71C4DA7438BA6C2E658D353D31A74A69C93A5CC665B23B4C59732C9638D2E394D6634CB8D9D1C69AD19693C59CA64F42E66A335C4EC723A72C4ED2558E2D9C533ACC4D65CA655B24CD59996336C369D0B4CD369C4DCA63964D370B5 (191) [RAW]
[V][mbus:053]: Frame: 6E44010629208200050758C37A560060859085A39321C647E7715892EB571E77D6A22DE66712462FCC78EA77A2A9A783430F16E6DF668D0835AA0408F0D1216A4D23FDD1FC23F83A0117CDFCFFE7F4DA0E2CD54EE0213B915B73B5C466C567D5EA7C3A6B14784BFED2C90235D728D154F90E71059B181FC3711FD1EA207743 (127) [with CRC]
[V][mbus:096]: Validating CRC for Block1
[V][crc:031]:     calculated: 0x58C3, read: 0x58C3
[V][mbus:116]: Validating CRC for Block2
[V][crc:031]:     calculated: 0xEB57, read: 0xEB57
[V][mbus:116]: Validating CRC for Block3
[V][crc:031]:     calculated: 0xA783, read: 0xA783
[V][mbus:116]: Validating CRC for Block4
[V][crc:031]:     calculated: 0x4D23, read: 0x4D23
[V][mbus:116]: Validating CRC for Block5
[V][crc:031]:     calculated: 0xD54E, read: 0xD54E
[V][mbus:116]: Validating CRC for Block6
[V][crc:031]:     calculated: 0x1478, read: 0x1478
[V][mbus:116]: Validating CRC for Block7
[V][crc:031]:     calculated: 0x1FC3, read: 0x1FC3
[V][mbus:116]: Validating CRC for Block8
[V][crc:031]:     calculated: 0x7743, read: 0x7743
[V][mbus:063]: Frame: 6E4401062920820005077A560060859085A39321C647E77158921E77D6A22DE66712462FCC78EA77A2A9430F16E6DF668D0835AA0408F0D1216AFDD1FC23F83A0117CDFCFFE7F4DA0E2CE0213B915B73B5C466C567D5EA7C3A6B4BFED2C90235D728D154F90E71059B18711FD1EA20 (111) [without CRC]
[V][Telegram.cpp:1131]: (wmbus) parseDLL @0 111
[V][Telegram.cpp:1178]: (wmbus) parseELL @10 101
[V][Telegram.cpp:1333]: (wmbus) parseNWL @10 101
[V][Telegram.cpp:1391]: (wmbus) parseAFL @10 101
[V][Telegram.cpp:2078]: (wmbus) parseTPL @10 101
[V][wmbus:063]: 01 - header przeparsowany
[V][wmbus:066]: 02 - driver wybrany

```
And disconnect:

[image]

After some time appear
```

WARNING wmbus-reader-v2 @ 192.168.13.108: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for wmbus-reader-v2 @ 192.168.13.108
WARNING Disconnected from API
INFO Successfully connected to wmbus-reader-v2 @ 192.168.13.108 in 0.014s
INFO Successful handshake with wmbus-reader-v2 @ 192.168.13.108 in 0.096s
```

SzczepanLeon, 2024-09-23:
> the same for apator08.

@czesiu20 but on what HW? Please share full YAML.

czesiu20, 2024-09-23:
HW is nodemcu esp32 and cc1101 from Mariusz Woszczynski.

My YAML:
```
substitutions:
  name: "wmbus-reader"
  friendly_name: "wmbus-reader"

esphome:
  name: "${name}"
  friendly_name: "${friendly_name}"
  project:
    name: esphome.wmbus_reader
    version: "1.0"
  # only for RPI 
  compile_process_limit: 1  
  
esp32:
  board: esp32dev
  framework:
    type: arduino

external_components:
  - source: github://SzczepanLeon/esphome-components@main
    components: [ wmbus ]
    refresh: 0d
    
time:
 - platform: sntp
   id: time_sntp

# Enable logging
logger:
  level: VERBOSE

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_key

ota:
  platform: esphome
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none
  fast_connect: true
  reboot_timeout: 10min
  manual_ip:
    static_ip: 10.0.0.135
    gateway: 10.0.0.1
    subnet: 255.255.255.0
  
web_server: 
  port: 80

wmbus:
  mosi_pin: GPIO32
  clk_pin:  GPIO33 
  miso_pin: GPIO19
  gdo2_pin: GPIO21
  gdo0_pin: GPIO22
  cs_pin:   GPIO23

  led_pin: GPIO2
  led_blink_time: "1s"

sensor:
  - platform: wmbus
    meter_id: 0x000ABFA4
    type: apator08
    sensors:
    - field: "rssi"
      name: "Woda - Licznik główny - RSSI"
      accuracy_decimals: 0
      unit_of_measurement: "dBm"
      device_class: "signal_strength"
      state_class: "measurement"
      entity_category: "diagnostic"
    - field: "total"
      id: "woda_licznik_glowny"
      name: "Woda - Licznik główny"
      icon: "mdi:water"
      accuracy_decimals: 3
      unit_of_measurement: "m3"
      device_class: "water"
      state_class: "total_increasing"
      filters:
        - offset: 83.078
  - platform: wmbus
    meter_id: 0x000B9CE7
    type: apator08
    sensors:
    - field: "rssi"
      name: "Woda - Licznik ogrodowy - RSSI"
      accuracy_decimals: 0
      unit_of_measurement: "dBm"
      device_class: "signal_strength"
      state_class: "measurement"
      entity_category: "diagnostic"
    - field: "total"
      id: "woda_licznik_ogrodowy"
      name: "Woda - Licznik ogrodowy"
      icon: "mdi:water"
      accuracy_decimals: 3
      unit_of_measurement: "m3"
      device_class: "water"
      state_class: "total_increasing"
      filters:
        - offset: 60.457
  - platform: wmbus
    meter_id: 0x00058794
    type: unismart
    key: "00000000000000000000000000000000"
    sensors:
    - field: "rssi"
      name: "Licznik gazu - RSSI"
      accuracy_decimals: 0
      unit_of_measurement: "dBm"
      device_class: "signal_strength"
      state_class: "measurement"
      entity_category: "diagnostic"
    - field: "total"
      id: "licznik_gazu"
      name: "Licznik gazu"
      icon: "mdi:gas-burner"
      accuracy_decimals: 3
      unit_of_measurement: "m3"
      device_class: "gas"
      state_class: "total_increasing"
      filters:
        - offset: 4337.090
```

SzczepanLeon, 2024-09-24:
Seems that crash is somewhere below:
```

        bool supported_link_mode{false};
        if (drv_info.linkModes().empty()) {
          supported_link_mode = true;
          ESP_LOGW(TAG, "Link modes not defined in driver %s. Processing anyway.",
                   mbus_data.mode,
                   drv_info.name().str().c_str());
        }
        else {
          supported_link_mode = ( ((mbus_data.mode == 'T') && (drv_info.linkModes().has(LinkMode::T1))) ||
                                  ((mbus_data.mode == 'C') && (drv_info.linkModes().has(LinkMode::C1))) );
        }
        bool meter_in_config = (this->wmbus_listeners_.count(meter_id) > 0) ? true : false;
        if ( this->log_all_ || meter_in_config) {
          this->led_blink();
          ESP_LOGV(TAG, "03 - mam w konfigu");
        }
        if (meter_in_config) {
          if (drv_info.name().str().empty()) {
            ESP_LOGW(TAG, "Can't find driver for T: %s", telegram.c_str());
          }
          else if (!supported_link_mode) {
            ESP_LOGW(TAG, "Link mode %c1 not supported in driver %s",
                     mbus_data.mode,
                     drv_info.name().str().c_str());
          }
```

SzczepanLeon, 2024-09-24:
Please try version **4.0.2**

kzajac83, 2024-09-24:
Is working, thank you

[image]

Only two warnings

```

INFO ESPHome 2024.9.1
INFO Reading configuration /config/esphome/wmbus-reader.yaml...
INFO Updating https://github.com/SzczepanLeon/esphome-components.git@main
INFO Detected timezone 'Europe/Warsaw'
INFO Generating C++ source...
INFO Compiling app...
Processing wmbus-reader-v2 (board: lolin_s2_mini; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.4
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.6
|-- ArduinoJson @ 6.18.5
|-- SPI @ 2.0.0
|-- SmartRC-CC1101-Driver-Lib @ 2.5.7
Compiling .pioenvs/wmbus-reader-v2/src/esphome/components/wmbus/wmbus.cpp.o
Compiling .pioenvs/wmbus-reader-v2/src/main.cpp.o
In file included from src/esphome/components/web_server_base/web_server_base.h:11,
                 from src/esphome/components/captive_portal/captive_portal.h:11,
                 from src/esphome/components/wmbus/wmbus.cpp:11:
.piolibdeps/wmbus-reader-v2/ESPAsyncWebServer-esphome/src/ESPAsyncWebServer.h:280:68: warning: 'warning_x' attribute directive ignored [-Wattributes]
     const String& ASYNCWEBSERVER_REGEX_ATTRIBUTE pathArg(size_t i) const;
                                                                    ^~~~~
Linking .pioenvs/wmbus-reader-v2/firmware.elf
```

and info "[10:00:46][W][wmbus:070]: Link modes not defined in driver . Processing anyway."

```
[10:00:46][V][mbus:116]: Validating CRC for Block14
[10:00:46][V][crc:031]:     calculated: 0x55BA, read: 0x55BA
[10:00:46][V][mbus:063]: Frame: CE4401063983590105077A0B00C085FA3427345CD43B7EB802BF590211C7C0D9085EE3ED3E9B5115AAB3F859322E968831FE13D9B83A1E36ED6A31C8C926CE338ED8551820E78A2E24AC9D6849A639FF018CE4CF152551BAF7567020C49A56976F28065EDCBDF6EF26D6DBCF1C9B718C12A46EEFCF776346A71C2456E5D403055C2F76E0EE710B4C25BB06F3B9D279C17BF1C5C0E23A591BA498830E34043EF5CA8051DE21068D59CD5CA74EE3BA7C6A7D02F0BC2881C117DD6B8A68330270752305F0EA7270FE783E00572467861B (207) [without CRC]
[10:00:46][V][Telegram.cpp:1131]: (wmbus) parseDLL @0 207
[10:00:46][V][Telegram.cpp:1178]: (wmbus) parseELL @10 197
[10:00:46][V][Telegram.cpp:1333]: (wmbus) parseNWL @10 197
[10:00:46][V][Telegram.cpp:1391]: (wmbus) parseAFL @10 197
[10:00:46][V][Telegram.cpp:2078]: (wmbus) parseTPL @10 197
[10:00:46][W][wmbus:070]: Link modes not defined in driver . Processing anyway.
[10:00:46][W][component:237]: Component wmbus took a long time for an operation (354 ms).
[10:00:46][W][component:238]: Components should block for at most 30 ms.
```

closed 2024-09-24 by SzczepanLeon

kzajac83, 2024-09-24:
And more lgs, only FYI

INFO ESPHome 2024.9.1
INFO Reading configuration /config/esphome/wmbus-reader.yaml...
INFO Updating https://github.com/SzczepanLeon/esphome-components.git@main
INFO Detected timezone 'Europe/Warsaw'
INFO Starting log output from 192.168.13.108 using esphome API
INFO Successfully connected to wmbus-reader-v2 @ 192.168.13.108 in 0.184s
INFO Successful handshake with wmbus-reader-v2 @ 192.168.13.108 in 0.108s
[10:04:24][I][app:100]: ESPHome version 2024.9.1 compiled on Sep 24 2024, 09:55:10
[10:04:24][I][app:102]: Project esphome.wmbus_reader_v2 version 1.0
[10:04:24][C][status_led:019]: Status LED:
[10:04:24][C][status_led:020]:   Pin: GPIO15
[10:04:24][C][wifi:600]: WiFi:
[10:04:24][C][wifi:428]:   Local MAC: 80:65:99:E2:30:6C
[10:04:24][C][wifi:433]:   SSID: [redacted]
[10:04:24][C][wifi:436]:   IP Address: 192.168.13.108
[10:04:24][C][wifi:440]:   BSSID: [redacted]
[10:04:24][C][wifi:441]:   Hostname: 'wmbus-reader-v2'
[10:04:24][C][wifi:443]:   Signal strength: -70 dB ▂▄▆█
[10:04:24][V][wifi:445]:   Priority: 0.0
[10:04:24][C][wifi:447]:   Channel: 8
[10:04:24][C][wifi:448]:   Subnet: 255.255.255.0
[10:04:24][C][wifi:449]:   Gateway: 192.168.13.1
[10:04:24][C][wifi:450]:   DNS1: 0.0.0.0
[10:04:24][C][wifi:451]:   DNS2: 0.0.0.0
[10:04:24][C][logger:185]: Logger:
[10:04:24][C][logger:186]:   Level: VERBOSE
[10:04:24][C][logger:188]:   Log Baud Rate: 115200
[10:04:24][C][logger:189]:   Hardware UART: USB_CDC
[10:04:24][C][psram:020]: PSRAM:
[10:04:24][C][psram:021]:   Available: YES
[10:04:24][C][psram:024]:   Size: 2047 KB
[10:04:24][C][homeassistant.time:010]: Home Assistant Time:
[10:04:24][C][homeassistant.time:011]:   Timezone: 'CET-1CEST,M3.5.0,M10.5.0/3'
[10:04:24][C][captive_portal:089]: Captive Portal:
[10:04:24][C][web_server:145]: Web Server:
[10:04:24][C][web_server:146]:   Address: 192.168.13.108:80
[10:04:24][C][mdns:116]: mDNS:
[10:04:24][C][mdns:117]:   Hostname: wmbus-reader-v2
[10:04:24][V][mdns:118]:   Services:
[10:04:24][V][mdns:120]:   - _esphomelib, _tcp, 6053
[10:04:24][V][mdns:122]:     TXT: friendly_name = wmbus reader v2
[10:04:25][V][mdns:122]:     TXT: version = 2024.9.1
[10:04:25][V][mdns:122]:     TXT: mac = 806599e2306c
[10:04:25][V][mdns:122]:     TXT: platform = ESP32
[10:04:25][V][mdns:122]:     TXT: board = lolin_s2_mini
[10:04:25][V][mdns:122]:     TXT: network = wifi
[10:04:25][V][mdns:122]:     TXT: api_encryption = Noise_NNpsk0_25519_ChaChaPoly_SHA256
[10:04:25][V][mdns:122]:     TXT: project_name = esphome.wmbus_reader_v2
[10:04:25][V][mdns:122]:     TXT: project_version = 1.0
[10:04:25][V][mdns:122]:     TXT: package_import_url = github://MariuszWoszczynski/wmbus-reader-v2/wmbus-reader-v2.yaml@main?full_config
[10:04:25][V][mdns:120]:   - _http, _tcp, 80
[10:04:25][C][esphome.ota:073]: Over-The-Air updates:
[10:04:25][C][esphome.ota:074]:   Address: 192.168.13.108:3232
[10:04:25][C][esphome.ota:075]:   Version: 2
[10:04:25][C][safe_mode:018]: Safe Mode:
[10:04:25][C][safe_mode:020]:   Boot considered successful after 60 seconds
[10:04:25][C][safe_mode:021]:   Invoke after 10 boot attempts
[10:04:25][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[10:04:25][C][api:139]: API Server:
[10:04:25][C][api:140]:   Address: 192.168.13.108:6053
[10:04:25][C][api:142]:   Using noise encryption: YES
[10:04:25][C][wmbus:297]: wM-Bus v4.0.2-1.17.1:
[10:04:25][C][wmbus:314]:   MAC: E2996580
[10:04:25][C][wmbus:315]:   CC1101 frequency: 868.950 MHz
[10:04:25][C][wmbus:316]:   CC1101 SPI bus:
[10:04:25][C][wmbus:320]:     MOSI Pin: GPIO35
[10:04:25][C][wmbus:321]:     MISO Pin: GPIO37
[10:04:25][C][wmbus:322]:     CLK Pin:  GPIO36
[... 2319 lines cut ...]
[10:07:07][D][sensor:094]: 'cold water from Apator NA-1': Sending state 650.09399 m³ with 3 decimals of accuracy
[10:07:07][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[10:07:07][V][json:058]: Size after shrink 96 bytes
[10:07:07][W][component:237]: Component wmbus took a long time for an operation (369 ms).
[10:07:07][W][component:238]: Components should block for at most 30 ms.
[10:07:07][V][rxLoop:167]: Have 191 bytes from CC1101 Rx, RSSI: -81 dBm LQI: 128
[10:07:07][V][rxLoop:172]: Synchronus mode enabled.
[10:07:07][D][mbus:035]: Received T1 A frame
[10:07:07][V][mbus:046]: Frame: 6B271C58D59A3AC35A96C58B599593B0D4D94E668B596696B199A534E6692E6B31A5C5A3596999C669724CE9999B48E3D263B4669C938DA8DA8E96A666571CA596539A68CBD0D95CD254F2663C74C6C6AC2F25A58F4672C6CC962D9B3439699A99CC4E6A6C4E64D6A32E38D66FC49D7258FB33DECB3B4ADA96C8D94E36797139715AC95C6719AB5A573C72996ED268F48DC8E6D2535A34D36334ECB22CB4F1D1ED2BCDC2D99AC95CA5A6714E3C9C9A6C66693B315A93B294ED25599D16B0B5 (191) [RAW]
[10:07:07][V][3of6:083]: Decode 3 out of 6 failed..
[10:07:07][W][component:237]: Component wmbus took a long time for an operation (51 ms).
[10:07:07][W][component:238]: Components should block for at most 30 ms.
[10:07:07][V][rxLoop:167]: Have 191 bytes from CC1101 Rx, RSSI: -81 dBm LQI: 130
[10:07:07][V][rxLoop:172]: Synchronus mode enabled.
[10:07:07][D][mbus:035]: Received T1 A frame
[10:07:07][V][mbus:046]: Frame: 6B271C58D59A96C69369958B5995933B29714E6C96596696B199969A698EC9C4CD70BB132F19966692F295ACA9969D1A7345B1CA6B32B16CA3A668E5A59399D3295364B64E8F2A71C9C5A6D254F2C694E564B34E732A71C4B665C71B266532E5A7464B69A95972594D2CBB0D64B699374B3294D68B4DC9ACA6559A35C593D34A4D9A3C99CA5C998F16A33A6C9658E2DC2DCD0B34D6696694E9A65C8DC7469AB1A399671A4ED34A4DB0D5998DC65C672C4B96C6669ACA663AC5B239A98B2D35 [... cut ...]
[10:07:07][V][mbus:053]: Frame: 6E4401069867650305072E9D7AE0006085A0AAA2E47143873DA05F3E96EF9FC64C0DEA8E80EBFAB9F525CE975352BEFDE40AC97EDF7953124EFDD359DD8A5739FC5366954991338153651C8E916374A8F9061407CCF1ABE5E9E5BD6B2AE0023434C3115F5F7FF9E1DC6686255DF2CCF18105B4545ED3985AA8FA280E26A337 (127) [with CRC]
[10:07:07][V][mbus:096]: Validating CRC for Block1
[10:07:07][V][crc:031]:     calculated: 0x2E9D, read: 0x2E9D
[10:07:07][V][mbus:116]: Validating CRC for Block2
[10:07:07][V][crc:031]:     calculated: 0x96EF, read: 0x96EF
[10:07:07][V][mbus:116]: Validating CRC for Block3
[10:07:07][V][crc:031]:     calculated: 0xBEFD, read: 0xBEFD
[10:07:07][V][mbus:116]: Validating CRC for Block4
[10:07

czesiu20, 2024-09-24:
Closed too fast.
Driver for apator08 is not detected:

[10:01:53][V][Telegram.cpp:2078]: (wmbus) parseTPL @10 93 [10:01:53][W][wmbus:070]: Link modes not defined in driver . Processing anyway. [10:01:53][I][wmbus:085]: [0x000b9ce7] RSSI: -25dBm T: 66441486E79C0B000303A0FBCB02009C400038110903180918346297D70305FF08176D02000300000000377C07304E464E1F000000002000000000000023B2FC810C2406340D160A1527000000280000000000002C0030151207172F0000000000003101000500 (103) T1 A [10:01:53][W][wmbus:089]: Can't find driver for T: 66441486E79C0B000303A0FBCB02009C400038110903180918346297D70305FF08176D02000300000000377C07304E464E1F000000002000000000000023B2FC810C2406340D160A1527000000280000000000002C0030151207172F0000000000003101000500 (103) [10:01:53][W][component:237]: Component wmbus took a long time for an operation (248 ms). [10:01:53][W][component:238]: Components should block for at most 30 ms. [10:02:11][V][rxLoop:167]: Have 191 bytes from CC1101 Rx, RSSI: -68 dBm LQI: 131 [10:02:11][D][mbus:035]: Received T1 A frame

SzczepanLeon, 2024-09-24:
> Closed too fast.
> Driver for apator08 is not detected:

Please share YAML.

czesiu20, 2024-09-24:
> HW is nodemcu esp32 and cc1101 from Mariusz Woszczynski.
>
> My YAML:
> ```
> substitutions:
>   name: "wmbus-reader"
>   friendly_name: "wmbus-reader"
>
> esphome:
>   name: "${name}"
>   friendly_name: [...]

The same