SzczepanLeon/esphome-components #126: [V4.0]  Hydrocalm3 no return quantity

kzajac83, 2024-09-24:
Hello.

I'd like focus on driver `hydrocalm3 `and fields `total_heating_m3` & `total_heating_kwh`

```
root@wyse:~# wmbusmeters --listfields=hydrocalm3
                  id  The meter id number.
                name  Your name for the meter.
               media  What does the meter measure?
               meter  Meter driver.
           timestamp  Timestamp when wmbusmeters received the telegram. Local time for hr/fields UTC for json.
        timestamp_ut  Unix timestamp when wmbusmeters received the telegram.
        timestamp_lt  Local time when wmbusmeters received the telegram.
       timestamp_utc  UTC time when wmbusmeters received the telegram.
              device  The wmbus device that received the telegram.
            rssi_dbm  The rssi for the received telegram as reported by the device.
              status  Meter status from tpl status field.
   total_heating_kwh  The total heating energy consumption recorded by this meter.
     device_datetime  The date time when the recording was made.
   total_cooling_kwh  The total cooling energy consumption recorded by this meter.
    total_heating_m3  Total heating volume of media.
    total_cooling_m3  Total cooling volume of media.
        c1_volume_m3  Supply c1 volume.
        c2_volume_m3  Return c2 volume.
supply_temperature_c  The supply t1 pipe temperature.
return_temperature_c  The return t2 pipe temperature.

```

Configuration YAML ESPHome for sensor only is:

```
sensor:
  - platform: wmbus
    meter_id: 0x05128041
    type: hydrocalm3
    #key: "00000000000000000000000000000000"
    sensors:
      - name: "HYDROCAL rssi"
        field: "rssi"
        accuracy_decimals: 0
        unit_of_measurement: "dBm"
        device_class: "signal_strength"
        state_class: "measurement"
        entity_category: "diagnostic"
      - name: "HYDROCAL total kWh"
        field: "total_heating"
        accuracy_decimals: 3
        unit_of_measurement: "kwh"
        device_class: "energy"
        state_class: "total_increasing"
      - name: "HYDROCAL m3"
        field: "total_heating"
        accuracy_decimals: 3
        unit_of_measurement: "m³"
        device_class: "water"
        state_class: "total_increasing"
```

Component return:

```
[10:16:02][V][rxLoop:167]: Have 104 bytes from CC1101 Rx, RSSI: -32 dBm LQI: 128
[10:16:02][V][rxLoop:172]: Synchronus mode enabled.
[10:16:02][D][mbus:035]: Received T1 A frame
[10:16:02][V][mbus:046]: Frame: 2E671C8DC5A570DB1634E5993665B14CBC4EB343968E64E669A34E59639659C6B15B49A636C2E55B45ACA666695965965965965B434B596596596596B343565AC59659659668D366596B3435634B5965965965965A969C58D356596596596596B1C371596A69A695 (104) [RAW]
[10:16:02][V][mbus:053]: Frame: 3A44B409418012051A0D73D28C20BA7A66120020046D0CAA18390C08FA5F000000000C13000000008C1008000000611A008C1013000000000F64011000000000841D00FFFF (69) [with CRC]
[10:16:02][V][mbus:096]: Validating CRC for Block1
[10:16:02][V][crc:031]:     calculated: 0x73D2, read: 0x73D2
[10:16:02][V][mbus:116]: Validating CRC for Block2
[... 53 lines cut ...]
[10:16:03][V][meters.cpp:2168]: (meter) DateTime device_datetime decoded datetime default datetime value 1.72717e+09 (scale 1)
[10:16:03][V][meters.cpp:986]: (meters) trying field info total_cooling(Energy)[3]...
[10:16:03][V][meters.cpp:986]: (meters) trying field info total_heating(Volume)[4]...
[10:16:03][V][meters.cpp:1008]: (meters) using field info total_heating(Volume)[4] to extract 0C13 at offset 32
[10:16:03][V][meters.cpp:2168]: (meter) Volume total_heating_m3 decoded m3 default m3 value 0 (scale 1)
[10:16:03][V][meters.cpp:986]: (meters) trying field info total_cooling(Volume)[5]...
[10:16:03][V][meters.cpp:986]: (meters) trying field info c1_volume(Volume)[6]...
[10:16:03][V][meters.cpp:986]: (meters) trying field info c2_volume(Volume)[7]...
[10:16:03][V][meters.cpp:986]: (meters) trying field info supply_temperature(Temperature)[8]...
[10:16:03][V][meters.cpp:986]: (meters) trying field info return_temperature(Temperature)[9]...
[10:16:03][V][sensor:043]: 'HYDROCAL rssi': Received new state -32.000000
[10:16:03][D][sensor:094]: 'HYDROCAL rssi': Sending state -32.00000 dBm with 0 decimals of accuracy
[10:16:03][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[10:16:03][V][json:058]: Size after shrink 80 bytes
[10:16:03][W][units.cpp:181]: Cannot convert between units! from 'kWh' to 'm³'
[10:16:03][W][wmbus:126]: Can't get requested field 'total_heating' with unit 'm³'
[10:16:03][W][component:237]: Component wmbus took a long time for an operation (702 ms).
[10:16:03][W][component:238]: Components should block for at most 30 ms.

```
The telegram:

3a44B409418012051a0d8c20Ba7a66120020046d0caa18390c08000000000c13000000008c1008000000008c1013000000000f6401100000000000

https://wmbusmeters.org/analyze/3A44B409418012051A0D8C20BA7A66120020046D0CAA18390C08000000000C13000000008C1008000000008C1013000000000F6401100000000000

czesiu20, 2024-09-24:
Driver for apator08 is not detected:

[10:01:53][V][Telegram.cpp:2078]: (wmbus) parseTPL @10 93 [10:01:53][W][wmbus:070]: Link modes not defined in driver . Processing anyway. [10:01:53][I][wmbus:085]: [0x000b9ce7] RSSI: -25dBm T: 66441486E79C0B000303A0FBCB02009C400038110903180918346297D70305FF08176D02000300000000377C07304E464E1F000000002000000000000023B2FC810C2406340D160A1527000000280000000000002C0030151207172F0000000000003101000500 (103) T1 A [10:01:53][W][wmbus:089]: Can't find driver for T: 66441486E79C0B000303A0FBCB02009C400038110903180918346297D70305FF08176D02000300000000377C07304E464E1F000000002000000000000023B2FC810C2406340D160A1527000000280000000000002C0030151207172F0000000000003101000500 (103) [10:01:53][W][component:237]: Component wmbus took a long time for an operation (248 ms). [10:01:53][W][component:238]: Components should block for at most 30 ms. [10:02:11][V][rxLoop:167]: Have 191 bytes from CC1101 Rx, RSSI: -68 dBm LQI: 131 [10:02:11][D][mbus:035]: Received T1 A frame

SzczepanLeon, 2024-09-24:
Please try version **4.0.3**

kzajac83, 2024-09-24:
Thank you, working fine now

```
[20:47:01][V][rxLoop:167]: Have 104 bytes from CC1101 Rx, RSSI: -29 dBm LQI: 128
[20:47:01][V][rxLoop:172]: Synchronus mode enabled.
[20:47:01][D][mbus:035]: Received T1 A frame
[20:47:01][V][mbus:046]: Frame: 2E671C8DC5A570DB1634E5993665B14CBC4EB343967164E6CB434E59639659C6B13A38DC36C2E55B45AC71C39A5965965965965B434B596596596596B343565AC59659659668D366596B3435634B5965965965965A969C58D356596596596596B1C371596A69A695 (104) [RAW]
[20:47:01][V][mbus:053]: Frame: 3A44B409418012051A0D73D28C20407AEC120020046D2BB418390C084426000000000C13000000008C1008000000611A008C1013000000000F64011000000000841D00FFFF (69) [with CRC]
[20:47:01][V][mbus:096]: Validating CRC for Block1
[20:47:01][V][crc:031]:     calculated: 0x73D2, read: 0x73D2
[20:47:01][V][mbus:116]: Validating CRC for Block2
[20:47:01][V][crc:031]:     calculated: 0x4426, read: 0x4426
[20:47:01][V][mbus:116]: Validating CRC for Block3
[20:47:01][V][crc:031]:     calculated: 0x611A, read: 0x611A
[20:47:01][V][mbus:116]: Validating CRC for Block4
[20:47:01][V][crc:031]:     calculated: 0x841D, read: 0x841D
[20:47:01][V][mbus:116]: Validating CRC for Block5
[20:47:01][V][crc:031]:     calculated: 0xFFFF, read: 0xFFFF
[20:47:01][V][mbus:063]: Frame: 3A44B409418012051A0D8C20407AEC120020046D2BB418390C08000000000C13000000008C1008000000008C1013000000000F6401100000000000 (59) [without CRC]
[20:47:01][V][Telegram.cpp:1131]: (wmbus) parseDLL @0 59
[20:47:01][V][Telegram.cpp:1178]: (wmbus) parseELL @10 49
[20:47:01][V][Telegram.cpp:1333]: (wmbus) parseNWL @13 46
[20:47:01][V][Telegram.cpp:1391]: (wmbus) parseAFL @13 46
[20:47:01][V][Telegram.cpp:2078]: (wmbus) parseTPL @13 46
[20:47:01][V][dvparser.cpp:1471]: [DVPARSER] entry 20: 046D Instantaneous vif=6d  st=0 ta=0 su=0
[20:47:01][V][dvparser.cpp:1471]: [DVPARSER] entry 26: 0C08 Instantaneous vif=8  st=0 ta=0 su=0
[20:47:01][V][dvparser.cpp:1471]: [DVPARSER] entry 32: 0C13 Instantaneous vif=13  st=0 ta=0 su=0
[20:47:01][V][dvparser.cpp:1471]: [DVPARSER] entry 39: 8C1008 Instantaneous vif=8  st=0 ta=1 su=0
[20:47:01][V][dvparser.cpp:1471]: [DVPARSER] entry 46: 8C1013 Instantaneous vif=13  st=0 ta=1 su=0
[20:47:01][I][wmbus:085]: hydrocalm3 [0x05128041] RSSI: -29dBm T: 3A44B409418012051A0D8C20407AEC120020046D2BB418390C08000000000C13000000008C1008000000008C1013000000000F6401100000000000 (59) T1 A
[20:47:01][D][meters.cpp:1993]: (meter) created ESPHome hydrocalm3 05128041 not-encrypted
[20:47:01][V][Telegram.cpp:1131]: (wmbus) parseDLL @0 59
[20:47:01][V][Telegram.cpp:1178]: (wmbus) parseELL @10 49
[20:47:01][V][Telegram.cpp:1333]: (wmbus) parseNWL @13 46
[20:47:01][V][Telegram.cpp:1391]: (wmbus) parseAFL @13 46
[20:47:01][V][Telegram.cpp:2078]: (wmbus) parseTPL @13 46
[20:47:01][V][dvparser.cpp:1471]: [DVPARSER] entry 20: 046D Instantaneous vif=6d  st=0 ta=0 su=0
[20:47:01][V][dvparser.cpp:1471]: [DVPARSER] entry 26: 0C08 Instantaneous vif=8  st=0 ta=0 su=0
[20:47:01][V][dvparser.cpp:1471]: [DVPARSER] entry 32: 0C13 Instantaneous vif=13  st=0 ta=0 su=0
[20:47:01][V][dvparser.cpp:1471]: [DVPARSER] entry 39: 8C1008 Instantaneous vif=8  st=0 ta=1 su=0
[20:47:01][V][dvparser.cpp:1471]: [DVPARSER] entry 46: 8C1013 Instantaneous vif=13  st=0 ta=1 su=0
[20:47:01][V][meters.cpp:635]: (meter) ESPHome: for me? 05128041.M=BMT.V=1a.T=0d,05128041.M=BMT.V=1a.T=0d in 05128041
[20:47:01][V][meters.cpp:699]: (meter) ESPHome: yes for me
[20:47:01][D][meters.cpp:922]: (meter) ESPHome(0) hydrocalm3  handling telegram from 05128041.M=BMT.V=1a.T=0d
[20:47:01][V][Telegram.cpp:1131]: (wmbus) parseDLL @0 59
[20:47:01][V][Telegram.cpp:545]: (telegram) DLL L=3a C=44 (from meter SND_NR) M=09b4 (BMT) A=05128041 VER=1a TYPE=0d (Heat/Cooling load meter) (driver hydrocalm3) DEV=ESPHome wM-Bus RSSI=-29
[20:47:01][V][Telegram.cpp:1178]: (wmbus) parseELL @10 49
[20:47:01][D][Telegram.cpp:564]: (telegram) ELL CI=8c CC=20 (slow_resp sync) ACC=40
[20:47:01][V][Telegram.cpp:1333]: (wmbus) parseNWL @13 46
[20:47:01][V][Telegram.cpp:1391]: (wmbus) parseAFL @13 46
[20:47:01][V][Telegram.cpp:2078]: (wmbus) parseTPL @13 46
[20:47:01][V][dvparser.cpp:1471]: [DVPARSER] entry 20: 046D Instantaneous vif=6d  st=0 ta=0 su=0
[20:47:01][V][dvparser.cpp:1471]: [DVPARSER] entry 26: 0C08 Instantaneous vif=8  st=0 ta=0 su=0
[20:47:01][V][dvparser.cpp:1471]: [DVPARSER] entry 32: 0C13 Instantaneous vif=13  st=0 ta=0 su=0
[20:47:01][V][dvparser.cpp:1471]: [DVPARSER] entry 39: 8C1008 Instantaneous vif=8  st=0 ta=1 su=0
[20:47:01][V][dvparser.cpp:1471]: [DVPARSER] entry 46: 8C1013 Instantaneous vif=13  st=0 ta=1 su=0
[20:47:01][V][Telegram.cpp:634]: (telegram) TPL CI=7a
[20:47:01][V][Telegram.cpp:640]:  ACC=ec STS=12 CFG=2000 (synchronous)
[20:47:01][V][meters.cpp:979]: (meters) skipping field without matcher status(Text)[0]...
[20:47:01][V][meters.cpp:986]: (meters) trying field info total_heating(Energy)[1]...
[20:47:01][V][meters.cpp:1008]: (meters) using field info total_heating(Energy)[1] to extract 0C08 at offset 26
[20:47:01][V][meters.cpp:2168]: (meter) AnyEnergyVIF total_heating_kwh decoded mj default kwh value 0 (scale 1)
[20:47:01][V][meters.cpp:986]: (meters) trying field info device(PointInTime)[2]...
[20:47:01][V][meters.cpp:1008]: (meters) using field info device(PointInTime)[2] to extract 046D at offset 20
[20:47:01][V][meters.cpp:2168]: (meter) DateTime device_datetime decoded datetime default datetime value 1.7272e+09 (scale 1)
[20:47:01][V][meters.cpp:986]: (meters) trying field info total_cooling(Energy)[3]...
[... cut ...]
[20:47:01][V][sensor:043]: 'HYDROCAL total kWh': Received new state 0.000000
[20:47:01][D][sensor:094]: 'HYDROCAL total kWh': Sending state 0.00000 kwh with 3 decimals of accuracy
[20:47:01][V][sensor:043]: 'HYDROCAL m3': Received new state 0.000000
[20:47:01][D][sensor:094]: 'HYDROCAL m3': Sending state 0.00000 m³ with 3 decimals of accuracy
[20:47:01][W][component:237]: Component wmbus took a long time for an operation (431 ms).
[20:47:01][W][component:238]: Components should block for at most 30 ms.

```

closed 2024-09-24 by SzczepanLeon