SzczepanLeon/esphome-components #127: [V4.0] Driver for apator08 is not detected

SzczepanLeon, 2024-09-24:
Driver for apator08 is not detected:

```
[10:01:53][V][Telegram.cpp:2078]: (wmbus) parseTPL @10 93 
[10:01:53][W][wmbus:070]: Link modes not defined in driver . Processing anyway. 
[10:01:53][I][wmbus:085]: [0x000b9ce7] RSSI: -25dBm T: 66441486E79C0B000303A0FBCB02009C400038110903180918346297D70305FF08176D02000300000000377C07304E464E1F000000002000000000000023B2FC810C2406340D160A1527000000280000000000002C0030151207172F0000000000003101000500 (103) T1 A 
[10:01:53][W][wmbus:089]: Can't find driver for T: 66441486E79C0B000303A0FBCB02009C400038110903180918346297D70305FF08176D02000300000000377C07304E464E1F000000002000000000000023B2FC810C2406340D160A1527000000280000000000002C0030151207172F0000000000003101000500 (103) 
[10:01:53][W][component:237]: Component wmbus took a long time for an operation (248 ms). 
[10:01:53][W][component:238]: Components should block for at most 30 ms. 
[10:02:11][V][rxLoop:167]: Have 191 bytes from CC1101 Rx, RSSI: -68 dBm LQI: 131 [10:02:11][D][mbus:035]: Received T1 A frame 
```

YAML
```
 sensor:
   - platform: wmbus
     meter_id: 0x000ABFA4
     type: apator08
     sensors:
     - field: "rssi"
       name: "Woda - Licznik główny - RSSI"
       accuracy_decimals: 0
       unit_of_measurement: "dBm"
       device_class: "signal_strength"
       state_class: "measurement"
       entity_category: "diagnostic"
     - field: "total"
       id: "woda_licznik_glowny"
       name: "Woda - Licznik główny"
       icon: "mdi:water"
       accuracy_decimals: 3
       unit_of_measurement: "m3"
       device_class: "water"
       state_class: "total_increasing"
       filters:
         - offset: 83.078
   - platform: wmbus
     meter_id: 0x000B9CE7
     type: apator08
     sensors:
     - field: "rssi"
       name: "Woda - Licznik ogrodowy - RSSI"
       accuracy_decimals: 0
       unit_of_measurement: "dBm"
       device_class: "signal_strength"
       state_class: "measurement"
       entity_category: "diagnostic"
     - field: "total"
       id: "woda_licznik_ogrodowy"
       name: "Woda - Licznik ogrodowy"
       icon: "mdi:water"
       accuracy_decimals: 3
       unit_of_measurement: "m3"
       device_class: "water"
       state_class: "total_increasing"
       filters:
         - offset: 60.457
   - platform: wmbus
     meter_id: 0x00058794
     type: unismart
     key: "00000000000000000000000000000000"
     sensors:
     - field: "rssi"
       name: "Licznik gazu - RSSI"
       accuracy_decimals: 0
       unit_of_measurement: "dBm"
       device_class: "signal_strength"
       state_class: "measurement"
       entity_category: "diagnostic"
     - field: "total"
       id: "licznik_gazu"
       name: "Licznik gazu"
       icon: "mdi:gas-burner"
       accuracy_decimals: 3
       unit_of_measurement: "m3"
       device_class: "gas"
       state_class: "total_increasing"
       filters:
         - offset: 4337.090
 ```

SzczepanLeon, 2024-09-24:
Please try version **4.0.3**

czesiu20, 2024-09-24:
Now collecting data from apatot08 as well. Great job. Coffee for you.

closed 2024-09-24 by SzczepanLeon