SzczepanLeon/esphome-components #132: Hydrocal M3 telegram parsing problem: "2F2F check after decrypting"

danielloczi, 2024-10-03:
I've a Hydrocal M3 meter with WMbus radio.
Just bought an ESP based WMbus reader with pre-installed firmware.
Configured it, but it seems that the reader is unable the parse the telegram message received from the reader.

An example telegram:
`7844B409094140020B0d7a800000000c0690230000046d282d033a0c13434363000c06241600000c13678680000c14634900000c14763500000a5a29020a5e24020f83ef0000ef0000ef0000ef0000ef0000ef0000e801000000000000000000000000000000005f0100000000000000000000000000000000`

This can be parsed with the WMbusmeters. See:
https://wmbusmeters.org/analyze/7844B409094140020B0D7A800000000C0690230000046D282D033A0C13434363000C06241600000C13678680000C14634900000C14763500000A5A29020A5E24020F83EF0000EF0000EF0000EF0000EF0000EF0000E801000000000000000000000000000000005F0100000000000000000000000000000000:hydrocalm3

Pls note that we don't use encryption, so I left the encryption key empty.

Here are the logs from the ESP:
```
[13:37:53][D][wmbus:097]: Using driver 'hydrocalm3' for ID [0x02404109] RSSI: -81 dBm LQI: 138 Frame: T1 A T: 7844B409094140020B0D7A760000000C0690230000046D1E2D033A0C13434363000C06241600000C13678680000C14634900000C14763500000A5A29020A5E24020F83EF0000EF0000EF0000EF0000EF0000EF0000E801000000000000000000000000000000005F0100000000000000000000000000000000 (121)
[13:37:53][D][utils:294]: 2F2F check after decrypting  !!!
[13:38:23][D][mbus:034]: Processing T1 A frame
[13:38:37][D][sensor:094]: 'WiFi Signal dB': Sending state -45.00000 dBm with 0 decimals of accuracy
[13:39:37][D][sensor:094]: 'WiFi Signal dB': Sending state -55.00000 dBm with 0 decimals of accuracy
[13:39:53][D][mbus:034]: Processing T1 A frame
[13:39:53][D][wmbus:097]: Using driver 'hydrocalm3' for ID [0x02404109] RSSI: -81 dBm LQI: 129 Frame: T1 A T: 7844B409094140020B0D7A7A0000000C0690230000046D282D033A0C13434363000C06241600000C13678680000C14634900000C14763500000A5A29020A5E24020F83EF0000EF0000EF0000EF0000EF0000EF0000E801000000000000000000000000000000005F0100000000000000000000000000000000 (121)
[13:39:53][D][utils:294]: 2F2F check after decrypting  !!!
[13:40:24][D][mbus:034]: Processing T1 A frame
[13:40:37][D][sensor:094]: 'WiFi Signal dB': Sending state -55.00000 dBm with 0 decimals of accuracy
[13:40:54][D][mbus:034]: Processing T1 A frame
[13:41:37][D][sensor:094]: 'WiFi Signal dB': Sending state -54.00000 dBm with 0 decimals of accuracy
[13:42:37][D][sensor:094]: 'WiFi Signal dB': Sending state -54.00000 dBm with 0 decimals of accuracy
[13:42:47][D][sntp:073]: Synchronized time: 2024-10-03 13:42:47
[13:42:54][D][mbus:034]: Processing T1 A frame
[13:42:54][D][wmbus:097]: Using driver 'hydrocalm3' for ID [0x02404109] RSSI: -82 dBm LQI: 132 Frame: T1 A T: 7844B409094140020B0D7A800000000C0690230000046D282D033A0C13434363000C06241600000C13678680000C14634900000C14763500000A5A29020A5E24020F83EF0000EF0000EF0000EF0000EF0000EF0000E801000000000000000000000000000000005F0100000000000000000000000000000000 (121)
[13:42:54][D][utils:294]: 2F2F check after decrypting  !!!
```

Here are the relevant section from config yaml:
```
#dashboard_import:
  #package_import_url: github://MariuszWoszczynski/WMBUS-reader/WMBUS-reader.yaml@main
  #import_full_config: true

substitutions:
  name: "wmbus-reader"
  friendly_name: "wmbus-reader"

esphome:
  name: "${name}"
  friendly_name: "${friendly_name}"
  project:
    name: esphome.wmbus_reader
    version: "1.0"
  # only for RPI 
#  compile_process_limit: 1
  
esp32:
  board: nodemcu-32s
  framework:
    type: arduino

external_components:
  - source: github://SzczepanLeon/esphome-components@3.2.3
    components: [ wmbus ]
    refresh: 0d

wmbus:
  mosi_pin: GPIO32
  clk_pin:  GPIO33 
  miso_pin: GPIO19
  gdo2_pin: GPIO21
  gdo0_pin: GPIO22
  cs_pin:   GPIO23
#  log_unknown: true
  led_pin: GPIO2
  led_blink_time: "1s"

text_sensor: 
    #diagnostic info
  - platform: wifi_info                   
    ip_address:
      name: 'IP Address'
      icon: mdi:wifi
    ssid:
      name: 'Connected SSID'
      icon: mdi:wifi-strength-2
      
sensor: 
    # Reports the WiFi signal strength/RSSI in dB
  - platform: wifi_signal 
    name: "WiFi Signal dB"
    id: wifi_signal_db
    update_interval: 60s
    entity_category: "diagnostic"

  - platform: wmbus
    meter_id: 0x02404109
    type: hydrocalm3
    key: "00000000000000000000000000000000"
    add_prefix: false
    total_heating_kwh:
      name: "Heating [kWh]"
```

What is the problem here or how to fix this issue?

SzczepanLeon, 2024-10-03:
As I see in YAML you defined key (all zeroes). Please remove **key** line from YAML.

closed 2024-10-03 by SzczepanLeon

danielloczi, 2024-10-03:
That was the missing bit!

Thank you for your help!

I've changed the config to (showing only the relevant parts), and all work as expected!

```
external_components:
  - source: github://SzczepanLeon/esphome-components@4.0.8
 
 - platform: wmbus
    meter_id: 0x02404109 
    type: hydrocalm3
    sensors:
      - name: "Heating [kWh]"
        field: "total_heating"
        unit_of_measurement: "kwh"
        device_class: "energy"
        state_class: "total_increasing"
      - name: "Cooling [kWh]"
        field: "total_cooling"
        unit_of_measurement: "kwh"
        device_class: "energy"
        state_class: "total_increasing"
      - name: "Cold Water [m3]"
        field: "c1_volume"
        unit_of_measurement: "m3"
        device_class: "water"
        state_class: "total_increasing"
        accuracy_decimals: 2
        icon: "mdi:water"
      - name: "Warm Water [m3]"
        field: "c2_volume"
        unit_of_measurement: "m3"
        device_class: "water"
        state_class: "total_increasing"
        accuracy_decimals: 2
        icon: "mdi:water"
      - name: "Supply temperature [C]"
        field: "supply_temperature"
        unit_of_measurement: "c"
        device_class: "temperature"
        state_class: "measurement"
        accuracy_decimals: 1
      - name: "Return temperature [C]"
        field: "return_temperature"
        unit_of_measurement: "c"
        device_class: "temperature"
        state_class: "measurement"
        accuracy_decimals: 1
```

SzczepanLeon, 2024-10-03:
Please also change **m3** and **C** to HomeAssistant supported  units.