SzczepanLeon/esphome-components #133: Guru Meditation Error: Core  1 panic'ed (LoadProhibited) - boot loop (wifi issues)

VivantSenior, 2024-10-03:
Hi,

After update to version 4.0 and ESPHome 2024.9.1 I noticed that my ESP32-WROOM-32 has issues connecting to the Wifi and warnings while decoding telegram messages. I found in the logs error: 
```
[C][wmbus:305]: wM-Bus v4.0.8-1.17.1-b8f4a945:
[C][wmbus:323]:   MAC: 68A8CC84
[C][wmbus:328]:   CC1101 frequency: 868.950 MHz
[C][wmbus:329]:   CC1101 SPI bus:
[C][wmbus:333]:     MOSI Pin: GPIO23
[C][wmbus:334]:     MISO Pin: GPIO19
[C][wmbus:335]:     CLK Pin:  GPIO18
[C][wmbus:336]:     CS Pin:   GPIO16
[C][wmbus:337]:     GDO0 Pin: GPIO33
[C][wmbus:338]:     GDO2 Pin: GPIO14
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400f7b7b  PS      : 0x00060330  A0      : 0x800f9a5b  A1      : 0x3ffda2c0  
A2      : 0x3ffb08e4  A3      : 0x3f41387b  A4      : 0x400dde38  A5      : 0x3ffda300  
A6      : 0x00000000  A7      : 0x3ffc582c  A8      : 0x800f7b7b  A9      : 0x3ffda2a0  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3f4134c7  A13     : 0x3f413aa1  
A14     : 0x3ffda31c  A15     : 0x408b2799  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x4008ac4d  LEND    : 0x4008ac5d  LCOUNT  : 0xfffffff3  

Backtrace:0x400f7b78:0x3ffda2c00x400f9a58:0x3ffda350 0x400f8ae9:0x3ffda370 0x400faf4e:0x3ffda3a0 0x4010521d:0x3ffda3c0 

ELF file SHA256: 0000000000000000

Rebooting...
```

Wifi issues are as follows:
- Auth Expired
- Authentication Failed

No changes related to the wifi or router settings has been done. I uploaded the firmware to different ESP32 but results are the same. I can provide more details logs if needed.

```
WARNING Decoded 0x400f7b7b: __gnu_cxx::__normal_iterator<DriverInfo**, std::vector<DriverInfo*, std::allocator<DriverInfo*> > >::__normal_iterator(DriverInfo** const&)
at toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/stl_iterator.h:784 (discriminator 3)
 (inlined by) std::vector<DriverInfo*, std::allocator<DriverInfo*> >::end()
at toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/stl_vector.h:717 (discriminator 3)
 (inlined by) esphome::wmbus::WMBusComponent::dump_config()
at esphome/components/wmbus/wmbus.cpp:340 (discriminator 3)
```

```
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x400f7b78: esphome::wmbus::WMBusComponent::dump_config() at esphome/components/wmbus/wmbus.cpp:340 (discriminator 3)
WARNING Decoded 0x400f9a58: esphome::Component::call_dump_config() at esphome/core/component.cpp:80
WARNING Decoded 0x400f8ae9: esphome::Application::loop() at esphome/core/application.cpp:106
WARNING Decoded 0x400faf4e: loop() at main.cpp:442
WARNING Decoded 0x4010521d: loopTask(void*) at framework-arduinoespressif32/cores/esp32/main.cpp:50
```

My setup:
ESP32-WROOM32
ESPHome: 2024.9.2
WM-Bus component version: v4.0.8-1.17.1-b8f4a945

Config:
```yaml
substitutions:
  name: watermeter-cc1101
  friendly_name: Watermeter CC1101

esphome:
  name: ${name}
  friendly_name: ${friendly_name}

esp32:
  board: esp32dev
  framework:
    type: arduino

external_components:
  - source: github://SzczepanLeon/esphome-components@4.0.8
    refresh: 0d
    components: [ wmbus ]

wmbus:
  mosi_pin: GPIO23
  miso_pin: GPIO19
  clk_pin:  GPIO18
  cs_pin:   GPIO16
  gdo0_pin: GPIO33
  gdo2_pin: GPIO14

  frequency: 868.950
  all_drivers: False
  sync_mode: True
  log_all: True

logger:
  level: VERY_VERBOSE

api:
  port: !secret api_port
  encryption:
    key: !secret api_encryption_key

ota:
  platform: esphome
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.50.38
    gateway: 192.168.50.1
    subnet: 255.255.255.0

time:
  - platform: sntp
    id: time_sntp

sensor:
  - platform: wmbus
    meter_id: 0x0001
    type: apator162
    key: "00000000000000000000000000000000"
    sensors:
      - name: "Main water RSSi"
        field: "rssi"
        accuracy_decimals: 0
        unit_of_measurement: "dBm"
        device_class: "signal_strength"
        state_class: "measurement"
        entity_category: "diagnostic"
      - name: "Main water total usage"
        field: "total"
        accuracy_decimals: 3
        unit_of_measurement: "m³"
        device_class: "water"
        state_class: "total_increasing"
        icon: "mdi:water"

  - platform: wmbus
    meter_id: 0x0002
    type: apator162
    key: "00000000000000000000000000000000"
    sensors:
      - name: "Garden water RSSi"
        field: "rssi"
        accuracy_decimals: 0
        unit_of_measurement: "dBm"
        device_class: "signal_strength"
        state_class: "measurement"
        entity_category: "diagnostic"
      - name: "Garden water total usage"
        field: "total"
        accuracy_decimals: 3
        unit_of_measurement: "m³"
        device_class: "water"
        state_class: "total_increasing"
        icon: "mdi:water"
```

I would much appreciate any help you can provide.

kam821, 2024-10-03:
+1, recent releases have brought weird behavior, e.g. previously rock-solid ESP32 DOIT Devkit is just bootlooping and ESPHome returns: Can't connect to ESPHome API.
Pinning the version to the 3.2.3 completely solves the issue.

SzczepanLeon, 2024-10-04:
I need more logs from serial port.

VivantSenior, 2024-10-06:
> I need more logs from serial port.
> 
> @VivantSenior the same place / log ? Please share YAML

Very verbose logs from the serial port sent via the PM. You can find yaml config at the end of the first post.

SzczepanLeon, 2024-10-06:
Please try with one sensor defined (one apator).
Both with VERY_VERBOSE and VERBOSE log level.

VivantSenior, 2024-10-06:
Logs send via the PM.

Config with just one Apator sensor added:
```yaml
substitutions:
  name: watermeter-cc1101
  friendly_name: Watermeter CC1101

esphome:
  name: ${name}
  friendly_name: ${friendly_name}

esp32:
  board: esp32dev
  framework:
    type: arduino

external_components:
  - source: github://SzczepanLeon/esphome-components@4.0.8
    refresh: 0d
    components: [ wmbus ]

wmbus:
  mosi_pin: GPIO23
  miso_pin: GPIO19
  clk_pin:  GPIO18
  cs_pin:   GPIO16
  gdo0_pin: GPIO33
  gdo2_pin: GPIO14

logger:
  level: VERBOSE

captive_portal:

api:
  port: !secret api_port
  encryption:
    key: !secret api_encryption_key

ota:
  platform: esphome
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.50.38
    gateway: 192.168.50.1
    subnet: 255.255.255.0

  ap:
    ssid: "${friendly_name} hotspot"
    password: !secret ap_password

time:
  - platform: homeassistant
    id: homeassistant_time

sensor:
  - platform: wmbus
    meter_id: 0x00001
    type: apator162
    key: "00000000000000000000000000000000"
    sensors:
      - name: "Main water total usage"
        field: "total"
        accuracy_decimals: 3
        unit_of_measurement: "m³"
        device_class: "water"
        state_class: "total_increasing"
        icon: "mdi:water-pump"
```

SzczepanLeon, 2024-10-06:
This is quite strange issue ;-)
Not related to C++ code but related to ESPHome python part.
To be more precise how platformio files (ini) are generated.

Please try with modified YAML.
Sections **wmbus** and **sensor** should be at the end.
First **wmbus** then **sensor** - without any other sections between those two.

Maybe there is another combination, but right now I don't have time to check it deeper.

VivantSenior, 2024-10-06:
You were right. Changing order of the nodes helped. 4.0.8 works :) Thank you

closed 2024-10-06 by VivantSenior