SzczepanLeon/esphome-components #134: Decode 3 out of 6 failed..

jpcs2012, 2024-10-05:
22:59:56	[V]	[rxLoop:167]	Have 45 bytes from CC1101 Rx, RSSI: -94 dBm LQI: 129
22:59:56	[D]	[mbus:034]	Processing T1 A frame
22:59:56	[V]	[mbus:045]	Frame: 36571C2D6734393C71B6638D96C58D5AC3B198E69D34B58634BA5372C36335364BD327E56938F1693C74F4B65A (45) [RAW]
22:59:56	[V]	[3of6:083]	Decode 3 out of 6 failed..

I've been using this code for over a year, but with the v4 update I've been having a lot of problems decoding the frame. I've already managed to get data after the update, but when I need to update the Esphome version, it always fails again. help plz.

ESPHome Current version: 2024.9.2 
Component: 3.2.3

Code:

esphome:
  name: "water-meter-esp"
  comment: "water-meter-esp"

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

external_components:
  - source: github://SzczepanLeon/esphome-components@3.2.3
    refresh: 0s
    components: [wmbus]

## ---------------------------------------------------
captive_portal:

## ---------------------------------------------------
## LOGGER COMPONENT
## ---------------------------------------------------
logger:
  id: appslogger
  level: "VERY_VERBOSE"
  baud_rate: "0"
  logs:
    wmbus: "VERY_VERBOSE"
    wMBus-lib: "VERY_VERBOSE"

## ---------------------------------------------------
## OTA COMPONENT
## ---------------------------------------------------
ota:
  platform: esphome
  password: !secret ota_pswd

## WEBSERVER COMPONENT
web_server:
  port: 80
## ---------------------------------------------------
## SNTP COMPONENT
## ---------------------------------------------------
time:
  - platform: sntp
    id: time_sntp
    timezone: Europe/Berlin
    servers:
      - 0.at.pool.ntp.org
      - 0.pool.ntp.org
      - 1.pool.ntp.org
    on_time_sync:
      then:
        - logger.log:
            tag: "system"
            level: INFO
            format: "Synchronized sntp clock"

wmbus:
  mosi_pin: GPIO23   
  miso_pin: GPIO19   
  clk_pin: GPIO18 
  cs_pin: GPIO05    
  gdo0_pin: GPIO16   
  gdo2_pin: GPIO17  

  log_unknown: True

## ---------------------------------------------------
## SWITCHES
## ---------------------------------------------------
switch:

  - platform: restart
    name: "Restart"
    id: restart_switch

## ---------------------------------------------------
## SENSORS
## ---------------------------------------------------
sensor:

  - platform: wmbus
    meter_id: "0x211ADD27"
    type: izar
    add_prefix: true

    # The LQI value reported by the CC1101 is a 7 bit unsigned number with a range from 0 to 127.
    # Note that a lower value indicates a better link.
    # The LQI of a received packet will be bad (higher number) when there is lot of interference.
    lqi:
      id: wmbus_cc1101_lqi
      name: "Watermeter CC1101 LQI"
      unit_of_measurement: "lqi"
      entity_category: "diagnostic"

    # The RSSI value reported by the CC1101 is a 8 bit signed number with an effective
    # range from -138 dBm to -10.5 dBm when the CC1101 is operating around 868 MHz.
    # RSSI stands for received signal strength (power) indication (in dBm).
    # A higher value indicates higher power.
    rssi:
      id: wmbus_cc1101_rssi
      name: "Watermeter CC1101 RSSI"
      unit_of_measurement: "%"
      entity_category: "diagnostic"

    # reports the watermeter display value
    total_water_m3:
      id: "waterdisplay"
      name: "Watermeter Display"
      unit_of_measurement: "m³"
      state_class: total_increasing
      device_class: "water"
      accuracy_decimals: 3
      icon: mdi:counter

    # get the last month total watermter m3 from the wmbus telegram  (wM-Bus 2.1.10)
    last_month_total_water_m3:
      name: "Water last month"
      id: "waterdisplay_lastmonth"
      unit_of_measurement: "m³"
      state_class: total_increasing
      device_class: "water"
      accuracy_decimals: 3
      icon: mdi:counter

    # water current month (wM-Bus v2.1.4)
    current_month_total_water_l:
      name: "Water current month"
      id: "watermonth"
      accuracy_decimals: 2
      unit_of_measurement: "L"
      state_class: total_increasing
      device_class: "water"

    # get the battery life time (wM-Bus v2.1.4)
    remaining_battery_life_y:
      name: "Watermeter Battery Life"
      id: "watermeter_batterie"
      accuracy_decimals: 2
      unit_of_measurement: "Years"
      icon: mdi:battery

    # get the last transmit periode (wM-Bus v2.1.4)
    transmit_period_s:
      name: "Watermeter transmit periode"
      id: "watermeter_transmit_periode"
      unit_of_measurement: "sec"
      accuracy_decimals: 2
      icon: mdi:timelapse

    ## get the current watermeter alarms and publish the text message (wM-Bus 2.1.10)
    current_alarms:
      id: "watermeter_current_alarms"
      name: "Watermeter current alarm code"
      icon: mdi:alarm-light
      entity_category: "diagnostic"

    ## get the prevois watermeter alarms and publish the text message (wM-Bus 2.1.10)
    previous_alarms:
      id: "watermeter_previous_alarms"
      name: "Watermeter pervious alarm code"
      icon: mdi:alarm-light
      entity_category: "diagnostic"

  # Wifi quality RSSI, internal used to calculate the Wifi quality RSSI in percentage
  - platform: wifi_signal
    id: wifi_signal_db
    update_interval: 60s
    internal: true
    disabled_by_default: true

  # Wifi quality RSSI in percentage
  # Received Signal Strength (RSSI) is a measure of incoherent
  ## (raw) RF power in a channel.
  - platform: copy # Reports the WiFi signal strength in %
    source_id: wifi_signal_db
    name: "Device WLAN Signal"
    filters:
      - lambda: return min(max(2 * (x + 100.0), 0.0), 100.0);
    unit_of_measurement: "%"
    entity_category: "diagnostic"

Thank you in advance for help.

SzczepanLeon, 2024-10-05:
Please try:
sync_mode = true

jpcs2012, 2024-10-06:
Well, no luck... I tried but the error remains the same.  The strange thing is that when I did the initial setup I was able to decode the frame on the page https://wmbusmeters.org, and now if I try to place this frame on the page I can't get any information. Could it be a problem with the meter?

SzczepanLeon, 2024-10-06:
> I was able to decode the frame on the page https://wmbusmeters.org, and now if I try to place this frame on the page I can't get any information

What frame? From trace:
```
22:59:56	[V]	[mbus:045]	Frame: 36571C2D6734393C71B6638D96C58D5AC3B198E69D34B58634BA5372C36335364BD327E56938F1693C74F4B65A (45) [RAW]
```

It is not possible to decode that in wmbusmeters because it is RAW frame.

In general every **Decode 3 out of 6 failed** means that there is something wrong with data received from radio.

VivantSenior, 2024-10-10:
I'm having the same error:
```
[V][3of6:083]: Decode 3 out of 6 failed..
```

SzczepanLeon, 2024-10-10:
This is not error. Just information that Decode failed - probably due to wrong data/data with error.

tomaszn, 2024-10-11:
I see this error when too far from the meter.

VivantSenior, 2024-10-13:
> This is not error. Just information that Decode failed - probably due to wrong data/data with error.

I understand by I don't recall this warning / information in the logs before 4.0.

> I have observed that this error occurs when the receiver is positioned too far from the meter. My meter is of `type: apator162`.

Too far or rather you meant the signal quality is poor and some data are lost therefor the frame is not complete? I have also apator162.

tomaszn, 2024-10-13:
> > This is not error. Just information that Decode failed - probably due to wrong data/data with error.
> 
> I understand by I don't recall this warning / information in the logs before 4.0.

Sir, I see it in the logs when using version 3.2.3.

OdoMarTus, 2024-10-22:
For me, too, with 3.2.3, this information appears next to Apator 16-2, but the distance is only a few centimeters.

yanoschik, 2024-12-20:
Any updates on this issue?
What can be done?

I'm having the same problem receiving Apator meters (AT-WMBUS-16-2 HW:3 FW:3), expected driver type apator162 ...
Running latest ESPHome v 2024.11.3 & wmbus v 4.1.0.

`
[09:02:23][V][rxLoop:167]: Have 191 bytes from CC1101 Rx, RSSI: -77 dBm LQI: 134
[09:02:23][D][mbus:035]: Received T1 A frame
[09:02:23][V][mbus:046]: Frame: 6B271C58D59A6997133995935995932DAB0B4E665C596696B192F42D6D1A959C72D0B69A9A564BC6635AB162F464E6994CD9968DA65AB23C4D4D98CD99398D6B2523A9271C58D59A6997133995935995932DAB0B4E665C596696B192F42D6D1A959C72D0B69A9A564BC6635AB162F464E6994CD9968DA65AB23C68B36CC8E8F4B2C7164F264B2E3CA697133995935995932DAB0B4E665C596696B192F42D6D1A959C72D0B69A9A564BC6635AB162F464E6994CD9968DA65AB23C68B36CC8E8 (191) [RAW]
[09:02:23][V][3of6:083]: Decode 3 out of 6 failed..
`
...getting alike 191 bytes raw all the time

sync_mode has no influence ...
Distance has no influence, tried 1m form meters ...
At the same time, I'm receiving data from other meters (heating - qcaloric) with no problem. So I'm not expecting issues with the RF module (CC1101) or reception.
Yes there are multiple transmissions in the environment, as the same kind of meters are used in all apartments in our building (water = apator, and heating = qcaloric), but I only have problem with the watermeters data...

Thanks

SzczepanLeon, 2024-12-20:
No update.
Maybe You can try switch from CC1101 to SX?

yanoschik, 2024-12-20:
Have no experience with SX modules... Is it a drop-in replacement working in esphome?

Something like CMT2300A / E49-900M20S ?

SzczepanLeon, 2024-12-20:
Should be possible with that chip - but I never tried.

DGrebe, 2025-02-15:
Hi, I also receive the Decode 3 of 6 failed error, but only for wmmeters version 4 on ESP32 with RSSI ~ 50 dBm and distance of 1 meter to sender

2 mbus senders in my home
- izar watermeter
- kamheat heatmeter

1. setup (should be replaced) wmmeter add-on on homeassistant with cc1011 chip via USB (5 meters and 2 floors to senders) - works for both meters

2. setup (test setup) wmmeter-gateway-v3 on ESP8266 D1 mini with CC1011 (1 meter, direct view to senders) works for both meters (wmbus-gateway-v3.yaml/.log)

- external_components:
  - source: github://SzczepanLeon/esphome-components@3.2.3  # due to esp8266 chip has not enough memory for v4

  3V3   -> VCC 
  GND -> GND
  D7    -> MOSI
  D5    -> SCK   
  D6    -> MISO
  D1    -> GDO2
  D2    -> GDO0
  D8    -> CSN   

  - platform: wmbus
    type: izar
    meter_id: 0x67980778

  - platform: wmbus
    type: kamheat
    meter_id: 0x82922976 
    key: "xxx"

    RSSI: -53 dBm

3. setup (target) wmmeter-gateway on ESP32 D1 mini32 with CC1011 (1 meter, direct view to senders) works for izar meter, kamheat meter displays  Decode 3 of 6 failed error for kamheat messages (wmbus-gateway.yaml/.log)
 
- external_components:
  - source: github://SzczepanLeon/esphome-components@main

  3V3   -> VCC   
  GND -> GND  
  D7    -> MOSI 
  D5    -> SCK    
  D6    -> MISO 
  D1    -> GDO2
  D2    -> GDO0
  D8    -> CSN   

  - platform: wmbus
    type: izar
    meter_id: 0x20446798

  - platform: wmbus
    type: kamheat

    RSSI: -47 dBm

kamheat RAW Messages differ after 51 bytes (second line)
V3: 
67271C3B13B44DA3A594EB0E71659C71958B4E65A95966563996B272CD232F48DC2CB2DC5A9A4EB3264B2E64CE8CB99CD0EB2C
4CBB316962E3B3439C58ECB259AD2C37272CC9998D98E4ECC5CB1A68D95C9669B14EC2E69539669A9C65CB2699CA39B1B0D72934B4F468DC9CB26A6CA5AC4D3632CBB1AC8D38E9719A639CA698CE65A70B35A58D58B2CBD26596D2634B2F2999A56B0D7348F23ACB0D4CED1699C2E69B2A

V4: 
67271C3B13B44DA3A594EB0E71659C71958B4E65A95966563996B272CD232F48DC2CB2DC5A9A4EB3264B2E64CE8CB99CD0EB2C
2C487261A2A946E0852103708B1776271C3B13B44DA3A594EB0E71659C71958B4E65A95966563996B272CD232F48DC2CB2DC5A9A4EB3264B2E64CE8CB99CD0EB2C4CBB316962E3B3439C58ECB259AD2C373B13B44DA3A594EB0E71659C71958B4E65A95966563996B272CD232F48DC2CB2

How far should the CC1011 board be away from the ESP32 board to avoid interferences?

----

I noted that the ID for the izar meter differs between v3 (0x67980778) and v4 (0x20446798). Any reason for that?
Therefore I kept the id out of the kamheat V4 config.

-----
config and Log files V3 and V4

[wmbus-gateway-v3.yaml.txt](https://github.com/user-attachments/files/18810991/wmbus-gateway-v3.yaml.txt)
[wmbus-gateway-v3.log](https://github.com/user-attachments/files/18810982/wmbus-gateway-v3.log)

[wmbus-gateway.yaml.txt](https://github.com/user-attachments/files/18810990/wmbus-gateway.yaml.txt)
[wmbus-gateway.log](https://github.com/user-attachments/files/18810981/wmbus-gateway.log)

DGrebe, 2025-02-15:
I have moved from D1 mini32 to Beetle ESP32C3, With this setup I got V4 working.

closed 2025-02-19 by SzczepanLeon

yanoschik, 2025-02-19:
Why was this marked completed??
The issue is not resolved at all... I see no solution for my case.

SzczepanLeon, 2025-02-19:
not related to software