SzczepanLeon/esphome-components #135: Boot loop on ESP32-C3 (v4.0)

ThomDietrich, 2024-10-07:
Hey again,
as described in https://github.com/SzczepanLeon/esphome-components/issues/120 I am trying to get v4.0 running on an ESP32-C3. Still working with the same minimal configuration as in #120.

After successful compilation the microcontroller goes into a boot loop. This is the log extract of the event:

```
[00:05:20]ESP-ROM:esp32c3-api1-20210207
[00:05:20]Build:Feb  7 2021
[00:05:20]rst:0x3 (RTC_SW_SYS_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
[00:05:20]Saved PC:0x4038217c
WARNING Decoded 0x4038217c: esp_restart_noos_dig at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:44 (discriminator 1)
[00:05:20]SPIWP:0xee
[00:05:20]mode:DIO, clock div:1
[00:05:20]load:0x3fcd6100,len:0x38c
[00:05:20]load:0x403ce000,len:0x6ac
[00:05:20]load:0x403d0000,len:0x24e4
[00:05:20]entry 0x403ce000
```

----

And here is the log with a  bumped framework version.

```yaml
esp32:
  framework:
    version: 2.0.17
    platform_version: 6.5.0
```

```
[10:54:41][C][wmbus:305]: wM-Bus v4.0.8-1.17.1-b8f4a945:
[10:54:41][C][wmbus:323]:   MAC: 8072B434
[10:54:41][C][wmbus:328]:   CC1101 frequency: 868.950 MHz
[10:54:41][C][wmbus:329]:   CC1101 SPI bus:
[10:54:41][C][wmbus:333]:     MOSI Pin: GPIO6
[10:54:41][C][wmbus:334]:     MISO Pin: GPIO7
[10:54:41][C][wmbus:335]:     CLK Pin:  GPIO5
[10:54:41][C][wmbus:336]:     CS Pin:   GPIO4
[10:54:41][C][wmbus:337]:     GDO0 Pin: GPIO3
[10:54:41][C][wmbus:338]:     GDO2 Pin: GPIO8
[10:54:41]Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.
[10:54:41]
[10:54:41]Core  0 register dump:
[10:54:41]MEPC    : 0x420319f6  RA      : 0x420319f6  SP      : 0x3fcae030  GP      : 0x3fc8e400
WARNING Decoded 0x420319f6: __gnu_cxx::__normal_iterator<DriverInfo**, std::vector<DriverInfo*, std::allocator<DriverInfo*> > >::__normal_iterator(DriverInfo** const&) at c:\users\thomas.dietrich\.platformio\packages\toolchain-riscv32-esp@8.4.0+2021r2-patch5\riscv32-esp-elf\include\c++\8.4.0\bits/stl_iterator.h:784 (discriminator 3)
 (inlined by) std::vector<DriverInfo*, std::allocator<DriverInfo*> >::end() at c:\users\thomas.dietrich\.platformio\packages\toolchain-riscv32-esp@8.4.0+2021r2-patch5\riscv32-esp-elf\include\c++\8.4.0\bits/stl_vector.h:717 (discriminator 3)
 (inlined by) esphome::wmbus::WMBusComponent::dump_config() at C:\Users\thomas.dietrich\Repos-Private\esphome-projects\.esphome\build\wasserzaehler/src/esphome/components/wmbus/wmbus.cpp:340 (discriminator 3)
WARNING Decoded 0x420319f6: __gnu_cxx::__normal_iterator<DriverInfo**, std::vector<DriverInfo*, std::allocator<DriverInfo*> > >::__normal_iterator(DriverInfo** const&) at c:\users\thomas.dietrich\.platformio\packages\toolchain-riscv32-esp@8.4.0+2021r2-patch5\riscv32-esp-elf\include\c++\8.4.0\bits/stl_iterator.h:784 (discriminator 3)
 (inlined by) std::vector<DriverInfo*, std::allocator<DriverInfo*> >::end() at c:\users\thomas.dietrich\.platformio\packages\toolchain-riscv32-esp@8.4.0+2021r2-patch5\riscv32-esp-elf\include\c++\8.4.0\bits/stl_vector.h:717 (discriminator 3)
 (inlined by) esphome::wmbus::WMBusComponent::dump_config() at C:\Users\thomas.dietrich\Repos-Private\esphome-projects\.esphome\build\wasserzaehler/src/esphome/components/wmbus/wmbus.cpp:340 (discriminator 3)
[10:54:42]TP      : 0x3fc7d754  T0      : 0x5b6d3533  T1      : 0x47202020  T2      : 0x3b305b1b  
[10:54:42]S0/FP   : 0x3fc9c00c  S1      : 0x3c107000  A0      : 0x00000000  A1      : 0x3c0f4770
[10:54:42]A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x3fc98000  
[10:54:42]A6      : 0x3a6e6950  A7      : 0x20324f44  S2      : 0x00001b55  S3      : 0x3fc98000
[10:54:42]S4      : 0x3fcbed70  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
[10:54:42]S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
[10:54:42]T3      : 0x20203a5d  T4      : 0x3833333a  T5      : 0x7375626d  T6      : 0x775b5d43  
[10:54:42]MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x00000004
[10:54:42]MHARTID : 0x00000000
[10:54:42]
[10:54:42]Stack memory:
[10:54:42]3fcae030: 0x3fc8f930 0x00000000 0x3fcbeb8c 0x40383b32 0x40389b7a 0x40389b64 0x3fcae0c0 0x3fc8e400
[10:54:42]3fcae050: 0x3fc7d754 0x3fcae05c 0x00000000 0xffffff00 0x3fc921f4 0x00001b52 0x00000001 0x00000001
[10:54:42]3fcae070: 0x00000000 0x00000001 0x600c0000 0x3fc98000 0xfa000000 0x3fcae08c 0x00000005 0x4f495047
[10:54:42]3fcae090: 0x3fcb0038 0x00000000 0x00000000 0x3dc7eaf0 0x00000000 0x00000000 0x3fcbed70 0x3fc9c00c
[10:54:42]3fcae0b0: 0x00001b55 0x00001b52 0x3fc9c00c 0x42033d4c 0x3fcbed70 0x00000102 0x3fc921f4 0x42032a46
[10:54:42]3fcae0d0: 0x303a6174 0x6f2e656d 0x00001b54 0x3fc9c00c 0x00000000 0x00000000 0x00000000 0x00000000
[10:54:42]3fcae0f0: 0x00000000 0x3fc98000 0x420454f0 0x42049214 0x00000000 0x00000000 0x00000000 0x40389aa8
[10:54:42]3fcae110: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
[... 85 lines cut ...]
  - platform: sntp
    id: time_sntp
    timezone: Europe/Berlin

external_components:
  - source: github://SzczepanLeon/esphome-components@main
    refresh: 0d
    components: [ wmbus ]

wmbus:
  # frequency: 868.950
  # Pin 1 - GND
  # Pin 2 - VCC - Power supply 1.8 V-3.6V DC
  gdo0_pin: GPIO3 # GDO0 - Information output pin
  cs_pin: GPIO4 # CSN - Chip selection pin, for starting a SPI communication
  clk_pin: GPIO5 # SCK - SPI clock pin
  mosi_pin: GPIO6 # MOSI - SPI data input pin
  miso_pin: GPIO7 # MISO/GDO1 - SPI data output pin
  gdo2_pin: GPIO8 # GDO2 - Information output pin
  # led_pin: GPIO0
  # led_blink_time: "1s"
  # all_drivers: False
  # sync_mode: True
  # log_all: True
```

SzczepanLeon, 2024-10-07:
Please check issue https://github.com/SzczepanLeon/esphome-components/issues/133 and YAML.

ThomDietrich, 2024-10-07:
Sadly no change. I did define a sensor and switched to the fix branch. Tested with standard and mentioned higher framework version.

```
[18:08:09][C][wmbus:305]: wM-Bus v4.0.9-1.17.1-b8f4a945:
[18:08:09][C][wmbus:323]:   MAC: 8072B434
[18:08:09][C][wmbus:325]:   CC1101 frequency: 868.950 MHz
[18:08:09][C][wmbus:326]:   CC1101 SPI bus:
[18:08:09][C][wmbus:330]:     MOSI Pin: GPIO6
[18:08:09][C][wmbus:331]:     MISO Pin: GPIO7
[18:08:09][C][wmbus:332]:     CLK Pin:  GPIO5
[18:08:09][C][wmbus:333]:     CS Pin:   GPIO4
[18:08:09][C][wmbus:334]:     GDO0 Pin: GPIO3
[18:08:09][C][wmbus:335]:     GDO2 Pin: GPIO8
[18:08:09]
[18:08:09]abort() was called at PC 0x420bf8d3 on core 0
[18:08:09]Core  0 register dump:
[18:08:09]MEPC    : 0x40382578  RA      : 0x40386e42  SP      : 0x3fcadcd0  GP      : 0x3fc8e400
WARNING Decoded 0x40382578: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:408
WARNING Decoded 0x40382578: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:408
[18:08:10]TP      : 0x3fc811dc  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
[18:08:10]S0/FP   : 0x3fcadcfc  S1      : 0x3fcadcfc  A0      : 0x3fcadd08  A1      : 0x3fcadcea
[18:08:10]A2      : 0x00000000  A3      : 0x3fcadd35  A4      : 0x00000001  A5      : 0x3fc98000
[18:08:10]A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x3fcbeff4  S3      : 0x3fc98000
[18:08:10]S4      : 0x3c0f5000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
[18:08:10]S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
[18:08:10]T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
[18:08:10]MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
[18:08:10]MHARTID : 0x00000000  
[18:08:10]
[18:08:10]Stack memory:
[18:08:10]3fcadcd0: 0x00000000 0x00000000 0x3fcadce8 0x4038c460 0x00000000 0x00000000 0x3fc90030 0x3fc90d60
[18:08:10]3fcadcf0: 0x3fcadcfc 0x3fc90d7c 0x3fcadce8 0x62303234 0x33643866 0x00000000 0x726f6261 0x20292874
[18:08:10]3fcadd10: 0x20736177 0x6c6c6163 0x61206465 0x43502074 0x34783020 0x66623032 0x20336438 0x63206e6f
[18:08:10]3fcadd30: 0x2065726f 0x00000030 0x40380000 0xfa5fa26f 0x00000000 0x3fcadda0 0x3fcbefd4 0x420bf8d6
[18:08:10]3fcadd50: 0x00000000 0x00000000 0x3fcbefd4 0x3fc982a8 0x3fcbeff4 0x3fcadda0 0x3fcbefd4 0x420bf93a
[18:08:10]3fcadd70: 0x3fcbefd4 0x3fcadda0 0x3fcbefd4 0x420bf87a 0x3c109db4 0x3fcadda0 0x420c1564 0x3c109f3c
[18:08:10]3fcadd90: 0x3c109db4 0x3fcadda0 0x3fcae000 0x420c1322 0x69736162 0x74735f63 0x676e6972 0x72653a3a
[18:08:10]3fcaddb0: 0x3a657361 0x705f5f20 0x2820736f 0x63696877 0x73692068 0x39323420 0x37363934 0x29343932
[18:08:10]3fcaddd0: 0x74203e20 0x2d736968 0x7a69733e 0x20292865 0x69687728 0x69206863 0x29302073 0x3fcbee00
[18:08:10]3fcaddf0: 0x0000002c 0x3fcbee0c 0x0000005c 0x0000005c 0x3fcbee10 0x3fcbefcc 0x3fc98c14 0x4038bab4
[18:08:10]3fcade10: 0x00001800 0x00000000 0x3fc98184 0x3c0f8000 0x00000000 0x00000000 0x00001800 0x3fc98000
[18:08:10]3fcade30: 0x0000002c 0x3fcbee10 0x3fc98c00 0x4038c05c 0x00000000 0x00000000 0x00060000 0xbaad5678
[18:08:10]3fcade50: 0x3fc98c00 0x3fcbee18 0x00000050 0x4038c1e8 0x00000020 0x3fc992f8 0x00001800 0x3fcbee10
[18:08:10]3fcade70: 0x3fc992f8 0x3fcbee18 0x00000050 0x40382bf6 0x00000000 0x0000001c 0x00000000 0x3fc98000
[18:08:10]3fcade90: 0x00001800 0x00000050 0x3fcadf8c 0x40382c26 0x3fcbef94 0x00000050 0x3fcadf8c 0x42040a74
[18:08:10]3fcadeb0: 0x3fc9df38 0x00000000 0x00000001 0xbaad5678 0x00000000 0x00000000 0x0000002f 0xfa5fa26f
[18:08:10]3fcaded0: 0x0000002f 0x0000004a 0x00000000 0x00000001 0x00000000 0x3fc98000 0x00000000 0x3fcbefc2
[18:08:10]3fcadef0: 0x3fc9dd24 0x3fcbef8c 0x3fc98c00 0x4038bf7e 0x3fc9dd24 0x3fcbef94 0x3fc98c00 0x4038c0d2
[18:08:10]3fcadf10: 0x3fc9dd24 0x3fcbef94 0x3fcadf8c 0x420381fa 0x00000000 0x00000000 0x3c0f587c 0x00000004
[18:08:10]3fcadf30: 0x3fc9dd24 0x3fcbefcc 0x3fc98c00 0x4038bf7e 0x3fc9dd24 0x3fcbefd4 0x3fc98c00 0x4038c0d2
[18:08:10]3fcadf50: 0x00000000 0x00000000 0x3fcadf8c 0x4204098c 0x420c12b8 0x420c0870 0x3fcadfe0 0x3fc8e400
[18:08:10]3fcadf70: 0x3fc811dc 0x5b6d3533 0x47202020 0x3b305b1b 0x3fcae064 0xfffffffe 0x3c109db4 0x3c109e28
[18:08:10]3fcadf90: 0xfffffffe 0x00000000 0x3fc98184 0xfffffffe 0x3a6e6950 0x20324f44 0xffffffff 0x3fc98000
[18:08:10]3fcadfb0: 0x3c0f5000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[18:08:10]3fcadfd0: 0x20203a5d 0x3533333a 0x7375626d 0x775b5d43 0x00000000 0x00000000 0x00000000 0x3fcae004
[18:08:10]3fcadff0: 0xffffffff 0xfffffffe 0x3fcae064 0x420c0870 0x00001800 0x3c109e28 0xfffffffe 0x00000000
[18:08:10]3fcae010: 0x3fc98184 0xfffffffe 0x3a6e6950 0x20324f44 0x3fc8f930 0x3c10a8f4 0x3fc97ddc 0x420c0b5c
[18:08:10]3fcae030: 0x00000000 0x3c0f6000 0x3fc9c1c0 0x4202e0c6 0x3fc8f930 0x00000000 0x3fcbef60 0x40383b32
[18:08:10]3fcae050: 0x40389b7a 0x40389b64 0x3fcae0d0 0x3fc8e400 0x3fc811dc 0x3fcae06c 0x00000000 0xffffff00
[18:08:10]3fcae070: 0x3fc921f4 0x00001ac3 0x00000001 0x00000001 0x00000000 0x00000001 0x600c0000 0x3fc98000
[18:08:10]3fcae090: 0xfa000000 0x3fcae09c 0x00000005 0x4f495047 0x3fcb0038 0x00000000 0x00000000 0xfa5fa26f
[18:08:10]3fcae0b0: 0x00000000 0x00000000 0x3fcbee70 0x3fc9c1c0 0x00001ac6 0x00001ac3 0x3fc9c1c0 0x4203030c
[18:08:10]
[18:08:10]
[18:08:10]
[... cut ...]
[18:08:11]entry 0x403cc710
[18:08:11]E (222) esp_core_dump_flash: No core dump partition found!
[18:08:11]E (222) esp_core_dump_flash: No core dump partition found!
[18:08:11][I][logger:156]: Log initialized
[18:08:11][C][safe_mode:079]: There have been 1 suspected unsuccessful boot attempts
[18:08:11][D][esp32.preferences:114]: Saving 1 preferences to flash...
[18:08:11][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[18:08:11][I][app:029]: Running through setup()...
```

SzczepanLeon, 2024-10-07:
Last run looks different.
What about **master** with sensor defined?

ThomDietrich, 2024-10-07:
I can see that too. Back on main branch:

```
[19:06:05][C][wmbus:305]: wM-Bus v4.0.8-1.17.1-b8f4a945:
[19:06:05][C][wmbus:323]:   MAC: 8072B434
[19:06:05][C][wmbus:328]:   CC1101 frequency: 868.950 MHz
[19:06:05][C][wmbus:329]:   CC1101 SPI bus:
[19:06:05][C][wmbus:333]:     MOSI Pin: GPIO6
[19:06:05][C][wmbus:334]:     MISO Pin: GPIO7
[19:06:05][C][wmbus:335]:     CLK Pin:  GPIO5
[19:06:05][C][wmbus:336]:     CS Pin:   GPIO4
[19:06:05][C][wmbus:337]:     GDO0 Pin: GPIO3
[19:06:05][C][wmbus:338]:     GDO2 Pin: GPIO8
[19:06:05]Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.
[19:06:05]
[19:06:05]Core  0 register dump:
[19:06:05]MEPC    : 0x4202e072  RA      : 0x4202e072  SP      : 0x3fcae040  GP      : 0x3fc8e400
WARNING Decoded 0x4202e072: __gnu_cxx::__normal_iterator<DriverInfo**, std::vector<DriverInfo*, std::allocator<DriverInfo*> > >::__normal_iterator(DriverInfo** const&) at c:\users\thomas.dietrich\.platformio\packages\toolchain-riscv32-esp@8.4.0+2021r2-patch5\riscv32-esp-elf\include\c++\8.4.0\bits/stl_iterator.h:
784 (discriminator 3)
 (inlined by) std::vector<DriverInfo*, std::allocator<DriverInfo*> >::end() at c:\users\thomas.dietrich\.platformio\packages\toolchain-riscv32-esp@8.4.0+2021r2-patch5\riscv32-esp-elf\include\c++\8.4.0\bits/stl_vector.h:717 (discriminator 3)
 (inlined by) esphome::wmbus::WMBusComponent::dump_config() at C:\Users\thomas.dietrich\Repos-Private\esphome-projects\.esphome\build\wasserzaehler/src/esphome/components/wmbus/wmbus.cpp:340 (discriminator 3)
WARNING Decoded 0x4202e072: __gnu_cxx::__normal_iterator<DriverInfo**, std::vector<DriverInfo*, std::allocator<DriverInfo*> > >::__normal_iterator(DriverInfo** const&) at c:\users\thomas.dietrich\.platformio\packages\toolchain-riscv32-esp@8.4.0+2021r2-patch5\riscv32-esp-elf\include\c++\8.4.0\bits/stl_iterator.h:
784 (discriminator 3)
 (inlined by) std::vector<DriverInfo*, std::allocator<DriverInfo*> >::end() at c:\users\thomas.dietrich\.platformio\packages\toolchain-riscv32-esp@8.4.0+2021r2-patch5\riscv32-esp-elf\include\c++\8.4.0\bits/stl_vector.h:717 (discriminator 3)
 (inlined by) esphome::wmbus::WMBusComponent::dump_config() at C:\Users\thomas.dietrich\Repos-Private\esphome-projects\.esphome\build\wasserzaehler/src/esphome/components/wmbus/wmbus.cpp:340 (discriminator 3)
[19:06:05]TP      : 0x3fc811dc  T0      : 0x5b6d3533  T1      : 0x47202020  T2      : 0x3b305b1b  
[19:06:05]S0/FP   : 0x3fc9c1c0  S1      : 0x3c0f6000  A0      : 0x00000000  A1      : 0x3c0e3298
[19:06:05]A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x3fc98000
[19:06:05]A6      : 0x3a6e6950  A7      : 0x20324f44  S2      : 0x00001b50  S3      : 0x3fc98000
[19:06:05]S4      : 0x3fcbef80  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
[19:06:05]S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
[19:06:05]T3      : 0x20203a5d  T4      : 0x3833333a  T5      : 0x7375626d  T6      : 0x775b5d43  
[19:06:05]MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x00000004
[19:06:05]MHARTID : 0x00000000
[19:06:05]
[19:06:05]Stack memory:
[19:06:05]3fcae040: 0x3c0e3000 0x00000017 0x3fc9ba34 0x403873ac 0x40389b7a 0x40389b64 0x3fcae0d0 0x3fc8e400
[19:06:05]3fcae060: 0x3fc811dc 0x3fcae06c 0x00000000 0x3b305b00 0x3fc921f4 0x00001b4e 0x00000001 0x00000001
[19:06:05]3fcae080: 0x00000000 0x00000001 0x600c0000 0x3fc98000 0xfa000000 0x3fcae09c 0x00000005 0x4f495047
[19:06:05]3fcae0a0: 0x3fcb0038 0x00000000 0x00000000 0xd96c5675 0x00000000 0x00000000 0x3fcbef80 0x3fc9c1c0
[19:06:05]3fcae0c0: 0x00001b50 0x00001b4e 0x3fc9c1c0 0x42030308 0x3fcbef80 0x00000102 0x3fc921f4 0x4202f016
[19:06:05]3fcae0e0: 0x00000000 0x00000000 0x00001b50 0x3fc9c1c0 0x55202020 0x3a5d3434 0x00000000 0x00000000
[19:06:05]3fcae100: 0x00000000 0x3fc98000 0x420401c6 0x42043ab8 0x00000000 0x00000000 0x00000000 0x40389aa8
[19:06:05]3fcae120: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
[19:06:05]3fcae140: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xbaad5678 0x00000a0c 0xabba1234 0x00000a00
[19:06:05]3fcae160: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
[19:06:05]3fcae180: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
[19:06:05]3fcae1a0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
[19:06:05]3fcae1c0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
[19:06:05]3fcae1e0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
[19:06:05]3fcae200: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
[19:06:05]3fcae220: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
[19:06:05]3fcae240: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
[19:06:05]3fcae260: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
[19:06:05]3fcae280: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
[19:06:05]3fcae2a0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
[19:06:05]3fcae2c0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
[... cut ...]
  clk_pin: GPIO5 # SCK - SPI clock pin
  mosi_pin: GPIO6 # MOSI - SPI data input pin
  miso_pin: GPIO7 # MISO/GDO1 - SPI data output pin
  gdo2_pin: GPIO8 # GDO2 - Information output pin

sensor:
  - platform: wmbus
    meter_id: !secret watermeter_id
    type: hydrus
    key: !secret watermeter_api_key

    sensors:
      - name: "my hot water"
        field: "total"
        accuracy_decimals: 3
        unit_of_measurement: "m³"
        device_class: "water"
        state_class: "total_increasing"
        icon: "mdi:water"
```

SzczepanLeon, 2024-10-08:
Can you share platformio.ini file for both branches?

ThomDietrich, 2024-10-08:
With default framework?

SzczepanLeon, 2024-10-08:
Yes.

ThomDietrich, 2024-10-08:
main branch:

```ini
; Auto generated code by esphome

[common]
lib_deps =
build_flags =
upload_flags =

; ========== AUTO GENERATED CODE BEGIN ===========
[platformio]
description = ESPHome 2024.9.2
[env:wasserzaehler]
board = esp32-c3-devkitm-1
board_build.flash_mode = dio
board_build.mcu = esp32c3
board_build.partitions = partitions.csv
board_upload.flash_size = 4MB
build_flags =
    -DARDUINO_USB_CDC_ON_BOOT=1
    -DARDUINO_USB_MODE=1
    -DESPHOME_LOG_LEVEL=ESPHOME_LOG_LEVEL_DEBUG
    -DUSE_ARDUINO
    -DUSE_ESP32
    -DUSE_ESP32_FRAMEWORK_ARDUINO
    -DUSE_ESP32_VARIANT_ESP32C3
    -Wno-sign-compare
    -Wno-unused-but-set-variable
    -Wno-unused-variable
    -fno-exceptions
build_src_filter =
    +<**/wmbus/driver_hydrus.cpp>
    +<*>
    -<.git/>
    -<.svn/>
    -<**/wmbus/driver_*.cpp>
extra_scripts =
    post:post_build.py
framework = arduino
lib_deps =
    esphome/AsyncTCP-esphome@2.1.4
    WiFi
    FS
    Update
    esphome/ESPAsyncWebServer-esphome@3.2.2
    DNSServer
    ESPmDNS
    bblanchon/ArduinoJson@6.18.5
    SPI
    LSatan/SmartRC-CC1101-Driver-Lib@2.5.7
    ${common.lib_deps}
lib_ldf_mode = off
platform = platformio/espressif32@5.4.0
platform_packages =
    platformio/framework-arduinoespressif32@~3.20005.0
; =========== AUTO GENERATED CODE END ============

```

fix branch:

```ini
; Auto generated code by esphome

[common]
lib_deps =
build_flags =
upload_flags =

; ========== AUTO GENERATED CODE BEGIN ===========
[platformio]
description = ESPHome 2024.9.2
[env:wasserzaehler]
board = esp32-c3-devkitm-1
board_build.flash_mode = dio
board_build.mcu = esp32c3
board_build.partitions = partitions.csv
board_upload.flash_size = 4MB
build_flags =
    -DARDUINO_USB_CDC_ON_BOOT=1
    -DARDUINO_USB_MODE=1
    -DESPHOME_LOG_LEVEL=ESPHOME_LOG_LEVEL_DEBUG
    -DUSE_ARDUINO
    -DUSE_ESP32
    -DUSE_ESP32_FRAMEWORK_ARDUINO
    -DUSE_ESP32_VARIANT_ESP32C3
    -Wno-sign-compare
    -Wno-unused-but-set-variable
    -Wno-unused-variable
    -fno-exceptions
build_src_filter =
    +<**/wmbus/driver_hydrus.cpp>
    +<*>
    -<.git/>
    -<.svn/>
    -<**/wmbus/driver_*.cpp>
    +<**/wmbus/driver_unknown.cpp
extra_scripts =
    post:post_build.py
framework = arduino
lib_deps =
    esphome/AsyncTCP-esphome@2.1.4
    WiFi
    FS
    Update
    esphome/ESPAsyncWebServer-esphome@3.2.2
    DNSServer
    ESPmDNS
    bblanchon/ArduinoJson@6.18.5
    SPI
    LSatan/SmartRC-CC1101-Driver-Lib@2.5.7
    ${common.lib_deps}
lib_ldf_mode = off
platform = platformio/espressif32@5.4.0
platform_packages =
    platformio/framework-arduinoespressif32@~3.20005.0
; =========== AUTO GENERATED CODE END ============

```

SzczepanLeon, 2024-10-08:
Thanks. Both with YAML included earlier?

closed 2024-10-08 by SzczepanLeon

ThomDietrich, 2024-10-08:
Hey! I can confirm that I got it to work now! 

However, there is still an issue or a bug to be solved here. You are correct, it's related to the order of sections. Additionally, I had to find out that my issue is related to my use of packages. I am still playing with the details but I'll post an MWE asap.

SzczepanLeon, 2024-10-08:
> However, there is still an issue or a bug to be solved here. 

I agree with you. It can be quickly fixed by one patch to ESPHome...

ThomDietrich, 2024-10-08:
Long story short: I gave up to understand the issue. 🤷‍♂️ I have a set of files which are include in my various ESPHome projects (via the `packages` directive). As soon as any of those packages is used the order of sections is changed in a way I couldn't figure out or influence.

I've  now composed a flat yaml config and everything is working as expected. Thanks @SzczepanLeon for your great work and the friendly exchange!

 > It can be quickly fixed by one patch to ESPHome...

Did you open or find an issue related to this?

----

Btw. I'll leave this for other users googling for a problem.

```
[22:11:03][D][mbus:035]: Received T1 A frame
[22:11:03][W][wmbus:076]: Link modes not defined in driver . Processing anyway.
[22:11:03][I][wmbus:091]:  [0x81755588] RSSI: -87dBm T: 5344A5119955758176078C008D900F002C25DA7107006E3A13ACA139D1E27A66003107107EF7B462CB7A313198C33202150CD031241DE488C09C48457D28AD83DA235DB6A8BF7E01FB92FC9FE04FF971AB4F4242 (84) T1
A[22:11:03][W][wmbus:095]: Can't find driver for T: 5344A5119955758176078C008D900F002C25DA7107006E3A13ACA139D1E27A66003107107EF7B462CB7A313198C33202150CD031241DE488C09C48457D28AD83DA235DB6A8BF7E01FB92FC9FE04FF971AB4F4242 (84)
```

This is an issue currently present in ESPHome. You need to restructure your yaml file.

SzczepanLeon, 2024-10-08:
> Did you open or find an issue related to this?

Probably yes, but don't know when.

> Btw. I'll leave this for other users googling for a problem.\n\n[22:11:03][D][mbus:035]: Received T1 A frame\n[22:11:03][W][wmbus:076]: Link modes not defined in driver . Processing anyway.

That trace means that You don't have inclyded driver for that telegram.

tOOmki, 2024-10-11:
> > Did you open or find an issue related to this?
> 
> Probably yes, I will create one, but don't know when.

I also have this message.
I swap sections, but no effect.
Can you write what you swapped?

ThomDietrich, 2024-10-11:
My working config:
https://github.com/ThomDietrich/esphome-projects/blob/master/wasserzaehler-hydrus-2-0.yaml