SzczepanLeon/esphome-components #138: text_sensor.wmbus - platform not found in Home Assistant

darek203, 2024-10-15:
Hello,

I'be been using your esp board to read my water meter. Everything worked well so far, until today my water meter was replaced and I wanted to update it's meter_id but it looks like code doesn't want to compile anymore. For some reason I'm getting an error text_sensor.wmbus - platform not found in Home Assistant. 

I'm on 2024.3.1... getting error compiling issue while trying to update it.

[image]

Any suggestions would be a great help!

SzczepanLeon, 2024-10-15:
Please read main github page. From v4.0 YAML is changed.

closed 2024-10-15 by SzczepanLeon

darek203, 2024-10-15:
Is there any most recent pdf instruction available which I could use to install everything from the scratch? I did receive one when I bought your meter, but it looks it's from 2023...

SzczepanLeon, 2024-10-15:
There is no PDF. Everything you have here:
https://github.com/SzczepanLeon/esphome-components