SzczepanLeon/esphome-components #140: Can't get sensor values from flowIQ 2200

shopro, 2024-10-20:
Trying to get readings from my new flowiq2200 watermeter, but no matter what I try to do, the sensors always stay N/A.

I can successfully decode the payload using the wmbusmeters analyzer when receiving the longer (70+) payload (not present in the log).

```
esphome:
  name: watermeter
  platform: ESP32
  board: esp32doit-devkit-v1

external_components:
  - source: github://SzczepanLeon/esphome-components@main
    refresh: 0d
    components: [wmbus]

wifi:
  ssid: !env_var IOT_WIFI_SSID
  password: !env_var IOT_WIFI_PASSWORD
  fast_connect: true
  use_address: "watermeter.local"

captive_portal:

logger:
  level: VERBOSE

api:

ota:
  platform: esphome

web_server:
  port: 80

wmbus:
  mosi_pin: GPIO23
  miso_pin: GPIO19
  clk_pin: GPIO18
  cs_pin: GPIO4
  gdo0_pin: GPIO32
  gdo2_pin: GPIO22

sensor:
  - platform: wmbus
    type: flowiq2200
    meter_id: !env_var METER_ID
    key: !env_var METER_KEY

    sensors:
      - name: "RSSi"
        field: "rssi"
        accuracy_decimals: 0
        unit_of_measurement: "dBm"
        device_class: "signal_strength"
        state_class: "measurement"
        entity_category: "diagnostic"
      - name: "Total water usage"
        field: "total"
        accuracy_decimals: 3
        unit_of_measurement: "m³"
        device_class: "water"
        state_class: "total_increasing"
        icon: "mdi:water"

time:
  - platform: sntp
    id: sntp_time

button:
  - platform: restart
[... 205 lines cut ...]
[05:02:46][V][Telegram.cpp:1131]: (wmbus) parseDLL @0 53
[05:02:46][V][Telegram.cpp:1178]: (wmbus) parseELL @10 43
[05:02:46][W][component:237]: Component wmbus took a long time for an operation (88 ms).
[05:02:46][W][component:238]: Components should block for at most 30 ms.
[05:04:21][V][rxLoop:167]: Have 57 bytes from CC1101 Rx, RSSI: -54 dBm LQI: 128
[05:04:21][D][mbus:024]: Received C1 B frame
[05:04:21][V][mbus:028]: Frame: 3644372C545594123C168D20B1182FA220C7C7C7C7F892B400B2DA5F0E00CE3E94A324BF6A24BACB8AEE2C55AE38ED755EB21135D4901A (55) [with CRC]
[05:04:21][V][mbus:166]: Validating CRC for Block1 + Block2
[05:04:21][V][crc:031]:     calculated: 0x901A, read: 0x901A
[05:04:21][V][mbus:063]: Frame: 3444372C545594123C168D20B1182FA220C7C7C7C7F892B400B2DA5F0E00CE3E94A324BF6A24BACB8AEE2C55AE38ED755EB21135D4 (53) [without CRC]
[05:04:21][V][Telegram.cpp:1131]: (wmbus) parseDLL @0 53
[05:04:21][V][Telegram.cpp:1178]: (wmbus) parseELL @10 43
[05:04:21][W][component:237]: Component wmbus took a long time for an operation (87 ms).
[05:04:21][W][component:238]: Components should block for at most 30 ms.
[05:05:56][V][rxLoop:167]: Have 57 bytes from CC1101 Rx, RSSI: -54 dBm LQI: 128
[05:05:56][D][mbus:024]: Received C1 B frame
[05:05:56][V][mbus:028]: Frame: 3644372C545594123C168D20B22C2FA22071D4792FD50EE2A63EABE74F8EAE7B10A0601EDC261ABCB15FFCD68C03728EB2E9CFA77CA66F (55) [with CRC]
[05:05:56][V][mbus:166]: Validating CRC for Block1 + Block2
[05:05:56][V][crc:031]:     calculated: 0xA66F, read: 0xA66F
[05:05:56][V][mbus:063]: Frame: 3444372C545594123C168D20B22C2FA22071D4792FD50EE2A63EABE74F8EAE7B10A0601EDC261ABCB15FFCD68C03728EB2E9CFA77C (53) [without CRC]
[05:05:56][V][Telegram.cpp:1131]: (wmbus) parseDLL @0 53
[05:05:56][V][Telegram.cpp:1178]: (wmbus) parseELL @10 43
[05:05:57][W][component:237]: Component wmbus took a long time for an operation (94 ms).
[05:05:57][W][component:238]: Components should block for at most 30 ms.
```

SzczepanLeon, 2024-10-21:
Telegrams from log needs key.

Please share logs for long ones.

shopro, 2024-10-22:
I upgraded ESPHome to 2024.10.0 and for some reason the longer payload stopped appearing and the time between short payloads increased, downgrading to the previous version 2024.5.4 I used, it again appears with the exact same config minus the ota platform: esphome part. No sensor data appears on either versions.

The `4F44372C545594123C168D20FCD4E9A22030FCA86D6443359486713D868140EE38EDA6C9486495092E32ABA9AE329E88E5CE878360B26DF66F2346301CE20BE648342F8F8A3ABA976DD346CE8EF736D2` is decoded into the following:

```
{
    "media":"cold water",
    "meter":"flowiq2200",
    "name":"",
    "id":,
    "flow_m3h":0.043,
    "max_flow_m3h":0.31,
    "max_flow_temperature_c":18,
    "min_external_temperature_c":16,
    "min_flow_m3h":0,
    "min_flow_temperature_c":12,
    "target_m3":52.276,
    "total_m3":59.806,
    "status":"OK",
    "target_date":"2024-10-01",
    "timestamp":"2024-10-22T07:06:22Z"
}
```

```[06:45:45][I][app:100]: ESPHome version 2024.5.4 compiled on Oct 22 2024, 06:43:05
[06:45:45][C][wifi:580]: WiFi:
[06:45:45][C][wifi:408]:   Local MAC: [redacted]
[06:45:45][C][wifi:413]:   SSID: 'IoT'[redacted]
[06:45:45][C][wifi:416]:   IP Address: 10.0.7.33
[06:45:45][C][wifi:420]:   BSSID: [redacted]
[06:45:45][C][wifi:421]:   Hostname: 'watermeter'
[06:45:45][C][wifi:423]:   Signal strength: -59 dB ▂▄▆█
[06:45:45][C][wifi:427]:   Channel: 1
[06:45:45][C][wifi:428]:   Subnet: 255.255.255.0
[06:45:45][C][wifi:429]:   Gateway: 10.0.7.1
[06:45:45][C][wifi:430]:   DNS1: 10.0.1.9
[06:45:45][C][wifi:431]:   DNS2: 0.0.0.0
[06:45:45][C][logger:185]: Logger:
[06:45:45][C][logger:186]:   Level: VERBOSE
[06:45:45][C][logger:188]:   Log Baud Rate: 115200
[06:45:45][C][logger:189]:   Hardware UART: UART0
[06:45:45][C][restart.button:017]: Restart Button 'Restart'
[06:45:45][C][captive_portal:088]: Captive Portal:
[06:45:45][C][web_server:173]: Web Server:
[06:45:45][C][web_server:174]:   Address: watermeter.local:80
[06:45:45][C][sntp:055]: SNTP Time:
[06:45:45][C][sntp:056]:   Server 1: '0.pool.ntp.org'
[06:45:45][C][sntp:057]:   Server 2: '1.pool.ntp.org'
[06:45:45][C][sntp:058]:   Server 3: '2.pool.ntp.org'
[06:45:45][C][sntp:059]:   Timezone: 'UTC0'
[06:45:45][C][mdns:115]: mDNS:
[06:45:45][C][mdns:116]:   Hostname: watermeter
[06:45:45][V][mdns:117]:   Services:
[06:45:45][V][mdns:119]:   - _esphomelib, _tcp, 6053
[06:45:45][V][mdns:121]:     TXT: version = 2024.5.4
[06:45:45][V][mdns:121]:     TXT: mac = [redacted]
[06:45:45][V][mdns:121]:     TXT: platform = ESP32
[06:45:45][V][mdns:121]:     TXT: board = esp32doit-devkit-v1
[06:45:45][V][mdns:121]:     TXT: network = wifi
[06:45:45][V][mdns:119]:   - _http, _tcp, 80
[06:45:45][C][ota:096]: Over-The-Air Updates:
[06:45:45][C][ota:097]:   Address: watermeter.local:3232
[06:45:45][C][ota:103]:   OTA version: 2.
[06:45:45][C][api:139]: API Server:
[06:45:45][C][api:140]:   Address: watermeter.local:6053
[06:45:45][C][api:144]:   Using noise encryption: NO
[06:45:45][C][wmbus:326]: wM-Bus v4.0.11-1.17.1-b8f4a945:
[06:45:45][C][wmbus:344]:   MAC: 15286F24
[06:45:45][C][wmbus:346]:   CC1101 frequency: 868.950 MHz
[06:45:45][C][wmbus:347]:   CC1101 SPI bus:
[... 59 lines cut ...]
[06:50:41][D][esp32.preferences:114]: Saving 1 preferences to flash...
[06:50:41][V][esp32.preferences:126]: sync: key: 233825507, len: 4
[06:50:41][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[06:50:41][W][component:237]: Component ota took a long time for an operation (108 ms).
[06:50:41][W][component:238]: Components should block for at most 30 ms.
[06:51:20][V][rxLoop:167]: Have 57 bytes from CC1101 Rx, RSSI: -55 dBm LQI: 128
[06:51:20][D][mbus:024]: Received C1 B frame
[06:51:20][V][mbus:028]: Frame: 3644372C545594123C168D20FBB0E9A220648C9C185E269CA7A830313DEA226337807D63644CAEDBFD1E33D18E401FD64028BBFDD319A8 (55) [with CRC]
[06:51:20][V][mbus:166]: Validating CRC for Block1 + Block2
[06:51:20][V][crc:031]:     calculated: 0x19A8, read: 0x19A8
[06:51:20][V][mbus:063]: Frame: 3444372C545594123C168D20FBB0E9A220648C9C185E269CA7A830313DEA226337807D63644CAEDBFD1E33D18E401FD64028BBFDD3 (53) [without CRC]
[06:51:20][V][Telegram.cpp:1131]: (wmbus) parseDLL @0 53
[06:51:20][V][Telegram.cpp:1178]: (wmbus) parseELL @10 43
[06:51:20][W][component:237]: Component wmbus took a long time for an operation (65 ms).
[06:51:20][W][component:238]: Components should block for at most 30 ms.
[06:52:58][V][rxLoop:167]: Have 82 bytes from CC1101 Rx, RSSI: -55 dBm LQI: 128
[06:52:58][D][mbus:024]: Received C1 B frame
[06:52:58][V][mbus:028]: Frame: 4F44372C545594123C168D20FCD4E9A22030FCA86D6443359486713D868140EE38EDA6C9486495092E32ABA9AE329E88E5CE878360B26DF66F2346301CE20BE648342F8F8A3ABA976DD346CE8EF736D2 (80) [with CRC]
[06:52:58][V][mbus:166]: Validating CRC for Block1 + Block2
[06:52:58][V][crc:031]:     calculated: 0x36D2, read: 0x36D2
[06:52:58][V][mbus:063]: Frame: 4D44372C545594123C168D20FCD4E9A22030FCA86D6443359486713D868140EE38EDA6C9486495092E32ABA9AE329E88E5CE878360B26DF66F2346301CE20BE648342F8F8A3ABA976DD346CE8EF7 (78) [without CRC]
[06:52:59][V][Telegram.cpp:1131]: (wmbus) parseDLL @0 78
[06:52:59][V][Telegram.cpp:1178]: (wmbus) parseELL @10 68
[06:52:59][W][component:237]: Component wmbus took a long time for an operation (77 ms).
[06:52:59][W][component:238]: Components should block for at most 30 ms.```

SzczepanLeon, 2024-10-24:
Please share long frame with key or decoded without key.

shopro, 2024-10-25:
The long frame doesn't appear in ESPHome 2024.10.0, this one is from 2024.5.4

Long Frame: 
```
[V][mbus:028]: Frame: 4F44372C545594123C168D20D4C4E5A3209E4D403E9B9980325A847B43A03377A73ADC09DA1C78980E8862A2C7F2EEE19DD4884467DEEEBB3BC914C088C3CB7F42B5424050CC2344422F40A1DC7D92A6 (80) [with CRC]
[V][mbus:166]: Validating CRC for Block1 + Block2
[V][crc:031]:     calculated: 0x92A6, read: 0x92A6
[V][mbus:063]: Frame: 4D44372C545594123C168D20D4C4E5A3209E4D403E9B9980325A847B43A03377A73ADC09DA1C78980E8862A2C7F2EEE19DD4884467DEEEBB3BC914C088C3CB7F42B5424050CC2344422F40A1DC7D (78) [without CRC]
```

Decoded payload:
```
{
    "media":"cold water",
    "meter":"flowiq2200",
    "name":"",
    "id":"",
    "flow_m3h":0,
    "max_flow_m3h":0.705,
    "max_flow_temperature_c":17,
    "min_external_temperature_c":16,
    "min_flow_m3h":0,
    "min_flow_temperature_c":12,
    "target_m3":52.276,
    "total_m3":60.598,
    "status":"OK",
    "target_date":"2024-10-01",
    "timestamp":"2024-10-25T02:06:33Z"
}
```

SzczepanLeon, 2024-10-25:
Not decoded payuload (YAML) but decoded frame (or frame with key).
On wmbusmeters page it is data after **telegram=|**
for [example](https://wmbusmeters.org/analyze/23442d2c216735241B168d20d4316159218d004cc4c563295905781B73c89Bdd4396a018:auto:0AF58C20AA94E0DC79B300DE3B00B61B) 

```

telegram=|23442D2C216735241B168D20D43161592136BA79EDA8504A_000069A20100D19A01000F0E|

Auto driver    : multical21
Similar driver : multical21 12/12
Using driver   : multical21 00/00
```

shopro, 2024-10-25:
I apologize, I misunderstood what you meant.
```
telegram=|4D44372C545594123C168D20D4C4E5A32014BE78_04FF23000000000413B6EC0000441334CC0000426C013A023B000092013BC102A2013B000006FF1B097000096000A1015B0C91015B11A1016710|

Auto driver    : flowiq2200
Similar driver : flowiq2200 23/29
Using driver   : flowiq2200 00/00
000   : 4d length (77 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : 372c dll-mfct (KAW)
004   : 54559412 dll-id ()
008   : 3c dll-version
009   : 16 dll-type (Cold water meter)
010   : 8d ell-ci-field (ELL: Extended Link Layer II (8 Byte))
011   : 20 ell-cc (slow_resp sync)
012   : d4 ell-acc
013   : c4e5a320 sn (AES_CTR)
017   : 14be payload crc (calculated 14be OK)
019   : 78 tpl-ci-field (EN 13757-3 Application Layer (no tplh))
020   : 04 dif (32 Bit Integer/Binary Instantaneous value)
021   : FF vif (Manufacturer specific)
022   : 23 vife (?)
023 C!: 00000000 ("status":"OK")
027   : 04 dif (32 Bit Integer/Binary Instantaneous value)
028   : 13 vif (Volume l)
029 C!: B6EC0000 ("total_m3":60.598)
033   : 44 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
034   : 13 vif (Volume l)
035 C!: 34CC0000 ("target_m3":52.276)
039   : 42 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
040   : 6C vif (Date type G)
041 C!: 013A ("target_date":"2024-10-01")
043   : 02 dif (16 Bit Integer/Binary Instantaneous value)
044   : 3B vif (Volume flow l/h)
045 C!: 0000 ("flow_m3h":0)
047   : 92 dif (16 Bit Integer/Binary Maximum value)
048   : 01 dife (subunit=0 tariff=0 storagenr=2)
049   : 3B vif (Volume flow l/h)
050 C!: C102 ("max_flow_m3h":0.705)
052   : A2 dif (16 Bit Integer/Binary Minimum value)
053   : 01 dife (subunit=0 tariff=0 storagenr=2)
054   : 3B vif (Volume flow l/h)
055 C!: 0000 ("min_flow_m3h":0)
057   : 06 dif (48 Bit Integer/Binary Instantaneous value)
058   : FF vif (Manufacturer specific)
059   : 1B vife (?)
060 C?: 097000096000
066   : A1 dif (8 Bit Integer/Binary Minimum value)
067   : 01 dife (subunit=0 tariff=0 storagenr=2)
068   : 5B vif (Flow temperature °C)
069 C!: 0C ("min_flow_temperature_c":12)
070   : 91 dif (8 Bit Integer/Binary Maximum value)
071   : 01 dife (subunit=0 tariff=0 storagenr=2)
072   : 5B vif (Flow temperature °C)
073 C!: 11 ("max_flow_temperature_c":17)
074   : A1 dif (8 Bit Integer/Binary Minimum value)
075   : 01 dife (subunit=0 tariff=0 storagenr=2)
076   : 67 vif (External temperature °C)
077 C!: 10 ("min_external_temperature_c":16)

{
    "media":"cold water",
    "meter":"flowiq2200",
    "name":"",
    "id":"",
    "flow_m3h":0,
    "max_flow_m3h":0.705,
    "max_flow_temperature_c":17,
    "min_external_temperature_c":16,
    "min_flow_m3h":0,
    "min_flow_temperature_c":12,
    "target_m3":52.276,
    "total_m3":60.598,
    "status":"OK",
    "target_date":"2024-10-01",
    "timestamp":"2024-10-25T06:49:33Z"
}

Using: wmbusmeters: 1.17.1-18-g26760f3
26760f3838aaffe0830a73316fc6b0a36ec2300b
```

The first post log is from 2024.10.0, it only seems to log short frames, downgrading to 2024.5.4 and the long frame appears again using the same config. But yes, you can see `Have 57 bytes from CC1101 Rx, RSSI: -54 dBm LQI: 128` appear in the log.

Here's the changed YAML based on the last comment with the working config, however still no sensor data.

```
esphome:
  name: watermeter

esp32:
  board: esp32doit-devkit-v1
  variant: esp32

external_components:
  - source: github://SzczepanLeon/esphome-components@main
    refresh: 0d
    components: [wmbus]

wifi:
  ssid: !env_var IOT_WIFI_SSID
  password: !env_var IOT_WIFI_PASSWORD
  fast_connect: true
  use_address: "watermeter.local"

captive_portal:

logger:
  level: VERBOSE

api:

ota:
  # platform: esphome

web_server:
  port: 80

time:
  - platform: sntp
    id: sntp_time

wmbus:
  mosi_pin: GPIO23
  miso_pin: GPIO19
  clk_pin: GPIO18
  cs_pin: GPIO4
  gdo0_pin: GPIO32
  gdo2_pin: GPIO22

  all_drivers: False
  log_all: False
  sync_mode: True

button:
  - platform: restart
    name: "Restart"

sensor:
  - platform: wmbus
    type: flowiq2200
    meter_id: XXX
    key: "XXX"

    sensors:
      - name: "Watermeter RSSi"
        field: "rssi"
        accuracy_decimals: 0
        unit_of_measurement: "dBm"
        device_class: "signal_strength"
        state_class: "measurement"
        entity_category: "diagnostic"
      - name: "Total water usage"
        field: "total"
        accuracy_decimals: 3
        unit_of_measurement: "m³"
        device_class: "water"
        state_class: "total_increasing"
        icon: "mdi:water"
```

shopro, 2024-10-25:
```
[07:27:42][I][app:100]: ESPHome version 2024.5.4 compiled on Oct 25 2024, 07:26:29
[07:27:42][C][wifi:580]: WiFi:
[07:27:42][C][wifi:408]:   Local MAC: 24:6F:28:15:86:D4
[07:27:42][C][wifi:413]:   SSID: [redacted]
[07:27:42][C][wifi:416]:   IP Address: 10.0.7.33
[07:27:43][C][wifi:420]:   BSSID: [redacted]
[07:27:43][C][wifi:421]:   Hostname: 'watermeter'
[07:27:43][C][wifi:423]:   Signal strength: -54 dB ▂▄▆█
[07:27:43][C][wifi:427]:   Channel: 1
[07:27:43][C][wifi:428]:   Subnet: 255.255.255.0
[07:27:43][C][wifi:429]:   Gateway: 10.0.7.1
[07:27:43][C][wifi:430]:   DNS1: 10.0.1.9
[07:27:43][C][wifi:431]:   DNS2: 0.0.0.0
[07:27:43][C][logger:185]: Logger:
[07:27:43][C][logger:186]:   Level: VERBOSE
[07:27:43][C][logger:188]:   Log Baud Rate: 115200
[07:27:43][C][logger:189]:   Hardware UART: UART0
[07:27:43][C][restart.button:017]: Restart Button 'Restart'
[07:27:43][C][captive_portal:088]: Captive Portal:
[07:27:43][C][web_server:173]: Web Server:
[07:27:43][C][web_server:174]:   Address: watermeter.local:80
[07:27:43][C][sntp:055]: SNTP Time:
[07:27:43][C][sntp:056]:   Server 1: '0.pool.ntp.org'
[07:27:43][C][sntp:057]:   Server 2: '1.pool.ntp.org'
[07:27:43][C][sntp:058]:   Server 3: '2.pool.ntp.org'
[07:27:43][C][sntp:059]:   Timezone: 'UTC0'
[07:27:43][C][mdns:115]: mDNS:
[07:27:43][C][mdns:116]:   Hostname: watermeter
[07:27:43][V][mdns:117]:   Services:
[07:27:43][V][mdns:119]:   - _esphomelib, _tcp, 6053
[07:27:43][V][mdns:121]:     TXT: version = 2024.5.4
[07:27:43][V][mdns:121]:     TXT: mac = 246f281586d4
[07:27:43][V][mdns:121]:     TXT: platform = ESP32
[07:27:43][V][mdns:121]:     TXT: board = esp32doit-devkit-v1
[07:27:43][V][mdns:121]:     TXT: network = wifi
[07:27:43][V][mdns:119]:   - _http, _tcp, 80
[07:27:43][C][ota:096]: Over-The-Air Updates:
[07:27:43][C][ota:097]:   Address: watermeter.local:3232
[07:27:43][C][ota:103]:   OTA version: 2.
[07:27:43][C][api:139]: API Server:
[07:27:43][C][api:140]:   Address: watermeter.local:6053
[07:27:43][C][api:144]:   Using noise encryption: NO
[07:27:43][C][wmbus:326]: wM-Bus v4.0.11-1.17.1-b8f4a945:
[07:27:43][C][wmbus:339]:   LED:
[07:27:43][C][wmbus:340]:     Pin: GPIO10
[07:27:43][C][wmbus:341]:     Duration: 1000 ms
[07:27:43][C][wmbus:344]:   MAC: 15286F24
[07:27:43][C][wmbus:346]:   CC1101 frequency: 868.950 MHz
[07:27:43][C][wmbus:347]:   CC1101 SPI bus:
[07:27:43][C][wmbus:351]:     MOSI Pin: GPIO23
[07:27:43][C][wmbus:352]:     MISO Pin: GPIO19
[07:27:43][C][wmbus:353]:     CLK Pin:  GPIO18
[07:27:43][C][wmbus:354]:     CS Pin:   GPIO4
[07:27:43][C][wmbus:355]:     GDO0 Pin: GPIO32
[07:27:43][C][wmbus:356]:     GDO2 Pin: GPIO22
[07:27:43][C][wmbus:362]:   Available drivers: unknown, flowiq2200
[07:27:43][C][wmbus:376]:   Meter:
[07:27:43][C][wmbus:377]:     ID: 
[07:27:43][C][wmbus:378]:     Type: flowiq2200
[07:27:43][C][wmbus:379]:     Key: ''
[07:27:43][C][wmbus:381]:     Field: 'rssi'
[07:27:43][C][wmbus:382]:      Name: 'Watermeter RSSi'
[07:27:43][C][wmbus:382]:        Device Class: 'signal_strength'
[07:27:43][C][wmbus:382]:        State Class: 'measurement'
[07:27:43][C][wmbus:382]:        Unit of Measurement: 'dBm'
[07:27:43][C][wmbus:382]:        Accuracy Decimals: 0
[07:27:43][C][wmbus:381]:     Field: 'total'
[07:27:43][C][wmbus:382]:      Name: 'Total water usage'
[07:27:43][C][wmbus:382]:        Device Class: 'water'
[... 237 lines cut ...]
[07:59:01][W][component:237]: Component wmbus took a long time for an operation (81 ms).
[07:59:01][W][component:238]: Components should block for at most 30 ms.
[08:00:37][V][rxLoop:167]: Have 57 bytes from CC1101 Rx, RSSI: -70 dBm LQI: 128
[08:00:37][V][rxLoop:172]: Synchronus mode enabled.
[08:00:37][D][mbus:024]: Received C1 B frame
[08:00:37][V][mbus:028]: Frame: 3644372C545594123C168D20B310FCA320404534C06B5DE7A4582E2D42EA465BA12CF0A755E9A3CFCA6425133A5AEBD1D8B09DF4E520DA (55) [with CRC]
[08:00:37][V][mbus:166]: Validating CRC for Block1 + Block2
[08:00:37][V][crc:031]:     calculated: 0x20DA, read: 0x20DA
[08:00:37][V][mbus:063]: Frame: 3444372C545594123C168D20B310FCA320404534C06B5DE7A4582E2D42EA465BA12CF0A755E9A3CFCA6425133A5AEBD1D8B09DF4E5 (53) [without CRC]
[08:00:37][V][Telegram.cpp:1131]: (wmbus) parseDLL @0 53
[08:00:37][V][Telegram.cpp:1178]: (wmbus) parseELL @10 43
[08:00:37][W][component:237]: Component wmbus took a long time for an operation (88 ms).
[08:00:37][W][component:238]: Components should block for at most 30 ms.
[08:02:12][V][rxLoop:167]: Have 82 bytes from CC1101 Rx, RSSI: -71 dBm LQI: 128
[08:02:12][V][rxLoop:172]: Synchronus mode enabled.
[08:02:12][D][mbus:024]: Received C1 B frame
[08:02:12][V][mbus:028]: Frame: 4F44372C545594123C168D20B424FCA320F9C43F5862B986812106C9F7607F5DBB98405978869A0708AE1F18F74734C0717EB126C8A7EF908C94B76D8B23BBEB7EE07FCBC63C524AD0F97162EC202718 (80) [with CRC]
[08:02:12][V][mbus:166]: Validating CRC for Block1 + Block2
[08:02:12][V][crc:031]:     calculated: 0x2718, read: 0x2718
[08:02:12][V][mbus:063]: Frame: 4D44372C545594123C168D20B424FCA320F9C43F5862B986812106C9F7607F5DBB98405978869A0708AE1F18F74734C0717EB126C8A7EF908C94B76D8B23BBEB7EE07FCBC63C524AD0F97162EC20 (78) [without CRC]
[08:02:12][V][Telegram.cpp:1131]: (wmbus) parseDLL @0 78
[08:02:12][V][Telegram.cpp:1178]: (wmbus) parseELL @10 68
[08:02:12][W][component:237]: Component wmbus took a long time for an operation (84 ms).
[08:02:12][W][component:238]: Components should block for at most 30 ms.
```

SzczepanLeon, 2024-10-25:
Please also add:
```
log_all: True
```

shopro, 2024-10-25:
It works, changing the meter_id to start with 0x did it. Thank you.

I did try the 0x before as well so it must have been it and the YAML order change. Now it even works with 2024.10.0.

```
[10:18:24][I][app:100]: ESPHome version 2024.10.0 compiled on Oct 25 2024, 10:15:45
[10:18:24][C][wifi:600]: WiFi:
[10:18:24][C][wifi:428]:   Local MAC: 24:6F:28:15:86:D4
[10:18:24][C][wifi:433]:   SSID: [redacted]
[10:18:24][C][wifi:436]:   IP Address: 10.0.7.33
[10:18:24][C][wifi:440]:   BSSID: [redacted]
[10:18:24][C][wifi:441]:   Hostname: 'watermeter'
[10:18:24][C][wifi:443]:   Signal strength: -53 dB ▂▄▆█
[10:18:24][C][wifi:447]:   Channel: 1
[10:18:24][C][wifi:448]:   Subnet: 255.255.255.0
[10:18:24][C][wifi:449]:   Gateway: 10.0.7.1
[10:18:24][C][wifi:450]:   DNS1: 10.0.1.9
[10:18:24][C][wifi:451]:   DNS2: 0.0.0.0
[10:18:24][C][logger:185]: Logger:
[10:18:24][C][logger:186]:   Level: VERBOSE
[10:18:24][C][logger:188]:   Log Baud Rate: 115200
[10:18:24][C][logger:189]:   Hardware UART: UART0
[10:18:24][C][restart.button:017]: Restart Button 'Restart'
[10:18:24][C][restart.button:017]:   Icon: 'mdi:restart'
[10:18:24][C][captive_portal:089]: Captive Portal:
[10:18:24][C][web_server:153]: Web Server:
[10:18:24][C][web_server:154]:   Address: watermeter.local:80
[10:18:24][C][sntp:048]: SNTP Time:
[10:18:24][C][sntp:049]:   Server 1: '0.pool.ntp.org'
[10:18:24][C][sntp:050]:   Server 2: '1.pool.ntp.org'
[10:18:24][C][sntp:051]:   Server 3: '2.pool.ntp.org'
[10:18:24][C][sntp:052]:   Timezone: 'UTC0'
[10:18:24][C][mdns:116]: mDNS:
[10:18:24][C][mdns:117]:   Hostname: watermeter
[10:18:24][V][mdns:118]:   Services:
[10:18:24][V][mdns:120]:   - _esphomelib, _tcp, 6053
[10:18:24][V][mdns:122]:     TXT: version = 2024.10.0
[10:18:24][V][mdns:122]:     TXT: mac = 246f281586d4
[10:18:24][V][mdns:122]:     TXT: platform = ESP32
[10:18:24][V][mdns:122]:     TXT: board = esp32doit-devkit-v1
[10:18:24][V][mdns:122]:     TXT: network = wifi
[10:18:24][V][mdns:120]:   - _http, _tcp, 80
[10:18:24][C][esphome.ota:073]: Over-The-Air updates:
[10:18:24][C][esphome.ota:074]:   Address: watermeter.local:3232
[10:18:24][C][esphome.ota:075]:   Version: 2
[10:18:24][C][safe_mode:018]: Safe Mode:
[10:18:24][C][safe_mode:020]:   Boot considered successful after 60 seconds
[10:18:24][C][safe_mode:021]:   Invoke after 10 boot attempts
[10:18:24][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[10:18:24][C][api:140]: API Server:
[10:18:24][C][api:141]:   Address: watermeter.local:6053
[10:18:24][C][api:145]:   Using noise encryption: NO
[10:18:24][C][wmbus:326]: wM-Bus v4.0.11-1.17.1-b8f4a945:
[10:18:24][C][wmbus:339]:   LED:
[10:18:24][C][wmbus:340]:     Pin: GPIO10
[10:18:24][C][wmbus:341]:     Duration: 1000 ms
[10:18:24][C][wmbus:344]:   MAC: 15286F24
[10:18:24][C][wmbus:346]:   CC1101 frequency: 868.950 MHz
[10:18:24][C][wmbus:347]:   CC1101 SPI bus:
[10:18:24][C][wmbus:351]:     MOSI Pin: GPIO23
[10:18:24][C][wmbus:352]:     MISO Pin: GPIO19
[10:18:24][C][wmbus:353]:     CLK Pin:  GPIO18
[10:18:24][C][wmbus:354]:     CS Pin:   GPIO4
[10:18:24][C][wmbus:355]:     GDO0 Pin: GPIO32
[10:18:24][C][wmbus:356]:     GDO2 Pin: GPIO22
[10:18:24][C][wmbus:362]:   Available drivers: unknown, flowiq2200
[10:18:24][C][wmbus:376]:   Meter:
[10:18:24][C][wmbus:377]:     ID: 311711060 [0x12945554]
[10:18:24][C][wmbus:378]:     Type: flowiq2200
[10:18:24][C][wmbus:379]:     Key: ''
[... 42 lines cut ...]
[10:18:34][D][Telegram.cpp:564]: (telegram) ELL CI=8d CC=20 (slow_resp sync) ACC=08
[10:18:34][D][Telegram.cpp:572]:  SN=b404a420 (AES_CTR session=4 time=671819) CRC=10a6
[10:18:34][V][Telegram.cpp:1333]: (wmbus) parseNWL @19 34
[10:18:34][V][Telegram.cpp:1391]: (wmbus) parseAFL @19 34
[10:18:34][V][Telegram.cpp:2078]: (wmbus) parseTPL @19 34
[10:18:34][D][Telegram.cpp:2032]: (wmbus) ignoring compressed telegram since format signature hash 0xa9f3 is yet unknown.
     this is not a problem, since you only need wait for at most 8 telegrams
     (8*16 seconds) until an full length telegram arrives and then we know
     the format giving this hash and start decoding the telegrams properly.
[10:18:34][V][sensor:043]: 'Watermeter RSSi': Received new state -70.000000
[10:18:34][D][sensor:094]: 'Watermeter RSSi': Sending state -70.00000 dBm with 0 decimals of accuracy
[10:18:34][W][wmbus:154]: Can't get requested field 'total' with unit 'm³'
[10:18:34][W][component:237]: Component wmbus took a long time for an operation (316 ms).
[10:18:34][W][component:238]: Components should block for at most 30 ms.
[10:19:14][D][esp32.preferences:114]: Saving 1 preferences to flash...
[10:19:14][V][esp32.preferences:163]: nvs_get_blob('871654188'): ESP_ERR_NVS_NOT_FOUND - the key might not be set yet
[10:19:14][V][esp32.preferences:126]: sync: key: 871654188, len: 7
[10:19:14][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[10:19:15][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[10:19:15][D][esp32.preferences:114]: Saving 1 preferences to flash...
[10:19:15][V][esp32.preferences:126]: sync: key: 233825507, len: 4
[10:19:15][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[10:20:13][V][rxLoop:167]: Have 57 bytes from CC1101 Rx, RSSI: -71 dBm LQI: 128
[10:20:13][V][rxLoop:172]: Synchronus mode enabled.
```

shopro, 2024-10-25:
It might have been an old value from the previous 2024.5.4 version that was displayed on the device webpage as after a reboot of the ESP32, it no longer shows the long frame at all in the log even after 30 minutes when using version 2024.10.0.

```
[11:10:15][I][app:100]: ESPHome version 2024.10.0 compiled on Oct 25 2024, 11:07:53
[11:10:15][C][wifi:600]: WiFi:
[11:10:15][C][wifi:428]:   Local MAC: 24:6F:28:15:86:D4
[11:10:15][C][wifi:433]:   SSID: [redacted]
[11:10:15][C][wifi:436]:   IP Address: 10.0.7.33
[11:10:15][C][wifi:440]:   BSSID: [redacted]
[11:10:15][C][wifi:441]:   Hostname: 'watermeter'
[11:10:15][C][wifi:443]:   Signal strength: -53 dB ▂▄▆█
[11:10:15][C][wifi:447]:   Channel: 1
[11:10:15][C][wifi:448]:   Subnet: 255.255.255.0
[11:10:15][C][wifi:449]:   Gateway: 10.0.7.1
[11:10:15][C][wifi:450]:   DNS1: 10.0.1.9
[11:10:15][C][wifi:451]:   DNS2: 0.0.0.0
[11:10:15][C][logger:185]: Logger:
[11:10:15][C][logger:186]:   Level: VERBOSE
[11:10:15][C][logger:188]:   Log Baud Rate: 115200
[11:10:15][C][logger:189]:   Hardware UART: UART0
[11:10:15][C][restart.button:017]: Restart Button 'Restart'
[11:10:15][C][restart.button:017]:   Icon: 'mdi:restart'
[11:10:15][C][captive_portal:089]: Captive Portal:
[11:10:15][C][web_server:153]: Web Server:
[11:10:15][C][web_server:154]:   Address: watermeter.local:80
[11:10:15][C][sntp:048]: SNTP Time:
[11:10:15][C][sntp:049]:   Server 1: '0.pool.ntp.org'
[11:10:15][C][sntp:050]:   Server 2: '1.pool.ntp.org'
[11:10:15][C][sntp:051]:   Server 3: '2.pool.ntp.org'
[11:10:15][C][sntp:052]:   Timezone: 'UTC0'
[11:10:15][C][mdns:116]: mDNS:
[11:10:15][C][mdns:117]:   Hostname: watermeter
[11:10:15][V][mdns:118]:   Services:
[11:10:16][V][mdns:120]:   - _esphomelib, _tcp, 6053
[11:10:16][V][mdns:122]:     TXT: version = 2024.10.0
[11:10:16][V][mdns:122]:     TXT: mac = 246f281586d4
[11:10:16][V][mdns:122]:     TXT: platform = ESP32
[11:10:16][V][mdns:122]:     TXT: board = esp32doit-devkit-v1
[11:10:16][V][mdns:122]:     TXT: network = wifi
[11:10:16][V][mdns:120]:   - _http, _tcp, 80
[11:10:16][C][esphome.ota:073]: Over-The-Air updates:
[11:10:16][C][esphome.ota:074]:   Address: watermeter.local:3232
[11:10:16][C][esphome.ota:075]:   Version: 2
[11:10:16][C][safe_mode:018]: Safe Mode:
[11:10:16][C][safe_mode:020]:   Boot considered successful after 60 seconds
[11:10:16][C][safe_mode:021]:   Invoke after 10 boot attempts
[11:10:16][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[11:10:16][C][api:140]: API Server:
[11:10:16][C][api:141]:   Address: watermeter.local:6053
[11:10:16][C][api:145]:   Using noise encryption: NO
[11:10:16][C][wmbus:326]: wM-Bus v4.0.11-1.17.1-b8f4a945:
[11:10:16][C][wmbus:339]:   LED:
[11:10:16][C][wmbus:340]:     Pin: GPIO10
[11:10:16][C][wmbus:341]:     Duration: 1000 ms
[11:10:16][C][wmbus:344]:   MAC: 15286F24
[11:10:16][C][wmbus:346]:   CC1101 frequency: 868.950 MHz
[11:10:16][C][wmbus:347]:   CC1101 SPI bus:
[11:10:16][C][wmbus:351]:     MOSI Pin: GPIO23
[11:10:16][C][wmbus:352]:     MISO Pin: GPIO19
[11:10:16][C][wmbus:353]:     CLK Pin:  GPIO18
[11:10:16][C][wmbus:354]:     CS Pin:   GPIO4
[11:10:16][C][wmbus:355]:     GDO0 Pin: GPIO32
[11:10:16][C][wmbus:356]:     GDO2 Pin: GPIO22
[11:10:16][C][wmbus:362]:   Available drivers: unknown, flowiq2200
[11:10:16][C][wmbus:376]:   Meter:
[11:10:16][C][wmbus:377]:     ID: 311711060 [0x12945554]
[11:10:16][C][wmbus:378]:     Type: flowiq2200
[11:10:16][C][wmbus:379]:     Key: ''
[11:10:16][C][wmbus:381]:     Field: 'rssi'
[11:10:16][C][wmbus:382]:      Name: 'Watermeter RSSi'
[... 84 lines cut ...]
[11:12:26][V][Telegram.cpp:1178]: (wmbus) parseELL @10 43
[11:12:26][V][Telegram.cpp:198]: (ELL) decrypting
[11:12:26][V][Telegram.cpp:217]: (ELL) IV 372C545594123C16200808A420000000
[11:12:26][V][Telegram.cpp:238]: (ELL) block 0 block_size 16 offset 0
[11:12:26][V][Telegram.cpp:242]: (ELL) decrypted
[11:12:26][V][Telegram.cpp:238]: (ELL) block 1 block_size 16 offset 16
[11:12:26][V][Telegram.cpp:242]: (ELL) decrypted
[11:12:26][V][Telegram.cpp:238]: (ELL) block 2 block_size 4 offset 32
[11:12:26][V][Telegram.cpp:242]: (ELL) decrypted
[11:12:26][V][Telegram.cpp:248]: (ELL) decrypted
[11:12:26][D][Telegram.cpp:564]: (telegram) ELL CI=8d CC=20 (slow_resp sync) ACC=29
[11:12:26][D][Telegram.cpp:572]:  SN=0808a420 (AES_CTR session=8 time=671872) CRC=5848
[11:12:26][V][Telegram.cpp:1333]: (wmbus) parseNWL @19 34
[11:12:26][V][Telegram.cpp:1391]: (wmbus) parseAFL @19 34
[11:12:26][V][Telegram.cpp:2078]: (wmbus) parseTPL @19 34
[11:12:26][D][Telegram.cpp:2032]: (wmbus) ignoring compressed telegram since format signature hash 0xa9f3 is yet unknown.
     this is not a problem, since you only need wait for at most 8 telegrams
     (8*16 seconds) until an full length telegram arrives and then we know
     the format giving this hash and start decoding the telegrams properly.
[11:12:26][V][sensor:043]: 'Watermeter RSSi': Received new state -71.000000
[11:12:26][D][sensor:094]: 'Watermeter RSSi': Sending state -71.00000 dBm with 0 decimals of accuracy
[11:12:26][W][wmbus:154]: Can't get requested field 'total' with unit 'm³'
[11:12:26][W][component:237]: Component wmbus took a long time for an operation (326 ms).
[11:12:26][W][component:238]: Components should block for at most 30 ms.
```

SzczepanLeon, 2024-10-25:
@shopro Can You collect more traces - at least one long frame and then short one?

shopro, 2024-10-25:
That's the thing, on version 2024.10.0 the log shows no long frames even after waiting 45 minutes.

On version 2024.5.4 the long frame comes along about every 12-13minutes when downgrading back to it.

SzczepanLeon, 2024-10-25:
V4.0 is more time-consuming. Maybe long frame is missed/transmitted when short one is parsed.

This is my explanation, RX/CC1101 part is common for V3.x and V4.0.

shopro, 2024-10-25:
The only part the YAML was changed was the `ota` part, besides that the only change  is to the ESPHome version. YAML is the 4.0 version from above.

2024.5.4
```
ota:
  # platform: esphome
```

2024.10.4
```
ota:
  platform: esphome
```

```
INFO ESPHome 2024.10.0
INFO Reading configuration /config/watermeter.yaml...
INFO Updating https://github.com/SzczepanLeon/esphome-components.git@main
INFO Detected timezone 'Etc/UTC'
INFO Generating C++ source...
INFO Compiling app...
Processing watermeter (board: esp32doit-devkit-v1; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.4
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- SPI @ 2.0.0
|-- SmartRC-CC1101-Driver-Lib @ 2.5.7
```

2024.5.4
```
INFO ESPHome 2024.5.4
INFO Reading configuration /config/watermeter.yaml...
INFO Updating https://github.com/SzczepanLeon/esphome-components.git@main
INFO Detected timezone 'Etc/UTC'
INFO Generating C++ source...
INFO Compiling app...
Processing watermeter (board: esp32doit-devkit-v1; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.3
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- SPI @ 2.0.0
|-- SmartRC-CC1101-Driver-Lib @ 2.5.7
```

Log from 2024.5.4, correctly shows the sensor.
```
SSI: -70 dBm LQI: 128
[12:30:50][V][rxLoop:172]: Synchronus mode enabled.
[12:30:50][D][mbus:024]: Received C1 B frame
[12:30:50][V][mbus:028]: Frame: 3644372C545594123C168D205AFC0CA4201E735A60D07789988339AA6E0EAEC0C8CA2A59474D892800DC4A59F7BDBB3C056689944EA4B6 (55) [with CRC]
[12:30:50][V][mbus:166]: Validating CRC for Block1 + Block2
[... 249 lines cut ...]
[12:35:35][V][meters.cpp:2155]: (meter) ExternalTemperature min_external_temperature_c decoded c default c value 16 (scale 1)
[12:35:35][V][meters.cpp:973]: (meters) trying field info max_flow(Flow)[8]...
[12:35:35][V][meters.cpp:995]: (meters) using field info max_flow(Flow)[8] to extract 92013B at offset 40
[12:35:35][V][meters.cpp:2155]: (meter) VolumeFlow max_flow_m3h decoded m3h default m3h value 0.705 (scale 1)
[12:35:35][V][meters.cpp:973]: (meters) trying field info min_flow(Flow)[9]...
[12:35:35][V][meters.cpp:995]: (meters) using field info min_flow(Flow)[9] to extract A2013B at offset 42
[12:35:35][V][meters.cpp:2155]: (meter) VolumeFlow min_flow_m3h decoded m3h default m3h value 0 (scale 1)
[12:35:35][V][meters.cpp:973]: (meters) trying field info max_external_temperature(Temperature)[10]...
[12:35:35][V][meters.cpp:973]: (meters) trying field info min_external_temperature(Temperature)[11]...
[12:35:35][V][meters.cpp:973]: (meters) trying field info max_flow(Flow)[12]...
[12:35:35][V][meters.cpp:973]: (meters) trying field info time_dry(Text)[13]...
[12:35:35][V][meters.cpp:973]: (meters) trying field info time_reversed(Text)[14]...
[12:35:35][V][meters.cpp:973]: (meters) trying field info time_leaking(Text)[15]...
[12:35:35][V][meters.cpp:973]: (meters) trying field info time_bursting(Text)[16]...
[12:35:35][V][sensor:043]: 'Watermeter RSSi': Received new state -71.000000
[12:35:35][D][sensor:094]: 'Watermeter RSSi': Sending state -71.00000 dBm with 0 decimals of accuracy
[12:35:35][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[12:35:35][V][json:058]: Size after shrink 80 bytes
[12:35:35][V][sensor:043]: 'Total water usage': Received new state 60.730000
[12:35:35][D][sensor:094]: 'Total water usage': Sending state 60.73000 m³ with 3 decimals of accuracy
[12:35:35][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[12:35:35][V][json:058]: Size after shrink 84 bytes
[12:35:35][W][component:237]: Component wmbus took a long time for an operation (850 ms).
[12:35:35][W][component:238]: Components should block for at most 30 ms.
```

SzczepanLeon, 2024-10-25:
Let's switch to VERY_VERBOSE logs - maybe ESPHome shows something around scheduing.

shopro, 2024-10-25:
Here's the log for the 2024.5.4 version, I'll get the 2024.10.0 later.

```
INFO ESPHome 2024.5.4
INFO Reading configuration /config/watermeter.yaml...
INFO Updating https://github.com/SzczepanLeon/esphome-components.git@main
INFO Detected timezone 'Etc/UTC'
WARNING Pin 10 (9-10) might already be used by the flash interface in QUAD IO flash mode.
INFO Starting log output from 10.0.7.33 using esphome API
INFO Successfully connected to watermeter @ 10.0.7.33 in 0.094s
INFO Successful handshake with watermeter @ 10.0.7.33 in 0.079s
[14:14:00][I][app:100]: ESPHome version 2024.5.4 compiled on Oct 25 2024, 13:53:09
[14:14:00][C][wifi:580]: WiFi:
[14:14:00][C][wifi:408]:   Local MAC: 24:6F:28:15:86:D4
[14:14:00][C][wifi:413]:   SSID: 'IoT'[redacted]
[14:14:00][C][wifi:416]:   IP Address: 10.0.7.33
[14:14:00][C][wifi:420]:   BSSID: 4A:D9:E7:91:67:A9[redacted]
[14:14:00][C][wifi:421]:   Hostname: 'watermeter'
[14:14:00][C][wifi:423]:   Signal strength: -53 dB ▂▄▆█
[14:14:00][C][wifi:427]:   Channel: 1
[14:14:00][C][wifi:428]:   Subnet: 255.255.255.0
[14:14:00][C][wifi:429]:   Gateway: 10.0.7.1
[14:14:00][C][wifi:430]:   DNS1: 10.0.1.9
[14:14:00][C][wifi:431]:   DNS2: 0.0.0.0
[14:14:00][C][logger:185]: Logger:
[14:14:00][C][logger:186]:   Level: VERY_VERBOSE
[14:14:00][C][logger:188]:   Log Baud Rate: 115200
[14:14:00][C][logger:189]:   Hardware UART: UART0
[14:14:00][C][restart.button:017]: Restart Button 'Restart'
[14:14:00][C][captive_portal:088]: Captive Portal:
[14:14:00][C][web_server:173]: Web Server:
[14:14:00][C][web_server:174]:   Address: watermeter.local:80
[14:14:00][C][sntp:055]: SNTP Time:
[14:14:00][C][sntp:056]:   Server 1: '0.pool.ntp.org'
[14:14:00][C][sntp:057]:   Server 2: '1.pool.ntp.org'
[14:14:00][C][sntp:058]:   Server 3: '2.pool.ntp.org'
[14:14:00][C][sntp:059]:   Timezone: 'UTC0'
[14:14:00][C][mdns:115]: mDNS:
[14:14:00][C][mdns:116]:   Hostname: watermeter
[14:14:00][V][mdns:117]:   Services:
[14:14:00][V][mdns:119]:   - _esphomelib, _tcp, 6053
[14:14:00][V][mdns:121]:     TXT: version = 2024.5.4
[14:14:00][V][mdns:121]:     TXT: mac = 246f281586d4
[14:14:00][V][mdns:121]:     TXT: platform = ESP32
[14:14:00][V][mdns:121]:     TXT: board = esp32doit-devkit-v1
[14:14:00][V][mdns:121]:     TXT: network = wifi
[14:14:00][V][mdns:119]:   - _http, _tcp, 80
[14:14:00][C][ota:096]: Over-The-Air Updates:
[14:14:00][C][ota:097]:   Address: watermeter.local:3232
[14:14:00][C][ota:103]:   OTA version: 2.
[14:14:00][C][api:139]: API Server:
[14:14:00][C][api:140]:   Address: watermeter.local:6053
[14:14:00][C][api:144]:   Using noise encryption: NO
[14:14:00][C][wmbus:326]: wM-Bus v4.0.11-1.17.1-b8f4a945:
[14:14:00][C][wmbus:339]:   LED:
[14:14:00][C][wmbus:340]:     Pin: GPIO10
[14:14:00][C][wmbus:341]:     Duration: 1000 ms
[14:14:00][C][wmbus:344]:   MAC: 15286F24
[14:14:00][C][wmbus:346]:   CC1101 frequency: 868.950 MHz
[14:14:00][C][wmbus:347]:   CC1101 SPI bus:
[14:14:00][C][wmbus:351]:     MOSI Pin: GPIO23
[14:14:00][C][wmbus:352]:     MISO Pin: GPIO19
[14:14:00][C][wmbus:353]:     CLK Pin:  GPIO18
[14:14:00][C][wmbus:354]:     CS Pin:   GPIO4
[14:14:00][C][wmbus:355]:     GDO0 Pin: GPIO32
[14:14:00][C][wmbus:356]:     GDO2 Pin: GPIO22
[14:14:00][C][wmbus:362]:   Available drivers: unknown, flowiq2200
[14:14:00][C][wmbus:376]:   Meter:
[14:14:00][C][wmbus:377]:     ID: 311711060 [0x12945554]
[14:14:00][C][wmbus:378]:     Type: flowiq2200
[... 228 lines cut ...]
  state: -70
  missing_state: NO
}
[14:15:39][V][sensor:043]: 'Total water usage': Received new state 60.757000
[14:15:39][D][sensor:094]: 'Total water usage': Sending state 60.75700 m³ with 3 decimals of accuracy
[14:15:39][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[14:15:39][V][json:058]: Size after shrink 84 bytes
[14:15:39][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: 39833189
  state: 60.757
  missing_state: NO
}
[14:15:39][W][component:237]: Component wmbus took a long time for an operation (907 ms).
[14:15:39][W][component:238]: Components should block for at most 30 ms.
[14:15:43][VV][scheduler:226]: Running interval '' with interval=10000 last_execution=1181829 (now=1191829)
[14:15:53][VV][scheduler:226]: Running interval '' with interval=10000 last_execution=1191829 (now=1201829)
[14:16:00][VV][api.connection:132]: Sending keepalive PING...
[14:16:00][VV][api.service:037]: send_ping_request: PingRequest {}
[14:16:00][VV][api.service:663]: on_ping_response: PingResponse {}
[14:16:03][VV][scheduler:226]: Running interval '' with interval=10000 last_execution=1201829 (now=1211829)
[14:16:13][VV][scheduler:226]: Running interval '' with interval=10000 last_execution=1211829 (now=1221829)
[14:16:15][VV][api.service:654]: on_ping_request: PingRequest {}
[14:16:15][VV][api.service:043]: send_ping_response: PingResponse {}

```

SzczepanLeon, 2024-10-25:
Maybe I have workaround. Please try this version:
```
external_components:
  - source: github://SzczepanLeon/esphome-components@fix_140
    refresh: 0d
    components: [wmbus]
```

shopro, 2024-10-26:
I think it's working with the workaround. This is with the same YAML as before, running version 2024.10.0.

Long frame appears once again in the logs and the water usage sensor is being updated.

```
INFO ESPHome 2024.10.0
INFO Reading configuration /config/watermeter.yaml...
INFO Updating https://github.com/SzczepanLeon/esphome-components.git@fix_140
INFO Detected timezone 'Etc/UTC'
WARNING Pin 10 (9-10) might already be used by the flash interface in QUAD IO flash mode.
INFO Starting log output from 10.0.7.33 using esphome API
INFO Successfully connected to watermeter @ 10.0.7.33 in 0.117s
INFO Successful handshake with watermeter @ 10.0.7.33 in 0.079s
[01:30:43][I][app:100]: ESPHome version 2024.10.0 compiled on Oct 26 2024, 01:12:23
[01:30:43][C][wifi:600]: WiFi:
[01:30:43][C][wifi:428]:   Local MAC: 24:6F:28:15:86:D4
[01:30:43][C][wifi:433]:   SSID: 'IoT'[redacted]
[01:30:43][C][wifi:436]:   IP Address: 10.0.7.33
[01:30:43][C][wifi:440]:   BSSID: 4A:D9:E7:91:67:A9[redacted]
[01:30:43][C][wifi:441]:   Hostname: 'watermeter'
[01:30:43][C][wifi:443]:   Signal strength: -53 dB ▂▄▆█
[01:30:43][C][wifi:447]:   Channel: 1
[01:30:43][C][wifi:448]:   Subnet: 255.255.255.0
[01:30:43][C][wifi:449]:   Gateway: 10.0.7.1
[01:30:43][C][wifi:450]:   DNS1: 10.0.1.9
[01:30:43][C][wifi:451]:   DNS2: 0.0.0.0
[01:30:43][C][logger:185]: Logger:
[01:30:43][C][logger:186]:   Level: VERY_VERBOSE
[01:30:43][C][logger:188]:   Log Baud Rate: 115200
[01:30:43][C][logger:189]:   Hardware UART: UART0
[01:30:43][C][restart.button:017]: Restart Button 'Restart'
[01:30:43][C][restart.button:017]:   Icon: 'mdi:restart'
[01:30:43][C][captive_portal:089]: Captive Portal:
[01:30:43][C][web_server:153]: Web Server:
[01:30:43][C][web_server:154]:   Address: watermeter.local:80
[01:30:43][C][sntp:048]: SNTP Time:
[01:30:43][C][sntp:049]:   Server 1: '0.pool.ntp.org'
[01:30:43][C][sntp:050]:   Server 2: '1.pool.ntp.org'
[01:30:43][C][sntp:051]:   Server 3: '2.pool.ntp.org'
[01:30:43][C][sntp:052]:   Timezone: 'UTC0'
[01:30:43][C][mdns:116]: mDNS:
[01:30:43][C][mdns:117]:   Hostname: watermeter
[01:30:43][V][mdns:118]:   Services:
[01:30:43][V][mdns:120]:   - _esphomelib, _tcp, 6053
[01:30:43][V][mdns:122]:     TXT: version = 2024.10.0
[01:30:43][V][mdns:122]:     TXT: mac = 246f281586d4
[01:30:43][V][mdns:122]:     TXT: platform = ESP32
[01:30:43][V][mdns:122]:     TXT: board = esp32doit-devkit-v1
[01:30:43][V][mdns:122]:     TXT: network = wifi
[01:30:43][V][mdns:120]:   - _http, _tcp, 80
[01:30:43][C][esphome.ota:073]: Over-The-Air updates:
[01:30:43][C][esphome.ota:074]:   Address: watermeter.local:3232
[01:30:43][C][esphome.ota:075]:   Version: 2
[01:30:43][C][safe_mode:018]: Safe Mode:
[01:30:43][C][safe_mode:020]:   Boot considered successful after 60 seconds
[01:30:43][C][safe_mode:021]:   Invoke after 10 boot attempts
[01:30:43][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[01:30:43][C][api:140]: API Server:
[01:30:43][C][api:141]:   Address: watermeter.local:6053
[01:30:43][C][api:145]:   Using noise encryption: NO
[01:30:43][C][wmbus:326]: wM-Bus v4.0.12-1.17.1-b8f4a945:
[01:30:43][C][wmbus:339]:   LED:
[01:30:43][C][wmbus:340]:     Pin: GPIO10
[01:30:44][C][wmbus:341]:     Duration: 1000 ms
[01:30:44][C][wmbus:344]:   MAC: 15286F24
[01:30:44][C][wmbus:346]:   CC1101 frequency: 868.950 MHz
[01:30:44][C][wmbus:347]:   CC1101 SPI bus:
[01:30:44][C][wmbus:351]:     MOSI Pin: GPIO23
[01:30:44][C][wmbus:352]:     MISO Pin: GPIO19
[01:30:44][C][wmbus:353]:     CLK Pin:  GPIO18
[... 1157 lines cut ...]
[01:46:36][VV][scheduler:226]: Running interval '' with interval=10000 last_execution=1954776 (now=1964776)
[01:46:37][VV][scheduler:226]: Running interval '' with interval=60000 last_execution=1905992 (now=1965992)
[01:46:44][VV][api.connection:133]: Sending keepalive PING...
[01:46:44][VV][api.service:037]: send_ping_request: PingRequest {}
[01:46:44][VV][api.service:706]: on_ping_response: PingResponse {}
[01:46:46][VV][scheduler:226]: Running interval '' with interval=10000 last_execution=1964776 (now=1974776)
[01:46:56][VV][scheduler:226]: Running interval '' with interval=10000 last_execution=1974776 (now=1984776)
[01:47:03][VV][api.service:697]: on_ping_request: PingRequest {}
[01:47:03][VV][api.service:043]: send_ping_response: PingResponse {}
[01:47:06][VV][scheduler:226]: Running interval '' with interval=10000 last_execution=1984776 (now=1994776)
[01:47:16][VV][scheduler:226]: Running interval '' with interval=10000 last_execution=1994776 (now=2004776)
[01:47:26][VV][scheduler:226]: Running interval '' with interval=10000 last_execution=2004776 (now=2014776)
[01:47:36][VV][scheduler:226]: Running interval '' with interval=10000 last_execution=2014776 (now=2024776)
[01:47:37][VV][scheduler:226]: Running interval '' with interval=60000 last_execution=1965992 (now=2025992)
[01:47:43][VV][api.service:697]: on_ping_request: PingRequest {}
[01:47:43][VV][api.service:043]: send_ping_response: PingResponse {}
[01:47:44][VV][api.connection:133]: Sending keepalive PING...
[01:47:44][VV][api.service:037]: send_ping_request: PingRequest {}
[01:47:44][VV][api.service:706]: on_ping_response: PingResponse {}
[01:47:46][VV][scheduler:226]: Running interval '' with interval=10000 last_execution=2024776 (now=2034777)
[01:47:56][VV][scheduler:226]: Running interval '' with interval=10000 last_execution=2034776 (now=2044776)
[01:48:06][VV][scheduler:226]: Running interval '' with interval=10000 last_execution=2044776 (now=2054776)
[01:48:16][VV][scheduler:226]: Running interval '' with interval=10000 last_execution=2054776 (now=2064776)

```

SzczepanLeon, 2024-10-26:
You are lucky. Workaround is for short frames. That long one just appeared without any reason (or maybe V_V logs helps?).

shopro, 2024-10-26:
Well whatever it is, it's working. Steady long frames coming even without verbose logging.

I'll try changing to the main again tomorrow and see if it stops, with this workaround it started straight away.

SzczepanLeon, 2024-10-26:
Miracle. 😉

shopro, 2024-10-27:
I tested with a brand new ESP32 board, the long frame appears in both the main and fix_140 versions.

Thanks for the help.

closed 2024-10-27 by shopro