SzczepanLeon/esphome-components #141: Brak przypisania wartosci z licznika Izar

mkruchy, 2024-10-22:
Telegram lapie poprawnie ale nie wiaze pozycji:
17:08:26][V][Telegram.cpp:545]: (telegram) DLL L=19 C=44 (from meter SND_NR) M=4c30 (SAP) A=21503016 VER=00 TYPE=07 (Water meter) (driver izar) DEV=ESPHome wM-Bus RSSI=-72
[17:08:26][V][Telegram.cpp:1178]: (wmbus) parseELL @10 16
[17:08:26][V][Telegram.cpp:1333]: (wmbus) parseNWL @10 16
[17:08:26][V][Telegram.cpp:1391]: (wmbus) parseAFL @10 16
[17:08:26][V][Telegram.cpp:2078]: (wmbus) parseTPL @10 16
[17:08:26][V][Telegram.cpp:634]: (telegram) TPL CI=a2
[17:08:26][V][meters.cpp:966]: (meters) skipping field without matcher prefix(Text)[0]...
[17:08:26][V][meters.cpp:966]: (meters) skipping field without matcher serial_number(Text)[1]...
[17:08:26][V][meters.cpp:966]: (meters) skipping field without matcher total(Volume)[2]...
[17:08:26][V][meters.cpp:966]: (meters) skipping field without matcher last_month_total(Volume)[3]...
[17:08:26][V][meters.cpp:966]: (meters) skipping field without matcher last_month_measure_date(Text)[4]...
[17:08:26][V][meters.cpp:966]: (meters) skipping field without matcher remaining_battery_life(Time)[5]...
[17:08:26][V][meters.cpp:966]: (meters) skipping field without matcher current_alarms(Text)[6]...
[17:08:26][V][meters.cpp:966]: (meters) skipping field without matcher previous_alarms(Text)[7]...
[17:08:26][V][meters.cpp:966]: (meters) skipping field without matcher transmit_period(Time)[8]...
[17:08:26][V][meters.cpp:966]: (meters) skipping field without matcher manufacture_year(Text)[9]...
[17:08:26][V][driver_izar.cpp:200]: (izar) Decoded PRIOS data: 
[17:08:26][W][driver_izar.cpp:206]: (izar) Decoding PRIOS data failed. Ignoring telegram.
[17:08:26][W][wmbus:154]: Can't get requested field 'last_month_measure_date' with unit 'm³'
[17:08:26][W][wmbus:154]: Can't get requested field 'last_month_total' with unit 'm3'
[17:08:26][W][wmbus:158]: Can't get proper unit from 'Y'
[17:08:26][V][sensor:043]: 'woda RSSi': Received new state -72.000000
[17:08:26][D][sensor:094]: 'woda RSSi': Sending state -72.00000 dBm with 0 decimals of accuracy
[17:08:26][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[17:08:26][V][json:058]: Size after shrink 76 bytes
[17:08:26][W][wmbus:154]: Can't get requested field 'total' with unit 'L'
[17:08:26][W][wmbus:154]: Can't get requested field 'transmit_period_s' with unit 's'

YAML:
sensor:
  - platform: wmbus
    meter_id: 0x21503016
    type: izar
    key: "00000000000000000000000000000000"
    sensors:
      - name: "woda RSSi"
        field: "rssi"
        accuracy_decimals: 0
        unit_of_measurement: "dBm"
        device_class: "signal_strength"
        state_class: "measurement"
        entity_category: "diagnostic"
      - name: "woda"
        field: "total"
        accuracy_decimals: 3
        unit_of_measurement: "L"
        device_class: "water"
        state_class: "total_increasing"
        icon: "mdi:water"
      - name: "poprzedni"
        field: "last_month_total"
        accuracy_decimals: 3
        unit_of_measurement: "m3"
        device_class: "water"
        state_class: "total_increasing"
        icon: "mdi:water"

SzczepanLeon, 2024-10-22:
Nie ma konwersji danych - Total daj w m3 (3 jako indeks górny)

mkruchy, 2024-10-23:
`sensor:
  - platform: wmbus
    meter_id: 0x21503016
    type: izar
    key: "00000000000000000000000000000000"
    sensors:
      - name: "woda RSSi"
        field: "rssi"
        accuracy_decimals: 0
        unit_of_measurement: "dBm"
        device_class: "signal_strength"
        state_class: "measurement"
        entity_category: "diagnostic"
      - name: "woda"
        field: "total"
        accuracy_decimals: 3
        unit_of_measurement: "m³"
        device_class: "water"
        state_class: "total_increasing"
        icon: "mdi:water"
      - name: "poprzedni"
        field: "last_month_total"
        accuracy_decimals: 3
        unit_of_measurement: "m³"
        device_class: "water"
        state_class: "total_increasing"
        icon: "mdi:water"
#      - name: "data"
#        field: "last_month_measure_date"
#        accuracy_decimals: 3
#        unit_of_measurement: "m³"
#        device_class: "water"
#        state_class: "total_increasing"
#        icon: "mdi:water"
      - name: "Okres nadawania"
        field: "transmit_period"
        accuracy_decimals: 0
        unit_of_measurement: "s"
#        device_class: "signal_strength"
#        state_class: "measurement"
#        entity_category: "diagnostic"
      - name: "bateria"
        field: "remaining_battery_life"
        accuracy_decimals: 0
        unit_of_measurement: "Y"
        device_class: "signal_strength"
        state_class: "measurement"
        entity_category: "diagnostic"`

`[08:52:46][I][wmbus:092]: Using selected driver izar (detected driver was izar)
[08:52:46][I][wmbus:104]: izar [0x21503016] RSSI: -67dBm T: 1944304C163050212C04A221100013DCD061D355501E8321F288 (26) T1 A
[08:52:46][D][meters.cpp:1980]: (meter) created ESPHome izar 21503016 encrypted
[08:52:46][V][Telegram.cpp:2136]: (diehl) preprocess necessary SAP_PRIOS
[08:52:46][V][manufacturer_specificities.cpp:156]: (diehl) Pre-processing: setting device type to water meter for SAP PRIOS
[08:52:46][V][Telegram.cpp:1131]: (wmbus) parseDLL @0 26
[08:52:46][V][Telegram.cpp:1178]: (wmbus) parseELL @10 16
[08:52:46][V][Telegram.cpp:1333]: (wmbus) parseNWL @10 16
[08:52:46][V][Telegram.cpp:1391]: (wmbus) parseAFL @10 16
[08:52:46][V][Telegram.cpp:2078]: (wmbus) parseTPL @10 16
[08:52:46][V][meters.cpp:635]: (meter) ESPHome: for me? 21503016.M=SAP.V=00.T=07,21503016.M=SAP.V=00.T=07 in 21503016
[08:52:46][V][meters.cpp:686]: (meter) ESPHome: yes for me
[08:52:46][D][meters.cpp:909]: (meter) ESPHome(0) izar  handling telegram from 21503016.M=SAP.V=00.T=07
[08:52:46][V][Telegram.cpp:2136]: (diehl) preprocess necessary SAP_PRIOS
[08:52:46][V][manufacturer_specificities.cpp:156]: (diehl) Pre-processing: setting device type to water meter for SAP PRIOS
[08:52:46][V][Telegram.cpp:1131]: (wmbus) parseDLL @0 26
[08:52:46][V][Telegram.cpp:545]: (telegram) DLL L=19 C=44 (from meter SND_NR) M=4c30 (SAP) A=21503016 VER=00 TYPE=07 (Water meter) (driver izar) DEV=ESPHome wM-Bus RSSI=-67
[08:52:46][V][Telegram.cpp:1178]: (wmbus) parseELL @10 16
[08:52:46][V][Telegram.cpp:1333]: (wmbus) parseNWL @10 16
[08:52:46][V][Telegram.cpp:1391]: (wmbus) parseAFL @10 16
[08:52:46][V][Telegram.cpp:2078]: (wmbus) parseTPL @10 16
[08:52:46][V][Telegram.cpp:634]: (telegram) TPL CI=a2
[08:52:46][V][meters.cpp:966]: (meters) skipping field without matcher prefix(Text)[0]...
[08:52:46][V][meters.cpp:966]: (meters) skipping field without matcher serial_number(Text)[1]...
[08:52:46][V][meters.cpp:966]: (meters) skipping field without matcher total(Volume)[2]...
[08:52:46][V][meters.cpp:966]: (meters) skipping field without matcher last_month_total(Volume)[3]...
[08:52:46][V][meters.cpp:966]: (meters) skipping field without matcher last_month_measure_date(Text)[4]...
[08:52:46][V][meters.cpp:966]: (meters) skipping field without matcher remaining_battery_life(Time)[5]...
[08:52:46][V][meters.cpp:966]: (meters) skipping field without matcher current_alarms(Text)[6]...
[08:52:46][V][meters.cpp:966]: (meters) skipping field without matcher previous_alarms(Text)[7]...
[08:52:46][V][meters.cpp:966]: (meters) skipping field without matcher transmit_period(Time)[8]...
[08:52:46][V][meters.cpp:966]: (meters) skipping field without matcher manufacture_year(Text)[9]...
[08:52:46][V][driver_izar.cpp:200]: (izar) Decoded PRIOS data: 
[08:52:46][W][driver_izar.cpp:206]: (izar) Decoding PRIOS data failed. Ignoring telegram.
[08:52:46][W][wmbus:154]: Can't get requested field 'last_month_total' with unit 'm³'
[08:52:46][W][wmbus:158]: Can't get proper unit from 'Y'
[08:52:46][V][sensor:043]: 'woda RSSi': Received new state -67.000000
[08:52:46][D][sensor:094]: 'woda RSSi': Sending state -67.00000 dBm with 0 decimals of accuracy
[08:52:46][W][wmbus:154]: Can't get requested field 'total' with unit 'm³'
[08:52:46][W][wmbus:154]: Can't get requested field 'transmit_period' with unit 's'`

`telegram=|1944304C163050212C04A2_21100013DCD061D355501E8321F288|

Auto driver    : izar
Similar driver : unknown 00/00
Using driver   : izar 00/00
000   : 19 length (25 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : 304c dll-mfct (SAP)
004   : 16305021 dll-id (21503016)
008   : 00 dll-version
009   : 07 dll-type (Water meter)
010   : a2 tpl-ci-field (Mfct specific)
011 C?: 21100013DCD061D355501E8321F288 mfct specific

{
    "media":"water",
    "meter":"izar",
    "name":"",
    "id":"21503016",
    "last_month_total_m3":265.096,
    "remaining_battery_life_y":8,
    "total_m3":274.096,
    "transmit_period_s":8,
    "current_alarms":"no_alarm",
    "last_month_measure_date":"2024-09-30",
    "manufacture_year":"2022",
    "prefix":"H22KA",
    "previous_alarms":"no_alarm",
    "serial_number":"032406",
    "timestamp":"2024-10-23T06:53:11Z"
}

Using: wmbusmeters: 1.17.1-18-g26760f3
26760f3838aaffe0830a73316fc6b0a36ec2300b`

SzczepanLeon, 2024-10-23:
Wywal klucz

mkruchy, 2024-10-23:
I teraz dziala elegancko :)

closed 2024-10-23 by mkruchy