SzczepanLeon/esphome-components #142: help with new code

vejvik, 2024-10-24:
Hello i had this code for sensor
sensor:
  - platform: wmbus
    meter_id: 0x24356721 # 0x173A771
    type: multical21
    key: "0AF58C20AA94E0DC79B300DE3B00B61B"
    mode: C1
    total_water_m3:
      name: "Aktualni hodnota vody"
    target_water_m3:
      name: "Pocatecni hodnota mesice vody"
    flow_temperature_c:
      name: "teplota vody" 
    external_temperature_c:
      name: "teplota okoli"  
      
  in new version esphome i have to change code to
  
  sensor:
  - platform: wmbus
    meter_id: 0x24356721 # 0x173A771
    type: multical21
    key: "0AF58C20AA94E0DC79B300DE3B00B61B"
    sensors:
      - name: "Aktualni hodnota vody"
        field: "total_water_m3"
        accuracy_decimals: 3
        unit_of_measurement: "m³"
        device_class: "water"
        state_class: "total_increasing"
        icon: "mdi:water"
      - name: "Pocatecni hodnota mesice vody"
        field: "target_water_m3"
        accuracy_decimals: 3
        unit_of_measurement: "m³"
        device_class: "water"
        state_class: "measurement"
        icon: "mdi:water"
      - name: "teplota vody" 
        field: "flow_temperature_c"
        accuracy_decimals: 1
        unit_of_measurement: "°C"
        device_class: "temperature"
        state_class: "measurement"
      - name: "teplota okoli"  
        field: "external_temperature_c"
        accuracy_decimals: 1
        unit_of_measurement: "°C"
        device_class: "temperature"
        state_class: "measurement"  
        
        now i do not have any value, all sensors are unknown, please help

SzczepanLeon, 2024-10-24:
Traces, please include VERBOSE traces.

vejvik, 2024-10-24:
my yaml:
esphome:
  name: vodomer
  friendly_name: vodomer

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger: 
  level: VERBOSE

# Enable Home Assistant API
api:
  encryption:
    key: "6OQlzejBK2ZhdK+oyVZJZOEvcTNibZmOXqVJXE17oek="

ota:
  - platform: esphome
    password: "f442e4f8dfdc81d520348879078f3c84"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Vodomer Fallback Hotspot"
    password: "hZvzkD4Ke2RE"

time:
  - platform: sntp
    id: time_sntp

external_components:
  - source: github://SzczepanLeon/esphome-components@main
    refresh: 0s
    components: [ wmbus ]

wmbus:
  frequency: 868.950
  mosi_pin: GPIO13
  miso_pin: GPIO5
  clk_pin:  GPIO2
  cs_pin:   GPIO14
  gdo0_pin: GPIO15
  gdo2_pin: GPIO16

  led_pin: GPIO0
  led_blink_time: "1s"

 # clients:
 #   - name: "wmbusmetersesp"
 #     ip_address: "192.168.1.22"
 #     port: 1883

sensor:
  - platform: wmbus
    meter_id: 0x24356721 # 0x173A771
    type: multical21
    key: "0AF58C20AA94E0DC79B300DE3B00B61B"
    sensors:
      - name: "Aktualni hodnota vody"
        field: "total_m3"  # Změněno z "total_water_m3" na "total_m3"
        accuracy_decimals: 3
        unit_of_measurement: "m³"
        device_class: "water"
        state_class: "total_increasing"
[... 164 lines cut ...]
[20:50:38][V][meters.cpp:2155]: (meter) Volume target_m3 decoded m3 default m3 value 105.169 (scale 1)
[20:50:38][V][meters.cpp:973]: (meters) trying field info flow_temperature(Temperature)[3]...
[20:50:38][V][meters.cpp:995]: (meters) using field info flow_temperature(Temperature)[3] to extract 615B at offset 34
[20:50:38][V][meters.cpp:2155]: (meter) FlowTemperature flow_temperature_c decoded c default c value 15 (scale 1)
[20:50:38][V][meters.cpp:973]: (meters) trying field info external_temperature(Temperature)[4]...
[20:50:38][V][meters.cpp:995]: (meters) using field info external_temperature(Temperature)[4] to extract 6167 at offset 35
[20:50:38][V][meters.cpp:2155]: (meter) ExternalTemperature external_temperature_c decoded c default c value 14 (scale 1)
[20:50:38][V][meters.cpp:973]: (meters) trying field info min_external_temperature(Temperature)[5]...
[20:50:38][V][meters.cpp:973]: (meters) trying field info max_flow(Flow)[6]...
[20:50:38][V][meters.cpp:973]: (meters) trying field info current_status(Text)[7]...
[20:50:38][V][meters.cpp:995]: (meters) using field info current_status(Text)[7] to extract 02FF20 at offset 24
[20:50:38][V][meters.cpp:973]: (meters) trying field info time_dry(Text)[8]...
[20:50:38][V][meters.cpp:995]: (meters) using field info time_dry(Text)[8] to extract 02FF20 at offset 24
[20:50:38][V][meters.cpp:973]: (meters) trying field info time_reversed(Text)[9]...
[20:50:38][V][meters.cpp:995]: (meters) using field info time_reversed(Text)[9] to extract 02FF20 at offset 24
[20:50:38][V][meters.cpp:973]: (meters) trying field info time_leaking(Text)[10]...
[20:50:38][V][meters.cpp:995]: (meters) using field info time_leaking(Text)[10] to extract 02FF20 at offset 24
[20:50:38][V][meters.cpp:973]: (meters) trying field info time_bursting(Text)[11]...
[20:50:38][V][meters.cpp:995]: (meters) using field info time_bursting(Text)[11] to extract 02FF20 at offset 24
[20:50:38][W][wmbus:154]: Can't get requested field 'external_temperature_c' with unit '°C'
[20:50:38][W][wmbus:154]: Can't get requested field 'flow_temperature_c' with unit '°C'
[20:50:38][W][wmbus:154]: Can't get requested field 'target_m3' with unit 'm³'
[20:50:38][W][wmbus:154]: Can't get requested field 'total_m3' with unit 'm³'
[20:50:38][W][component:237]: Component wmbus took a long time for an operation (572 ms).
[20:50:38][W][component:238]: Components should block for at most 30 ms.

vejvik, 2024-10-24:
esp sees telegrams and decodes them correctly, but in HA all sensors are unknown

vejvik, 2024-10-24:
problem solved
as you wrote in the instructions
i changed first total_water_m3 to total_m3 but right is only total

closed 2024-10-24 by SzczepanLeon