SzczepanLeon/esphome-components #143: [V4] All telegrams have CRC errors after migrating

ndandanov, 2024-10-25:
Hi and many thanks for the great component!

As soon as v4 was released, I migrated my configuration to it. Back then I noticed that no readings from sensors were ever received.

I rolled back to my working v3 config, and a few days ago I assembled a second ESP32 board with CC1101.

I kept the two boards running for a few days and my previous observations were confirmed: the new board with v4 did not read successfully **any** package for at least 48 hours!

I managed to capture the output of both boards. Here a snippet of the working one (v3) with successful decoding:
```
[01:10:43][V][rxLoop:167]: Have 68 bytes from CC1101 Rx, RSSI: -64 dBm LQI: 129
[01:10:43][D][mbus:013]: Processing C1 A frame
[01:10:43][V][mbus:017]: Frame: 39449344708648131707D26C7A5F0000200C13351600004C13351600E65F00426CFF2CCC081335160000C2086C1E3F0A3902BB560000326CFFFF046D0717193A3955 (66) [with CRC]
[01:10:43][V][mbus:095]: Validating CRC for Block1
[01:10:43][V][crc:031]:     calculated: 0xD26C, read: 0xD26C
[01:10:43][V][mbus:115]: Validating CRC for Block2
[01:10:43][V][crc:031]:     calculated: 0xE65F, read: 0xE65F
[01:10:43][V][mbus:115]: Validating CRC for Block3
[01:10:43][V][crc:031]:     calculated: 0x3F0A, read: 0x3F0A
[01:10:43][V][mbus:115]: Validating CRC for Block4
[01:10:43][V][crc:031]:     calculated: 0x3955, read: 0x3955
[01:10:43][V][mbus:062]: Frame: 394493447086481317077A5F0000200C13351600004C1335160000426CFF2CCC081335160000C2086C1E3902BB560000326CFFFF046D0717193A (58) [without CRC]
[01:10:43][D][wmbus:182]: Meter ID [0x13488670] RSSI: -64 dBm LQI: 129 Frame: C1 A not found in configuration T: 394493447086481317077A5F0000200C13351600004C1335160000426CFF2CCC081335160000C2086C1E3902BB560000326CFFFF046D0717193A (58)
[01:10:43][V][wmbus:228]: Will send RTLWMBUS telegram to 192.168.17.99:9022 via TCP
```

And this is the same telegram received by the board with v4 and CRC errors:

```
[01:10:43][V][rxLoop:167]: Have 68 bytes from CC1101 Rx, RSSI: -80 dBm LQI: 145
[01:10:43][D][mbus:014]: Received C1 A frame
[01:10:43][V][mbus:018]: Frame: 39449344708648131707D26C7A5F0800200C13251600004C13351600E65F00426CFF2CCC081335160000C2086C1E3F0A393906183600003BD88C050609310000CD39 (66) [with CRC]
[01:10:43][V][mbus:096]: Validating CRC for Block1
[01:10:43][V][crc:031]:     calculated: 0xD26C, read: 0xD26C
[01:10:43][V][mbus:116]: Validating CRC for Block2
[01:10:43][D][crc:035]:     calculated: 0xD16B, read: 0xE65F  !!!
```
Unfortunately, `sync_mode` did not make any difference.

ndandanov, 2024-10-25:
Please note that all around me, the meters are all `qundis` (`qwater` and `qheat_55_us` according to https://wmbusmeters.org/).

Below is my config for v4:
```yaml
substitutions:
  ## device settings
  device_name_short: water-meter-esp32-s3
  device_description: Water- and heatmeters ESP32, CUL - CC1101
  projectname: Water and heat meters
  friendly_name: Water and heat meters
  appversion: "2.0.9"

  ## logger settings
  log_level: "VERBOSE" # not that logging need memory, so in production mode use "WARN"
  log_wmbus: "VERBOSE" # Loglevel for wmbus meters component
  log_baudrate: "115200" # use 0 to disable serial UART port logging.

  ## data service (optional, default: False)
  ## use only if you have a service host present
  service_enabled: "false"
  service_url: !secret service_host
  service: "izar.watermeter"

## -----------------------------------------------------------------
## HARDWARE esp32-s3-devkitc-1
## -----------------------------------------------------------------
esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino
    version: latest
  flash_size: 16MB

# Customize the frequency in which data is flushed to the flash.
preferences:
  flash_write_interval: 10min

## ----------------------------------------------------------------
## APPLICATION ESPHOME
## ----------------------------------------------------------------
esphome:
  name: ${device_name_short}
  comment: ${device_description}
  platformio_options:
    build_flags: -DBOARD_HAS_PSRAM
    board_build.arduino.memory_type: qio_opi

## ----------------------------------------------------------------
## EXTERNAL COMPONENTS
## ----------------------------------------------------------------
external_components:
  # uses the latest version from SzczepanLeon (github)
  # https://github.com/SzczepanLeon/esphome-components
  # You can make ESPHome check the repository every time by setting this option to 0s
  - source: github://SzczepanLeon/esphome-components@main
    refresh: 0s
    components: [wmbus]

  # optinal: use local component from
  # https://github.com/zdzichu6969/esphome-components
  - source:
      type: local
      path: custom_components
    components: [backup]

  # optinal: use local component from
  # https://github.com/TheStaticTurtle/esphome_syslog
  - source:
      type: local
      path: custom_components
[... 171 lines cut ...]
  - platform: wmbus
    meter_id: !secret meterHeatId
    type: qheat_55_us
    sensors:
      - name: "Heat Meter RSSI"
        field: "rssi"
        accuracy_decimals: 0
        unit_of_measurement: "dBm"
        device_class: "signal_strength"
        state_class: "measurement"
        entity_category: "diagnostic"
      - name: "Heat Meter LQI"
        field: "lqi"
        unit_of_measurement: "lqi"
        state_class: "measurement"
        entity_category: "diagnostic"
      - name: "Heat"
        field: "total"
        accuracy_decimals: 3
        unit_of_measurement: "kWh"
        device_class: "energy"
        state_class: "total_increasing"
        icon: "mdi:counter"
  - platform: wmbus
  ```

SzczepanLeon, 2024-10-26:
I don't have idea why You have CRC error - CC1101 part is common for V3.x and V4.0.

Maybe You should try V4.0 with minimum YAML.

ndandanov, 2024-10-28:
Thank you, @SzczepanLeon!

I tried a minimal config for the past few days, but still there are almost 0 telegrams without CRC errors.

I have both devices (v3 and v4) sitting next to each other, and capture their logs. I will try to inspect them in the next days to see if I can find any clue regarding the CRC errors.

Do you think that any change introduced in v4 can be the culprit?

For what it's worth, all meters around me operate in C1 mode.

SzczepanLeon, 2024-10-28:
> Do you think that any change introduced in v4 can be the culprit?

Not really. As I mentioned before CC1101 part is common.
You canntry without sensors defined. Just VERBOSE log.

SzczepanLeon, 2024-10-28:
Please turn on **SYNC**hronous mode.

ndandanov, 2024-10-28:
Thank you, I enabled sync mode. No evident change so far, unfortunately, with all telegrams having CRC errors (in block 3 mainly).

I will leave the board to collect some logs and will report how it went in the next days.

ndandanov, 2024-11-27:
After about a month of collected logs, I confirm that only **8 out of 390077** frames have been decoded without CRC errors. Most errors seem to be in blocks 2 and 3.

Today I reflashed the board with version 4.1.0, but there were no successfully decoded frames since then.

To compare, the board running version 3 has decoded without CRC error **283579 out of 356833** frames. The two boards were 30 cm apart next to each other and antennas were oriented similarly. The different number of frames is mainly due to slight differences in deployment time.

I think all the frames are from `qundis` devices (`qwater` and `qheat_55_us`). All 8 successfully decoded frames in version 4 are from `qwater` meters.

Any guidance on how to debug those CRC errors? I already have some additional ESP32 and CC1101 boards, I can deploy a third device if needed :smiley:

SzczepanLeon, 2025-02-13:
> For some reason, I was under the impression that those pins are not so important, but it turns out I had misremembered that.

Those pins are very important.

closed 2025-02-19 by SzczepanLeon

ndandanov, 2025-02-20:
Sorry, I forgot to confirm that everything works nicely so far. So, the issue is resolved. Thank you! :)