SzczepanLeon/esphome-components #144: Problem with units

pizang, 2024-10-26:
I had a working esp32 that was reading my central heat ing cost dividers through wmbus.After upgrading both ESP32Home and wmbus it still reads data but I have following errors in the log:

[14:59:40][W][wmbus:144]: Fields without unit not supported yet!
[14:59:40][W][wmbus:154]: Can't get requested field 'temp_room_avg_c' with unit '°C'

Sensor definition:

sensor:
  - platform: wmbus
    meter_id: 0x123456789
    type: apatoreitn
    sensors:
      - name: "Kuchnia Current heat"
        field: "current_hca"
        unit_of_measurement: ""
      - name: "Kuchnia Prevous year heat"
        field: "previous_hca"
        unit_of_measurement: ""
      - name: "Kuchnia Room avg temp"
        field: "temp_room_avg_c"
        unit_of_measurement: "°C"
      - name: "Kuchnia Previous Room avg temp"
        field: "temp_room_prev_avg_c"
        unit_of_measurement: "°C"

HCA is a value without an unit - it is kind of "tick" counter. I also tried None in the unit field. Celsius degrees also do not work.

Propably I have an error in the config but cannot figure it out.

SzczepanLeon, 2024-10-26:
RTFM

pizang, 2024-10-30:
I RTFM (sorry for that) and yes, temperature works flawlessly. The removal of HCA also works!

[12:05:54][D][mbus:035]: Received T1 A frame
[12:05:54][I][wmbus:092]: Using selected driver apatoreitn (detected driver was apatoreitn)
[12:05:54][I][wmbus:104]: apatoreitn [0x12345678] RSSI: -74dBm T: REDACTED (26) T1 A
[12:05:54][W][wmbus:111]: Link modes not defined in driver apatoreitn. Processing anyway.
[12:05:54][D][meters.cpp:1980]: (meter) created ESPHome apatoreitn 12345678 not-encrypted
[12:05:54][D][meters.cpp:909]: (meter) ESPHome(0) apatoreitn  handling telegram from 12345678 .M=APA.V=04.T=08
[12:05:54][D][sensor:094]: 'Kuchnia Current heat': Sending state 326.00000 hca with 0 decimals of accuracy
[12:05:54][D][sensor:094]: 'Kuchnia Prevous year heat': Sending state 3532.00000 hca with 0 decimals of accuracy
[12:05:54][D][sensor:094]: 'Kuchnia Room avg temp': Sending state 21.39062 °C with 0 decimals of accuracy
[12:05:54][D][sensor:094]: 'Kuchnia Previous Room avg temp': Sending state 20.87500 °C with 0 decimals of accuracy
[12:05:54][W][component:237]: Component wmbus took a long time for an operation (110 ms).
[12:05:54][W][component:238]: Components should block for at most 30 ms.

Config:

sensor:
  - platform: wmbus
    meter_id: 0x0123445566
    type: apatoreitn
    sensors:
      - name: "Kuchnia Current heat"
        field: "current"
        unit_of_measurement: "hca"
      - name: "Kuchnia Prevous year heat"
        field: "previous"
        unit_of_measurement: "hca"
      - name: "Kuchnia Room avg temp"
        field: "temp_room_avg"
        unit_of_measurement: "°C"
      - name: "Kuchnia Previous Room avg temp"
        field: "temp_room_prev_avg"
        unit_of_measurement: "°C"

Thanks a lot!

closed 2024-10-30 by pizang