SzczepanLeon/esphome-components #145: Sensor for correctly connected CC1101 module

ThomDietrich, 2024-10-30:
Hey there!
I wonder if it would be useful to provide the connectivity of the CC1101 module as a sensor. Currently it can not be distinguished whether the module connection loosened or whether no telegrams were received (e.g. distance or interference). The RSSI sensor shows "unknown" in both cases and the user needs to check the console or randomly move the device.

Thanks and best! Thomas

SzczepanLeon, 2024-11-01:
You can try to use lambda from ESPHome and function **is_failed()** to create sensor in YAML.
I don't have any working example, yet.

ThomDietrich, 2024-11-02:
Hey, thanks for this! I can confirm it's working at boot, however not quite deterministic when I start pulling wires at runtime. Still, seems good enough to identify issues. Thanks!

If you like, add this to the README example:
```yaml
  - platform: template
    name: CC1101 I2C
    device_class: connectivity
    entity_category: diagnostic
    lambda: "return !id(wmbus_component).is_failed();"
```

closed 2024-11-02 by ThomDietrich