SzczepanLeon/esphome-components #148: Not able to find the ID of my iPerl Watermeter

smai86, 2024-11-01:
Hey Guys,
I´m not knowing what I´m doing wrong... 
I cant get the ID of my iPerl Watermeter... I am using an ESP32 D1 Mini with a C1101 Board.

Here is my Config in HA:
_esphome:
  name: watermeter
  friendly_name: watermeter

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  level: VERBOSE

# Enable Home Assistant API
api:
  encryption:
    key: "XXXXX"

ota:
  - platform: esphome
    password: "XXXXX"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Watermeter Fallback Hotspot"
    password: "XXXXX"

captive_portal:

external_components:
  - source: github://SzczepanLeon/esphome-components@main

wmbus:
  mosi_pin: GPIO19
  miso_pin: GPIO23
  clk_pin: GPIO18
  cs_pin: GPIO05
  gdo0_pin: GPIO21
  gdo2_pin: GPIO22
  log_all: True

time:
  - platform: sntp
    id: time_sntp_

Here is what I get from the ESPHome Log:
_[08:46:47][V][rxLoop:167]: Have 152 bytes from CC1101 Rx, RSSI: -99 dBm LQI: 226
[08:46:47][D][mbus:035]: Received T1 A frame
[08:46:47][V][mbus:046]: Frame: 66C4EC2231BCE5B206714B549E646593FD0C157C75780EC4EF9682F4B8B958D831E69D831A4396AE262A805D7723F902894FB43B3780A43D033A767AAABA7107EDAA19F35DDFAEB6B8DE63FCB0B62F7FBF9EA57DDD026EB27B3DF4D32F54ED04768FC450FBFFF8D7F8CA4356E2CCA51BC0C3F4B089D6E0948EF30B97F37DDDF390B788C9BB55C7C48DA8B0C9FAA11E7EBB8C206BC48C021C (152) [RAW]
[08:46:47][V][3of6:083]: Decode 3 out of 6 failed.._

The board is placed 10cm in front of the iPerl.

Anyone got an idea?

Greetings,
Steffen

SzczepanLeon, 2024-11-01:
RSSI is quite low for CC1101. I would check antena.

gomble, 2024-11-13:
> Thanks for your fast reply. I just resoldered the antena, but doesnt get any better. Now I just ordered an other C1101 module from amazon, maybe mine is broken. Im getting an update on monday.
> 
> Thanks for your help!

and was it the antenna?

smai86, 2024-11-13:
> > Thanks for your fast reply. I just resoldered the antena, but doesnt get any better. Now I just ordered an other C1101 module from amazon, maybe mine is broken. Im getting an update on monday.
> > Thanks for your help!
> 
> and was it the antenna?

idk... with the new C1101 board it was working like a charm. but now i´ve got the problem that i cant get the aes key from my water supplier and the "default key" isnt working for me :-(

closed 2024-11-13 by SzczepanLeon