SzczepanLeon/esphome-components #149: DoIt ESP32 devkit v1 version with 36 GPIOS - boot loop

Mareckil, 2024-11-01:
hi I am having problem with running WmBus component on my ESP32 it looks like it have some issue while connecting to my WiFi.
Board is fine it was running webradio and didn't have any issues with connection to the same WiFi network.

log:
[logs (2).txt](https://github.com/user-attachments/files/17603107/logs.2.txt)

config:
[liczniki.txt](https://github.com/user-attachments/files/17603128/liczniki.txt)

SzczepanLeon, 2024-11-01:
Please start with minimum YAML.

Mareckil, 2024-11-01:
removed wmbus from yaml and it connecting to wifi without any problems even with cc1101 module still connected, connected and configured i2c sensor vl53l0x and works without any problems. Compiled with only wmbus module - fine, compiled with watermeter driver - crashes and boot loop

SzczepanLeon, 2024-11-01:
Please collect VERY_VERBOSE logs.

Mareckil, 2024-11-02:
There are yaml config and log file. I've commented i2c device so it is only wmbus compiled.
[logs (5).txt](https://github.com/user-attachments/files/17606333/logs.5.txt)
[esp_yaml.txt](https://github.com/user-attachments/files/17606335/esp_yaml.txt)

SzczepanLeon, 2024-11-02:
Please try this version:

```
external_components:
  - source: github://SzczepanLeon/esphome-components@wdt_loop
```

and collect **VERBOSE** logs.

Mareckil, 2024-11-02:
new log:
[logs (6).txt](https://github.com/user-attachments/files/17607970/logs.6.txt)

SzczepanLeon, 2024-11-02:
Are you 100% sure that pins are defined properly? 
In logs I see at least two CC1101 versions - should be one.

Mareckil, 2024-11-02:
new log:

Mareckil, 2024-11-02:
Well, I think you're right, probably one of the connectors in the socket was unreliable, I soldered it permanently and it seems to be fine. My mistake, what lost me was that without any drivers the module seemed to work fine.
Great thanks for your very quick response and help.

closed 2024-11-02 by Mareckil