SzczepanLeon/esphome-components #151: No messages send to homeassistant

code4me2all, 2024-11-08:
Hi,

I have some trouble to get messages to hass. My hardware (esp32 woom)  is up and running and receive telegrams from the watermeter. 
[I][wmbus:104]: hydrus [0x71236535] RSSI: -78dBm T: *********7170077A150030051***** (63) T1 A //shorten with!

The analyzing of the telegram with  https://wmbusmeters.org/analyze gave me this
{
    "media":"water",
    "meter":"hydrus",
    "name":"",
    "id":"01234567",
    "flow_temperature_c":20.8,
    "remaining_battery_life_y":12.734005,
    "target_m3":484.945,
    "target_datetime":"2024-10-31 23:59",
    "total_m3":490.992,
    "status":"OK",
    "timestamp":"2024-11-08T18:46:19Z"
} 

then i add in esphome.yaml:

mqtt:
  broker: *.*.*.* #ip of my mqtt server
  username: mqttuser
  password: !secret mqtt_password

sensor:
  - platform: wmbus
    meter_id: 1234567
    type: hydrus
    key: "51728910e66d83f851728910e66d83f8"
    sensors:
      - name: "Hauptwasserzaehler RSSi"
        field: "rssi"
        accuracy_decimals: 0
        unit_of_measurement: "dBm"
        device_class: "signal_strength"
        state_class: "measurement"
        entity_category: "diagnostic"

      - name: "Hauptwasserzaehler_total"
        field: "total"
        accuracy_decimals: 3
        unit_of_measurement: "m³"
        device_class: "water"
        state_class: "total_increasing"
        icon: "mdi:water"

I get on mqtt server log:
2024-11-08 21:01:14: New client connected from *.*.*.*:51623 as watermeter-********** (p2, c0, k15, u'mqttuser').

But no values are transmitted in HASS over MQTT
Some hint why not?

SzczepanLeon, 2024-11-08:
In VERBOSE logs do you see something MQTT related?

code4me2all, 2024-11-08:
Not really , only at the beginning:

[18:31:08][W][component:170]: Component mqtt cleared Warning flag
[18:31:08][I][mqtt:286]: MQTT Connected!
[18:31:08][W][component:237]: Component mqtt took a long time for an operation (117 ms).
[18:31:08][W][component:238]: Components should block for at most 30 ms.

SzczepanLeon, 2024-11-08:
I'm not mqtt component expert. It is not related to my component.

closed 2024-11-08 by SzczepanLeon

code4me2all, 2024-11-08:
I also do not see the values on the webif -> State NA
Is there an other possability to get the values into hass?

SzczepanLeon, 2024-11-08:
Hass=HomeAssistant??

Please include full DEBUG log.

code4me2all, 2024-11-09:
Found a solution. Send to hass addon wmbusmeter with 
  clients:
    - name: "wmbusmeters"
      ip_address: "x.x.x.x"
      port: 9022
      transport: TCP

In wmbusmeter configuration set _netcat-listen-tcp_ to 9022 and in _wmbusmeters HA addon configuration_ in the device section _rtlwmbus:CMD(nc -lk 9022)_

Also need full meter setup (driver,id,key,name)

SzczepanLeon, 2024-11-09:
This is not solution. It is workaround (rather ugly one).

If You want help then please include DEBUG or VERBOSE log from time period when telegram is received and processed.

code4me2all, 2024-11-09:
Here it is the VERBOSE log:

`[15:15:10]ets Jul 29 2019 12:21:46
[15:15:10]
[15:15:10]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[15:15:10]configsip: 0, SPIWP:0xee
[15:15:10]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[15:15:10]mode:DIO, clock div:2
[15:15:10]load:0x3fff0030,len:1184
[15:15:10]load:0x40078000,len:13132
[15:15:10]load:0x40080400,len:3036
[15:15:10]entry 0x400805e4
[15:15:10][    16][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[15:15:10][I][logger:156]: Log initialized
[15:15:10][C][safe_mode:079]: There have been 2 suspected unsuccessful boot attempts
[15:15:10][D][esp32.preferences:114]: Saving 1 preferences to flash...
[15:15:10][V][esp32.preferences:126]: sync: key: 233825507, len: 4
[15:15:10][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[15:15:10][I][app:029]: Running through setup()...
[15:15:10][V][app:030]: Sorting components by setup priority...
[15:15:10][C][wifi:048]: Setting up WiFi...
[15:15:10][C][wifi:061]: Starting WiFi...
[15:15:10][C][wifi:062]:   Local MAC: D8:BC:38:E5:FE:64
[15:15:10][V][esp32.preferences:059]: nvs_get_blob('1802229350'): ESP_ERR_NVS_NOT_FOUND - the key might not be set yet
[15:15:10][V][wifi_esp32:072]: Enabling STA.
[15:15:10][   110][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
[15:15:10][V][wifi_esp32:491][arduino_events]: Event: WiFi ready
[15:15:10][   205][V][WiFiGeneric.cpp:338] _arduino_event_cb(): STA Started
[15:15:10][   205][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 2 - STA_START
[15:15:10][V][wifi_esp32:502][arduino_events]: Event: WiFi STA start
[15:15:16][  5826][V][WiFiGeneric.cpp:381] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 2
[15:15:16][  5826][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 1 - SCAN_DONE
[15:15:16][V][wifi_esp32:496][arduino_events]: Event: WiFi Scan Done status=0 number=2 scan_id=128
[15:15:16][D][wifi:497]: Found networks:
[15:15:16][I][wifi:541]: - 'Media' (16:CC:20:E1:40:26) [redacted]▂▄▆█
[15:15:16][D][wifi:542]:     Channel: 13
[15:15:16][D][wifi:543]:     RSSI: -44 dB
[15:15:16][D][wifi:546]: - 'monster.at'[redacted] (14:CC:20:E1:40:26) [redacted]▂▄▆█
[15:15:16][I][wifi:313]: WiFi Connecting to 'Media'...
[15:15:16][V][wifi:315]: Connection Params:
[15:15:16][V][wifi:316]:   SSID: 'Media'
[15:15:16][V][wifi:319]:   BSSID: 16:CC:20:E1:40:26
[15:15:16][V][wifi:349]:   Password: '***'[redacted]
[15:15:16][V][wifi:354]:   Channel: 13
[15:15:16][V][wifi:363]:   Using DHCP IP
[15:15:16][V][wifi:365]:   Hidden: NO
[15:15:17][  6546][V][WiFiGeneric.cpp:353] _arduino_event_cb(): STA Connected: SSID: Media, BSSID: 16:cc:20:e1:40:26, Channel: 13, Auth: WPA2_PSK
[15:15:17][  6547][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[15:15:17][V][wifi_esp32:521][arduino_events]: Event: Connected ssid='Media' bssid=16:CC:20:E1:40:26[redacted] channel=13, authmode=WPA2 PSK
[15:15:17][  6560][V][WiFiGeneric.cpp:367] _arduino_event_cb(): STA Got New IP:172.31.66.218
[15:15:17][  6576][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[15:15:17][  6583][D][WiFiGeneric.cpp:991] _eventCallback(): STA IP: 172.31.66.218, MASK: 255.255.255.0, GW: 172.31.66.1
[15:15:17][V][wifi_esp32:575][arduino_events]: Event: Got IP static_ip=172.31.66.218 gateway=172.31.66.1
[15:15:17][I][wifi:617]: WiFi Connected!
[15:15:17][C][wifi:428]:   Local MAC: D8:BC:38:E5:FE:64
[15:15:17][C][wifi:433]:   SSID: 'Media'[redacted]
[15:15:17][C][wifi:436]:   IP Address: 172.31.66.218
[15:15:17][C][wifi:440]:   BSSID: 16:CC:20:E1:40:26[redacted]
[15:15:17][C][wifi:441]:   Hostname: 'watermetermqtt-e5fe64'
[15:15:17][C][wifi:443]:   Signal strength: -42 dB ▂▄▆█
[15:15:17][V][wifi:445]:   Priority: 0.0
[15:15:17][C][wifi:447]:   Channel: 13
[15:15:17][C][wifi:448]:   Subnet: 255.255.255.0
[15:15:17][C][wifi:449]:   Gateway: 172.31.66.1
[15:15:17][C][wifi:450]:   DNS1: 172.31.66.1
[15:15:17][C][wifi:451]:   DNS2: 0.0.0.0
[15:15:17][D][wifi:626]: Disabling AP...
[15:15:17][C][web_server:100]: Setting up web server...
[15:15:17][C][sntp:023]: Setting up SNTP...
[15:15:17][C][mqtt:037]: Setting up MQTT...
[... 428 lines cut ...]
[15:17:49][V][rxLoop:167]: Have 110 bytes from CC1101 Rx, RSSI: -58 dBm LQI: 128
[15:17:49][V][rxLoop:172]: Synchronus mode enabled.
[15:17:49][D][mbus:035]: Received T1 A frame
[15:17:49][V][mbus:046]: Frame: 2F271C99934D2D969938B4CD4D659336C5B44E66AC5962D6599732B31B26D0BA72963A65C65A717234F2CB1C4B3A6A5AD1C9656718F1D13A6359C94B65CC598CDA5CA5A72C2CE65A2ECCB2C65996A6CB16CA694E8DCD0B3AC4E598D3ACCB29963B134D2E6CB15A398BB0DD32B328 (110) [RAW]
[15:17:49][V][mbus:053]: Frame: 3E44A511356523717007180C7A680030054E8D8AC3FE9BF9D9FD4B7EEDD32AF6C4995DBDC7FB049354D5B1F4F648325638EED9A0F880EA92B4C32879A128EEA02D113AED0BA381CE8E (73) [with CRC]
[15:17:49][V][mbus:096]: Validating CRC for Block1
[15:17:49][V][crc:031]:     calculated: 0x180C, read: 0x180C
[15:17:49][V][mbus:116]: Validating CRC for Block2
[15:17:49][V][crc:031]:     calculated: 0xEDD3, read: 0xEDD3
[15:17:49][V][mbus:116]: Validating CRC for Block3
[15:17:49][V][crc:031]:     calculated: 0x3256, read: 0x3256
[15:17:49][V][mbus:116]: Validating CRC for Block4
[15:17:49][V][crc:031]:     calculated: 0x2D11, read: 0x2D11
[15:17:49][V][mbus:116]: Validating CRC for Block5
[15:17:49][V][crc:031]:     calculated: 0xCE8E, read: 0xCE8E
[... cut ...]
[15:17:49][V][Telegram.cpp:1333]: (wmbus) parseNWL @10 53
[15:17:49][V][Telegram.cpp:1391]: (wmbus) parseAFL @10 53
[15:17:49][V][Telegram.cpp:2078]: (wmbus) parseTPL @10 53
[15:17:49][I][wmbus:104]: hydrus [0x71236535] RSSI: -58dBm T: 3E44A5113565237170077A680030054E8D8AC3FE9BF9D9FD4B7E2AF6C4995DBDC7FB049354D5B1F4F64838EED9A0F880EA92B4C32879A128EEA03AED0BA381 (63) T1 A
[15:17:49][W][component:237]: Component wmbus took a long time for an operation (180 ms).
[15:17:49][W][component:238]: Components should block for at most 30 ms.
`

code4me2all, 2024-11-09:
here the log file
[watermetermqtt_logs (2).txt](https://github.com/user-attachments/files/17688480/watermetermqtt_logs.2.txt)

SzczepanLeon, 2024-11-09:
You have wrong ID in YAML. Should be 0x71236535 or 1898145077.

code4me2all, 2024-11-09:
I changed the ID and it works, but i do not understand why it is here a hex value.
On my linux system i have this working config:
/etc/wmbusmeters.d/MyTapWater
_name=Hauptwasserzaehler
id=71236535
key=51728910e66d83f851728910e66d83f8
driver=hydrus_ 
Anyway 

Great thank's for your help.

I want to sponsore your project a little bit, if possible.

SzczepanLeon, 2024-11-09:
> but i do not understand why it is here a hex value.

Because it is in hex - see RAW frame.