SzczepanLeon/esphome-components #153: DiehlAddressTransformMethod

nikito7, 2024-11-12:
How to configure this?

I need "SWAPPING"

SzczepanLeon, 2024-11-12:
It should works. Do you have VERBOSE logs?

nikito7, 2024-11-12:
id needs swap. "0782" it's version field

```
[12:16:51][I][wmbus:092]: Using selected driver izar (detected driver was )
[12:16:51][I][wmbus:104]: izar [0x61620782] RSSI: -69dBm T: 19442434820762619119A2BD0E0013C5D9A3961185257575F162 (26) T1 A
[12:16:51][V][Telegram.cpp:1178]: (wmbus) parseELL @10 16
[12:16:51][V][Telegram.cpp:1333]: (wmbus) parseNWL @10 16
[12:16:51][V][Telegram.cpp:1391]: (wmbus) parseAFL @10 16
[12:16:51][V][Telegram.cpp:2078]: (wmbus) parseTPL @10 16
[12:16:51][V][meters.cpp:635]: (meter) ESPHome: for me? 61620782.M=MAD.V=91.T=19,61620782.M=MAD.V=91.T=19 in 61620782
[12:16:51][V][meters.cpp:681]: (meter) ESPHome: meter detection did not match the selected driver 'izar', detected driver is: 'unknown!'
[12:16:51][V][meters.cpp:686]: (meter) ESPHome: yes for me
[12:16:51][D][meters.cpp:909]: (meter) ESPHome(0) izar  handling telegram from 61620782.M=MAD.V=91.T=19
[12:16:51][V][Telegram.cpp:1131]: (wmbus) parseDLL @0 26
[12:16:51][V][Telegram.cpp:545]: (telegram) DLL L=19 C=44 (from meter SND_NR) M=3424 (MAD) A=61620782 VER=91 TYPE=19 (A/D converter) (driver unknown!) DEV=ESPHome wM-Bus RSSI=-69
[12:16:51][V][Telegram.cpp:1178]: (wmbus) parseELL @10 16
[12:16:51][V][Telegram.cpp:1333]: (wmbus) parseNWL @10 16
[12:16:51][V][Telegram.cpp:1391]: (wmbus) parseAFL @10 16
[12:16:51][V][Telegram.cpp:2078]: (wmbus) parseTPL @10 16
[12:16:51][V][Telegram.cpp:634]: (telegram) TPL CI=a2
[12:16:51][V][meters.cpp:966]: (meters) skipping field without matcher prefix(Text)[0]...
[12:16:51][V][meters.cpp:966]: (meters) skipping field without matcher serial_number(Text)[1]...
[12:16:51][V][meters.cpp:966]: (meters) skipping field without matcher total(Volume)[2]...
[12:16:51][V][meters.cpp:966]: (meters) skipping field without matcher last_month_total(Volume)[3]...
[12:16:51][V][meters.cpp:966]: (meters) skipping field without matcher last_month_measure_date(Text)[4]...
[12:16:51][V][meters.cpp:966]: (meters) skipping field without matcher remaining_battery_life(Time)[5]...
[12:16:51][V][meters.cpp:966]: (meters) skipping field without matcher current_alarms(Text)[6]...
[12:16:51][V][meters.cpp:966]: (meters) skipping field without matcher previous_alarms(Text)[7]...
[12:16:51][V][meters.cpp:966]: (meters) skipping field without matcher transmit_period(Time)[8]...
[12:16:51][V][meters.cpp:966]: (meters) skipping field without matcher manufacture_year(Text)[9]...
[12:16:51][V][driver_izar.cpp:200]: (izar) Decoded PRIOS data:
[12:16:51][W][driver_izar.cpp:206]: (izar) Decoding PRIOS data failed. Ignoring telegram.
[12:16:51][V][sensor:043]: 'Signal': Received new state -69.000000
[12:16:51][D][sensor:094]: 'Signal': Sending state -69.00000 dBm with 0 decimals of accuracy
[12:16:51][W][wmbus:154]: Can't get requested field 'total' with unit 'm³'
[12:16:51][W][component:237]: Component wmbus took a long time for an operation (72 ms).
[12:16:51][W][component:238]: Components should block for at most 30 ms.
```

nikito7, 2024-11-12:
from wmbusmeters:

Tested with key and manual swap:
```
Auto driver    : not found!
Similar driver : unknown 00/00
Using driver   : izar 00/00
000   : 19 length (25 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : 2434 dll-mfct (MAD)
004   : ****** dll-id (******)
008   : 82 dll-version 
009   : 07 dll-type (Water meter) 
010   : a2 tpl-ci-field (Mfct specific)
011 C?: DD0E0013C5F54523EB41CDA1AFA76D mfct specific

{
    "media":"water",
    "meter":"izar",
    "name":"",
    "id":"******",
    "last_month_total_m3":474.768,
    "remaining_battery_life_y":7,
    "total_m3":477.668,
    "transmit_period_s":32768,  (0x8000, maybe 8s)
    "current_alarms":"no_alarm",
    "last_month_measure_date":"2024-11-01",
    "previous_alarms":"no_alarm",
    "timestamp":"2024-11-12T03:51:05Z"
}
```

SzczepanLeon, 2024-11-12:
Sounds like issue in wmbusmeters -- new interpretation should be added.

nikito7, 2024-11-12:
How to convert key used in wmbusmeters format to hex in esphome?

SzczepanLeon, 2024-11-12:
It should be the same key.

nikito7, 2024-11-12:
Default izars key

```
Failed config
...

  Key must consist of 16 hexadecimal numbers.
  key: 39BC8A10E66D83F8
```

SzczepanLeon, 2024-11-12:
Those keys are used in code - you don't need to add it in YAML.
If you have another key you can add 0s at the begining (I never tested this).

nikito7, 2024-11-13:
Aaaaaaa0000000000. (right zeros).

Thank you for the tips :-)

closed 2024-11-13 by nikito7