SzczepanLeon/esphome-components #158: Zapytanie w sprawie kompatybilności z Radio Evo

Damianod27, 2024-12-03:
Cześć. Nie mogę znaleść informacji a jeżeli jest gdzieś podana to przepraszam. Czy będzie działać może z Radio Evo ? Pozdrawiam

SzczepanLeon, 2024-12-03:
Konkret prosze, zdjecie itp.

Damianod27, 2024-12-06:
> Konkret prosze, zdjecie itp.

Cześć. Przepraszam Cię najmocniej ale po aktualizacji aplikacji GitHub zostałem wylogowany i nie dostałem powiadomienia o Twojej odpowiedzi. Dopiero teraz zauważyłem. Przesyłam zdjęcie. Dziękuję [image]

SzczepanLeon, 2024-12-06:
Na 99,9% bedzie działać. Jest to polska brandowana Maddalena Evo868.
[image]

closed 2024-12-06 by SzczepanLeon

Damianod27, 2024-12-06:
@SzczepanLeon OK dzięki. Sprawdzę jak przyjdą elementy do budowy. Te 8 cyfr na zdjęciu to nie kod ?

SzczepanLeon, 2024-12-06:
Normalnie to powinien być numer ID wodomierza. Ale tę nakładkę można zaprogramować dowolnie i to co na naklejce nie musi się zgadzać z tym co na radiu jest wysyłane.