SzczepanLeon/esphome-components #159: Selected driver apator162 doesn't exist

bladecell, 2024-12-04:
Hey I have fully function config running for two months on v4 and wanted to split the config to more files. So I did, but the drivers dont load properly and I am getting error 

```
[20:38:28][W][wmbus:088]: Selected driver apator162 doesn't exist, using 
[20:38:28][I][wmbus:104]: Unknown! [0x06732044] RSSI: -70dBm T: 6E4401064420730605077A8C006085248EDE59D3BF37A3C6C44F37C1E57A81CA677B3612FD6E389DC0DD7B8402E2CA812212B231AE9C750ACF15ADB085D75F865359EE2D883CD8E21973DB370C191F0DE0D7A1CFBA8653E3A82CD265D556F3BF314CCC4449E2B1F1DBF2087506FDE9 (111) T1 A
[20:38:28][W][wmbus:111]: Link modes not defined in driver Unknown!. Processing anyway.
[20:38:28][W][wmbus:119]: Can't find driver for T: 6E4401064420730605077A8C006085248EDE59D3BF37A3C6C44F37C1E57A81CA677B3612FD6E389DC0DD7B8402E2CA812212B231AE9C750ACF15ADB085D75F865359EE2D883CD8E21973DB370C191F0DE0D7A1CFBA8653E3A82CD265D556F3BF314CCC4449E2B1F1DBF2087506FDE9 (111)
```

When I set the `all_drivers: True`, everything works fine with the split config

<details>
  <summary>esph_wmbusmeter.yaml</summary>

      packages:
        device: !include devices/wmbusmeter.yaml
      
      sensor:
        - platform: wmbus
          meter_id: "0x06738709"
          type: apator162
          key: "00000000000000000000000000000000"
          sensors:
            - name: "Cold Water RSSi"
              field: "rssi"
              accuracy_decimals: 0
              unit_of_measurement: "dBm"
              device_class: "signal_strength"
              state_class: "measurement"
              entity_category: "diagnostic"
            - name: "Cold water meter"
              field: "total"
              accuracy_decimals: 3
              unit_of_measurement: "m3"
              device_class: "water"
              state_class: "total_increasing"
              icon: "mdi:water"
        - platform: wmbus
          meter_id: "0x06732044"
          type: apator162
          key: "00000000000000000000000000000000"
          sensors:
            - name: "Hot Water RSSi"
              field: "rssi"
              accuracy_decimals: 0
              unit_of_measurement: "dBm"
              device_class: "signal_strength"
              state_class: "measurement"
              entity_category: "diagnostic"
            - name: "Hot water meter"
              field: "total"
              accuracy_decimals: 3
              unit_of_measurement: "m3"
              device_class: "water"
              state_class: "total_increasing"
              icon: "mdi:water"
        - platform: wmbus
          meter_id: "0x30560356"
          type: qcaloric
          sensors:
            - name: "Study room RSSi"
              field: "rssi"
              accuracy_decimals: 0
              unit_of_measurement: "dBm"
              device_class: "signal_strength"
              state_class: "measurement"
              entity_category: "diagnostic"
            - name: "Study room Heat"
              field: "current_consumption" 
              unit_of_measurement: "hca"
              device_class: "Energy"
              state_class: "total_increasing"
              icon: "mdi:fire-station"
        - platform: wmbus
          meter_id: "0x30560364"
          type: qcaloric
          sensors:
            - name: "Bedroom RSSi"
              field: "rssi"
              accuracy_decimals: 0
              unit_of_measurement: "dBm"
              device_class: "signal_strength"
              state_class: "measurement"
              entity_category: "diagnostic"
            - name: "Bedroom Heat"
              field: "current_consumption" 
              unit_of_measurement: "hca"
              device_class: "Energy"
              state_class: "total_increasing"
              icon: "mdi:fire-station"
        - platform: wmbus
          meter_id: "0x30343806"
          type: qcaloric
          sensors:
            - name: "Kitchen Heat RSSi"
              field: "rssi"
              accuracy_decimals: 0
              unit_of_measurement: "dBm"
              device_class: "signal_strength"
              state_class: "measurement"
              entity_category: "diagnostic"
            - name: "Kitchen Heat"
              field: "current_consumption" 
              unit_of_measurement: "hca"
              device_class: "Energy"
              state_class: "total_increasing"
              icon: "mdi:fire-station"
        - platform: wmbus
          meter_id: "0x30560357"
          type: qcaloric
          sensors:
            - name: "Living room Heat RSSi"
              field: "rssi"
              accuracy_decimals: 0
              unit_of_measurement: "dBm"
              device_class: "signal_strength"
              state_class: "measurement"
              entity_category: "diagnostic"
            - name: "Living room Heat"
              field: "current_consumption" 
              unit_of_measurement: "hca"
              device_class: "Energy"
              state_class: "total_increasing"
              icon: "mdi:fire-station"
 </details>

<details>
  <summary>devices/wmbusmeter.yaml</summary>

    packages:
      <<: !include_dir_named ../components/common
      board: !include ../boards/esp32-c3.yaml
    
    external_components:
      - source: github://SzczepanLeon/esphome-components@main
        refresh: 0s
        components: [wmbus]
    
    wmbus:
      mosi_pin: GPIO4
      miso_pin: GPIO0
      clk_pin:  GPIO1
      cs_pin:   GPIO5
      gdo0_pin: GPIO10
      gdo2_pin: GPIO8
    
    
      frequency: 868.950
      all_drivers: False
      sync_mode: True
      log_all: True

</details>

SzczepanLeon, 2024-12-04:
Please check https://github.com/SzczepanLeon/esphome-components/issues/135#issuecomment-2400807307

You can also add **all_drivers = true**

SzczepanLeon, 2024-12-04:
Please check https://github.com/SzczepanLeon/esphome-components/issues/135#issuecomment-2400807307

You can also add **all_drivers = true** or merge  into one YAML

closed 2024-12-04 by SzczepanLeon