SzczepanLeon/esphome-components #161: On my request to wmbusmeters

jm314159, 2025-01-02:
Hi!

Since I have the Sensus Metering Systems water meter unencoded, I [asked wmbusmeters](https://github.com/wmbusmeters/wmbusmeters/issues/1395) to add support for that based on my readings and logs. I have just been encouraged by the statement in logs: `please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/ to add support for this unknown mfct,media,version combination`. The response, however, did not satisfy the request.

I wonder your opinion on that discussion. Can I expect more from wmbusmeters or is their attitude right? Please advise.

SzczepanLeon, 2025-01-02:
But as I see in logs you can decode fields. I don't understand what you want to achieve.

closed 2025-01-02 by SzczepanLeon

jm314159, 2025-01-02:
Yes, I can. I would like to have a driver just for Sensus as it is currently unavailable for that meter type. Of course, I am happy I can use another driver successfully.