SzczepanLeon/esphome-components #164: reading issues

super-bedst, 2025-01-13:
I have tried ypur component, but something seems to be problematic, i have installed on esp32s i can complle etc, but then it shold be ready for readings nothings happens, here is the logoutput froom  ootup:
[20:30:55][I][wifi:617]: WiFi Connected!
[20:30:55][C][wifi:428]:   Local MAC: 40:91:51:FD:AB:70
[20:30:55][C][wifi:433]:   SSID: [redacted]
[20:30:55][C][wifi:436]:   IP Address: 10.250.250.35
[20:30:55][C][wifi:440]:   BSSID: [redacted]
[20:30:55][C][wifi:441]:   Hostname: 'esp_wmbus'
[20:30:55][C][wifi:443]:   Signal strength: -40 dB ▂▄▆█
[20:30:55][C][wifi:447]:   Channel: 1
[20:30:55][C][wifi:448]:   Subnet: 255.255.255.0
[20:30:55][C][wifi:449]:   Gateway: 10.250.250.1
[20:30:55][C][wifi:450]:   DNS1: 10.250.250.1
[20:30:55][C][wifi:451]:   DNS2: 0.0.0.0
[20:30:55][D][wifi:626]: Disabling AP...
[20:30:55][C][web_server:100]: Setting up web server...
[20:30:55][C][sntp:018]: Setting up SNTP...
[20:30:55][D][rxLoop:031]: Set CC1101 frequency to 868.950MHz [21 6B D0]
[20:30:55][D][rxLoop:043]: CC1101 version '20'
[20:30:55][D][rxLoop:045]: CC1101 initialized
[20:30:55][I][app:062]: setup() finished successfully!
[20:30:55][W][component:170]: Component wifi cleared Warning flag
[20:30:55][I][app:100]: ESPHome version 2024.12.2 compiled on Jan 13 2025, 19:22:19
[20:30:55][C][wifi:600]: WiFi:
[20:30:55][C][wifi:428]:   Local MAC: 40:91:51:FD:AB:70
[20:30:55][C][wifi:433]:   SSID: [redacted]
[20:30:55][C][wifi:436]:   IP Address: 10.250.250.35
[20:30:55][C][wifi:440]:   BSSID: [redacted]
[20:30:55][C][wifi:441]:   Hostname: 'esp_wmbus'
[20:30:55][C][wifi:443]:   Signal strength: -40 dB ▂▄▆█
[20:30:55][C][wifi:447]:   Channel: 1
[20:30:55][C][wifi:448]:   Subnet: 255.255.255.0
[20:30:55][C][wifi:449]:   Gateway: 10.250.250.1
[20:30:55][C][wifi:450]:   DNS1: 10.250.250.1
[20:30:55][C][wifi:451]:   DNS2: 0.0.0.0
[20:30:55][C][logger:185]: Logger:
[20:30:55][C][logger:186]:   Level: DEBUG
[20:30:55][C][logger:188]:   Log Baud Rate: 115200
[20:30:55][C][logger:189]:   Hardware UART: UART0
[20:30:55][C][restart:068]: Restart Switch 'esp_wmbus Restart'
[20:30:55][C][restart:070]:   Icon: 'mdi:restart'
[20:30:55][C][restart:091]:   Restore Mode: always OFF
[20:30:55][C][captive_portal:089]: Captive Portal:
[20:30:55][C][web_server:153]: Web Server:
[20:30:55][C][web_server:154]:   Address: esp_wmbus.local:80
[20:30:55][C][sntp:042]: SNTP Time:
[20:30:55][C][sntp:045]:   Server 0: '0.pool.ntp.org'
[20:30:55][C][sntp:045]:   Server 1: '1.pool.ntp.org'
[20:30:55][C][sntp:045]:   Server 2: '2.pool.ntp.org'
[20:30:55][C][mdns:116]: mDNS:
[20:30:55][C][mdns:117]:   Hostname: esp_wmbus
[20:30:55][C][esphome.ota:073]: Over-The-Air updates:
[20:30:55][C][esphome.ota:074]:   Address: esp_wmbus.local:3232
[20:30:55][C][esphome.ota:075]:   Version: 2
[20:30:55][C][esphome.ota:078]:   Password configured
[20:30:55][C][safe_mode:018]: Safe Mode:
[20:30:55][C][safe_mode:020]:   Boot considered successful after 60 seconds
[20:30:55][C][safe_mode:021]:   Invoke after 10 boot attempts
[20:30:55][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[20:30:55][C][wmbus:338]: wM-Bus v4.1.1-1.17.1-b8f4a945:
[20:30:55][C][wmbus:351]:   LED:
[20:30:55][C][wmbus:352]:     Pin: GPIO0
[20:30:55][C][wmbus:353]:     Duration: 1000 ms
[20:30:55][C][wmbus:356]:   Chip ID: 70ABFD519140
[20:30:55][C][wmbus:358]:   CC1101 frequency: 868.950 MHz
[20:30:55][C][wmbus:359]:   CC1101 SPI bus:
[20:30:55][C][wmbus:363]:     MOSI Pin: GPIO13
[20:30:55][C][wmbus:364]:     MISO Pin: GPIO12
[20:30:55][C][wmbus:365]:     CLK Pin:  GPIO14
[20:30:55][C][wmbus:366]:     CS Pin:   GPIO2
[20:30:55][C][wmbus:367]:     GDO0 Pin: GPIO5
[20:30:55][C][wmbus:368]:     GDO2 Pin: GPIO4
[20:30:55][C][wmbus:374]:   Available drivers: wme5, weh_07, watertech, waterstarm, vario451mid, vario451, vario411, unknown, unismart, ultrimis, ultraheat, tsd2, topaseskr, supercom587, sontex868, sharky774, sharky, sensostar, rfmtx1, rfmamb, qwater, qualcosonic, qsmoke, qheat_55_us, qheat, qcaloric, q400, pollucomf, piigth, omnipower, nemo, munia, multical21, mkradio4a, mkradio4, mkradio3, minomess, microclima, lse_08, lse_07_17, lansenth, lansensm, lansenrp, lansenpu, lansendw, kampress, kamheat, izar, iwm
[20:31:00][D][sntp:067]: Synchronized time: 2025-01-13 20:31:01
[20:31:48][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[20:31:48][D][esp32.preferences:114]: Saving 1 preferences to flash...
[20:31:48][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[20:32:27][D][mbus:035]: Received T1 A frame

configuration:
esphome:
  name: esp_wmbus

esp32:
  board: nodemcu-32s
  framework:
    type: arduino
  
# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
#api:
 # encryption:
  #  key: "xxxxxxxxxxxxxxxxxxxxxx"

ota:
  - platform: esphome
    password: "xxxxxxxxxxxxxxxxxxxxxxxxxxx"

external_components:

  - source: github://SzczepanLeon/esphome-components@main
    refresh: 0d
    components: [ wmbus ]

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp-Wmbus Fallback Hotspot"
    password: "redacted"

captive_portal:
web_server:
  port: 80
time:
  platform: sntp

wmbus:

  led_pin: GPIO0
  led_blink_time: "1s"

  frequency: 868.950
  all_drivers: True
  log_all: True

switch:
  - platform: restart
    name: "esp_wmbus Restart"

i am in a area with a lot of kamstrup meters, so am am wondring why i only recieve one 

[mbus:035] | Received T1 A frame
 but without content, i have tried with anoter cc1101 module also with same result..
 
 the wiring is by default and have been checked

SzczepanLeon, 2025-01-13:
Try VERBOSE logs.

super-bedst, 2025-01-13:
here it is:
[21:45:18][V][rxLoop:167]: Have 369 bytes from CC1101 Rx, RSSI: -93 dBm LQI: 201
[21:45:18][V][rxLoop:172]: Synchronus mode enabled.
[21:45:18][D][mbus:035]: Received T1 A frame
[21:45:18][V][mbus:041]: Frame: C65CA92F494399C6BCB43E03758F1BA423FADC0E0CC5AA2D5AF25311EFF46B227161AA81807318097565BD5C585C0016226CE5940488903942B40680A06B03E97806E6FE16FF74CA398251F9281A355152129327FAB0E350706FB046723189F4336E8AC7800884463E398010844A91E165120821C7BB46821842273BDB90966197AD4F02E064A7FD8C137ED344C52560CBC2ACF90CA0F4DCC9BB903C4B70085514CB38D096EC69F51C427823341A928A0762B4077D431070063B000244A64775010932583B1C1E72 [RAW]
[21:45:18][V][mbus:043]:        BC01D00850E1F4D2602390008D611C2392492940CC981F289FD41966A12940A91A5A01FD17D381506CE8994212A2A65B5A1A683E7FDACD354A10BA42D659049932D73782B8A88F1620D8A2E10A6803924007C20B8AA474F159817108134A8343D1305801400C04D5084909F2934870904354D6E2A6742A2D2C2B0A01D80787181106C1096670A14C34C574AE28968C83A5E606198E3496A435844B4DE30BC5D23DD0B4C0CE05E42129 (369) [RAW]
[21:45:18][V][3of6:083]: Decode 3 out of 6 failed..
[21:45:18][W][component:237]: Component wmbus took a long time for an operation (119 ms).
[21:45:18][W][component:238]: Components should block for at most 30 ms.
[21:47:03][D][sntp:067]: Synchronized time: 2025-01-13 21:47:04
[21:55:26][V][rxLoop:167]: Have 266 bytes from CC1101 Rx, RSSI: -98 dBm LQI: 255
[21:55:26][V][rxLoop:172]: Synchronus mode enabled.
[21:55:26][D][mbus:035]: Received T1 A frame
[21:55:26][V][mbus:041]: Frame: 9669998A270B44E7BA16D8D8EDCE3278992AAB04559875C0530062BA560274E32A8015B89ECE51AC0287C72878378644A702D743F8A6B68328C71E5ED39CDC2B66978A224A195C8049A000825C0AA30EB9E05BE4057D759C1BAE0726985D8F5F812678E459076AC0207069800DE830554F86A7121EDFDB12BF7F8DA0DDD90046BAF5021A80D4D43E6BA891B9914540C04841E6E2628F6C72053220E05C446C3315F638244EFA170CC421A5807DEA50030D3300180764888086AB5F0574057BA0350419CD13759159 [RAW]
[21:55:26][V][mbus:043]:        14A40A01072AF0783F5000843988283B8BD7B0AC6B75BCFC8542355970FC1B021A339C358A02824120D0C8A290CA616817A42C15CF6C3D9946BAA67775004A3FDBA3 (266) [RAW]
[21:55:26][V][3of6:083]: Decode 3 out of 6 failed..
[21:55:26][W][component:237]: Component wmbus took a long time for an operation (92 ms).
[21:55:26][W][component:238]: Components should block for at most 30 ms.

SzczepanLeon, 2025-01-13:
CC1101 received something but there was error in decoding.

How far CC1101 is from meter?

super-bedst, 2025-01-13:
2-3 meters

SzczepanLeon, 2025-01-13:
RSSI is quite low.

super-bedst, 2025-01-13:
ok, will try tomorrow with another placement of the unit

nikito7, 2025-01-22:
```led_blink_time: "1s"```

Change to 50ms

closed 2025-02-19 by SzczepanLeon