SzczepanLeon/esphome-components #166: Not possible to read telegrams after restart.

marianbezeg, 2025-01-31:
Hello,

I tried to configure reading on multiple ESP boards with multiple CC1101 chips, but always after some time it get broken and somehow i cannot fetch data. For example i just powered off /on ESP and its not working anymore.... any hints ? I even try to compile code again but without any luck.

INFO ESPHome 2024.12.4
INFO Reading configuration /config/esphome/izar-watermeter.yaml...
INFO Detected timezone 'Europe/Budapest'
INFO Generating C++ source...
INFO Compiling app...
Processing izar-watermeter (board: esp01_1m; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.6
|-- SPI @ 1.0
|-- SmartRC-CC1101-Driver-Lib @ 2.5.7+sha.b8c6af4
|-- wmbus-drivers @ 0.0.0+20250129101029.sha.c856f7d
Compiling .pioenvs/izar-watermeter/src/main.cpp.o
Linking .pioenvs/izar-watermeter/firmware.elf
RAM:   [=====     ]  55.0% (used 45048 bytes from 81920 bytes)
Flash: [======    ]  57.7% (used 590465 bytes from 1023984 bytes)
Building .pioenvs/izar-watermeter/firmware.bin
esp8266_copy_factory_bin([".pioenvs/izar-watermeter/firmware.bin"], [".pioenvs/izar-watermeter/firmware.elf"])
esp8266_copy_ota_bin([".pioenvs/izar-watermeter/firmware.bin"], [".pioenvs/izar-watermeter/firmware.elf"])
========================= [SUCCESS] Took 21.32 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.1.221 port 8266...
INFO Connected to 192.168.1.221
INFO Uploading /data/build/izar-watermeter/.pioenvs/izar-watermeter/firmware.bin (594624 bytes)
INFO Compressed to 406373 bytes
Uploading: [============================================================] 100% Done...

INFO Upload took 3.42 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.1.221 using esphome API
INFO Successfully connected to izar-watermeter @ 192.168.1.221 in 19.379s
INFO Successful handshake with izar-watermeter @ 192.168.1.221 in 2.416s
[19:09:11][I][app:100]: ESPHome version 2024.12.4 compiled on Jan 31 2025, 19:08:27
[19:09:11][C][wifi:600]: WiFi:
[19:09:11][C][wifi:428]:   Local MAC: E8:DB:84:F3:AF:32
[19:09:11][C][wifi:433]:   SSID: [redacted]
[19:09:11][C][wifi:436]:   IP Address: 192.168.1.221
[19:09:11][C][wifi:439]:   BSSID: [redacted]
[19:09:11][C][wifi:441]:   Hostname: 'izar-watermeter'
[19:09:11][C][wifi:443]:   Signal strength: -44 dB ▂▄▆█
[19:09:11][V][wifi:445]:   Priority: 0.0
[19:09:11][C][wifi:447]:   Channel: 5
[19:09:11][C][wifi:448]:   Subnet: 255.255.255.0
[19:09:11][C][wifi:449]:   Gateway: 192.168.1.1
[19:09:11][C][wifi:450]:   DNS1: 0.0.0.0
[19:09:11][C][wifi:451]:   DNS2: 0.0.0.0
[19:09:11][C][logger:185]: Logger:
[19:09:11][C][logger:186]:   Level: VERY_VERBOSE
[19:09:11][C][logger:188]:   Log Baud Rate: 115200
[19:09:11][C][logger:189]:   Hardware UART: UART0
[19:09:11][C][captive_portal:089]: Captive Portal:
[19:09:11][C][sntp:042]: SNTP Time:
[19:09:11][C][sntp:045]:   Server 0: '0.pool.ntp.org'
[19:09:11][C][sntp:045]:   Server 1: '1.pool.ntp.org'
[19:09:11][C][sntp:045]:   Server 2: '2.pool.ntp.org'
[19:09:11][C][mdns:116]: mDNS:
[19:09:11][C][mdns:117]:   Hostname: izar-watermeter
[19:09:11][V][mdns:118]:   Services:
[19:09:11][V][mdns:120]:   - _esphomelib, _tcp, 6053
[19:09:11][V][mdns:122]:     TXT: friendly_name = izar watermeter
[19:09:11][V][mdns:122]:     TXT: version = 2024.12.4
[19:09:11][V][mdns:122]:     TXT: mac = e8db84f3af32
[19:09:11][V][mdns:122]:     TXT: platform = ESP8266
[19:09:11][V][mdns:122]:     TXT: board = esp01_1m
[19:09:11][V][mdns:122]:     TXT: network = wifi
[19:09:11][V][mdns:122]:     TXT: api_encryption = Noise_NNpsk0_25519_ChaChaPoly_SHA256
[19:09:11][C][esphome.ota:073]: Over-The-Air updates:
[19:09:11][C][esphome.ota:074]:   Address: 192.168.1.221:8266
[19:09:11][C][esphome.ota:075]:   Version: 2
[19:09:11][C][esphome.ota:078]:   Password configured
[19:09:11][C][safe_mode:018]: Safe Mode:
[19:09:11][C][safe_mode:019]:   Boot considered successful after 60 seconds
[19:09:11][C][safe_mode:021]:   Invoke after 10 boot attempts
[19:09:11][C][safe_mode:022]:   Remain in safe mode for 300 seconds
[19:09:11][C][api:140]: API Server:
[19:09:11][C][api:141]:   Address: 192.168.1.221:6053
[19:09:11][C][api:143]:   Using noise encryption: YES
[19:09:11][C][wmbus:372]: wM-Bus v3.2.3:
[19:09:11][C][wmbus:389]:   CC1101 frequency: 434.475 MHz
[19:09:11][C][wmbus:390]:   CC1101 SPI bus:
[19:09:11][C][wmbus:391]:     MOSI Pin: GPIO13
[19:09:11][C][wmbus:392]:     MISO Pin: GPIO12
[19:09:11][C][wmbus:393]:     CLK Pin:  GPIO14
[19:09:11][C][wmbus:394]:     CS Pin:   GPIO15
[19:09:11][C][wmbus:395]:     GDO0 Pin: GPIO4
[19:09:11][C][wmbus:396]:     GDO2 Pin: GPIO5
[19:09:11][C][wmbus:403]:   Available drivers:  amiplus, apator08, apator162, apatoreitn, bmeters, c5isf, compact5, dme07, elf, evo868, fhkvdataiii, flowiq2200, hydrocalm3, hydrus, iperl, itron, izar, kamheat, mkradio3, mkradio4, mkradio4a, multical21, qheat, qwater, rfmtx1, sharky774, topaseskr, ultrimis, unismart, vario451
[19:09:11][C][wmbus:421]:   Meter:
[19:09:11][C][wmbus:422]:     ID: 389482360 [0x17370778]
[19:09:11][C][wmbus:423]:     Type: izar
[19:09:11][C][wmbus:424]:     Mode: T1
[19:09:11][C][wmbus:425]:     Key: ''
[19:09:11][C][wmbus:427]:     Sensor '0x17370778 Watermeter Display'
[19:09:11][C][wmbus:427]:       Device Class: 'water'
[19:09:11][C][wmbus:427]:       State Class: 'total_increasing'
[19:09:11][C][wmbus:427]:       Unit of Measurement: 'm³'
[19:09:11][C][wmbus:427]:       Accuracy Decimals: 3
[19:09:11][C][wmbus:427]:       Icon: 'mdi:water'
[19:09:45][VV][scheduler:225]: Running interval '' with interval=60000 last_execution=4294964186 (now=56890)
[19:09:47][VV][api.service:697]: on_ping_request: PingRequest {}
[19:09:47][VV][api.service:043]: send_ping_response: PingResponse {}
[19:09:54][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[19:10:11][VV][api.connection:133]: Sending keepalive PING...
[19:10:11][VV][api.service:037]: send_ping_request: PingRequest {}
[19:10:11][VV][api.service:706]: on_ping_response: PingResponse {}
[19:10:27][VV][api.service:697]: on_ping_request: PingRequest {}
[19:10:27][VV][api.service:043]: send_ping_response: PingResponse {}
[19:10:45][VV][scheduler:225]: Running interval '' with interval=60000 last_execution=56890 (now=116890)
[19:11:07][VV][api.service:697]: on_ping_request: PingRequest {}
[19:11:07][VV][api.service:043]: send_ping_response: PingResponse {}
[19:11:11][VV][api.connection:133]: Sending keepalive PING...
[19:11:11][VV][api.service:037]: send_ping_request: PingRequest {}
[19:11:11][VV][api.service:706]: on_ping_response: PingResponse {}
[19:11:45][VV][scheduler:225]: Running interval '' with interval=60000 last_execution=116890 (now=176890)
[19:11:47][VV][api.service:697]: on_ping_request: PingRequest {}
[19:11:47][VV][api.service:043]: send_ping_response: PingResponse {}
[19:12:11][VV][api.connection:133]: Sending keepalive PING...
[19:12:11][VV][api.service:037]: send_ping_request: PingRequest {}
[19:12:11][VV][api.service:706]: on_ping_response: PingResponse {}

SzczepanLeon, 2025-01-31:
Please try to change GDIO0 and GDOO2.

marianbezeg, 2025-01-31:
I can try to change but it worked before, and i am using original shield board which you have on your github, i was able to get data for whole week but once i restarted it it want to connect again. I didnt change anything in terms of config or place where it is..

I can't go to version4 as its 8266, i faced same isue before with esp32 and CC1101 shield...

SzczepanLeon, 2025-01-31:
I don't have any idea why it stopped.

marianbezeg, 2025-01-31:
@SzczepanLeon is there any way to get more logs or verify if wmbus is at least working ?

SzczepanLeon, 2025-01-31:
VV or V should be enough.
You can try another CC1101 or TV stick or something based on SX chip or SDR radio to see if meter is transmitting.

marianbezeg, 2025-01-31:
@SzczepanLeon it is i had SDR but was overkill to me so moved ot ESP ..but somehow its not stable ...i just pick new cc1101  and d1 mini so going to solder it together..

SzczepanLeon, 2025-01-31:
I would recomend to switch to ESP32 or ESP32 +SX radio.

closed 2025-02-19 by SzczepanLeon