SzczepanLeon/esphome-components #167: SmartMeter NES 83334-xIABMA installed by LinzNetz AG

HannesOOE, 2025-02-02:
Hello,
I live in Linz / Austria and had a SmartMeter NES 83334-xIABMA installed by LinzNetz AG.
This SmartMeter can be read via WModbus.
Unfortunately, this model is not known in your software, so reading does not work.

The protocol description is available here:
https://www.linznetz.at/media/linz_netz_website/netz_documents/Bescription-Wireless_M-Bus-Interface.pdf

Do you have an idea for me on how I could connect this SmartMeter?
Thank you in advance for your help and information!

Best regards
Hannes

SzczepanLeon, 2025-02-02:
> The protocol description is available here:

Fehler 404 - Seite nicht gefunden

HannesOOE, 2025-02-02:
[Beschreibung-Wireless_M-Bus-Schnittstelle.pdf](https://github.com/user-attachments/files/18632071/Beschreibung-Wireless_M-Bus-Schnittstelle.pdf)

SzczepanLeon, 2025-02-02:
Great documentation. On last page I see DIF VIF so I would sugest to capture some telegrams and try to decode it on wmbusmeters.org Then You will have information what driver is the best for that meter.

HannesOOE, 2025-02-02:
Thanks for your quick reply!
I have now tried this and “amiplus” is displayed and evaluated correctly.
Unfortunately, this doesn't seem to work in your implementation of version 4.
I tried it in version 3, here I can at least read out the 2 energy and the 2 performance values.

Best regards,
Hannes

[image]

SzczepanLeon, 2025-02-02:
Please include logs and YAML from version 4.

HannesOOE, 2025-02-02:
Hello,
I cleaned up my ESPHome environment (Docker) and updated it to the latest version.
Now I can also read the data with version 4.
Please excuse me for bothering you and thank you for your help!
Best regards,
Hannes

closed 2025-02-02 by HannesOOE