SzczepanLeon/esphome-components #171: update_interval

pdrobek, 2025-02-10:
How can i read sensor parameters more frequently?

statement update_interval: 120s for example 

[update_interval] is an invalid option for [sensor.wmbus]. Please check the indentation.

SzczepanLeon, 2025-02-10:
What sensor?
Only T1/C1 meter are supported. It means that transmission interval is configured by meter owner and You can't change it.

pdrobek, 2025-02-10:
sensor:
  - platform: wmbus
    meter_id: 0xedited
    type: apator162
    key: "00000000000000000000000000000000"
    lqi:
      name: "My lqi"
    rssi:
      name: "My RSSI"
    total_water_m3:
      name: "Current m^3"
      filters:
        - offset: 935.925

stuck on offset value and not moving even when i use some water

SzczepanLeon, 2025-02-10:
Please update to v4.