SzczepanLeon/esphome-components #179: Problem with new version esphome and ESP8266

Wilu88, 2025-02-19:
After update esphome today i have problem with compile. I see new version 4.1.3 but i need version on esp8266.

SzczepanLeon, 2025-02-19:
Please use:
@version_2
or
@version_3

closed 2025-02-19 by SzczepanLeon

Wilu88, 2025-02-20:
Yeah i use version 3 but i have problem

ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/esphome/esphome/__main__.py", line 1049, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1027, in run_esphome
    config = read_config(dict(args.substitution) if args.substitution else {})
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 1095, in read_config
    res = load_config(command_line_substitutions)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 949, in load_config
    return _load_config(command_line_substitutions)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 939, in _load_config
    return validate_config(config, command_line_substitutions)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 835, in validate_config
    target_platform = core_config.preload_core_config(config, result)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/core/config.py", line 222, in preload_core_config
    if _is_target_platform(domain):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/core/config.py", line 184, in _is_target_platform
    if get_component(name, True).is_target_platform:
       ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/loader.py", line 205, in get_component
    return _lookup_module(domain, exception)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/loader.py", line 181, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/data/external_components/0e75e1a4/components/wmbus/__init__.py", line 62, in <module>
    cv.Required(CONF_IP_ADDRESS): cv.ipv4,
                                  ^^^^^^^
AttributeError: module 'esphome.config_validation' has no attribute 'ipv4'

SzczepanLeon, 2025-02-20:
> Yeah i use version 3 but i have problem

On what ESPHome and what component version?

SzczepanLeon, 2025-02-20:
for that ESPHome, please use:
github://SzczepanLeon/esphome-components@version_3

Wilu88, 2025-02-20:
Wow super Works :)

Thanks ;) Dziękuje bardzo kolego ;) Leci kawka :)