SzczepanLeon/esphome-components #181: Selected driver hydrodigit doesn't exist

reaper7, 2025-02-20:
When I set **all_drivers** to **false** then selected driver "hydrodigit" is not recognized and the _driver_hydrodigit.cpp_ file is not compiled.
When the frame from my water meter is received I get the message:
```
[17:54:23][I][wmbus:104]: Unknown! [0x12345678] RSSI: -55dBm T: 4E44B40......
```

but when **all_drivers** is set to **true** then everything works fine!
```
[17:46:11][I][wmbus:092]: Using selected driver hydrodigit (detected driver was hydrodigit)
[17:46:11][I][wmbus:104]: hydrodigit [0x12345678] RSSI: -32dBm T: 4E44B40......
```
Unfortunately, I can't compile with all drivers because it takes up all the memory and I don't want to give up other features.

Part of my config:
```yaml
wmbus:
  mosi_pin: GPIO23
  miso_pin: GPIO19
  clk_pin:  GPIO18
  cs_pin:   GPIO5
  gdo0_pin: GPIO33
  gdo2_pin: GPIO32

#  all_drivers: true
  log_all: true
  all_drivers: false
#  log_all: false
  frequency: 868.925
  sync_mode: true

  led_pin:
    number: GPIO2
    inverted: false
    allow_other_uses: true
  led_blink_time: 500ms

  - platform: wmbus
    meter_id: 0x12345678
    type: hydrodigit
    key: "00000000000000000000000000000000"
    sensors:
      - name: "RSSi WWB"
        field: "rssi"
        accuracy_decimals: 0
        unit_of_measurement: "dBm"
        device_class: "signal_strength"
        state_class: "measurement"
        entity_category: "diagnostic"
      - name: "waterbathroomwarm"
        id: "wwb"
        field: "total"
        accuracy_decimals: 3
        unit_of_measurement: "m³"
        device_class: "water"
        state_class: "total_increasing"
        icon: "mdi:water"
```

SzczepanLeon, 2025-02-20:
Please check https://github.com/SzczepanLeon/esphome-components/issues/135#issuecomment-2400807307 and/or share full YAML.

reaper7, 2025-02-20:
Yes, it was a section order problem.
Currently, my working configuration is as follows:
```
substitutions:
esphome:
esp32:
external_components:
wifi:
captive_portal:
logger:
ota:
web_server:
time:
mqtt:
wmbus:

binary_sensor:
button:
light:
sensor:
```

Previously, the **wmbus** section was right after external_components,
but it didn't raise my concern because in 3.x versions it didn't cause any problem

tnx!

BTW
How about writing about the ORDER in the README at the beginning of section 2.1.2 ?
I think a lot of people with low logging levels only see a 3 out of 6 decoding error and don't understand what is the problem.

closed 2025-02-20 by reaper7