SzczepanLeon/esphome-components #182: Timestamp as sensor

IntExCZ, 2025-02-20:
There would be good to get timestamp of received telegram to detect if/when the telegram arrived. Automations in HA/ESPhome reacts only on value changes so there acutally is no posibility to detect if the telegrams are comming (only when meter value changes).

SzczepanLeon, 2025-02-20:
You can use everything from ESPHome [sensor](https://esphome.io/components/sensor/index.html).

IntExCZ, 2025-02-20:
Not everytime when telelegram arrives, because "on_value" gets executed only when value changes.

SzczepanLeon, 2025-02-20:
> Not everytime when telelegram arrives, because "on_value" gets executed only when value changes.

This is not related to component - this is ESPHome/HA behaviour

IntExCZ, 2025-02-20:
But why involve the MQTT when the ESPhome integration works well? Why is timestamp available only in MQTT?

SzczepanLeon, 2025-02-20:
Try text_sensor

IntExCZ, 2025-02-20:
Tried, but "[wmbus:168]: Can't get requested field 'timestamp'". :-/

IntExCZ, 2025-02-20:
Just to understand, for functioning ESPhome's "on_value" automation is required changed value from a previous one and the timestamp is the best option for that - I think. When the meter is steady for a few hours, the telegrams arrives but there is no option to know/detect that from the ESPhome/HA (instead of reading ESPhome live logs).

SzczepanLeon, 2025-02-20:
Please update component - version 4.1.4.
**timestamp** text sensor should be present.

IntExCZ, 2025-02-20:
Should and is, thank you! 👍 
[image]

closed 2025-02-20 by SzczepanLeon