SzczepanLeon/esphome-components #184: Drivers are not added to the build when all_drivers: False

Arman-GeDon, 2025-02-22:
I have just changed the version from the old 3.2.3 to the newest (4.1.3?) and adjusted the code.
Esphome 2025.2.0.

I have two sensors of the amiplus and unismart type, which are properly declared as sensor types.
If I have declared in the code:
```
wmbus:
  all_drivers: False
```

Then no drivers are added to the compilation list (I can see it in the logs during compilation), only driver_unknown. The code compiles correctly, but of course the device reports an error after receiving the telegram:

```
[13:12:27][D][mbus:035]: Received T1 A frame
[13:12:27][W][wmbus:088]: Selected driver amiplus doesn't exist, using 
[13:12:27][I][wmbus:104]: Unknown! [0x...1] RSSI: -48dBm T: ...telegram... (159) T1 A
[13:12:27][W][wmbus:111]: Link modes not defined in driver Unknown!. Processing anyway.
[13:12:27][W][wmbus:119]: Can't find driver for T: ...telegram... (159) T1 A

[13:13:32][D][mbus:035]: Received T1 A frame
[13:13:32][W][wmbus:088]: Selected driver unismart doesn't exist, using 
[13:13:32][I][wmbus:104]: Unknown! [0x...2] RSSI: -66dBm T: ...telegram... (111) T1 A
[13:13:32][W][wmbus:111]: Link modes not defined in driver Unknown!. Processing anyway.
[13:13:32][W][wmbus:119]: Can't find driver for T: ...telegram... (111) T1 A
```

Of course everything works when I declare:
```
wmbus:
  all_drivers: True
```

Meter IDs and telegrams edited above.

SzczepanLeon, 2025-02-22:
The latest version is 4.1.4
Please check this issue https://github.com/SzczepanLeon/esphome-components/issues/135 and adjust component.order in YAML.

Arman-GeDon, 2025-02-22:
Thanks @SzczepanLeon !
It is not possible to set the correct order of components with packages. At least in my case.
I either have to use flat yaml (without packages) or declare all_drivers: True.
Does "all_drivers: True" mean only a larger file to flash? Does it have any disadvantages?
Alternatively, the solution could be to optionally declare the necessary drivers by the user in the wmbus component.

SzczepanLeon, 2025-02-22:
Please share YAML.

SzczepanLeon, 2025-02-22:
> Does \"all_drivers: True\" mean only a larger file to flash? Does it have any disadvantages?

No, only large flash usage.

zamojski, 2025-03-13:
I've just upgraded from v3 to v4.1.4 and I'm facing the same issue. What's the correct order of components?