SzczepanLeon/esphome-components #187: ESPHome Device Builder 2025.2.0

riro-at, 2025-02-24:
Hier ist die englische Übersetzung deines Textes:

Hello,

After updating the ESPHome Device Builder to 2025.2.0, ha-watermeter from zibous no longer works.
I changed external_components to:

- source: github://SzczepanLeon/esphome-components@main

I also updated the sensors to the new syntax.
For example, from:

lqi:
    id: wmbus_cc1101_lqi
    name: "${friendly_name} CC1101 LQI"
    entity_category: "diagnostic"
    unit_of_measurement: "lqi"
    state_class: "measurement"

to:

- name: "${friendly_name} CC1101 LQI"
   id: wmbus_cc1101_lqi
   entity_category: "diagnostic"
   unit_of_measurement: "lqi"
   state_class: "measurement"

However, the sensor is unknown in Home Assistant and does not receive any values.

I would appreciate any help.
Thanks,
riro

SzczepanLeon, 2025-02-24:
LQI was removed in version 4.

riro-at, 2025-02-24:
Thank you for the quick response.
None of the sensors are working anymore, for example: wmbus_cc1101_rssi, waterdisplay, watermonth, waterdisplay_lastmonth, etc.

closed 2025-02-24 by SzczepanLeon

riro-at, 2025-02-24:
If I use version_3 instead of main, I get the following error message:

INFO ESPHome 2025.2.0
INFO Reading configuration /config/esphome/watermeter.yaml...
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/esphome/esphome/__main__.py", line 1049, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1027, in run_esphome
    config = read_config(dict(args.substitution) if args.substitution else {})
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 1095, in read_config
    res = load_config(command_line_substitutions)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 949, in load_config
    return _load_config(command_line_substitutions)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 939, in _load_config
    return validate_config(config, command_line_substitutions)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 835, in validate_config
    target_platform = core_config.preload_core_config(config, result)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/core/config.py", line 222, in preload_core_config
    if _is_target_platform(domain):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/core/config.py", line 184, in _is_target_platform
    if get_component(name, True).is_target_platform:
       ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/loader.py", line 205, in get_component
    return _lookup_module(domain, exception)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/loader.py", line 181, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/data/external_components/0e75e1a4/components/wmbus/__init__.py", line 62, in <module>
    cv.Required(CONF_IP_ADDRESS): cv.ipv4,
                                  ^^^^^^^
AttributeError: module 'esphome.config_validation' has no attribute 'ipv4'

If I use main, none of the sensors work anymore.

SzczepanLeon, 2025-02-24:
Do you have
**refresh: 0d**
in YAML ?

riro-at, 2025-02-24:
no, i don`t have