SzczepanLeon/esphome-components #191: ESPhome not able to decode the Wmbus Telegram with provided Key.

MUTEtheCyberwolf, 2025-03-04:
### This is the result i get from the Logs of ESPHome

I did provide a valid Key for decoding through the key attribute. 

Copying the Frame into wmbusmeters.org with my key results in the Telegram being shown without any problems.

If i am just stupid then i do have to admit it. Still have to say @SzczepanLeon awesome Repo :]

____________________________________________________________________________________________
[19:24:05][V][rxLoop:167]: Have 171 bytes from CC1101 Rx, RSSI: -60 dBm LQI: 128
[19:24:05][D][mbus:035]: Received T1 A frame
[19:24:05][V][mbus:046]: Frame: 68B71C99934DB0EB1635C69A4D65936934DCB345963999565A95963B43993B4A71B1C5969532E6999D1A58B9B1696B0E8D98CB4E63B45967165933564DC71CC8D5A3356A4E9B1396B26C5364BD19B1334B73173264D3AC8EC716C9966C94E94BC6C2F16A98F1A4B65AA5A59A8F26A63AC34BD1364D6B1B168E6C66C63972A59D1A69998D9597346A55B16694CE9698E62EC6A99B4D2C8CB4F4953B1339A8CB959CB22CDCB16A33A539C68B (171) [RAW]
[19:24:05][V][mbus:053]: Frame: 6344A51182801466700767748C0025900F002C252CFD8400973AA5C603AD6082B5B37A2C004007107444E10B10F2AD208AD753C587134D4E5128B840E5589293D83D6FBDF356F606BE6A2813C7516D80BADADB9EF5C665A1954C690D5F729FBA386FACC8B37C978726B395EE31ED6B292463 (114) [with CRC]
[19:24:05][V][mbus:096]: Validating CRC for Block1
[19:24:05][V][crc:031]:     calculated: 0x6774, read: 0x6774
[19:24:05][V][mbus:116]: Validating CRC for Block2
[19:24:05][V][crc:031]:     calculated: 0x03AD, read: 0x03AD
[19:24:05][V][mbus:116]: Validating CRC for Block3
[19:24:05][V][crc:031]:     calculated: 0xAD20, read: 0xAD20
[19:24:05][V][mbus:116]: Validating CRC for Block4
[19:24:05][V][crc:031]:     calculated: 0xD83D, read: 0xD83D
[19:24:05][V][mbus:116]: Validating CRC for Block5
[19:24:05][V][crc:031]:     calculated: 0xDB9E, read: 0xDB9E
[19:24:05][V][mbus:116]: Validating CRC for Block6
[19:24:05][V][crc:031]:     calculated: 0xB37C, read: 0xB37C
[19:24:05][V][mbus:116]: Validating CRC for Block7
[19:24:05][V][crc:031]:     calculated: 0x2463, read: 0x2463
[19:24:05][V][mbus:063]: Frame: 6344A5118280146670078C0025900F002C252CFD8400973AA5C66082B5B37A2C004007107444E10B10F28AD753C587134D4E5128B840E55892936FBDF356F606BE6A2813C7516D80BADAF5C665A1954C690D5F729FBA386FACC8978726B395EE31ED6B29 (100) [without CRC]
[19:24:05][V][Telegram.cpp:1131]: (wmbus) parseDLL @0 100
[19:24:05][V][Telegram.cpp:1178]: (wmbus) parseELL @10 90
[19:24:05][V][Telegram.cpp:1333]: (wmbus) parseNWL @13 87
[19:24:05][V][Telegram.cpp:1391]: (wmbus) parseAFL @13 87
[19:24:05][V][Telegram.cpp:2078]: (wmbus) parseTPL @30 70
[19:24:05][V][Telegram.cpp:1546]: (wmbus) input to kdf for enc
[19:24:05][V][Telegram.cpp:1555]: (wmbus) no key, thus cannot execute kdf.

SzczepanLeon, 2025-03-04:
Please include VERY_VERBOSE logs and YAML.

MUTEtheCyberwolf, 2025-03-04:
### Here is the Log:

[19:46:36][I][app:100]: ESPHome version 2025.2.2 compiled on Mar  4 2025, 19:45:49
[19:46:36][I][app:102]: Project Diehl Hydrus RC 868.Watermeter version 2.0.7
[19:46:36][C][logger:177]: Logger:
[19:46:36][C][logger:178]:   Max Level: VERY_VERBOSE
[19:46:36][C][logger:179]:   Initial Level: VERY_VERBOSE
[19:46:36][C][logger:181]:   Log Baud Rate: 0
[19:46:36][C][logger:182]:   Hardware UART: UART0
[19:46:36][C][logger:186]:   Level for 'wMBus-lib': VERY_VERBOSE
[19:46:36][C][logger:186]:   Level for 'wmbus': VERY_VERBOSE
[19:46:36][C][template.sensor:022]: Template Sensor 'Device Boot counter'
[19:46:36][C][template.sensor:022]:   State Class: 'measurement'
[19:46:36][C][template.sensor:022]:   Unit of Measurement: ''
[19:46:36][C][template.sensor:022]:   Accuracy Decimals: 0
[19:46:36][C][template.sensor:023]:   Update Interval: 60.0s
[19:46:36][C][template.sensor:022]: Template Sensor 'Device Last Boot'
[19:46:36][C][template.sensor:022]:   State Class: ''
[19:46:36][C][template.sensor:022]:   Unit of Measurement: ''
[19:46:36][C][template.sensor:022]:   Accuracy Decimals: 1
[19:46:36][C][template.sensor:022]:   Icon: 'mdi:clock-start'
[19:46:36][C][template.sensor:023]:   Update Interval: 60.0s
[19:46:36][C][template.switch:068]: Template Switch 'Device Boot Counter reset'
[19:46:36][C][template.switch:091]:   Restore Mode: always OFF
[19:46:36][C][template.switch:057]:   Optimistic: NO
[19:46:36][C][restart:068]: Restart Switch 'Device Restart'
[19:46:36][C][restart:070]:   Icon: 'mdi:restart'
[19:46:36][C][restart:091]:   Restore Mode: always OFF
[19:46:36][C][ethernet:326]: Ethernet:
[19:46:36][C][ethernet:515]:   IP Address: 192.168.100.90
[19:46:36][C][ethernet:516]:   Hostname: 'water-meter-esp'
[19:46:36][C][ethernet:517]:   Subnet: 255.255.252.0
[19:46:36][C][ethernet:518]:   Gateway: 192.168.100.1
[19:46:36][C][ethernet:523]:   DNS1: 192.168.100.1
[19:46:36][C][ethernet:524]:   DNS2: 0.0.0.0
[19:46:36][C][ethernet:536]:   MAC Address: C4:DE:E2:B2:81:AC
[19:46:36][C][ethernet:537]:   Is Full Duplex: YES
[19:46:36][C][ethernet:538]:   Link Speed: 100
[19:46:36][C][ethernet:345]:   Power Pin: 16
[19:46:36][C][ethernet:347]:   MDC Pin: 23
[19:46:36][C][ethernet:348]:   MDIO Pin: 18
[19:46:36][C][ethernet:349]:   PHY addr: 1
[19:46:36][C][ethernet:351]:   Type: LAN8720
[19:46:36][C][sntp:042]: SNTP Time:
[19:46:36][C][sntp:045]:   Server 0: '0.at.pool.ntp.org'
[19:46:36][C][sntp:045]:   Server 1: '0.pool.ntp.org'
[19:46:36][C][mdns:116]: mDNS:
[19:46:36][C][mdns:117]:   Hostname: water-meter-esp
[19:46:36][V][mdns:118]:   Services:
[19:46:36][V][mdns:120]:   - _esphomelib, _tcp, 6053
[19:46:36][V][mdns:122]:     TXT: version = 2025.2.2
[19:46:36][V][mdns:122]:     TXT: mac = c4dee2b281ac
[19:46:36][V][mdns:122]:     TXT: platform = ESP32
[19:46:36][V][mdns:122]:     TXT: board = wt32-eth01
[19:46:36][V][mdns:122]:     TXT: network = ethernet
[19:46:36][V][mdns:122]:     TXT: api_encryption = Noise_NNpsk0_25519_ChaChaPoly_SHA256
[19:46:36][V][mdns:122]:     TXT: project_name = Diehl Hydrus RC 868.Watermeter
[19:46:36][V][mdns:122]:     TXT: project_version = 2.0.7
[19:46:36][C][esphome.ota:073]: Over-The-Air updates:
[19:46:36][C][esphome.ota:074]:   Address: water-meter-esp.local:3232
[19:46:36][C][esphome.ota:075]:   Version: 2
[19:46:36][C][esphome.ota:078]:   Password configured
[19:46:36][C][safe_mode:018]: Safe Mode:
[19:46:36][C][safe_mode:020]:   Boot considered successful after 60 seconds
[19:46:36][C][safe_mode:021]:   Invoke after 10 boot attempts
[19:46:36][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[19:46:36][C][api:140]: API Server:
[19:46:36][C][api:141]:   Address: water-meter-esp.local:6053
[19:46:36][C][api:143]:   Using noise encryption: YES
[19:46:36][C][wmbus:338]: wM-Bus v4.1.4-1.17.1-b8f4a945:
[... 257 lines cut ...]
        unit_of_measurement: "dBm"
        device_class: "signal_strength"
        state_class: "measurement"
        entity_category: "diagnostic"
      - name: "Zählerstand Wasser TOTAL"
        field: "total"
        accuracy_decimals: 3
        unit_of_measurement: "m³"
        device_class: "water"
        state_class: "total_increasing"
        icon: "mdi:water"
  - platform: template
    name: Device Boot counter
    id: bootcounter
    accuracy_decimals: 0
    state_class: "measurement"
    entity_category: "diagnostic"
    lambda: return (id(boot_counter));

  # Last boot timestamp
  - platform: template
    name: "Device Last Boot"
    disabled_by_default: true
    id: device_lastBoot_time
    icon: mdi:clock-start`

SzczepanLeon, 2025-03-04:
I need logs with received telegrams, and level should be VERY_VERDOSE

MUTEtheCyberwolf, 2025-03-04:
### Here is the Log that i got from the ESPhome node. The Logger Level is set to VERY_VERBOSE.

`INFO ESPHome 2025.2.2
INFO Reading configuration /config/esphome/watermeter.yaml...
WARNING GPIO2 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO12 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO15 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from 192.168.100.90 using esphome API
INFO Successfully connected to water-meter-esp @ 192.168.100.90 in 0.002s
INFO Successful handshake with water-meter-esp @ 192.168.100.90 in 0.053s
[00:12:26][I][app:100]: ESPHome version 2025.2.2 compiled on Mar  4 2025, 19:45:49
[00:12:26][I][app:102]: Project Diehl Hydrus RC 868.Watermeter version 2.0.7
[00:12:26][C][logger:177]: Logger:
[00:12:26][C][logger:178]:   Max Level: VERY_VERBOSE
[00:12:26][C][logger:179]:   Initial Level: VERY_VERBOSE
[00:12:26][C][logger:181]:   Log Baud Rate: 0
[00:12:26][C][logger:182]:   Hardware UART: UART0
[00:12:26][C][logger:186]:   Level for 'wMBus-lib': VERY_VERBOSE
[00:12:26][C][logger:186]:   Level for 'wmbus': VERY_VERBOSE
[00:12:26][C][template.sensor:022]: Template Sensor 'Device Boot counter'
[00:12:26][C][template.sensor:022]:   State Class: 'measurement'
[00:12:26][C][template.sensor:022]:   Unit of Measurement: ''
[00:12:26][C][template.sensor:022]:   Accuracy Decimals: 0
[00:12:26][C][template.sensor:023]:   Update Interval: 60.0s
[00:12:26][C][template.sensor:022]: Template Sensor 'Device Last Boot'
[00:12:26][C][template.sensor:022]:   State Class: ''
[00:12:26][C][template.sensor:022]:   Unit of Measurement: ''
[00:12:26][C][template.sensor:022]:   Accuracy Decimals: 1
[00:12:26][C][template.sensor:022]:   Icon: 'mdi:clock-start'
[00:12:26][C][template.sensor:023]:   Update Interval: 60.0s
[00:12:26][C][template.switch:068]: Template Switch 'Device Boot Counter reset'
[00:12:26][C][template.switch:091]:   Restore Mode: always OFF
[00:12:26][C][template.switch:057]:   Optimistic: NO
[00:12:26][C][restart:068]: Restart Switch 'Device Restart'
[00:12:26][C][restart:070]:   Icon: 'mdi:restart'
[00:12:26][C][restart:091]:   Restore Mode: always OFF
[00:12:26][C][ethernet:326]: Ethernet:
[00:12:26][C][ethernet:515]:   IP Address: 192.168.100.90
[00:12:26][C][ethernet:516]:   Hostname: 'water-meter-esp'
[00:12:26][C][ethernet:517]:   Subnet: 255.255.252.0
[00:12:26][C][ethernet:518]:   Gateway: 192.168.100.1
[00:12:26][C][ethernet:523]:   DNS1: 192.168.100.1
[00:12:26][C][ethernet:524]:   DNS2: 0.0.0.0
[00:12:26][C][ethernet:536]:   MAC Address: C4:DE:E2:B2:81:AC
[00:12:26][C][ethernet:537]:   Is Full Duplex: YES
[00:12:26][C][ethernet:538]:   Link Speed: 100
[00:12:26][C][ethernet:345]:   Power Pin: 16
[00:12:26][C][ethernet:347]:   MDC Pin: 23
[00:12:26][C][ethernet:348]:   MDIO Pin: 18
[00:12:26][C][ethernet:349]:   PHY addr: 1
[00:12:26][C][ethernet:351]:   Type: LAN8720
[00:12:26][C][sntp:042]: SNTP Time:
[00:12:26][C][sntp:045]:   Server 0: '0.at.pool.ntp.org'
[00:12:26][C][sntp:045]:   Server 1: '0.pool.ntp.org'
[00:12:26][C][mdns:116]: mDNS:
[00:12:26][C][mdns:117]:   Hostname: water-meter-esp
[00:12:26][V][mdns:118]:   Services:
[00:12:26][V][mdns:120]:   - _esphomelib, _tcp, 6053
[00:12:26][V][mdns:122]:     TXT: version = 2025.2.2
[00:12:26][V][mdns:122]:     TXT: mac = c4dee2b281ac
[00:12:26][V][mdns:122]:     TXT: platform = ESP32
[00:12:26][V][mdns:122]:     TXT: board = wt32-eth01
[00:12:26][V][mdns:122]:     TXT: network = ethernet
[00:12:26][V][mdns:122]:     TXT: api_encryption = Noise_NNpsk0_25519_ChaChaPoly_SHA256
[... 52 lines cut ...]
[00:12:31][VV][3of6:095]: Decode 3 out of 6 OK.
[00:12:31][V][mbus:053]: Frame: 6344A51182801466700767748C0038900F002C251902850029A5864C3E7F565288DB7A1900400710B8E02A628D31F0601296A033C03926C2A907F77B7607B64D01C67748B28431720EABDF6C909D76C4C153EFCEED977DECAEF2746371129727C72D124F9AE61E63F086461971CB5FE247C9 (114) [with CRC]
[00:12:31][V][mbus:096]: Validating CRC for Block1
[00:12:31][V][crc:031]:     calculated: 0x6774, read: 0x6774
[00:12:31][V][mbus:116]: Validating CRC for Block2
[00:12:31][V][crc:031]:     calculated: 0x3E7F, read: 0x3E7F
[00:12:31][V][mbus:116]: Validating CRC for Block3
[00:12:31][V][crc:031]:     calculated: 0xF060, read: 0xF060
[00:12:31][V][mbus:116]: Validating CRC for Block4
[00:12:31][V][crc:031]:     calculated: 0x01C6, read: 0x01C6
[00:12:31][V][mbus:116]: Validating CRC for Block5
[00:12:31][V][crc:031]:     calculated: 0xEFCE, read: 0xEFCE
[00:12:31][V][mbus:116]: Validating CRC for Block6
[00:12:31][V][crc:031]:     calculated: 0x9AE6, read: 0x9AE6
[00:12:31][V][mbus:116]: Validating CRC for Block7
[00:12:31][V][crc:031]:     calculated: 0x47C9, read: 0x47C9
[00:12:31][V][mbus:063]: Frame: 6344A5118280146670078C0038900F002C251902850029A5864C565288DB7A1900400710B8E02A628D311296A033C03926C2A907F77B7607B64D7748B28431720EABDF6C909D76C4C153ED977DECAEF2746371129727C72D124F1E63F08646197 [... cut ...] E2 (100) [without CRC]
[00:12:31][VV][wmbus:056]: Have data from RF ...
[00:12:31][V][Telegram.cpp:1131]: (wmbus) parseDLL @0 100
[00:12:31][V][Telegram.cpp:1178]: (wmbus) parseELL @10 90
[00:12:31][V][Telegram.cpp:1333]: (wmbus) parseNWL @13 87
[00:12:31][V][Telegram.cpp:1391]: (wmbus) parseAFL @13 87
[00:12:31][V][Telegram.cpp:2078]: (wmbus) parseTPL @30 70
[00:12:31][V][Telegram.cpp:1546]: (wmbus) input to kdf for enc
[00:12:31][V][Telegram.cpp:1555]: (wmbus) no key, thus cannot execute kdf.`

MUTEtheCyberwolf, 2025-05-04:
Any Idea by now what may cause this ?

AndrCarl, 2025-06-26:
Hi have the same issue with my hydrus

> [10:52:54][C][wmbus:413]: wM-Bus v4.2.1-1.17.1-b8f4a945:
> [10:52:54][C][wmbus:431]:   Chip ID: C8565540C86C
> [10:52:54][C][wmbus:433]:   CC1101 frequency: 868.950 MHz
> [10:52:54][C][wmbus:440]:     CLK Pin:  GPIO18
> [10:52:54][C][wmbus:464]:     ID: 87865106 [0x053CB712]
> [10:52:54][C][wmbus:466]:     Key: '' (emptied)
> [10:52:54][C][wmbus:468]:     Field: 'flow_temperature'
> [10:52:55][C][wmbus:469]:      Name: 'Water Temperature'
> [10:52:55][C][wmbus:469]:        Device Class: 'temperature'
> [10:52:55][C][wmbus:469]:        State Class: 'measurement'
> [10:52:55][C][wmbus:469]:        Unit of Measurement: '°C'
> [10:52:55][C][wmbus:469]:        Accuracy Decimals: 1
> [10:52:55][C][wmbus:469]:        Icon: 'mdi:thermometer-water'
> [10:52:55][C][wmbus:468]:     Field: 'remaining_battery_life'
> [10:52:57][C][wmbus:469]:        Accuracy Decimals: 3
> [10:52:57][C][wmbus:469]:      Name: 'RSSI'
> [10:53:00][C][wmbus:469]:        Device Class: 'signal_strength'
> [10:53:00][C][wmbus:469]:        State Class: 'measurement'
> [10:53:00][C][wmbus:469]:        Unit of Measurement: 'dBm'
> [10:53:00][C][wmbus:469]:        Accuracy Decimals: 0
> [10:53:00][C][wmbus:468]:     Field: 'total'
> [10:53:00][C][wmbus:469]:        Device Class: 'water'
> [10:53:00][C][wmbus:469]:        Icon: 'mdi:water'

> [10:57:16][V][rxLoop:167]: Have 144 bytes from CC1101 Rx, RSSI: -80 dBm LQI: 128
> [10:57:19][V][rxLoop:172]: Synchronus mode enabled.
> [10:57:19][D][mbus:035]: Received T1 A frame
> [10:57:19][V][mbus:046]: Frame: 64B71C99934D59A64DB1AB134DA5932D335AB3459658E9565A95963B439958E34B593596C933A5D2969AD25B0DC69C4E4E63B24E6D265962CD5933569A366CB1ACA39AC98E99C8F25963A5A5C8DA95CB1C9ACA56B268F264D3B23656B271C9A97266A9363C537196599A3B313A9C4B68EC96D32993CB2719CB18E6D0BCB19593A35B4D0E96C996C8D2CB4E58D3C5638E (144) [RAW]
> [10:57:19][VV][3of6:095]: Decode 3 out of 6 OK.
> [10:57:19][V][mbus:053]: Frame: 5344A51106518687760737168C0002900F002C2502130700E729CF66C981DFD27A2E7ACA00310710AB5886EBA8A2A4BE0029F4B69484A8F08ABE512E196E44AF4A6F1BD74555AB8D2FD362E0CEA7EE45EDBAC3ED952B0CC298A0E13379B7D022 (96) [with CRC]
> [10:57:19][V][mbus:096]: Validating CRC for Block1
> [10:57:19][V][crc:031]:     calculated: 0x3716, read: 0x3716
> [10:57:19][V][mbus:116]: Validating CRC for Block2
> [10:57:19][V][crc:031]:     calculated: 0xC981, read: 0xC981
> [10:57:19][V][mbus:116]: Validating CRC for Block3
> [10:57:19][V][crc:031]:     calculated: 0xA4BE, read: 0xA4BE
> [10:57:19][V][mbus:116]: Validating CRC for Block4
> [10:57:19][V][crc:031]:     calculated: 0x4A6F, read: 0x4A6F
> [10:57:19][V][mbus:116]: Validating CRC for Block5
> [10:57:19][V][crc:031]:     calculated: 0xC3ED, read: 0xC3ED
> [10:57:20][V][mbus:116]: Validating CRC for Block6
> [10:57:20][V][Telegram.cpp:1555]: (wmbus) no key, thus cannot execute kdf.
> [10:57:20][I][wmbus:106]: hydrus [0x87865106] RSSI: -80dBm T: 5344A5110651868776078C0002900F002C2502130700E729CF66DFD27A2E7ACA00310710AB5886EBA8A20029F4B69484A8F08ABE512E196E44AF1BD74555AB8D2FD362E0CEA7EE45EDBA952B0CC298A0E13379B7 (84) T1 A

Telegram with 96 with CRC is decodeing good in wmbusmeters.org/analyze

> {
>     "_":"telegram",
>     "media":"water",
>     "meter":"hydrus",
>     "name":"",
>     "id":"87865106",
>     "at_date":"2024-12-31",
>     "flow_temperature_c":19.4,
>     "remaining_battery_life_y":12.107025,
>     "total_m3":8.951,
>     "total_at_date_m3":0.2,
>     "status":"OK",
>     "timestamp":"2025-06-26T09:28:49Z"
> }

I tried key in secret file and also directly in yaml with quotes, but no sucess in decrypting.

yaml is nearly same like [MUTEtheCyberwolf], only sync_mode: true and version4 as I have cc1101