SzczepanLeon/esphome-components #194: wmbus 3.2.3 on ESPHome 2025.2.2

HausnerR, 2025-03-09:
Hi,
wmbus component 3.2.3 (I'm using ESP8266) is not compiling on ESPHome 2025.2.2:

```
INFO ESPHome 2025.2.2
INFO Reading configuration home_apator.yaml...
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "/opt/homebrew/bin/esphome", line 8, in <module>
    sys.exit(main())
  File "/opt/homebrew/lib/python3.10/site-packages/esphome/__main__.py", line 1049, in main
    return run_esphome(sys.argv)
  File "/opt/homebrew/lib/python3.10/site-packages/esphome/__main__.py", line 1027, in run_esphome
    config = read_config(dict(args.substitution) if args.substitution else {})
  File "/opt/homebrew/lib/python3.10/site-packages/esphome/config.py", line 1095, in read_config
    res = load_config(command_line_substitutions)
  File "/opt/homebrew/lib/python3.10/site-packages/esphome/config.py", line 949, in load_config
    return _load_config(command_line_substitutions)
  File "/opt/homebrew/lib/python3.10/site-packages/esphome/config.py", line 939, in _load_config
    return validate_config(config, command_line_substitutions)
  File "/opt/homebrew/lib/python3.10/site-packages/esphome/config.py", line 835, in validate_config
    target_platform = core_config.preload_core_config(config, result)
  File "/opt/homebrew/lib/python3.10/site-packages/esphome/core/config.py", line 222, in preload_core_config
    if _is_target_platform(domain):
  File "/opt/homebrew/lib/python3.10/site-packages/esphome/core/config.py", line 184, in _is_target_platform
    if get_component(name, True).is_target_platform:
  File "/opt/homebrew/lib/python3.10/site-packages/esphome/loader.py", line 205, in get_component
    return _lookup_module(domain, exception)
  File "/opt/homebrew/lib/python3.10/site-packages/esphome/loader.py", line 181, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/opt/homebrew/Cellar/python@3.10/3.10.16/Frameworks/Python.framework/Versions/3.10/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/Users/jakub/Repos/esphome-projects/.esphome/external_components/0e75e1a4/components/wmbus/__init__.py", line 62, in <module>
    cv.Required(CONF_IP_ADDRESS): cv.ipv4,
AttributeError: module 'esphome.config_validation' has no attribute 'ipv4'
```

Are you plan to fix this issue to be able to still use this component on ESP8266?

Best,
Kuba

SzczepanLeon, 2025-03-09:
Already fixed long time ago
https://github.com/SzczepanLeon/esphome-components/issues/189#issuecomment-2679873456

closed 2025-03-09 by SzczepanLeon

HausnerR, 2025-03-09:
Thanks! Did not find that topic before :)