SzczepanLeon/esphome-components #204: Receive no Information from QHeat

Icesoldier7, 2025-04-18:
Hi Szczepan, 

i hope you can help me, i try to receive information from a qheat 5.5 us 
wmbuster says: 

telegram=|58449344185645673C3778_07791856456793443C040DFF5F350082B200359CA6AAB1D75D8D875454F8567D9E1B00EDB7DE6BE9CB991DE3BC07FFA797CC4D555CD35DE2A9088437B7EA303F8BA28502FD170000046D2D0F3234|

Auto driver    : not found!
Similar driver : relhca 04/67
Using driver   : qheat 00/00
000   : 58 length (88 bytes)(OK) ("total_energy_consumption_kwh":-854548.7) ("last_year_date":"2058-13-22") ("last_year_energy_consumption_kwh":23000291.1) ("last_month_date":"2109-14-23") ("last_month_energy_consumption_kwh":10110197.8)
001   : 44 dll-c (from meter SND_NR)
002   : 9344 dll-mfct (QDS)
004   : 18564567 dll-id (67455618)
008   : 3c dll-version
009   : 37 dll-type (Radio converter (meter side))
010   : 78 tpl-ci-field (EN 13757-3 Application Layer (no tplh))
011   : 07 dif (64 Bit Integer/Binary Instantaneous value)
012   : 79 vif (Enhanced identification)
013 C?: 1856456793443C04
013 C!: *** 18564567 tpl-id (67455618)
017   : *** 9344 tpl-mfct (QDS)
019   : *** 3C tpl-version
020   : *** 04 tpl-type (Heat meter)
021   : 0D dif (variable length Instantaneous value)
022   : FF vif (Manufacturer specific)
023   : 5F vife (?)
024   : 35 varlen=53
025 C?: 0082B200359CA6AAB1D75D8D875454F8567D9E1B00EDB7DE6BE9CB991DE3BC07FFA797CC4D555CD35DE2A9088437B7EA303F8BA285
037 C!: *** 875454F8 ("total_energy_consumption_kwh":-854548.7)
041 C!: *** 567D ("last_year_date":"2058-13-22")
043 C!: *** 9E1B00ED ("last_year_energy_consumption_kwh":23000291.1)
047 C!: *** B7DE ("last_month_date":"2109-14-23")
049 C!: *** 6BE9CB99 ("last_month_energy_consumption_kwh":10110197.8)
078   : 02 dif (16 Bit Integer/Binary Instantaneous value)
079   : FD vif (Second extension FD of VIF-codes)
080   : 17 vife (Error flags (binary))
081 C!: 0000 ("status":"OK")
083   : 04 dif (32 Bit Integer/Binary Instantaneous value)
084   : 6D vif (Date and time type)
085 C!: 2D0F3234 ("device_date_time":"2025-04-18 15:45")

{
    "_":"telegram",
    "media":"heat",
    "meter":"qheat",
    "name":"",
    "id":"67455618",
    "last_month_energy_consumption_kwh":10110197.8,
    "last_year_energy_consumption_kwh":23000291.1,
    "total_energy_consumption_kwh":-854548.7,
    "device_date_time":"2025-04-18 15:45",
    "last_month_date":"2109-14-23",
    "last_year_date":"2058-13-22",
    "status":"OK",
    "timestamp":"2025-04-18T14:22:07Z"

But if i try with esphome text_sensor:

  - platform: version
    name: "ESPHome Version"
  - platform: template
    name: "Kaltwasserzähler Letztes Update"
    id: cold_water_last_update
  - platform: template
    name: "Heizungswasser Letztes Update"
    id: hot_water_last_update
  - platform: wmbus
    meter_id: 0x67455618 
    type: qheat
    sensors:
    - name: "Heizungswasser"
      field: "total_energy_consumption_kwh"

it doesent work ][meters.cpp:913]: (meter) ESPHome(0) qheat  handling telegram from 67455618.M=QDS.V=3c.T=37
                           [W][wmbus:170]: Can't get requested field 'total_energy_consumption_kwh'

What is here my mistake ?

SzczepanLeon, 2025-04-18:
share YAML, but I will start with FAQ and split field and unit

Icesoldier7, 2025-04-18:
esphome:
  name: kalt-und-warmwasser
  friendly_name: Kalt und Warmwasser

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "##Secretkey##"

ota:
  - platform: esphome
    password: "##password##"

wifi:
  ssid: ssid
  password: Password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Kalt-Und-Warmwasser"
    password: "Password"

captive_portal:
    
web_server:
  port: 80

external_components:
  - source: github://SzczepanLeon/esphome-components@main
    components: [wmbus]

wmbus:
  mosi_pin: GPIO23  # D23 auf ESP32
  miso_pin: GPIO19  # D19 auf ESP32
  clk_pin: GPIO18   # D18 auf ESP32
  cs_pin: GPIO5     # D5 auf ESP32
  gdo0_pin: GPIO16  # RX2 auf ESP32
  gdo2_pin: GPIO17  # TX2 auf ESP32
  led_pin: GPIO2
  all_drivers: true
  log_all: true

sensor:
  - platform: wmbus
    type: qwater
    #add_prefix: true
    # mode: C1

  - platform: wmbus
    meter_id: 0x51416968 
    type: qwater 
    
    sensors:  
     - name: "Kaltwasser"
       field: "total"
       accuracy_decimals: 3
       unit_of_measurement: "m³"
       state_class: "total_increasing"
       device_class: "water"
       icon: "mdi:water"
       on_value:
         then:
           - text_sensor.template.publish:
               id: cold_water_last_update
               state: !lambda 'return id(sntp_time).now().strftime("%Y-%m-%d %H:%M:%S");'
                   
time:
  - platform: sntp
    id: sntp_time
    timezone: Europe/Sofia
    servers:
     - 0.pool.ntp.org
     - 1.pool.ntp.org
     - 2.pool.ntp.org

text_sensor:
  - platform: version
    name: "ESPHome Version"
  - platform: template
    name: "Kaltwasserzähler Letztes Update"
    id: cold_water_last_update
  - platform: template
    name: "Heizungswasser Letztes Update"
    id: hot_water_last_update
  - platform: wmbus
    meter_id: 0x67455618 
    type: qheat
    sensors:
    - name: "Heizungswasser"
      field: "total_energy_consumption_kwh"

Icesoldier7, 2025-04-18:
The Coldwater sends Informations, but this qheat doesent work

SzczepanLeon, 2025-04-18:
Please check example in github, correct YAML and try again.

total_energy_consumption_kwh is sensor not text_sensor. Also:
field = total_energy_consumption
unik = kWh

Icesoldier7, 2025-04-18:
**You are my Hero of the day :)  .... I receive data :)** 
But its strang: 
meters.cpp:913]: (meter) ESPHome(0) qheat  handling telegram from 67455618.M=QDS.V=3c.T=37
[sensor:094]: 'Heizungswasser': Sending state 8109523.00000 kwh with 3 decimals of accuracy
[D][meters.cpp:913]: (meter) ESPHome(0) qheat  handling telegram from 67455618.M=QDS.V=3c.T=37
[sensor:094]: 'Heizungswasser': Sending state 2533817.00000 kwh with 3 decimals of accuracy
[meters.cpp:913]: (meter) ESPHome(0) qheat  handling telegram from 67455618.M=QDS.V=3c.T=37
[D][sensor:094]: 'Heizungswasser': Sending state -991758.87500 kwh with 3 decimals of accuracy
[D][meters.cpp:913]: (meter) ESPHome(0) qheat  handling telegram from 67455618.M=QDS.V=3c.T=37
[D][sensor:094]: 'Heizungswasser': Sending state 19920212.00000 kwh with 3 decimals of accuracy
[D][meters.cpp:913]: (meter) ESPHome(0) qheat  handling telegram from 67455618.M=QDS.V=3c.T=37
[D][sensor:094]: 'Heizungswasser': Sending state 4754103.00000 kwh with 3 decimals of accuracy
[D][sensor:094]: 'Heizungswasser': Sending state 21127230.00000 kwh with 3 decimals of accuracy
[D][text_sensor:064]: 'Heizungswasser Letztes Update': Sending state '2025-04-18 20:32:28'

??? why this sends diferrent informations ?