SzczepanLeon/esphome-components #206: ESP32 & CC1101 : no data wmbusmeter

Anoxym, 2025-04-27:
Hi,

I successfully compiled the program on ESP32 and communication with CC1101 works. I first compiled the test program and get the watermeter address (0x21202341)

```
[23:18:06][D][sensor:094]: '0x00 Watermeter CC1101 LQI': Sending state 128.00000 lqi with 0 decimals of accuracy
[23:18:06][D][sensor:094]: '0x00 Watermeter CC1101 RSSI': Sending state -58.00000 % with 0 decimals of accuracy
[23:18:06][D][wmbus:182]: Meter ID [0x21202341] RSSI: -58 dBm LQI: 128 Frame: T1 A not found in configuration T: 1E44A511412320217B077A8B001005181A382891BBA9777E31D03543F7A72F (31)
[23:18:20][D][mbus:034]: Processing T1 A frame
[23:18:20][D][sensor:094]: '0x00 Watermeter CC1101 LQI': Sending state 128.00000 lqi with 0 decimals of accuracy
[23:18:20][D][sensor:094]: '0x00 Watermeter CC1101 RSSI': Sending state -50.00000 % with 0 decimals of accuracy
[23:18:20][D][wmbus:182]: Meter ID [0x21202341] RSSI: -50 dBm LQI: 128 Frame: T1 A not found in configuration T: 1E44A511412320217B077A8B001005181A382891BBA9777E31D03543F7A72F (31)
[23:18:24][D][sensor:094]: 'Device Boot counter': Sending state 2.00000  with 0 decimals of accuracy
[23:18:32][D][sensor:094]: 'wifi_signal_db': Sending state -70.00000 dBm with 0 decimals of accuracy
[23:18:32][D][sensor:094]: 'Device WLAN Signal': Sending state 60.00000 % with 0 decimals of accuracy
[23:18:33][D][mbus:034]: Processing T1 A frame
[23:18:33][D][sensor:094]: '0x00 Watermeter CC1101 LQI': Sending state 128.00000 lqi with 0 decimals of accuracy
[23:18:33][D][sensor:094]: '0x00 Watermeter CC1101 RSSI': Sending state -48.00000 % with 0 decimals of accuracy
[23:18:33][D][wmbus:182]: Meter ID [0x21202341] RSSI: -48 dBm LQI: 128 Frame: T1 A not found in configuration T: 3E44A511412320217B077A8B0030056FA7C61077767ECC6B9701BD272F5D3A61308913459C34B9937B05C788DE6755A63F32EE78C373EE95D892A887A448A0 (63)
[23:18:46][D][mbus:034]: Processing T1 A frame
[23:18:46][D][sensor:094]: '0x00 Watermeter CC1101 LQI': Sending state 128.00000 lqi with 0 decimals of accuracy
[23:18:46][D][sensor:094]: '0x00 Watermeter CC1101 RSSI': Sending state -47.00000 % with 0 decimals of accuracy
[23:18:46][D][wmbus:182]: Meter ID [0x21202341] RSSI: -47 dBm LQI: 128 Frame: T1 A not found in configuration T: 1E44A511412320217B077A8B001005181A382891BBA9777E31D03543F7A72F (31)
[23:18:58][D][mbus:034]: Processing T1 A frame
[23:18:58][D][sensor:094]: '0x00 Watermeter CC1101 LQI': Sending state 128.00000 lqi with 0 decimals of accuracy
[23:18:58][D][sensor:094]: '0x00 Watermeter CC1101 RSSI': Sending state -48.00000 % with 0 decimals of accuracy
[23:18:58][D][wmbus:182]: Meter ID [0x21202341] RSSI: -48 dBm LQI: 128 Frame: T1 A not found in configuration T: 1E44A511412320217B077A8B001005181A382891BBA9777E31D03543F7A72F (31)
[23:19:11][D][mbus:034]: Processing T1 A frame
[23:19:11][D][sensor:094]: '0x00 Watermeter CC1101 LQI': Sending state 128.00000 lqi with 0 decimals of accuracy
[23:19:11][D][sensor:094]: '0x00 Watermeter CC1101 RSSI': Sending state -48.00000 % with 0 decimals of accuracy
[23:19:11][D][wmbus:182]: Meter ID [0x21202341] RSSI: -48 dBm LQI: 128 Frame: T1 A not found in configuration T: 1E44A511412320217B077A8B001005181A382891BBA9777E31D03543F7A72F (31)
[23:19:23][D][mbus:034]: Processing T1 A frame
[23:19:23][D][sensor:094]: '0x00 Watermeter CC1101 LQI': Sending state 128.00000 lqi with 0 decimals of accuracy
[23:19:23][D][sensor:094]: '0x00 Watermeter CC1101 RSSI': Sending state -56.00000 % with 0 decimals of accuracy
[23:19:23][D][wmbus:182]: Meter ID [0x21202341] RSSI: -56 dBm LQI: 128 Frame: T1 A not found in configuration T: 1E44A511412320217B077A8B001005181A382891BBA9777E31D03543F7A72F (31)
[23:19:24][D][sensor:094]: 'Device Boot counter': Sending state 2.00000  with 0 decimals of accuracy
[23:19:32][D][sensor:094]: 'wifi_signal_db': Sending state -66.00000 dBm with 0 decimals of accuracy
[23:19:32][D][sensor:094]: 'Device WLAN Signal': Sending state 68.00000 % with 0 decimals of accuracy
[23:19:36][D][mbus:034]: Processing T1 A frame
[23:19:36][D][sensor:094]: '0x00 Watermeter CC1101 LQI': Sending state 129.00000 lqi with 0 decimals of accuracy
[23:19:36][D][sensor:094]: '0x00 Watermeter CC1101 RSSI': Sending state -55.00000 % with 0 decimals of accuracy
[23:19:36][D][wmbus:182]: Meter ID [0x21202341] RSSI: -55 dBm LQI: 129 Frame: T1 A not found in configuration T: 1E44A511412320217B077A8B001005181A382891BBA9777E31D03543F7A72F (31)
```

And then, complied this program : https://github.com/zibous/ha-watermeter/issues/60#issuecomment-2817070527 (I am on ESPHome via Home assistant)

But I didn't get any data, here is what I have in log :

```
[00:17:30][D][wmbus:549]: cc1101 state message: Init, error code: 0
[00:18:06][D][mbus:034]: Processing T1 A frame
[00:18:06][D][wmbus:097]: Using driver 'izar' for ID [0x21202341] RSSI: -66 dBm LQI: 129 Frame: T1 A T: 1E44A511412320217B077A8F0010054B59096DB937F54B8E362B048AABED11 (31)
[00:18:07][W][wmbus:923]: WARNING Alarm message: Fehler, Überschwemung, error code: 130
[00:18:07][D][wmbus:943]: Alarm message: \xec$\xfb?�, error code: 0
[00:18:30][D][wmbus:549]: cc1101 state message: Init, error code: 0
[00:18:34][D][mbus:034]: Processing T1 A frame
[00:19:23][D][mbus:034]: Processing T1 A frame
[00:19:24][D][wmbus:097]: Using driver 'izar' for ID [0x21202341] RSSI: -67 dBm LQI: 140 Frame: T1 A T: 1E44A511412320217B077A8F0010054B59096DB937F54B8E362B048AABED11 (31)
[00:19:24][W][wmbus:923]: WARNING Alarm message: Fehler, Überschwemung, error code: 130
[00:19:24][D][wmbus:943]: Alarm message: \xec$\xfb?�, error code: 0
[00:19:30][D][wmbus:549]: cc1101 state message: Init, error code: 0
[00:19:36][D][mbus:034]: Processing T1 A frame
[00:19:36][D][wmbus:097]: Using driver 'izar' for ID [0x21202341] RSSI: -68 dBm LQI: 130 Frame: T1 A T: 1E44A511412320217B077A8F0010054B59096DB937F54B8E362B048AABED11 (31)
[00:19:37][W][wmbus:923]: WARNING Alarm message: Fehler, Überschwemung, error code: 130
[00:19:37][D][wmbus:943]: Alarm message: \xec$\xfb?�, error code: 0
[00:19:49][D][mbus:034]: Processing T1 A frame
[00:19:49][D][wmbus:097]: Using driver 'izar' for ID [0x21202341] RSSI: -65 dBm LQI: 128 Frame: T1 A T: 1E44A511412320217B077A8F0010054B59096DB937F54B8E362B048AABED11 (31)
[00:19:50][W][wmbus:923]: WARNING Alarm message: Fehler, Überschwemung, error code: 130
[00:19:50][D][wmbus:943]: Alarm message: \xec$\xfb?�, error code: 0
[00:20:03][D][mbus:034]: Processing T1 A frame
[00:20:03][D][wmbus:097]: Using driver 'izar' for ID [0x21202341] RSSI: -65 dBm LQI: 133 Frame: T1 A T: 1E44A511412320217B077A8F0010054B59096DB937F54B8E362B048AABED11 (31)
[00:20:03][W][wmbus:923]: WARNING Alarm message: Fehler, Überschwemung, error code: 130
[00:20:03][D][wmbus:943]: Alarm message: \xec$\xfb?�, error code: 0
[00:20:07][D][sntp:067]: Synchronized time: 2025-04-27 00:20:07
[00:20:07][W][system:471]: Synchronized sntp clock
[00:20:07][D][wmbus:549]: cc1101 state message: Init, error code: 0
[00:20:30][D][wmbus:549]: cc1101 state message: Init, error code: 0
[00:20:42][D][mbus:034]: Processing T1 A frame
[00:20:54][D][mbus:034]: Processing T1 A frame
[00:20:54][D][wmbus:097]: Using driver 'izar' for ID [0x21202341] RSSI: -63 dBm LQI: 132 Frame: T1 A T: 1E44A511412320217B077A8F0010054B59096DB937F54B8E362B048AABED11 (31)
[00:20:54][W][wmbus:923]: WARNING Alarm message: Fehler, Überschwemung, error code: 130
[00:20:54][D][wmbus:943]: Alarm message: \xec$\xfb?�, error code: 0
[00:21:06][D][mbus:034]: Processing T1 A frame
[00:21:06][D][wmbus:097]: Using driver 'izar' for ID [0x21202341] RSSI: -68 dBm LQI: 130 Frame: T1 A T: 1E44A511412320217B077A8F0010054B59096DB937F54B8E362B048AABED11 (31)
[00:21:07][W][wmbus:923]: WARNING Alarm message: Fehler, Überschwemung, error code: 130
[00:21:07][D][wmbus:943]: Alarm message: \xec$\xfb?�, error code: 0
[00:21:19][D][mbus:034]: Processing T1 A frame
[00:21:30][D][wmbus:549]: cc1101 state message: Init, error code: 0
[00:21:32][D][mbus:034]: Processing T1 A frame
[00:21:32][D][wmbus:097]: Using driver 'izar' for ID [0x21202341] RSSI: -67 dBm LQI: 130 Frame: T1 A T: 3E44A511412320217B077A8F003005BB15441123859046292A3987BBF052222ACF4FF3B5BEB88F0C490633D033F7BBB4D2513D71D38ABA6E7472DB54CC9F84 (63)
[00:21:32][W][wmbus:923]: WARNING Alarm message: Fehler, Wasser Überlauf, Überschwemung, error code: 194
[00:21:33][D][wmbus:943]: Alarm message: \xec$\xfb?�, error code: 0
```

https://wmbusmeters.org/ seems to be able to read data :

[image]

Can someone help me?
Thanks

farmio, 2025-05-09:
Hi 👋! 
Disclaimer: Be  aware that I'm very new to all of ESPHome and this component - so not speaking from experience.

It looks like you are using the wrong driver for your meter. 
> [00:18:06][D][wmbus:097]: Using driver 'izar' for ID [0x21202341]

The wmbusmeters screenshot says `dme_07`, not `izar`.
Also note that the linked ESPHome config uses version 3 of this component, not the current version 4 - so docs may not fit.

Anoxym, 2025-05-11:
Hi!

Yes, I’ve seen that the current driver is dme_07, but changing it doesn’t seem to make any difference.
As for version 3, from what I understand, it’s the only one that works with ESPHome on Home Assistant.
Or maybe someone have a code that works on the last version?

SzczepanLeon, 2025-05-11:
Every version is based on ESPHome so it works with Home Assistant.

Anoxym, 2025-05-11:
Like the example code ? I'm not very comfortable with writing the functional code. I want to retrieving the water meter data to add it to the Energy section in Home Assistant.

SzczepanLeon, 2025-05-11:
On main page.

Anoxym, 2025-05-11:
Like this ? 

```
# Global config
esphome:
  name: "water"
  friendly_name: "water"

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  level: INFO
  # level: DEBUG

wifi:

  networks:

    # default wifi iot
    - ssid: !secret wifi_ssid
      password: !secret wifi_password

  on_connect:
    - logger.log:
        level: INFO
        tag: "WIFI-INFO"
        format: "Device has connected to wifi"

  on_disconnect:
    - logger.log:
        level: WARN
        tag: "WIFI-INFO"
        format: "Client disconnected from wifi, try recconect"

  domain: !secret domain
  reboot_timeout: 5min

  power_save_mode: none
  output_power: 12db
  fast_connect: false
  passive_scan: false
  enable_on_boot: true

time:
  - platform: sntp
    id: time_sntp

external_components:
  - source: github://SzczepanLeon/esphome-components@main
    refresh: 0d
    components: [ wmbus ]

wmbus:
  mosi_pin: GPIO23
  miso_pin: GPIO19
  clk_pin:  GPIO18
  cs_pin:   GPIO21
  gdo0_pin: GPIO16
  gdo2_pin: GPIO17

  led_pin: GPIO2
  led_blink_time: "1s"

  frequency: 868.950
  sync_mode: True

sensor:
# add driver to compile list (will be available for autodetect), don't create sensor
  - platform: wmbus
    type: dme_07

# add sensor with defined type (driver will be also added to compile list)
  - platform: wmbus
    meter_id: 0x43430778
    type: dme_07
    key: ""
    sensors:
      - name: "my hot water RSSi"
        field: "rssi"
        accuracy_decimals: 0
        unit_of_measurement: "dBm"
        device_class: "signal_strength"
        state_class: "measurement"
        entity_category: "diagnostic"
      - name: "my hot water"
        field: "total"
        accuracy_decimals: 3
        unit_of_measurement: "m³"
        device_class: "water"
        state_class: "total_increasing"
        icon: "mdi:water"
```

I have this error here : 

[image]

Anoxym, 2025-05-29:
Hi!

```
# Global config
esphome:
  name: "water"
  friendly_name: "water"

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  level: INFO
  # level: DEBUG

wifi:

  networks:

    # default wifi iot
    - ssid: !secret wifi_ssid
      password: !secret wifi_password

  on_connect:
    - logger.log:
        level: INFO
        tag: "WIFI-INFO"
        format: "Device has connected to wifi"

  on_disconnect:
    - logger.log:
        level: WARN
        tag: "WIFI-INFO"
        format: "Client disconnected from wifi, try recconect"

  domain: !secret domain
  reboot_timeout: 5min

  power_save_mode: none
  output_power: 12db
  fast_connect: false
  passive_scan: false
  enable_on_boot: true

time:
  - platform: sntp
    id: time_sntp

external_components:
  - source: github://SzczepanLeon/esphome-components@version_4
    refresh: 0d
    components: [ wmbus ]

wmbus:
  mosi_pin: GPIO23
  miso_pin: GPIO19
  clk_pin:  GPIO18
  cs_pin:   GPIO21
  gdo0_pin: GPIO16
  gdo2_pin: GPIO17

  led_pin: GPIO2
  led_blink_time: "1s"

  frequency: 868.950
  sync_mode: True

sensor:
# add sensor with defined type (driver will be also added to compile list)
  - platform: wmbus
    meter_id: 0x43430778
    type: dme_07
    key: ""
    sensors:
      - name: "my hot water RSSi"
        field: "rssi"
        accuracy_decimals: 0
        unit_of_measurement: "dBm"
        device_class: "signal_strength"
        state_class: "measurement"
        entity_category: "diagnostic"
      - name: "my hot water"
        field: "total"
        accuracy_decimals: 3
        unit_of_measurement: "m³"
        device_class: "water"
        state_class: "total_increasing"
        icon: "mdi:water"
```

Error on line "sensor" : [sensors] is an invalid option for [sensor.wmbus]. Please check the indentation.

hakelgg, 2025-06-04:
Unfortunately, I am not getting any data transferred to Homeassistant or Webbroser.

I have tested the various ESP32s - the error remains.
HASS cleaned. - Error remains
Working under Proxmox VM Homeassistant - Hass reinstalled - the error remains

I have attached below everything I thought might be useful for troubleshooting. 

[watermeter-waterstarm_yaml.txt](https://github.com/user-attachments/files/20597419/watermeter-waterstarm_yaml.txt)
[logs_watermeter-waterstarm_logs.txt](https://github.com/user-attachments/files/20597355/logs_watermeter-waterstarm_logs.txt)
[logs_watermeter-waterstarm_run.txt](https://github.com/user-attachments/files/20597356/logs_watermeter-waterstarm_run.txt)
[image]
![Image](https://github.com/user-attachments/assets/671d18

SzczepanLeon, 2025-06-04:
Because in YAML you have ID as DEC but it should be in HEX.
Just add 0x before numbers and everything should be OK.

Everything is in logs (screenshot that you attached).

hakelgg, 2025-06-04:
Moin
Yes, it works.
Thank you very much.