SzczepanLeon/esphome-components #208: `wmbus` conflicts with `ethernet` component on ESP32-S3-ETH

farmio, 2025-05-07:
Hi 👋! 

First of all, thanks for this component! I'm trying to use it to read my Diehl Hydrus and managed to receive my first telegrams 😃 However, I'm using a [Waveshare ESP32-S3-ETH](https://www.waveshare.com/wiki/ESP32-S3-ETH) board, but it only works with `wifi` enabled. Whenever I use `ethernet` instead, the board gets stuck a boot loop.

When commenting out the `wmbus` component, the ethernet connection works fine.

<details><summary>Logs</summary>

```log
INFO Successfully uploaded program.
INFO Starting log output from /dev/cu.usbmodem101 with baud rate 115200
[15:24:28][I][logger:171]: Log initialized
[15:24:28][C][safe_mode:079]: There have been 3 suspected unsuccessful boot attempts
[15:24:28][D][esp32.preferences:114]: Saving 1 preferences to flash...
[15:24:28][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[15:24:28][I][app:029]: Running through setup()...
[15:24:28][C][ethernet:041]: Setting up Ethernet...
[15:24:28][I][ethernet:248]: Starting ethernet connection
[15:24:28][W][component:157]: Component ethernet set Warning flag: waiting for IP configuration
[15:24:34][I][ethernet:259]: Connected via Ethernet!
[15:24:34][C][ethernet:515]:   IP Address: 10.1.1.179
[15:24:34][C][ethernet:516]:   Hostname: 'wasserzaehler'
[15:24:34][C][ethernet:517]:   Subnet: 255.255.0.0
[15:24:34][C][ethernet:518]:   Gateway: 10.1.0.1
[15:24:34][C][ethernet:523]:   DNS1: 10.1.0.10
[15:24:34][C][ethernet:524]:   DNS2: 0.0.0.0
[15:24:34][C][ethernet:536]:   MAC Address: CE:BA:97:10:53:1D
[15:24:34][C][ethernet:537]:   Is Full Duplex: YES
[15:24:34][C][ethernet:538]:   Link Speed: 100
[15:24:34][W][component:172]: Component ethernet cleared Warning flag
[15:24:34][C][sntp:018]: Setting up SNTP...
[15:24:34][C][api:026]: Setting up Home Assistant API server...
[15:24:34][D][rxLoop:031]: Set CC1101 frequency to 868.950MHz [21 6B D0]
[15:24:34][D][rxLoop:043]: CC1101 version '4'
[15:24:34][D][rxLoop:045]: CC1101 initialized
[15:24:34][I][app:062]: setup() finished successfully!
[15:24:34][W][component:157]: Component api set Warning flag: unspecified
[15:24:34][I][app:100]: ESPHome version 2025.4.1 compiled on May  7 2025, 15:24:09
[15:24:34][C][logger:177]: Logger:
[15:24:34][C][logger:178]:   Max Level: DEBUG
[15:24:34][C][logger:179]:   Initial Level: DEBUG
[15:24:34][C][logger:181]:   Log Baud Rate: 115200
[15:24:34][C][logger:182]:   Hardware UART: USB_CDC
[15:24:34][C][ethernet:326]: Ethernet:
[15:24:34][C][ethernet:515]:   IP Address: 10.1.1.179
[15:24:34][C][ethernet:516]:   Hostname: 'wasserzaehler'
[15:24:34][C][ethernet:517]:   Subnet: 255.255.0.0
[15:24:34][C][ethernet:518]:   Gateway: 10.1.0.1
[15:24:34][C][ethernet:523]:   DNS1: 10.1.0.10
[15:24:34][C][ethernet:524]:   DNS2: 0.0.0.0
[15:24:34][C][ethernet:536]:   MAC Address: CE:BA:97:10:53:1D
[15:24:34][C][ethernet:537]:   Is Full Duplex: YES
[15:24:34][C][ethernet:538]:   Link Speed: 100
[15:24:34][C][ethernet:329]:   CLK Pin: 13
[15:24:34][C][ethernet:330]:   MISO Pin: 12
[15:24:34][C][ethernet:331]:   MOSI Pin: 11
[15:24:34][C][ethernet:332]:   CS Pin: 14
[15:24:34][C][ethernet:339]:   IRQ Pin: 10
[15:24:34][C][ethernet:341]:   Reset Pin: 9
[15:24:34][C][ethernet:342]:   Clock Speed: 26 MHz
[15:24:34][C][ethernet:351]:   Type: W5500
[15:24:34][C][sntp:042]: SNTP Time:
[15:24:34][C][sntp:045]:   Server 0: '0.pool.ntp.org'
[15:24:34][C][sntp:045]:   Server 1: '1.pool.ntp.org'
[15:24:34][C][sntp:045]:   Server 2: '2.pool.ntp.org'
[15:24:34][C][mdns:116]: mDNS:
[15:24:34][C][mdns:117]:   Hostname: wasserzaehler
[15:24:34][C][esphome.ota:073]: Over-The-Air updates:
[15:24:34][C][esphome.ota:074]:   Address: wasserzaehler.local:3232
[15:24:34][C][esphome.ota:075]:   Version: 2
[15:24:34][C][esphome.ota:078]:   Password configured
[15:24:34][C][safe_mode:018]: Safe Mode:
[15:24:34][C][safe_mode:020]:   Boot considered successful after 60 seconds
[15:24:34][C][safe_mode:021]:   Invoke after 10 boot attempts
[15:24:34][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[15:24:34][W][safe_mode:029]: Last reset occurred too quickly; safe mode will be invoked in 7 restarts
[15:24:34][C][api:140]: API Server:
[15:24:34][C][api:141]:   Address: wasserzaehler.local:6053
[15:24:34][C][api:145]:   Using noise encryption: NO
[15:24:34][C][wmbus:413]: wM-Bus v4.2.1-1.17.1-b8f4a945:
[15:24:34][C][wmbus:431]:   Chip ID: 1C531097BACC
[15:24:34][C][wmbus:433]:   CC1101 frequency: 868.950 MHz
[15:24:34][C][wmbus:434]:   CC1101 SPI bus:
[15:24:34][C][wmbus:438]:     MOSI Pin: GPIO46
[15:24:34][C][wmbus:439]:     MISO Pin: GPIO42
[15:24:34][C][wmbus:440]:     CLK Pin:  GPIO45
[15:24:34][C][wmbus:441]:     CS Pin:   GPIO39
[15:24:34][C][wmbus:442]:     GDO0 Pin: GPIO40
[15:24:34][C][wmbus:443]:     GDO2 Pin: GPIO41
[15:24:34][C][wmbus:449]:   Available drivers: unknown, hydrus
[15:24:34][C][wmbus:463]:   Meter:
[15:24:34][C][wmbus:464]:     ID: 2282968643 [0x88135643]
[15:24:34][C][wmbus:465]:     Type: hydrus
[15:24:34][C][wmbus:466]:     Key: ''
[15:24:34][C][wmbus:468]:     Field: 'flow_temperature'
[15:24:34][C][wmbus:469]:      Name: 'Wasser Temperatur'
[15:24:34][C][wmbus:469]:        Device Class: 'temperature'
[15:24:34][C][wmbus:469]:        State Class: 'measurement'
[15:24:34][C][wmbus:469]:        Unit of Measurement: '°C'
[15:24:34][C][wmbus:469]:        Accuracy Decimals: 1
[15:24:34][C][wmbus:469]:        Icon: 'mdi:water'
[15:24:34][C][wmbus:468]:     Field: 'remaining_battery_life'
[15:24:34][C][wmbus:469]:      Name: 'Wasserzähler verbleibende Batterielaufzeit'
[15:24:34][C][wmbus:469]:        State Class: 'measurement'
[15:24:34][C][wmbus:469]:        Unit of Measurement: 'y'
[15:24:34][C][wmbus:469]:        Accuracy Decimals: 5
[15:24:34][C][wmbus:469]:        Icon: 'mdi:water'
[15:24:34][C][wmbus:468]:     Field: 'rssi'
[15:24:34][C][wmbus:469]:      Name: 'Wasserzähler RSSi'
[15:24:34][C][wmbus:469]:        Device Class: 'signal_strength'
[15:24:34][C][wmbus:469]:        State Class: 'measurement'
[15:24:34][C][wmbus:469]:        Unit of Measurement: 'dBm'
[15:24:34][C][wmbus:469]:        Accuracy Decimals: 0
[15:24:34][C][wmbus:468]:     Field: 'total'
[15:24:34][C][wmbus:469]:      Name: 'Wasser Verbrauch'
[15:24:34][C][wmbus:469]:        Device Class: 'water'
[15:24:34][C][wmbus:469]:        State Class: 'total_increasing'
[15:24:34][C][wmbus:469]:        Unit of Measurement: 'm³'
[15:24:34][C][wmbus:469]:        Accuracy Decimals: 3
[15:24:34][C][wmbus:469]:        Icon: 'mdi:water'
[15:24:34]ESP-ROM:esp32s3-20210327
[15:24:34]Build:Mar 27 2021
[15:24:34]rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[15:24:34]Saved PC:0x403770cc
WARNING Decoded 0x403770cc: esp_restart_noos at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/port/soc/esp32s3/system_internal.c:143 (discriminator 1)
[15:24:34]SPIWP:0xee
[15:24:34]mode:DIO, clock div:1
[15:24:34]load:0x3fce3808,len:0x43c
[15:24:34]load:0x403c9700,len:0xbec
[15:24:34]load:0x403cc700,len:0x2a3c
[15:24:34]entry 0x403c98d8
[15:24:35][I][logger:171]: Log initialized
[15:24:35][C][safe_mode:079]: There have been 4 suspected unsuccessful boot attempts
[15:24:35][D][esp32.preferences:114]: Saving 1 preferences to flash...
[15:24:35][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[15:24:35][I][app:029]: Running through setup()...
[15:24:35][C][ethernet:041]: Setting up Ethernet...
```

</details> 

<details><summary>ESPHome config </summary>

```yaml
esphome:
  name: wasserzaehler

esp32:
  board: esp32s3box
  framework:
    type: arduino
  flash_size: 16MB

# Enable logging
logger:

api:
  password: ...

ota:
  - platform: esphome
    password: ...

# wifi:
#   ssid: ...
#   password: ...

ethernet:
  type: W5500
  clk_pin: GPIO13
  mosi_pin: GPIO11
  miso_pin: GPIO12
  cs_pin: GPIO14
  interrupt_pin: GPIO10
  reset_pin: GPIO9

time:
  - platform: sntp
    id: time_sntp

external_components:
  - source: github://SzczepanLeon/esphome-components@main
    refresh: 0d
    components: [ wmbus ]

wmbus:
  mosi_pin: GPIO46
  miso_pin: GPIO42
  clk_pin:  GPIO45
  cs_pin:   GPIO39
  gdo0_pin: GPIO40
  gdo2_pin: GPIO41

sensor:
  - platform: wmbus
    type: hydrus
    ...
```

</details> 

Is there anything I can do to get the wmbus component with a CC1101 running with an ethernet connection?

SzczepanLeon, 2025-05-22:
Is it possible to run that board on ethernet and without wmbus component?

farmio, 2025-05-22:
Yes, ethernet works fine when commenting out the wmbus component.

SzczepanLeon, 2025-05-22:
Probably hardcoded SPI channel for CC1101 is the same as for ETH.

Please try this in YAML:
```
external_components:
  - source: github://SzczepanLeon/esphome-components@main
    refresh: 0d
    components: [ wmbus, ethernet ]
```

farmio, 2025-05-22:
Yes that works perfectly fine! Thank you very much!

Can that SPI channel be configured somehow or would it be possible to do so in a future release?
If you need any further testing, I'd be happy to help.

SzczepanLeon, 2025-05-22:
As I mentioned it is hardcoded, but in v5 it will be configurable.

closed 2025-05-22 by SzczepanLeon

farmio, 2025-05-22:
Nice! Thank you for your work!