SzczepanLeon/esphome-components #209: Can´t get values from telegram

josefkrtil, 2025-05-18:
I can receive telegram from my new IPERL watermeter.

`[19:45:58][D][mbus:034]: Processing T1 A frame
[19:45:58][D][wmbus:097]: Using driver 'iperl' for ID [0x22845882] RSSI: -64 dBm LQI: 128 Frame: T1 A T: 1E44AE4C8258842268077AC10010054F1050607C747FB981F5EB91067F7336 (31)
[19:45:58][D][wmbus:161]: Can't get value(s) from telegram for ID [0x22845882]`
`[16:47:52][D][mbus:034]: Processing T1 A frame
[16:47:52][D][wmbus:097]: Using driver 'iperl' for ID [0x22845882] RSSI: -52 dBm LQI: 128 Frame: T1 A T: 1E44AE4C8258842268077ABE0010053761648B2FA60680A4FB1A438B2FB577 (31)
[16:47:52][D][wmbus:161]: Can't get value(s) from telegram for ID [0x22845882]`

But all telegrams, are the same - no data inside - only ID and the same timestamp : 1970-01-01T00:00:00Z

`Auto driver    : iperl
Similar driver : unknown 00/00
Using driver   : iperl 00/00
000   : 1e length (30 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : ae4c dll-mfct (SEN)
004   : 82588422 dll-id (22845882)
008   : 68 dll-version
009   : 07 dll-type (Water meter)
010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
011   : c1 tpl-acc-field
012   : 00 tpl-sts-field (OK)
013   : 1005 tpl-cfg 0510 (AES_CBC_IV nb=1 cntn=0 ra=0 hc=0 )
015 CE: 4F1050607C747FB981F5EB91067F7336 encrypted

{
    "_":"telegram",
    "media":"water",
    "meter":"iperl",
    "name":"",
    "id":"22845882",
    "timestamp":"1970-01-01T00:00:00Z"
}

Using: wmbusmeters: 1.18.0-31-g6d3b3e5
6d3b3e5f233cbfc8cfd9bac99d3f845496330e5f`

I dont know if it is problem from watermeter or something other :-(

SzczepanLeon, 2025-05-18:
You need key.

josefkrtil, 2025-05-18:
Sorry for stupid question, but where can I find the key?

SzczepanLeon, 2025-05-18:
Try this one:
https://github.com/wmbusmeters/wmbusmeters/issues/877#issuecomment-1442513839

closed 2025-05-18 by SzczepanLeon

josefkrtil, 2025-05-18:
Thanks. It works!