SzczepanLeon/esphome-components #210: ESP8266 with components v. 3.2.3 and ESPHome 2025.3.3 ipv4 error

gitwoy, 2025-05-18:
as in topic.
Log below:

INFO ESPHome 2025.3.3
INFO Reading configuration /config/esphome/licznik.yaml...
INFO Updating https://github.com/SzczepanLeon/esphome-components.git@3.2.3
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/esphome/esphome/__main__.py", line 1057, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1035, in run_esphome
    config = read_config(dict(args.substitution) if args.substitution else {})
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 1095, in read_config
    res = load_config(command_line_substitutions)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 949, in load_config
    return _load_config(command_line_substitutions)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 939, in _load_config
    return validate_config(config, command_line_substitutions)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 835, in validate_config
    target_platform = core_config.preload_core_config(config, result)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/core/config.py", line 231, in preload_core_config
    if _is_target_platform(domain):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/core/config.py", line 192, in _is_target_platform
    return get_component(name, True).is_target_platform
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/loader.py", line 205, in get_component
    return _lookup_module(domain, exception)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/loader.py", line 181, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/data/external_components/0e75e1a4/components/wmbus/__init__.py", line 62, in <module>
    cv.Required(CONF_IP_ADDRESS): cv.ipv4,
                                  ^^^^^^^
AttributeError: module 'esphome.config_validation' has no attribute 'ipv4'

SzczepanLeon, 2025-05-18:
Please share YAML, but this issue is already fixed. Please check history or use latest component and latest ESPHome.

closed 2025-05-18 by SzczepanLeon

peetereczek, 2025-05-18:
Issue is not fixed.

`esphome:
  name: wmbus-sniffer
  friendly_name: wmbus-sniffer
  comment: WMBUS sniffing device
  compile_process_limit: 1 # tylko w przypadku RPI

esp8266:
  board: nodemcuv2

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: "69DXzhY3jAwVtNrlM8pBWxQHkeeX3ub6jtNG4MUI2Vo="

ota:
  - platform: esphome
    password: !secret

wifi:
  ssid: !secret
  password: !secret

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Wmbus-Sniffer Fallback Hotspot"
    password: !secret

captive_portal:

web_server:
  port: 80

external_components:
  source: github://SzczepanLeon/esphome-components@3.2.3
  components: [ wmbus ]
  refresh: 0d

time:
  platform: sntp
  id: time_sntp

wmbus:
  mosi_pin: GPIO13
  miso_pin: GPIO12
  clk_pin: GPIO14
  cs_pin: GPIO2
  gdo0_pin: GPIO5
  gdo2_pin: GPIO4
  led_pin: GPIO0
  led_blink_time: "1s"
  all_drivers: True
  log_all: True`

SzczepanLeon, 2025-05-18:
Because you are using not latest v3.
Please change YAML from tag to branch.