SzczepanLeon/esphome-components #217: Get frame but dont decript

oomichalekoo, 2025-05-23:
Get frame but dont decript version 4 normal work

`[09:01:31][C][wmbus:469]:      Name: '0x6956659_woda'
[09:01:31][C][wmbus:469]:        Device Class: 'water'
[09:01:31][C][wmbus:469]:        State Class: 'total_increasing'
[09:01:31][C][wmbus:469]:        Unit of Measurement: 'm³'
[09:01:31][C][wmbus:469]:        Accuracy Decimals: 3
[09:01:31][C][wmbus:469]:        Icon: 'mdi:water'
[09:01:35][D][mbus:035]: Received T1 A frame`

SzczepanLeon, 2025-05-23:
With what YAML?
please use sync_mode = true

oomichalekoo, 2025-05-23:
I have true

oomichalekoo, 2025-05-23:
`INFO ESPHome 2025.5.0
INFO Reading configuration /config/esphome/licznik.yaml...
INFO Updating https://github.com/SzczepanLeon/esphome-components.git@main
WARNING 'woda_szczepan': Using the '_' (underscore) character in the hostname is discouraged as it can cause problems with some DHCP and local name services. For more information, see https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name
INFO Detected timezone 'Europe/Warsaw'
WARNING GPIO12 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO15 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from 192.168.1.166 using esphome API
INFO Successfully connected to woda_szczepan @ 192.168.1.166 in 0.113s
INFO Successful handshake with woda_szczepan @ 192.168.1.166 in 0.057s`

SzczepanLeon, 2025-05-23:
And what about VERBOSE one?
What YAML?

oomichalekoo, 2025-05-23:
`substitutions:
  name: "woda_szczepan"
  friendly_name: Woda

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.6.0
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: dev

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

# Poprawiona sekcja OTA – nowa składnia
ota:
  platform: esphome

# Allow provisioning Wi-Fi via serial
improv_serial:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

# Umożliwia konfigurację Wi-Fi przez AP
captive_portal:

# Aby można było wejść na urządzenie przez IP
web_server:

# Poprawna wersja komponentu WMBUS z GitHub
external_components:
  - source: github://SzczepanLeon/esphome-components@main
    refresh: 0d
    components: [wmbus]

# Czas przez NTP
time:
  - platform: sntp
    id: sntp_time

# Konfiguracja WMBUS (z CC1101)
wmbus:
  mosi_pin: GPIO13  # blue
  clk_pin:  GPIO14  # purple
  miso_pin: GPIO12  # green
  gdo2_pin: GPIO22  # gray
  gdo0_pin: GPIO21  # yellow
  cs_pin:   GPIO15  # orange
  frequency: 868.950
  all_drivers: false
  log_all: false
  sync_mode: true 

# Sensory z Apatora
sensor:
  - platform: wmbus
    meter_id: 0x06954659
    type: apator162
    key: "00000000000000000000000000000000"
    sensors:
      - name: "0x6954659_My_RSSI"
        field: "rssi"
        accuracy_decimals: 0
        unit_of_measurement: "dBm"
        device_class: "signal_strength"
        state_class: "measurement"
        entity_category: "diagnostic"
      - name: "0x6956659_woda"
        field: "total"
        accuracy_decimals: 3
        unit_of_measurement: "m³"
        device_class: "water"
        state_class: "total_increasing"
        icon: "mdi:water"
`

oomichalekoo, 2025-05-23:
i get frame why becouse i have verbose ? i i can delete it ?
`logger:
  level: VERBOSE`

SzczepanLeon, 2025-05-23:
You can change VERBOSE to anything you want.

closed 2025-05-23 by SzczepanLeon

oomichalekoo, 2025-05-23:
but verbose help get frame ? i thing its only log

SzczepanLeon, 2025-05-23:
It should be log only.