SzczepanLeon/esphome-components #229: Compile error - missing ELECHOUSE_CC1101_SRC_DRV.h

pawkrk, 2025-06-24:
ESPHome: 2025.6.0

Compiling fails both on ESP32C3 and C6. I have fresh installation of ESPhome. 

external_components:
  - source: github://SzczepanLeon/esphome-components@version_4
    refresh: 0d
    components: [ wmbus ]

Error:
Compiling .pioenvs/water-meter-c6/src/esphome/components/wmbus/translatebits.cpp.o
In file included from src/esphome/components/wmbus/rf_cc1101.h:9,
                 from src/esphome/components/wmbus/rf_cc1101.cpp:1:
src/esphome/components/wmbus/cc1101_rf_settings.h:3:10: fatal error: ELECHOUSE_CC1101_SRC_DRV.h: No such file or directory

****************************************************************************************
* Looking for ELECHOUSE_CC1101_SRC_DRV.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ELECHOUSE_CC1101_SRC_DRV.h"
* Web  > https://registry.platformio.org/search?q=header:%1B%5Bm%1B%5BKELECHOUSE_CC1101_SRC_DRV.h
*
****************************************************************************************

    3 | #include <ELECHOUSE_CC1101_SRC_DRV.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pioenvs/water-meter-c6/src/esphome/components/wmbus/types.cpp.o
*** [.pioenvs/water-meter-c6/src/esphome/components/wmbus/rf_cc1101.cpp.o] Error 1
========================= [FAILED] Took 396.34 seconds =========================

platformio installed packages:

# platformio lib --global list

WARNING: This command is deprecated and will be removed in the next releases.
Please use `pio pkg list` instead.

Libraries
├── ArduinoJson @ 7.0.0 (required: bblanchon/ArduinoJson)
├── ArduinoJson @ 6.18.5 (required: bblanchon/ArduinoJson)
├── AsyncMqttClient-esphome @ 1.0.0 (required: heman/AsyncMqttClient-esphome)
│   ├── AsyncTCP-esphome @ 1.2.2 (required: esphome/AsyncTCP-esphome @ ^1.1.1)
│   └── ESPAsyncTCP-esphome @ 2.0.0 (required: esphome/ESPAsyncTCP-esphome @ ^2.0.0)
├── AsyncTCP-esphome @ 2.1.4 (required: esphome/AsyncTCP-esphome)
├── Crypto @ 0.4.0 (required: rweather/Crypto)
├── Dsmr @ 0.7.0 (required: glmnet/Dsmr)
├── ESP32-audioI2S @ 2.2.0 (required: esphome/ESP32-audioI2S)
├── ESPAsyncWebServer-esphome @ 3.3.0 (required: esphome/ESPAsyncWebServer-esphome)
├── ESPMicroSpeechFeatures @ 1.1.0 (required: kahrendt/ESPMicroSpeechFeatures)
├── FastLED @ 3.9.16 (required: fastled/FastLED)
├── HaierProtocol @ 0.9.31 (required: pavlodn/HaierProtocol)
├── HeatpumpIR @ 1.0.32 (required: tonia/HeatpumpIR)
│   └── IRremoteESP8266 @ 2.8.6 (required: crankyoldgit/IRremoteESP8266 @ ~2.8.4)
├── Improv @ 1.2.4 (required: improv/Improv)
├── MideaUART @ 1.1.9 (required: dudanov/MideaUART)
├── NeoPixelBus @ 2.7.3 (required: makuna/NeoPixelBus)
├── TM1651 @ 1.0.1 (required: freekode/TM1651)
├── TinyGPSPlus @ 1.1.0 (required: mikalhart/TinyGPSPlus)
├── arduino-MLX90393 @ 1.0.2 (required: functionpointer/arduino-MLX90393)
├── esp-audio-libs @ 1.1.4 (required: esphome/esp-audio-libs)
├── esp_mbedtls_esp8266 @ 2.22300.2 (required: droscy/esp_mbedtls_esp8266)
├── esp_wireguard @ 0.4.2 (required: droscy/esp_wireguard)
│   └── libsodium @ 1.10018.1 (required: esphome/libsodium @ ^1.10018.1)
├── lvgl @ 8.4.0 (required: lvgl/lvgl)
├── noise-c @ 0.1.4 (required: esphome/noise-c)
├── noise-c @ 0.1.1 (required: esphome/noise-c)
├── pngle @ 1.1.0 (required: kikuchan98/pngle)
└── qr-code-generator-library @ 1.7.0 (required: wjtje/qr-code-generator-library)

damadmax, 2025-07-09:
did you have any success compiling recently? getting the same error here.

SzczepanLeon, 2025-07-23:
Not planed for v4.

closed 2025-07-23 by SzczepanLeon