SzczepanLeon/esphome-components #231: version5(main) not for me: no match on every receveid data

lordp1, 2025-06-28:
so i have a water-meter with id 20412874 (is on my waterbill).
i didn't receive any more data on my old setup with version4 (could be because there are also multple modbus readouts on it, maybe time/cpu-cycle problem?).

so i bought an sx1276 module. for version 5 on a new esp32 board.
here the config
[esp-watermeter-2.txt](https://github.com/user-attachments/files/20964965/esp-watermeter-2.txt)

but i always get the message
`[00:46:48][V][wmbusmeters:563]: (meter) izar-013779ca: not for me: no match`
here the logs
[logs_esp-watermeter_run-3.txt](https://github.com/user-attachments/files/20965004/logs_esp-watermeter_run-3.txt)

tried meter-id's '0x013779ca #0xCA793701 #20412874' but none of them decode

but on the online docoder it works:
[https://wmbusmeters.org/analyze/2944A511780774284120A2010F00132E764E1ABC4E69C0B9B30464D0031BD69F23C087B8603BE9DEFA59](url)
`{
    "_":"telegram",
    "media":"water",
    "meter":"izar",
    "name":"",
    "id":"20412874",
    "last_month_total_m3":19.099,
    "remaining_battery_life_y":7.5,
    "total_m3":19.623,
    "transmit_period_s":8,
    "current_alarms":"no_alarm",
    "last_month_measure_date":"2025-05-31",
    "previous_alarms":"no_alarm",
    "timestamp":"2025-06-28T22:50:28Z"
}`

lordp1, 2025-06-29:
thank that works.
_[didn't consider taking decimal number, and just put 0x in front of it without conversion to hex]_

closed 2025-06-29 by lordp1

SzczepanLeon, 2025-06-29:
It is not decimals. It only looks like DEC in this case.