SzczepanLeon/esphome-components #233: V4 in esphome 2025.6 does not work

kaczkalolo, 2025-07-04:
Before 2025.6 everything worked fine. Now i've updated my code to include 
  - source: github://SzczepanLeon/esphome-components@version_4
and now i don't get any readouts. Nothing at all. Nothing in the logs. 

This is was my esphome code:

`esphome:
  name: watermeter
  friendly_name: Watermeter

esp32:
  board: esp32dev
  framework:
    type: arduino

external_components:
  - source: github://SzczepanLeon/esphome-components@main
    refresh: 0d
    components: [ wmbus ]

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxxx"

ota:
  - platform: esphome
    password: "xxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Watermeter Fallback Hotspot"
    password: "xxxx"

captive_portal:

web_server:
  port: 80
  version: 2

time:
  platform: sntp

wmbus:
  mosi_pin: GPIO13
  miso_pin: GPIO12
  clk_pin:  GPIO14
  cs_pin:   GPIO15
  gdo0_pin: GPIO5
  gdo2_pin: GPIO4
  all_drivers: False
  log_all: False
  frequency: 868.950
  # sync_mode: True

sensor:
  - platform: wmbus
    meter_id: 0x01218653
    type: apator162
    key: "00000000000000000000000000000000"
    sensors:
      - name: "Watermeter Home RSSI"
        field: "rssi"
        accuracy_decimals: 0
        unit_of_measurement: "dBm"
        device_class: "signal_strength"
        state_class: "measurement"
        entity_category: "diagnostic"  
      - name: "Watermeter Home"
        field: "total"
        accuracy_decimals: 3
        unit_of_measurement: "m³"
        device_class: "water"
        state_class: "total_increasing"
        icon: "mdi:water"   
        filters:
          - offset: 62.440
  - platform: wmbus
    meter_id: 0x01333362
    type: apator162
    key: "00000000000000000000000000000000"
    sensors:
      - name: "Watermeter Garden RSSI"
        field: "rssi"
        accuracy_decimals: 0
        unit_of_measurement: "dBm"
        device_class: "signal_strength"
        state_class: "measurement"
        entity_category: "diagnostic"
      - name: "Watermeter Garden"
        field: "total"
        accuracy_decimals: 3
        unit_of_measurement: "m³"
        device_class: "water"
        state_class: "total_increasing"
        icon: "mdi:water"  
        filters:
          - offset: -1.539     `

kaczkalolo, 2025-07-04:
Nothing at all

SzczepanLeon, 2025-07-04:
Stay on old version or update HW and switch to v5.

closed 2025-07-04 by SzczepanLeon